Koenkk/zigbee2mqtt #107: Missing device after updating

kiucheng, 2018-06-09:
So i have done an update following wiki, after that all my device (only 1 actually) are lost and there is nothing in configuration.yaml. So I proceed to pairing it again (Aqara motion and lux sensor), this happens

What is the correct way to update zigbee2mqtt? Do I have to back-up configuration.yaml first?

'''Jun 09 13:35:42 hassbian npm[4411]: 2018-6-9 13:35:42 WARN Message without device!
Jun 09 13:35:45 hassbian npm[4411]: spinlock: false []
Jun 09 13:36:19 hassbian npm[4411]: /opt/zigbee2mqtt/node_modules/zcl-packet/lib/foundation.js:297
Jun 09 13:36:19 hassbian npm[4411]:         dataTypeStr = zclId.dataType(dataType).key;
Jun 09 13:36:19 hassbian npm[4411]:                                               ^
Jun 09 13:36:19 hassbian npm[4411]: TypeError: Cannot read property 'key' of undefined
Jun 09 13:36:19 hassbian npm[4411]:     at ensureDataTypeString (/opt/zigbee2mqtt/node_modules/zcl-packet/lib/foundation.js:297:47)
Jun 09 13:36:19 hassbian npm[4411]:     at getDataType (/opt/zigbee2mqtt/node_modules/zcl-packet/lib/foundation.js:307:16)
Jun 09 13:36:19 hassbian npm[4411]:     at Dissolve.<anonymous> (/opt/zigbee2mqtt/node_modules/zcl-packet/lib/foundation.js:646:24)
Jun 09 13:36:19 hassbian npm[4411]:     at Dissolve._transform (/opt/zigbee2mqtt/node_modules/dissolve/index.js:80:16)
Jun 09 13:36:19 hassbian npm[4411]:     at Dissolve.Transform._read (/opt/zigbee2mqtt/node_modules/dissolve/node_modules/readable-stream/lib/_stream_transform.js:184:10)
Jun 09 13:36:19 hassbian npm[4411]:     at Dissolve.Transform._write (/opt/zigbee2mqtt/node_modules/dissolve/node_modules/readable-stream/lib/_stream_transform.js:172:12)
Jun 09 13:36:19 hassbian npm[4411]:     at doWrite (/opt/zigbee2mqtt/node_modules/dissolve/node_modules/readable-stream/lib/_stream_writable.js:279:12)
Jun 09 13:36:19 hassbian npm[4411]:     at writeOrBuffer (/opt/zigbee2mqtt/node_modules/dissolve/node_modules/readable-stream/lib/_stream_writable.js:266:5)
Jun 09 13:36:19 hassbian npm[4411]:     at Dissolve.Writable.write (/opt/zigbee2mqtt/node_modules/dissolve/node_modules/readable-stream/lib/_stream_writable.js:211:11)
Jun 09 13:36:19 hassbian npm[4411]:     at Dissolve.Writable.end (/opt/zigbee2mqtt/node_modules/dissolve/node_modules/readable-stream/lib/_stream_writable.js:426:10)
Jun 09 13:36:19 hassbian npm[4411]: npm ERR! code ELIFECYCLE
Jun 09 13:36:19 hassbian npm[4411]: npm ERR! errno 1
Jun 09 13:36:19 hassbian npm[4411]: npm ERR! zigbee2mqtt@0.1.0 start: `node index.js`
Jun 09 13:36:19 hassbian npm[4411]: npm ERR! Exit status 1
Jun 09 13:36:19 hassbian npm[4411]: npm ERR!
Jun 09 13:36:19 hassbian npm[4411]: npm ERR! Failed at the zigbee2mqtt@0.1.0 start script.
Jun 09 13:36:19 hassbian npm[4411]: npm ERR! This is probably not a problem with npm. There is likely additional logging output above.
Jun 09 13:36:19 hassbian npm[4411]: npm ERR! A complete log of this run can be found in:
Jun 09 13:36:19 hassbian npm[4411]: npm ERR!     /home/pi/.npm/_logs/2018-06-09T13_36_19_468Z-debug.log
Jun 09 13:36:19 hassbian systemd[1]: zigbee2mqtt.service: Main process exited, code=exited, status=1/FAILURE
Jun 09 13:36:19 hassbian systemd[1]: zigbee2mqtt.service: Unit entered failed state.
Jun 09 13:36:19 hassbian systemd[1]: zigbee2mqtt.service: Failed with result 'exit-code'.
Jun 09 13:36:19 hassbian systemd[1]: zigbee2mqtt.service: Service hold-off time over, scheduling restart.
Jun 09 13:36:19 hassbian systemd[1]: Stopped zigbee2mqtt.
Jun 09 13:36:19 hassbian systemd[1]: Started zigbee2mqtt.
'''

kiucheng, 2018-06-09:
```git checkout HEAD -- npm-shrinkwrap.json
git pull
rm -rf node_modules
npm install```

ciotlosm, 2018-06-09:
You should back you data/configuration.yaml and data/database.db and restore them after. A git pull could potentially break those files.

ciotlosm, 2018-06-09:
I get the same error starting from scratch using the addon:

```
2018-6-9 18:51:43 INFO MQTT publish, topic: 'zigbee2mqtt/bridge/state', payload: 'online'
/app/node_modules/zcl-packet/lib/foundation.js:297
        dataTypeStr = zclId.dataType(dataType).key;
                                              ^
TypeError: Cannot read property 'key' of undefined
    at ensureDataTypeString (/app/node_modules/zcl-packet/lib/foundation.js:297:47)
    at getDataType (/app/node_modules/zcl-packet/lib/foundation.js:307:16)
    at Dissolve.<anonymous> (/app/node_modules/zcl-packet/lib/foundation.js:646:24)
    at Dissolve._transform (/app/node_modules/dissolve/index.js:80:16)
    at Dissolve.Transform._read (/app/node_modules/dissolve/node_modules/readable-stream/lib/_stream_transform.js:184:10)
    at Dissolve.Transform._write (/app/node_modules/dissolve/node_modules/readable-stream/lib/_stream_transform.js:172:12)
    at doWrite (/app/node_modules/dissolve/node_modules/readable-stream/lib/_stream_writable.js:279:12)
    at writeOrBuffer (/app/node_modules/dissolve/node_modules/readable-stream/lib/_stream_writable.js:266:5)
    at Dissolve.Writable.write (/app/node_modules/dissolve/node_modules/readable-stream/lib/_stream_writable.js:211:11)
    at Dissolve.Writable.end (/app/node_modules/dissolve/node_modules/readable-stream/lib/_stream_writable.js:426:10)
npm
 
ERR! code ELIFECYCLE
npm
 ERR! errno 1
npm
 ERR! zigbee2mqtt@0.1.0 start: `node index.js`
npm ERR! Exit status 1
npm ERR! 
npm ERR! Failed at the zigbee2mqtt@0.1.0 start script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.
npm ERR! A complete log of this run can be found in:
npm ERR!     /root/.npm/_logs/2018-06-09T15_51_53_228Z-debug.log
```

ciotlosm, 2018-06-09:
@kiucheng do you have any other serial device connected?

Koenkk, 2018-06-09:
@ciotlosm does #7bf5b2fd767fa46f319abfc1c568a178f5cc1be4 fix it?

ciotlosm, 2018-06-09:
@Koenkk yes, that fixes the problem.

Koenkk, 2018-06-09:
That is automatically done, build is currently in progress.

ciotlosm, 2018-06-10:
This can be closed.

Koenkk, 2018-06-10:
I've included backing-up the configuration in the update instructions: https://github.com/Koenkk/zigbee2mqtt/wiki/Running-the-bridge#6-for-later-update-zigbee2mqtt-to-the-latest-version. This should prevent any devices from being lost when updating. 

The OP exception also has been fixed.

In case you still have problems feel free to reopen.

closed 2018-06-10 by Koenkk