Koenkk/zigbee2mqtt #108: Battery value problem

ciotlosm, 2018-06-09:
I see the battery value is calculated based on fixed values. My buttons show 100% and I'm sure that is not true. While this is a possible approach, I've checked smartthings and it seems they somehow get battery directly from zigbee messages?

https://github.com/a4refillpad/Xiaomi/blob/master/devicetypes/a4refillpad/xiaomi-zigbee-button.src/xiaomi-zigbee-button.groovy

Koenkk, 2018-06-09:
What is the voltage of your battery? (it should report that together with the battery %)

ciotlosm, 2018-06-09:
3015 which is correctly calculated to 100%, but that can't be. I will pair this device with my gateway and see what battery % does it report.

Koenkk, 2018-06-09:
so it seems that the gateway calculates the percentage with a different algorithm. We have to check at what voltage a new battery starts and at which points it stops working.

ciotlosm, 2018-06-09:
Is there no way to get inspiration from the smartthings setup? How did @kirovilya extract battery %?

ciotlosm, 2018-06-09:
it seems `xiaomi_aqara` component that was reporting my value also calculates it:

```
    def parse_voltage(self, data):
        """Parse battery level data sent by gateway."""
        if 'voltage' not in data:
            return False
        max_volt = 3300
        min_volt = 2800
        voltage = data['voltage']
        voltage = min(voltage, max_volt)
        voltage = max(voltage, min_volt)
        percent = ((voltage - min_volt) / (max_volt - min_volt)) * 100
        self._device_state_attributes[ATTR_BATTERY_LEVEL] = round(percent, 1)
        return True
```

Koenkk, 2018-06-09:
It uses a different range so that explains it. We have to do some test what the value is when we put in a new battery.

ciotlosm, 2018-06-09:
Agree. But for transition would be nice to have it consistent. Our range is wrong anyway because we have 3000 as 100% and I already have on a used battery 3015. 

I recommend we adopt the xiaomi_aqara component approach and do more math when getting a new battery.

ciotlosm, 2018-06-10:
I've created https://github.com/Koenkk/zigbee-shepherd-converters/pull/13 to improve a bit, but we can readjust once we have more samples and details.

closed 2018-06-10 by Koenkk