Koenkk/zigbee2mqtt #111: Cube message for missing converter

ciotlosm, 2018-06-11:
Unsure how I missed this, but this appeared in the logs. Maybe someone has time to correct before I wrap my head around it. 
```
2018-6-11 22:07:52 WARN No converter available for 'MFKZQ01LM' with cid 'genBasic' and type 'devChange'
2018-6-11 22:07:52 WARN Please see: https://github.com/Koenkk/zigbee2mqtt/wiki/How-to-support-new-devices.
```

Disposaber, 2018-06-17:
I got this too. The cube seems to report fine its states though
```
2018-6-17 17:30:56 INFO Connecting with device...
2018-6-17 17:30:56 INFO MQTT publish, topic: 'zigbee2mqtt/bridge/log', payload: '{"type":"pairing","message":"connecting with device"}'
2018-6-17 17:30:57 DEBUG Recieved zigbee message with data {"cid":"genBasic","data":{"modelId":"lumi.sensor_cube"}}
2018-6-17 17:30:57 DEBUG Recieved zigbee message with data {"cid":"genBasic","data":{"modelId":"lumi.sensor_cube"}}
2018-6-17 17:30:57 WARN No converter available for 'MFKZQ01LM' with cid 'genBasic' and type 'devChange'
2018-6-17 17:30:57 WARN Please see: https://github.com/Koenkk/zigbee2mqtt/wiki/How-to-support-new-devices.
2018-6-17 17:30:58 DEBUG Recieved zigbee message with data {"cid":"genBasic","data":{"modelId":"lumi.sensor_cube"}}
2018-6-17 17:30:59 DEBUG Recieved zigbee message with data "0x00158d0001100fa7"
2018-6-17 17:30:59 INFO Connecting with device...
```

Koenkk, 2018-06-17:
This is fixed in https://github.com/Koenkk/zigbee-shepherd-converters/commit/ac324c707fdf1e34b68c95a3af0782f8c581daea, please update zigbee2mqtt to the latest version.

closed 2018-06-17 by Koenkk