Koenkk/zigbee2mqtt #112: Disconnecting issues

alekslyse, 2018-06-11:
I setup a sniffer dongle as stated in the basic instructions and installed a fresh hassio on a rp3. 

Everything seems to be working fine, and tried with an RGB IKEA bulb, it pairs fine and I can control it with Hass, but after some minutes it get stuck on one colour and stop responding before I delete it and re pair it. 

Also tried a hue RGB rev2, but it can't pair it.

I'm not sure if it's anything I'm doing wrong as it seems strange it works some minutes then just stops on a random colour. I see in the log if I try to turn off the light it sends the command but no response.

ciotlosm, 2018-06-12:
@alekslyse try to start zigbee2mqtt in debug mode and post a log sample when you encounter the problem. Please use https://hastebin.com/ to post the log here as a link

ciotlosm, 2018-06-18:
@alekslyse were you able to fix your problem or it's still in issue?

alekslyse, 2018-06-18:
No, I temporary switched to a dezconf setup for zigbee, but I would assume the solution for philips hue bulbs that does not connect is how they are forced to be paired to the hue network.

To solve it I had to use the touchlink interface to reset them before they would pair with any "non hue" hubs. I would think this regards this project too, that people need to do that reset instead of the remote controller reset (that wont put them out of the hue network mode).

ciotlosm, 2018-06-18:
@alekslyse there is some documentation around Philips Hue here: https://github.com/Koenkk/zigbee2mqtt/wiki/Pairing-devices#philips-hue I'm not sure I understand exactly the problem, as I don't have any hue setup. 

Do you have some device IDs? Maybe we can check what other people are experiencing and update documentation accordingly.

alekslyse, 2018-06-18:
The thing is that you cant just factory reset Philips HUE bulbs when they are connected to a HUE hub (I dont know about new bulbs), so I think what most people would face is not being able to detect HUE bulbs, even when "factory reset" them. From what I know Hue put the bulbs into their own network mode (not regular zigbee) when they are connected, and they will NOT go out of this mode with factory reset.

To get the bulbs out of this mode, the user need to use the HUE api (touchlink) to actively remove them from the network. This is a integration the user need to do directly on the bulbs, but its also something you can implement in the software, to search for HUE lights via their API and then reset them - then people do not need to do the remote as it wont work anyways (at least with a V2 bulb).

ciotlosm, 2018-06-29:
@Koenkk Is this a different type of device?

Koenkk, 2018-06-30:
@ciotlosm what do you mean?

ciotlosm, 2018-06-30:
I see in the docs we have support for some ikea bulbs. I would have assumed procedure is similar for all types. Is this a bulb we support and we just need more docs on how to pair or a new device that we don't support yet?

Koenkk, 2018-06-30:
I also have an Ikea RGB bulb and could pair the same way as the other bulbs.

alfatag, 2018-07-13:
@alekslyse 

Ive done the touchlink reset,  confirmed by the light blinking, but had no success with zigbee2mqtt finding it as a new device.

Already tried turning off the HUE HUB as fast as I can after reset in any hope as the possible cause of no detection, but again, without success.

Did you take any specific steps to add the bulbs as new devices, after touchlink?

Thanks in advance!

alekslyse, 2018-07-13:
@alfatag If you have a remote (for example with IKEA remote) you could do the following 

1. Reset the bulb with touchlink to make sure it exists Hue network
2. Reset the bulb with the remote holding the reset button for 10 seconds to ensure the bulb is properly reset
3. Start searching for a light
4. Power switch the light once

alfatag, 2018-07-13:
Hi alekslyse!

Thanks for answering!

Ive just ordered the HUE Dimmer, so I will update as soon I can try your steps.

Best regards!

closed 2018-09-18 by stale[bot]