Koenkk/zigbee2mqtt #113: Could not open specified HEX file

eflemingtra, 2018-06-12:
Hi,

Sorry, newbie here. When trying to flash my CC2531 via the SmartRF Flash Programmer, I get the error:

Could not open specified HEX file

[image]

After this, the USB dongle green light goes out and I have to reinsert the dongle for it to reactivate.

When downloading the .hex file, all I am doing is right clicking and saving it. There's nothing else required, correct?  Windows device manager shows both the CC debugger and USB dongle drivers installed correctly.

eflemingtra, 2018-06-12:
Fixed with alternate download options.

closed 2018-06-12 by eflemingtra

joonhshin, 2018-11-10:
> Fixed with alternate download options.

I'm having the same issue. What alternate download options did you use?

Koenkk, 2018-11-10:
Download from here: https://github.com/Koenkk/Z-Stack-firmware/tree/optimize/coordinator/CC2531/bin