Koenkk/zigbee2mqtt #114: Xiaomi wireless button doesn't work after restart or reboot

subzero79, 2018-06-12:
Just today i got my cc debugger. cc231 dongle is flashed, and I have two spare devices to test (xiaomi's: smart plug and a round switch). Both join zigbee2mqtt easily with the instructions on the wiki.

However seems like the button gets lost after a service restart or system reboot. This is the debug log

```
homeassistant@ha:/opt/zigbee2mqtt/zigbee2mqtt$ DEBUG=zigbee-shepherd* /opt/zigbee2mqtt/node_8.11.2/bin/node index.js 
2018-6-12 19:38:41 INFO Starting zigbee-shepherd
  zigbee-shepherd:init zigbee-shepherd booting... +0ms
  zigbee-shepherd:request REQ --> SYS:osalNvRead +13ms
  zigbee-shepherd:msgHdlr IND <-- ZDO:mgmtPermitJoinRsp +10ms
  zigbee-shepherd:request RSP <-- SYS:osalNvRead, status: 0 +13ms
  zigbee-shepherd:request REQ --> SAPI:readConfiguration +11ms
  zigbee-shepherd:request RSP <-- SAPI:readConfiguration, status: 0 +22ms
  zigbee-shepherd:request REQ --> SAPI:readConfiguration +11ms
  zigbee-shepherd:request RSP <-- SAPI:readConfiguration, status: 0 +8ms
  zigbee-shepherd:request REQ --> SAPI:readConfiguration +11ms
  zigbee-shepherd:request RSP <-- SAPI:readConfiguration, status: 0 +17ms
  zigbee-shepherd:request REQ --> SAPI:readConfiguration +11ms
  zigbee-shepherd:request RSP <-- SAPI:readConfiguration, status: 0 +12ms
  zigbee-shepherd:request REQ --> SAPI:getDeviceInfo +11ms
  zigbee-shepherd:request RSP <-- SAPI:getDeviceInfo +5ms
  zigbee-shepherd:init Now the ZNP is a coordinator. +0ms
  zigbee-shepherd:request REQ --> SAPI:getDeviceInfo +1ms
  zigbee-shepherd:request RSP <-- SAPI:getDeviceInfo +4ms
  zigbee-shepherd:request REQ --> SAPI:getDeviceInfo +0ms
  zigbee-shepherd:request RSP <-- SAPI:getDeviceInfo +5ms
  zigbee-shepherd:request REQ --> SAPI:getDeviceInfo +0ms
  zigbee-shepherd:request RSP <-- SAPI:getDeviceInfo +5ms
  zigbee-shepherd:request REQ --> SAPI:getDeviceInfo +0ms
  zigbee-shepherd:request RSP <-- SAPI:getDeviceInfo +5ms
  zigbee-shepherd:request REQ --> SAPI:getDeviceInfo +0ms
  zigbee-shepherd:request RSP <-- SAPI:getDeviceInfo +5ms
  zigbee-shepherd:request REQ --> SAPI:getDeviceInfo +1ms
  zigbee-shepherd:request RSP <-- SAPI:getDeviceInfo +4ms
  zigbee-shepherd:request REQ --> ZDO:nodeDescReq +1ms
  zigbee-shepherd:msgHdlr IND <-- ZDO:nodeDescRsp +14ms
  zigbee-shepherd:request REQ --> ZDO:activeEpReq +1ms
  zigbee-shepherd:msgHdlr IND <-- ZDO:activeEpRsp +13ms
  zigbee-shepherd:request REQ --> ZDO:simpleDescReq +1ms
  zigbee-shepherd:msgHdlr IND <-- ZDO:simpleDescRsp +13ms
  zigbee-shepherd:request REQ --> ZDO:simpleDescReq +1ms
  zigbee-shepherd:msgHdlr IND <-- ZDO:simpleDescRsp +12ms
  zigbee-shepherd:request REQ --> ZDO:simpleDescReq +1ms
  zigbee-shepherd:msgHdlr IND <-- ZDO:simpleDescRsp +15ms
  zigbee-shepherd:request REQ --> ZDO:simpleDescReq +0ms
  zigbee-shepherd:msgHdlr IND <-- ZDO:simpleDescRsp +16ms
  zigbee-shepherd:request REQ --> ZDO:simpleDescReq +0ms
  zigbee-shepherd:msgHdlr IND <-- ZDO:simpleDescRsp +17ms
  zigbee-shepherd:request REQ --> ZDO:simpleDescReq +1ms
  zigbee-shepherd:msgHdlr IND <-- ZDO:simpleDescRsp +12ms
  zigbee-shepherd:request REQ --> ZDO:nodeDescReq +2ms
  zigbee-shepherd:msgHdlr IND <-- ZDO:nodeDescRsp +13ms
  zigbee-shepherd:request REQ --> ZDO:activeEpReq +0ms
  zigbee-shepherd:msgHdlr IND <-- ZDO:activeEpRsp +13ms
  zigbee-shepherd:request REQ --> ZDO:mgmtPermitJoinReq +1ms
  zigbee-shepherd:msgHdlr IND <-- ZDO:mgmtPermitJoinRsp +13ms
  zigbee-shepherd:init Loading devices from database done. +15ms
  zigbee-shepherd:init zigbee-shepherd is up and ready. +0ms
  zigbee-shepherd:init Network information: +0ms
  zigbee-shepherd:init  >> State:      Coordinator +1ms
  zigbee-shepherd:init  >> Channel:    11 +0ms
  zigbee-shepherd:init  >> PanId:      0x1a62 +0ms
  zigbee-shepherd:init  >> Nwk Addr:   0 +0ms
  zigbee-shepherd:init  >> Ieee Addr:  0x00124b0014bdea7a +0ms
  zigbee-shepherd:init  >> Ext. PanId: 0xdddddddddddddddd +0ms
  zigbee-shepherd:request REQ --> ZDO:nodeDescReq +0ms
  zigbee-shepherd:request REQ --> ZDO:nodeDescReq +1ms
2018-6-12 19:38:41 INFO zigbee-shepherd started
2018-6-12 19:38:41 INFO Currently 2 devices are joined:
2018-6-12 19:38:41 INFO 0x00158d0001a4ddae (0x00158d0001a4ddae): ZNCZ02LM - Xiaomi Mi power plug ZigBee (Router)
[... 49 lines cut ...]
  zigbee-shepherd:msgHdlr IND <-- AF:incomingMsg, transId: 0 +206ms
  zigbee-shepherd:msgHdlr IND <-- AF:incomingMsg, transId: 0 +2s
  zigbee-shepherd:msgHdlr IND <-- AF:incomingMsg, transId: 0 +209ms
  zigbee-shepherd:msgHdlr IND <-- AF:incomingMsg, transId: 0 +2s
  zigbee-shepherd:msgHdlr IND <-- AF:incomingMsg, transId: 0 +209ms
  zigbee-shepherd:msgHdlr IND <-- AF:incomingMsg, transId: 0 +1s
  zigbee-shepherd:msgHdlr IND <-- AF:incomingMsg, transId: 0 +209ms
  zigbee-shepherd:request REQ --> ZDO:nodeDescReq +25s
  zigbee-shepherd:msgHdlr IND <-- ZDO:nodeDescRsp +37ms
  zigbee-shepherd:msgHdlr IND <-- AF:incomingMsg, transId: 0 +728ms
  zigbee-shepherd:msgHdlr IND <-- AF:incomingMsg, transId: 0 +206ms
  zigbee-shepherd:msgHdlr IND <-- AF:incomingMsg, transId: 0 +2s
  zigbee-shepherd:msgHdlr IND <-- AF:incomingMsg, transId: 0 +204ms
  zigbee-shepherd:msgHdlr IND <-- AF:incomingMsg, transId: 0 +347ms
  zigbee-shepherd:msgHdlr IND <-- AF:incomingMsg, transId: 0 +3s
  zigbee-shepherd:msgHdlr IND <-- AF:incomingMsg, transId: 0 +209ms
  zigbee-shepherd:msgHdlr IND <-- AF:incomingMsg, transId: 0 +3s
2018-6-12 19:41:50 INFO MQTT publish, topic: 'zigbee2mqtt/0x00158d0001a4ddae', payload: '{"state":"ON","power":0,"voltage":234,"consumption":3.56,"temperature":33}'
  zigbee-shepherd:msgHdlr IND <-- AF:incomingMsg, transId: 0 +2s
  zigbee-shepherd:msgHdlr IND <-- AF:incomingMsg, transId: 0 +207ms
^C2018-6-12 19:41:57 INFO MQTT publish, topic: 'zigbee2mqtt/bridge/state', payload: 'offline'
2018-6-12 19:41:57 INFO Disconnecting from MQTT server
  zigbee-shepherd:request REQ --> ZDO:mgmtPermitJoinReq +5s
2018-6-12 19:41:57 INFO zigbee-shepherd stopped
```

subzero79, 2018-06-12:
This is the configuration.yaml

```
homeassistant: false
permit_join: true
mqtt:
  base_topic: zigbee2mqtt
  server: 'mqtt://10.10.0.152'
  user: homeassistant
  password: EdT3zJoy7sESStp16TMd5dQxpFwGINaq
serial:
  port: /dev/ttyACM0
devices:
  '0x00158d0002108072':
    friendly_name: '0x00158d0002108072'
    retain: false
  '0x00158d0001a4ddae':
    friendly_name: '0x00158d0001a4ddae'
    retain: false
```

And the database.db

```
{"id":4,"type":"EndDevice","ieeeAddr":"0x00158d0002108072","nwkAddr":10251,"manufId":4151,"manufName":"LUMI","powerSource":"Battery","modelId":"lumi.sensor_switch","epList":[1],"status":"offline","joinTime":null,"endpoints":{"1":{"profId":260,"epId":1,"devId":260,"inClusterList":[0,3,25,65535],"outClusterList":[0,3,4,5,6,8,25],"clusters":{"genBasic":{"dir":{"value":3},"attrs":{}},"genIdentify":{"dir":{"value":3},"attrs":{}},"genGroups":{"dir":{"value":2},"attrs":{}},"genScenes":{"dir":{"value":2},"attrs":{}},"genOnOff":{"dir":{"value":2},"attrs":{}},"genLevelCtrl":{"dir":{"value":2},"attrs":{}},"genOta":{"dir":{"value":3},"attrs":{}},"manuSpecificCluster":{"dir":{"value":1},"attrs":{}}}}},"_id":"9vJTGsQcrB3nRemB"}
{"id":3,"type":"Router","ieeeAddr":"0x00158d0001a4ddae","nwkAddr":1721,"manufId":4447,"manufName":"LUMI","powerSource":"Mains (single phase)","modelId":"lumi.plug","epList":[1,2,3,100],"status":"offline","joinTime":null,"endpoints":{"1":{"profId":260,"epId":1,"devId":81,"inClusterList":[0,1,2,3,4,5,6,10,16],"outClusterList":[10,25],"clusters":{}},"2":{"profId":260,"epId":2,"devId":9,"inClusterList":[12],"outClusterList":[4,12],"clusters":{}},"3":{"profId":260,"epId":3,"devId":83,"inClusterList":[12],"outClusterList":[12],"clusters":{}},"100":{"profId":260,"epId":100,"devId":263,"inClusterList":[15],"outClusterList":[4,15],"clusters":{}}},"_id":"AUHTQqiLjMjyuNfI"}
{"id":1,"type":"Coordinator","ieeeAddr":"0x00124b0014bdea7a","nwkAddr":0,"manufId":0,"epList":[1,2,3,4,5,6],"status":"online","joinTime":1528796321,"endpoints":{"1":{"profId":260,"epId":1,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{}},"2":{"profId":257,"epId":2,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{}},"3":{"profId":261,"epId":3,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{}},"4":{"profId":263,"epId":4,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{}},"5":{"profId":264,"epId":5,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{}},"6":{"profId":265,"epId":6,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{}}},"_id":"mms2bGLGd9e8iM95"}
{"$$indexCreated":{"fieldName":"id","unique":true,"sparse":false}}
{"id":1,"type":"Coordinator","ieeeAddr":"0x00124b0014bdea7a","nwkAddr":0,"manufId":0,"epList":[1,2,3,4,5,6],"status":"online","joinTime":1528796881,"endpoints":{"1":{"profId":260,"epId":1,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{}},"2":{"profId":257,"epId":2,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{}},"3":{"profId":261,"epId":3,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{}},"4":{"profId":263,"epId":4,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{}},"5":{"profId":264,"epId":5,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{}},"6":{"profId":265,"epId":6,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{}}},"_id":"mms2bGLGd9e8iM95"}
{"id":4,"type":"EndDevice","ieeeAddr":"0x00158d0002108072","nwkAddr":10251,"manufId":4151,"manufName":"LUMI","powerSource":"Battery","modelId":"lumi.sensor_switch","epList":[1],"status":"offline","joinTime":null,"endpoints":{"1":{"profId":260,"epId":1,"devId":260,"inClusterList":[0,3,25,65535],"outClusterList":[0,3,4,5,6,8,25],"clusters":{"genBasic":{"dir":{"value":3},"attrs":{}},"genIdentify":{"dir":{"value":3},"attrs":{}},"genGroups":{"dir":{"value":2},"attrs":{}},"genScenes":{"dir":{"value":2},"attrs":{}},"genOnOff":{"dir":{"value":2},"attrs":{}},"genLevelCtrl":{"dir":{"value":2},"attrs":{}},"genOta":{"dir":{"value":3},"attrs":{}},"manuSpecificCluster":{"dir":{"value":1},"attrs":{}}}}},"_id":"9vJTGsQcrB3nRemB"}
{"id":3,"type":"Router","ieeeAddr":"0x00158d0001a4ddae","nwkAddr":1721,"manufId":4447,"manufName":"LUMI","powerSource":"Mains (single phase)","modelId":"lumi.plug","epList":[1,2,3,100],"status":"offline","joinTime":null,"endpoints":{"1":{"profId":260,"epId":1,"devId":81,"inClusterList":[0,1,2,3,4,5,6,10,16],"outClusterList":[10,25],"clusters":{}},"2":{"profId":260,"epId":2,"devId":9,"inClusterList":[12],"outClusterList":[4,12],"clusters":{}},"3":{"profId":260,"epId":3,"devId":83,"inClusterList":[12],"outClusterList":[12],"clusters":{}},"100":{"profId":260,"epId":100,"devId":263,"inClusterList":[15],"outClusterList":[4,15],"clusters":{}}},"_id":"AUHTQqiLjMjyuNfI"}
```

Koenkk, 2018-06-12:
Is that the `datbase.db` after the reboot (when devices are lost)?

subzero79, 2018-06-12:
Yes, correct that's the database after device is lost. Only the button is lost atm

subzero79, 2018-06-12:
I re-added the button, seems working now across reboots.....but i think i know the problem presents when i disconnect the xiaomi plug, i cannot see any response for the button in topic, i plug again and then the button comes alive again.

ciotlosm, 2018-06-12:
Is your button far away? Maybe you don't have signal all the way to your CC2531

subzero79, 2018-06-12:
Not at all maybe 4 meters, with the xiaomi plug disconnected i go next to the cc2531 dongle and no response, i pull the plug and response comes back again.

Now I cleaned the db and configuration file, this time I add just the single button first, no plug.
I get perfect response from the same position. Reboot, is still working. 
I add the xiaomi plug, joins network, button still responding. Reboot still working.

Gonna leave it like that and test in a few more hours.

Any ideas why duplicate entries in db? for the xiaomi plug i keep pressing on/off the reset button as per wiki until it joins.

ciotlosm, 2018-06-12:
@subzero79 duplicates in DB are expected. Unsure how the item structure is maintained by zigbee-shepherd but it's not something to worry about. Ideally we should never touch the database.db directly, only through zigbee-shepherd API.

ciotlosm, 2018-06-13:
Closing. Feel free to reopen if it reoccurs.

closed 2018-06-13 by ciotlosm

subzero79, 2018-06-13:
Seems stable for now after 12 hours came back home, both plug and button still responding.