Koenkk/zigbee2mqtt #116: Crashes with CC2531 router paired

kimonm, 2018-06-13:
zigbee2mqtt is working fine, except if I pair a CC2531 flashed as a router (using CC2531-router.hex code). It pairs fine, however when I restart zigbee2mqtt, I get the following output and npm-debug.log. Thanks in advance for any help!

6/12/2018, 9:38:28 PM INFO Starting zigbee-shepherd
6/12/2018, 9:38:29 PM INFO zigbee-shepherd started
6/12/2018, 9:38:29 PM INFO Currently 6 devices are joined:
6/12/2018, 9:38:29 PM INFO xiaomi_sw1 (0x00158d0001f43047): WXKG01LM - Xiaomi MiJia wireless switch (EndDevice)
6/12/2018, 9:38:29 PM INFO office_lamp (0xb0ce18140310b553): E11-G13 - Sengled Sengled Element Classic bulb (EndDevice)
6/12/2018, 9:38:29 PM INFO xiaomi_cube (0x00158d00024a477d): MFKZQ01LM - Xiaomi Mi smart home cube (EndDevice)
6/12/2018, 9:38:29 PM INFO ge_plug1 (0x0022a300001af090): 45852 - GE GE wall switch 45852 (Router)
6/12/2018, 9:38:29 PM INFO xiaomi_sw2 (0x00158d0002133e41): WXKG11LM - Xiaomi Aqara wireless switch (EndDevice)
6/12/2018, 9:38:29 PM INFO cc2531_router1 (0x00124b0012013c50): CC2530.ROUTER - Texas Instruments [CC2530 router](http://ptvo.info/cc2530-based-zigbee-coordinator-and-router-112/) (Router)
6/12/2018, 9:38:29 PM INFO Connecting to MQTT server at mqtt://localhost
/home/kimon/Services/zigbee2mqtt/lib/zigbee.js:78
                this.shepherd.find(device.ieeeAddr, 1).getDevice().update({
                                                      ^

TypeError: Cannot read property 'getDevice' of undefined
    at devices.forEach (/home/kimon/Services/zigbee2mqtt/lib/zigbee.js:78:55)
    at Array.forEach (<anonymous>)
    at Zigbee.handleReady (/home/kimon/Services/zigbee2mqtt/lib/zigbee.js:76:17)
    at emitNone (events.js:106:13)
    at ZShepherd.emit (events.js:208:7)
    at Immediate._onImmediate (/home/kimon/Services/zigbee2mqtt/node_modules/zigbee-shepherd/lib/shepherd.js:89:18)
    at runCallback (timers.js:794:20)
    at tryOnImmediate (timers.js:752:5)
    at processImmediate [as _immediateCallback] (timers.js:729:5)
	
npm ERR! Linux 4.15.0-23-generic
npm ERR! argv "/usr/bin/node" "/usr/bin/npm" "start"
npm ERR! node v8.10.0
npm ERR! npm  v3.5.2
npm ERR! code ELIFECYCLE
npm ERR! zigbee2mqtt@0.1.0 start: `node index.js`
npm ERR! Exit status 1
npm ERR! 
npm ERR! Failed at the zigbee2mqtt@0.1.0 start script 'node index.js'.

**npm-debug.log**
0 info it worked if it ends with ok
1 verbose cli [ '/usr/bin/node', '/usr/bin/npm', 'start' ]
2 info using npm@3.5.2
3 info using node@v8.10.0
4 verbose run-script [ 'prestart', 'start', 'poststart' ]
5 info lifecycle zigbee2mqtt@0.1.0~prestart: zigbee2mqtt@0.1.0
6 silly lifecycle zigbee2mqtt@0.1.0~prestart: no script for prestart, continuing
7 info lifecycle zigbee2mqtt@0.1.0~start: zigbee2mqtt@0.1.0
8 verbose lifecycle zigbee2mqtt@0.1.0~start: unsafe-perm in lifecycle true
9 verbose lifecycle zigbee2mqtt@0.1.0~start: PATH: /usr/share/npm/bin/node-gyp-bin:/home/kimon/Services/zigbee2mqtt/node_modules/.bin:/home/kimon/bin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/usr/games:/usr/local/games:/snap/bin:/home/kimon/Go/bin
10 verbose lifecycle zigbee2mqtt@0.1.0~start: CWD: /home/kimon/Services/zigbee2mqtt
11 silly lifecycle zigbee2mqtt@0.1.0~start: Args: [ '-c', 'node index.js' ]
12 silly lifecycle zigbee2mqtt@0.1.0~start: Returned: code: 1  signal: null
13 info lifecycle zigbee2mqtt@0.1.0~start: Failed to exec start script
14 verbose stack Error: zigbee2mqtt@0.1.0 start: `node index.js`
14 verbose stack Exit status 1
14 verbose stack     at EventEmitter.<anonymous> (/usr/share/npm/lib/utils/lifecycle.js:232:16)
14 verbose stack     at emitTwo (events.js:126:13)
14 verbose stack     at EventEmitter.emit (events.js:214:7)
14 verbose stack     at ChildProcess.<anonymous> (/usr/share/npm/lib/utils/spawn.js:24:14)
14 verbose stack     at emitTwo (events.js:126:13)
14 verbose stack     at ChildProcess.emit (events.js:214:7)
14 verbose stack     at maybeClose (internal/child_process.js:925:16)
14 verbose stack     at Process.ChildProcess._handle.onexit (internal/child_process.js:209:5)
15 verbose pkgid zigbee2mqtt@0.1.0
16 verbose cwd /home/kimon/Services/zigbee2mqtt
17 error Linux 4.15.0-23-generic
18 error argv "/usr/bin/node" "/usr/bin/npm" "start"
19 error node v8.10.0
20 error npm  v3.5.2
21 error code ELIFECYCLE
22 error zigbee2mqtt@0.1.0 start: `node index.js`
22 error Exit status 1
23 error Failed at the zigbee2mqtt@0.1.0 start script 'node index.js'.
23 error Make sure you have the latest version of node.js and npm installed.
23 error If you do, this is most likely a problem with the zigbee2mqtt package,
23 error not with npm itself.
23 error Tell the author that this fails on your system:
23 error     node index.js
23 error You can get information on how to open an issue for this project with:
23 error     npm bugs zigbee2mqtt
23 error Or if that isn't available, you can get their info via:
23 error     npm owner ls zigbee2mqtt
23 error There is likely additional logging output above.
24 verbose exit [ 1, true ]

kirovilya, 2018-06-13:
This is because the cc2531-router pretended to be "lumi" and we sent a message to it at the start (I told the  firmware author about it). But this cc2531-router does not have 1 endpoint - so it drops.

ptvoinfo, 2018-06-13:
So, is it necessary to change a model ID and remove "lumi" from ID?

kirovilya, 2018-06-13:
@ptvoinfo Yes, I think so. And look at current endpoint on interview - (https://github.com/Koenkk/zigbee2mqtt/issues/52#issuecomment-392872524).
By the way, we checked the work of the cc2531 as a router - it works!

ptvoinfo, 2018-06-13:
What is wrong in the interview process? The endpoint number can be any. The sample application from the ZStack SDK used this endpoint number. I didn't change it.

Also, the reported problem is not directly related with "lumi" prefix.

kirovilya, 2018-06-13:
@ptvoinfo yes, endpoint number does not related to the problem with "lumi" :)
But since you will change the firmware, then fix it ... :)
Maybe I'm wrong, but it seems to me in the interview the enum of the endpoint in the "current" and the total count of endpoints in "total" is indicated. And in your firmware endpoint number is "8" (I expected 1).
Can, if you fix the firmware with 1, then you could leave "lumi" :))

Koenkk, 2018-06-13:
Aside from any firmware changes, zigbee2mqtt shouldn't crash on this. I will add an `if device` statement here: https://github.com/Koenkk/zigbee2mqtt/blob/master/lib/zigbee.js#L79

ptvoinfo, 2018-06-13:
@Koenkk yes, correct.
@kirovilya if I'll change the firmware (exclude the "lumi" prefix) then we need to change converters.

ptvoinfo, 2018-06-13:
@kirovilya I've checked all ZStack examples and they have only one endpoint with the number "8". So, I think it shouldn't be a problem. 

The reported problem is related with something else.

kirovilya, 2018-06-13:
@ptvoinfo Hmm. Stop! Not need remove "lumi" from modelID ... Main problem for this issue was "manufId = 4447" in your router firmware (not "lumi.router")!

ptvoinfo, 2018-06-13:
@kirovilya why? zigbee2mqtt distinguishes device by modelID. Why does manufId matter?

kirovilya, 2018-06-13:
@ptvoinfo manufId is used in the expression https://github.com/Koenkk/zigbee2mqtt/blob/master/lib/zigbee.js#L78 and finds that it is xiaomi and send them message! :)

ptvoinfo, 2018-06-13:
@kirovilya, No problems here. .update() sends the "node description request" and the router answers to this message.

ciotlosm, 2018-06-13:
@kimonm can you post a link to the firmware you used?

kimonm, 2018-06-13:
I downloaded the firmware: cc2531_1.2.2a.44539_firmware.zip 
from the supported sniffers wiki page: https://github.com/Koenkk/zigbee2mqtt/wiki/Supported-sniffer-devices
I flashed CC2531-router.hex

Koenkk, 2018-06-14:
This issue should be fixed now.

kimonm, 2018-06-14:
Unfortunately, not for me. I just did a clean install of zigbee2mqtt & tried to pair the CC2531 (same router firmware). The output is:

6/14/2018, 10:26:35 AM INFO Starting zigbee-shepherd
6/14/2018, 10:26:35 AM INFO zigbee-shepherd started
6/14/2018, 10:26:35 AM INFO Currently 0 devices are joined:
6/14/2018, 10:26:35 AM WARN `permit_join` set to  `true` in configuration.yaml.
6/14/2018, 10:26:35 AM WARN Allowing new devices to join.
6/14/2018, 10:26:35 AM WARN Set `permit_join` to `false` once you joined all devices.
6/14/2018, 10:26:35 AM INFO Zigbee: allowing new devices to join.
6/14/2018, 10:26:35 AM INFO Connecting to MQTT server at mqtt://localhost
6/14/2018, 10:26:35 AM INFO zigbee-shepherd ready
6/14/2018, 10:26:36 AM INFO Connected to MQTT server
6/14/2018, 10:26:36 AM INFO MQTT publish, topic: 'zigbee2mqtt/bridge/state', payload: 'online'
spinlock: false []
/home/kimon/Services/test/node_modules/zcl-packet/lib/foundation.js:683
        ru[dataType](name)(this);
                    ^

TypeError: ru[dataType] is not a function
    at Dissolve.<anonymous> (/home/kimon/Services/test/node_modules/zcl-packet/lib/foundation.js:683:21)
    at Dissolve._transform (/home/kimon/Services/test/node_modules/dissolve/index.js:80:16)
    at Dissolve.Transform._read (/home/kimon/Services/test/node_modules/dissolve/node_modules/readable-stream/lib/_stream_transform.js:184:10)
    at Dissolve.Transform._write (/home/kimon/Services/test/node_modules/dissolve/node_modules/readable-stream/lib/_stream_transform.js:172:12)
    at doWrite (/home/kimon/Services/test/node_modules/dissolve/node_modules/readable-stream/lib/_stream_writable.js:279:12)
    at writeOrBuffer (/home/kimon/Services/test/node_modules/dissolve/node_modules/readable-stream/lib/_stream_writable.js:266:5)
    at Dissolve.Writable.write (/home/kimon/Services/test/node_modules/dissolve/node_modules/readable-stream/lib/_stream_writable.js:211:11)
    at Dissolve.Writable.end (/home/kimon/Services/test/node_modules/dissolve/node_modules/readable-stream/lib/_stream_writable.js:426:10)
    at FoundPayload._getObj (/home/kimon/Services/test/node_modules/zigbee-shepherd/node_modules/zcl-packet/lib/foundation.js:242:12)
    at getObjCB (/home/kimon/Services/test/node_modules/zigbee-shepherd/node_modules/zcl-packet/lib/foundation.js:51:22)

Koenkk, 2018-06-14:
This is a different error then before, can you do a 

```
rm -rf node_modules
npm install
```

kimonm, 2018-06-14:
No change, still crashes. Below is stdout, npm-debug.log, and then full debug output.

> zigbee2mqtt@0.1.0 start /home/kimon/Services/test
> node index.js

6/14/2018, 11:48:44 AM INFO Starting zigbee-shepherd
6/14/2018, 11:48:44 AM INFO zigbee-shepherd started
6/14/2018, 11:48:44 AM INFO Currently 0 devices are joined:
6/14/2018, 11:48:44 AM WARN `permit_join` set to  `true` in configuration.yaml.
6/14/2018, 11:48:44 AM WARN Allowing new devices to join.
6/14/2018, 11:48:44 AM WARN Set `permit_join` to `false` once you joined all devices.
6/14/2018, 11:48:44 AM INFO Zigbee: allowing new devices to join.
6/14/2018, 11:48:44 AM INFO Connecting to MQTT server at mqtt://localhost
6/14/2018, 11:48:45 AM INFO zigbee-shepherd ready
6/14/2018, 11:48:45 AM INFO Connected to MQTT server
6/14/2018, 11:48:45 AM INFO MQTT publish, topic: 'zigbee2mqtt/bridge/state', payload: 'online'
spinlock: false []

**npm-debug.log**

0 info it worked if it ends with ok
1 verbose cli [ '/usr/bin/node', '/usr/bin/npm', 'start' ]
2 info using npm@3.5.2
3 info using node@v8.10.0
4 verbose run-script [ 'prestart', 'start', 'poststart' ]
5 info lifecycle zigbee2mqtt@0.1.0~prestart: zigbee2mqtt@0.1.0
6 silly lifecycle zigbee2mqtt@0.1.0~prestart: no script for prestart, continuing
7 info lifecycle zigbee2mqtt@0.1.0~start: zigbee2mqtt@0.1.0
8 verbose lifecycle zigbee2mqtt@0.1.0~start: unsafe-perm in lifecycle true
9 verbose lifecycle zigbee2mqtt@0.1.0~start: PATH: /usr/share/npm/bin/node-gyp-bin:/home/kimon/Services/test/node_modules/.bin:/home/kimon/bin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/usr/games:/usr/local/games:/snap/bin:/home/kimon/Go/bin
10 verbose lifecycle zigbee2mqtt@0.1.0~start: CWD: /home/kimon/Services/test
11 silly lifecycle zigbee2mqtt@0.1.0~start: Args: [ '-c', 'node index.js' ]
12 silly lifecycle zigbee2mqtt@0.1.0~start: Returned: code: 1  signal: null
13 info lifecycle zigbee2mqtt@0.1.0~start: Failed to exec start script
14 verbose stack Error: zigbee2mqtt@0.1.0 start: `node index.js`
14 verbose stack Exit status 1
14 verbose stack     at EventEmitter.<anonymous> (/usr/share/npm/lib/utils/lifecycle.js:232:16)
14 verbose stack     at emitTwo (events.js:126:13)
14 verbose stack     at EventEmitter.emit (events.js:214:7)
14 verbose stack     at ChildProcess.<anonymous> (/usr/share/npm/lib/utils/spawn.js:24:14)
14 verbose stack     at emitTwo (events.js:126:13)
14 verbose stack     at ChildProcess.emit (events.js:214:7)
14 verbose stack     at maybeClose (internal/child_process.js:925:16)
14 verbose stack     at Process.ChildProcess._handle.onexit (internal/child_process.js:209:5)
15 verbose pkgid zigbee2mqtt@0.1.0
16 verbose cwd /home/kimon/Services/test
17 error Linux 4.15.0-23-generic
18 error argv "/usr/bin/node" "/usr/bin/npm" "start"
19 error node v8.10.0
20 error npm  v3.5.2
21 error code ELIFECYCLE
22 error zigbee2mqtt@0.1.0 start: `node index.js`
22 error Exit status 1
23 error Failed at the zigbee2mqtt@0.1.0 start script 'node index.js'.
23 error Make sure you have the latest version of node.js and npm installed.
23 error If you do, this is most likely a problem with the zigbee2mqtt package,
23 error not with npm itself.
23 error Tell the author that this fails on your system:
23 error     node index.js
23 error You can get information on how to open an issue for this project with:
23 error     npm bugs zigbee2mqtt
23 error Or if that isn't available, you can get their info via:
23 error     npm owner ls zigbee2mqtt
23 error There is likely additional logging output above.
24 verbose exit [ 1, true ]

**Debug output**
Thu, 14 Jun 2018 18:48:44 GMT zigbee-shepherd:init zigbee-shepherd booting...
Thu, 14 Jun 2018 18:48:44 GMT cc-znp The serialport /dev/ttyACM0 is opened.
Thu, 14 Jun 2018 18:48:44 GMT zigbee-shepherd:request REQ --> SYS:osalNvRead
[... 196 lines cut ...]
    at FoundPayload._getObj (/home/kimon/Services/test/node_modules/zigbee-shepherd/node_modules/zcl-packet/lib/foundation.js:242:12)
    at getObjCB (/home/kimon/Services/test/node_modules/zigbee-shepherd/node_modules/zcl-packet/lib/foundation.js:51:22)

npm ERR! Linux 4.15.0-23-generic
npm ERR! argv "/usr/bin/node" "/usr/bin/npm" "start"
npm ERR! node v8.10.0
npm ERR! npm  v3.5.2
npm ERR! code ELIFECYCLE
npm ERR! zigbee2mqtt@0.1.0 start: `node index.js`
npm ERR! Exit status 1
npm ERR! 
npm ERR! Failed at the zigbee2mqtt@0.1.0 start script 'node index.js'.
npm ERR! Make sure you have the latest version of node.js and npm installed.
npm ERR! If you do, this is most likely a problem with the zigbee2mqtt package,
npm ERR! not with npm itself.
npm ERR! Tell the author that this fails on your system:
npm ERR!     node index.js
npm ERR! You can get information on how to open an issue for this project with:
npm ERR!     npm bugs zigbee2mqtt
npm ERR! Or if that isn't available, you can get their info via:
npm ERR!     npm owner ls zigbee2mqtt
npm ERR! There is likely additional logging output above.

npm ERR! Please include the following file with any support request:
npm ERR!     /home/kimon/Services/test/npm-debug.log

Koenkk, 2018-06-14:
Can you try clearing your database.db and removing all `devices` from `configuration.yaml`?

kimonm, 2018-06-14:
For the new error, all output is from clean install with no devices in the configuration.yaml and no database file.

ciotlosm, 2018-06-14:
@kimonm if you started fresh, can you please let us know the steps?

* which device did you pair first?
* when did you add the router?
* when did it crash exactly?

I see in your first log you interrupted at `spinlock: false []`? Why?

kimonm, 2018-06-14:
Started zigbee2mqtt with the debug option and nothing paired, no databasefile
First device I attempted to pair is the CC2531.
It crashes at/after the attempted pairing - the router changes from a rapid red flash to a once every 4 second flash, so it thinks it's paired.
I didn't interrupt. It crashed out on it's own.

ptvoinfo, 2018-06-15:
@ciotlosm I'm using 3 CC2530 based routers and one CC2531 based. No problems at all. 
@kimonm Is it a standard CC2531 USB stick without any RF front-end?

Koenkk, 2018-06-15:
@kimonm what if you don't pair the CC2531 router? Does it crash then?

kimonm, 2018-06-15:
I'm out of town through the weekend, so I can't run experiments right now. 
The router is the "typical" CC2531 w/o, e.g., any external antenna.
Starting zigbee2mqtt without anything paired is fine. It doesn't crash until I attempt to pair the router.

@coitlosm - which firmware do you use? I can try re-flashing the device.

ciotlosm, 2018-06-18:
@kimonm I don't use a router yet. I don't have enough CC2531, but in a week time I will have more to test it out.

ciotlosm, 2018-06-19:
@kimonm Can you please let us know exactly which version of NPM and NODE you are running now?

The reports every minute should be normal for an active device, but @Koenkk might be able to explain better what `state: false` means.

kimonm, 2018-06-19:
kimon@Pebbles:~$ node --version
v8.11.3
kimon@Pebbles:~$ npm --version
5.6.0

Thanks all for your help; it's great software.

ciotlosm, 2018-06-19:
Just so it's documented on the issue, you moved from:
```
node v8.10.0
npm  v3.5.2
```
to 

```
node v8.11.3
npm v5.6.0
```

It's unfortunate we haven't noticed your versions, as there is documentation on the required version:
```
node --version  # Should output v8.X.X (at time of writing v8.11.2)
npm --version  # Should output > 5.X.X (at time of writing 5.6.0)
```

I will close the issue.

closed 2018-06-19 by ciotlosm