Koenkk/zigbee2mqtt #117: Aqara wall switch neutral line

Disposaber, 2018-06-13:
### Aqara wall switch neutral line - Single (QBKG11LM):
Log: 
`2018-06-13T12:50:44.338Z - warn: Device with modelID 'lumi.ctrl_ln1.aq1' is not supported.
2018-06-13T12:50:44.341Z - warn: Please see: https://github.com/Koenkk/zigbee2mqtt/wiki/How-to-support-new-devices`

Database:
`{"id":3,"type":"Router","ieeeAddr":"0x00158d00022XXXXX","nwkAddr":46XXX,"manufId":4447,"manufName":"LUMI","powerSource":"Mains (single phase)","modelId":"lumi.ctrl_ln1.aq1","epList":[1,2,3,5],"status":"offline","joinTime":null,"endpoints":{"1":{"profId":260,"epId":1,"devId":81,"inClusterList":[0,1,2,3,4,5,6,10,16],"outClusterList":[10,25],"clusters":{"genBasic":{"dir":{"value":1},"attrs":{"zclVersion":1,"appVersion":31,"stackVersion":2,"hwVersion":18,"manufacturerName":"LUMI","modelId":"lumi.ctrl_ln1.aq1","dateCode":"10-11-2017","powerSource":1}},"genPowerCfg":{"dir":{"value":1},"attrs":{"mainsVoltage":2145}},"genDeviceTempCfg":{"dir":{"value":1},"attrs":{"currentTemperature":25}},"genIdentify":{"dir":{"value":1},"attrs":{"identifyTime":0}},"genGroups":{"dir":{"value":1},"attrs":{"nameSupport":0}},"genScenes":{"dir":{"value":1},"attrs":{"count":0,"currentScene":0,"currentGroup":0,"sceneValid":0,"nameSupport":0,"lastCfgBy":"0xffffffffffffffff"}},"genOnOff":{"dir":{"value":1},"attrs":{"onOff":0}},"genTime":{"dir":{"value":3},"attrs":{}},"genBinaryOutput":{"dir":{"value":1},"attrs":{"outOfService":0,"presentValue":0,"statusFlags":0}},"genOta":{"dir":{"value":2},"attrs":{}}}},"2":{"profId":260,"epId":2,"devId":9,"inClusterList":[12],"outClusterList":[4,12],"clusters":{"genGroups":{"dir":{"value":2},"attrs":{}},"genAnalogInput":{"dir":{"value":3},"attrs":{"261":0,"262":0,"outOfService":0,"presentValue":0,"statusFlags":0,"applicationType":589824}}}},"3":{"profId":260,"epId":3,"devId":83,"inClusterList":[12],"outClusterList":[12],"clusters":{"genAnalogInput":{"dir":{"value":3},"attrs":{"261":0,"262":0,"outOfService":0,"presentValue":0.09716890007257462,"statusFlags":0,"applicationType":720896}}}},"5":{"profId":260,"epId":5,"devId":0,"inClusterList":[16,18],"outClusterList":[],"clusters":{"genBinaryOutput":{"dir":{"value":1},"attrs":{"outOfService":0,"presentValue":0,"statusFlags":0}},"genMultistateInput":{"dir":{"value":1},"attrs":{"numberOfStates":6,"outOfService":0,"presentValue":0,"statusFlags":0}}}}},"_id":"XAHPAlQ64SXXXXXX"}`

### Aqara wall switch neutral line - Double (QBKG12LM):
Log:
`2018-06-13T15:50:35.677Z - warn: Device with modelID 'lumi.ctrl_ln2.aq1' is not supported.
2018-06-13T15:50:35.681Z - warn: Please see: https://github.com/Koenkk/zigbee2mqtt/wiki/How-to-support-new-devices`

Database:
`{"id":4,"type":"Router","ieeeAddr":"0x00158d00022aXXXX","nwkAddr":49XXX,"manufId":4447,"manufName":"LUMI","powerSource":"Mains (single phase)","modelId":"lumi.ctrl_ln2.aq1","epList":[1,2,3,4,5,6,7],"status":"online","joinTime":152890XXXX,"endpoints":{"1":{"profId":260,"epId":1,"devId":81,"inClusterList":[0,1,2,3,4,5,6,10,16],"outClusterList":[10,25],"clusters":{"genBasic":{"dir":{"value":1},"attrs":{"zclVersion":1,"appVersion":31,"stackVersion":2,"hwVersion":18,"manufacturerName":"LUMI","modelId":"lumi.ctrl_ln2.aq1","dateCode":"10-12-2017","powerSource":1}},"genPowerCfg":{"dir":{"value":1},"attrs":{"mainsVoltage":2210}},"genDeviceTempCfg":{"dir":{"value":1},"attrs":{"currentTemperature":31}},"genIdentify":{"dir":{"value":1},"attrs":{"identifyTime":0}},"genGroups":{"dir":{"value":1},"attrs":{"nameSupport":128}},"genScenes":{"dir":{"value":1},"attrs":{"count":0,"currentScene":0,"currentGroup":0,"sceneValid":0,"nameSupport":0,"lastCfgBy":"0xffffffffffffffff"}},"genOnOff":{"dir":{"value":1},"attrs":{"onOff":0}},"genTime":{"dir":{"value":3},"attrs":{}},"genBinaryOutput":{"dir":{"value":1},"attrs":{"outOfService":0,"presentValue":0,"statusFlags":0}},"genOta":{"dir":{"value":2},"attrs":{}}}},"2":{"profId":260,"epId":2,"devId":81,"inClusterList":[6,16],"outClusterList":[],"clusters":{"genOnOff":{"dir":{"value":1},"attrs":{"onOff":0}},"genBinaryOutput":{"dir":{"value":1},"attrs":{"outOfService":0,"presentValue":0,"statusFlags":0}}}},"3":{"profId":260,"epId":3,"devId":9,"inClusterList":[12],"outClusterList":[4,12],"clusters":{"genGroups":{"dir":{"value":2},"attrs":{}},"genAnalogInput":{"dir":{"value":3},"attrs":{"261":0,"262":0,"outOfService":0,"presentValue":0,"statusFlags":0,"applicationType":589824}}}},"4":{"profId":260,"epId":4,"devId":83,"inClusterList":[12],"outClusterList":[12],"clusters":{"genAnalogInput":{"dir":{"value":3},"attrs":{"261":0,"262":0,"outOfService":0,"presentValue":1.5031226873397827,"statusFlags":0,"applicationType":720896}}}},"5":{"profId":260,"epId":5,"devId":0,"inClusterList":[16,18],"outClusterList":[],"clusters":{"genBinaryOutput":{"dir":{"value":1},"attrs":{"outOfService":0,"presentValue":0,"statusFlags":0}},"genMultistateInput":{"dir":{"value":1},"attrs":{"numberOfStates":6,"outOfService":0,"presentValue":0,"statusFlags":0}}}},"6":{"profId":260,"epId":6,"devId":0,"inClusterList":[16,18],"outClusterList":[],"clusters":{"genBinaryOutput":{"dir":{"value":1},"attrs":{"outOfService":0,"presentValue":0,"statusFlags":0}},"genMultistateInput":{"dir":{"value":1},"attrs":{"numberOfStates":6,"outOfService":0,"presentValue":0,"statusFlags":0}}}},"7":{"profId":260,"epId":7,"devId":0,"inClusterList":[16,18],"outClusterList":[],"clusters":{"genBinaryOutput":{"dir":{"value":1},"attrs":{"outOfService":0,"presentValue":0,"statusFlags":0}},"genMultistateInput":{"dir":{"value":1},"attrs":{"numberOfStates":6,"outOfService":0,"presentValue":0,"statusFlags":0}}}}},"_id":"Zg8TBMujEwXXXXXX"}`

Koenkk, 2018-06-13:
Can you try adding the following to your `node_modules/zigbee-shepherd-converters/devices.js`?

```js
    {
        zigbeeModel: ['lumi.ctrl_ln1.aq1],
        model: 'QBKG11LM',
        vendor: 'Xiaomi',
        description: 'Aqara single key wired wall switch',
        supports: 'on/off',
        fromZigbee: [fz.QBKG04LM_state, fz.ignore_onoff_change],
        toZigbee: [tz.onoff],
        ep: {'': 2},
    },
    {
        zigbeeModel: ['lumi.ctrl_ln2.aq1'],
        model: 'QBKG12LM',
        vendor: 'Xiaomi',
        description: 'Aqara double key wired wall switch',
        supports: 'l1 and l2 on/off',
        fromZigbee: [fz.QBKG03LM_state, fz.QBKG03LM_buttons],
        toZigbee: [tz.onoff],
        ep: {'left': 2, 'right': 3},
    },
```

Koenkk, 2018-06-14:
Whats the output when running with `DEBUG=zigbee2mqtt:controller npm start` (then we can see the message contents)

Disposaber, 2018-06-14:
This is the output

`2018-6-14 11:37:25 INFO Starting zigbee-shepherd
2018-6-14 11:37:26 INFO zigbee-shepherd started
2018-6-14 11:37:26 INFO Currently 3 devices are joined:
2018-6-14 11:37:26 INFO motion_hall (0x00158d0001e5fec8): RTCGQ11LM - Xiaomi Aqara human body movement and illuminance sensor (EndDevice)
2018-6-14 11:37:26 INFO wall_switch_bathroom (0x00158d00022a8278): QBKG12LM - Xiaomi Aqara double key wired wall switch (Router)
2018-6-14 11:37:26 INFO wall_switch_hall (0x00158d0002289982): QBKG11LM - Xiaomi Aqara single key wired wall switch (Router)
2018-6-14 11:37:26 WARN `permit_join` set to  `true` in configuration.yaml.
2018-6-14 11:37:26 WARN Allowing new devices to join.
2018-6-14 11:37:26 WARN Set `permit_join` to `false` once you joined all devices.
2018-6-14 11:37:26 INFO Zigbee: allowing new devices to join.
2018-6-14 11:37:26 INFO Connecting to MQTT server at mqtt://127.0.0.1
2018-6-14 11:37:27 INFO zigbee-shepherd ready
2018-6-14 11:37:27 INFO Connected to MQTT server
2018-6-14 11:37:27 INFO MQTT publish, topic: 'zigbee2mqtt/bridge/state', payload: 'online'
  zigbee2mqtt:controller Recieved zigbee message with data { cid: 'genBasic',
  data: 
   { '65281': 
      { '3': 27,
        '5': 14,
        '7': '0x0000000000000000',
        '8': 4383,
        '9': 512,
        '100': 0,
        '149': 0.10862531512975693,
        '152': 0,
        '154': 16 } } } +0ms
2018-6-14 11:38:26 WARN No converter available for 'QBKG11LM' with cid 'genBasic' and type 'attReport'
2018-6-14 11:38:26 WARN Please see: https://github.com/Koenkk/zigbee2mqtt/wiki/How-to-support-new-devices.
  zigbee2mqtt:controller Recieved zigbee message with data { cid: 'genBasic',
  data: 
   { '65281': 
      [ <3 empty items>,
        27,
        <1 empty item>,
        14,
        <1 empty item>,
        '0x0000000000000000',
        4383,
        512,
        <90 empty items>,
        0,
        <48 empty items>,
        0.10862531512975693,
        <2 empty items>,
        0,
        <1 empty item>,
        16 ] } } +33ms
2018-6-14 11:38:26 WARN No converter available for 'QBKG11LM' with cid 'genBasic' and type 'devChange'
2018-6-14 11:38:26 WARN Please see: https://github.com/Koenkk/zigbee2mqtt/wiki/How-to-support-new-devices.
  zigbee2mqtt:controller Recieved zigbee message with data { cid: 'genOnOff', data: { '61440': 65555200, onOff: 1 } } +24s
2018-6-14 11:38:50 INFO MQTT publish, topic: 'zigbee2mqtt/wall_switch_hall', payload: '{"state":"ON"}'
  zigbee2mqtt:controller Recieved zigbee message with data { cid: 'genOnOff', data: { '61440': 65555200, onOff: 1 } } +17ms
  zigbee2mqtt:controller Recieved zigbee message with data { cid: 'genMultistateInput', data: { presentValue: 1 } } +328ms
2018-6-14 11:38:51 WARN No converter available for 'QBKG11LM' with cid 'genMultistateInput' and type 'attReport'
2018-6-14 11:38:51 WARN Please see: https://github.com/Koenkk/zigbee2mqtt/wiki/How-to-support-new-devices.
  zigbee2mqtt:controller Recieved zigbee message with data { cid: 'genMultistateInput', data: { presentValue: 1 } } +15ms
2018-6-14 11:38:51 WARN No converter available for 'QBKG11LM' with cid 'genMultistateInput' and type 'devChange'
2018-6-14 11:38:51 WARN Please see: https://github.com/Koenkk/zigbee2mqtt/wiki/How-to-support-new-devices.
  zigbee2mqtt:controller Recieved zigbee message with data { cid: 'genOnOff', data: { '61440': 65555200, onOff: 0 } } +1s
2018-6-14 11:38:52 INFO MQTT publish, topic: 'zigbee2mqtt/wall_switch_hall', payload: '{"state":"OFF"}'
  zigbee2mqtt:controller Recieved zigbee message with data { cid: 'genOnOff', data: { onOff: 0 } } +15ms
  zigbee2mqtt:controller Recieved zigbee message with data { cid: 'genMultistateInput', data: { presentValue: 1 } } +331ms
2018-6-14 11:38:52 WARN No converter available for 'QBKG11LM' with cid 'genMultistateInput' and type 'attReport'
2018-6-14 11:38:52 WARN Please see: https://github.com/Koenkk/zigbee2mqtt/wiki/How-to-support-new-devices.
  zigbee2mqtt:controller Recieved zigbee message with data { cid: 'genOnOff', data: { '61440': 63009792, onOff: 1 } } +50s
2018-6-14 11:39:42 INFO MQTT publish, topic: 'zigbee2mqtt/wall_switch_bathroom', payload: '{"state_undefined":"ON"}'
  zigbee2mqtt:controller Recieved zigbee message with data { cid: 'genOnOff', data: { '61440': 63009792, onOff: 1 } } +14ms
  zigbee2mqtt:controller Recieved zigbee message with data { cid: 'genMultistateInput', data: { presentValue: 1 } } +345ms
[... 19 lines cut ...]
  data: { presentValue: 10.600000381469727 } } +15ms
2018-6-14 11:39:46 WARN No converter available for 'QBKG12LM' with cid 'genAnalogInput' and type 'devChange'
2018-6-14 11:39:46 WARN Please see: https://github.com/Koenkk/zigbee2mqtt/wiki/How-to-support-new-devices.
  zigbee2mqtt:controller Recieved zigbee message with data { cid: 'genOnOff', data: { '61440': 63009792, onOff: 0 } } +2s
2018-6-14 11:39:48 INFO MQTT publish, topic: 'zigbee2mqtt/wall_switch_bathroom', payload: '{"state_undefined":"OFF","state_left":"ON"}'
  zigbee2mqtt:controller Recieved zigbee message with data { cid: 'genOnOff', data: { onOff: 0 } } +13ms
  zigbee2mqtt:controller Recieved zigbee message with data { cid: 'genMultistateInput', data: { presentValue: 1 } } +336ms
2018-6-14 11:39:48 WARN No converter available for 'QBKG12LM' with cid 'genMultistateInput' and type 'attReport'
2018-6-14 11:39:48 WARN Please see: https://github.com/Koenkk/zigbee2mqtt/wiki/How-to-support-new-devices.
[... cut ...]
  zigbee2mqtt:controller Recieved zigbee message with data { cid: 'genMultistateInput', data: { presentValue: 1 } } +266ms
2018-6-14 11:39:49 WARN No converter available for 'QBKG12LM' with cid 'genMultistateInput' and type 'attReport'
2018-6-14 11:39:49 WARN Please see: https://github.com/Koenkk/zigbee2mqtt/wiki/How-to-support-new-devices.
^C2018-6-14 11:40:13 INFO MQTT publish, topic: 'zigbee2mqtt/bridge/state', payload: 'offline'
2018-6-14 11:40:13 INFO Disconnecting from MQTT server
2018-6-14 11:40:13 INFO zigbee-shepherd stopped
`

Koenkk, 2018-06-15:
I've added initial support in the QBKG11LM_QBKG12LM branch.

```bash
cd node_modules
rm -rf zigbee-shepherd-converters
git clone https://github.com/Koenkk/zigbee-shepherd-converters.git -b QBKG11LM_QBKG12LM
```

Can you check if everything is ok?

Disposaber, 2018-06-15:
I updated to the latest zigbee2mqtt and test it out the converters you said but it doesn't start.
```
> zigbee2mqtt@0.1.0 start /opt/zigbee2mqtt
> node index.js

module.js:549
    throw err;
    ^

Error: Cannot find module 'debounce'
    at Function.Module._resolveFilename (module.js:547:15)
    at Function.Module._load (module.js:474:25)
    at Module.require (module.js:596:17)
    at require (internal/module.js:11:18)
    at Object.<anonymous> (/opt/zigbee2mqtt/node_modules/zigbee-shepherd-converters/converters/fromZigbee.js:3:18)
    at Module._compile (module.js:652:30)
    at Object.Module._extensions..js (module.js:663:10)
    at Module.load (module.js:565:32)
    at tryModuleLoad (module.js:505:12)
    at Function.Module._load (module.js:497:3)
npm ERR! code ELIFECYCLE
npm ERR! errno 1
npm ERR! zigbee2mqtt@0.1.0 start: `node index.js`
npm ERR! Exit status 1
npm ERR! 
npm ERR! Failed at the zigbee2mqtt@0.1.0 start script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.

npm ERR! A complete log of this run can be found in:
npm ERR!     /home/pi/.npm/_logs/2018-06-15T15_58_49_590Z-debug.log
```

The npm log:
```
0 info it worked if it ends with ok
1 verbose cli [ '/usr/bin/node', '/usr/bin/npm', 'start' ]
2 info using npm@5.6.0
3 info using node@v8.11.2
4 verbose run-script [ 'prestart', 'start', 'poststart' ]
5 info lifecycle zigbee2mqtt@0.1.0~prestart: zigbee2mqtt@0.1.0
6 info lifecycle zigbee2mqtt@0.1.0~start: zigbee2mqtt@0.1.0
7 verbose lifecycle zigbee2mqtt@0.1.0~start: unsafe-perm in lifecycle true
8 verbose lifecycle zigbee2mqtt@0.1.0~start: PATH: /usr/lib/node_modules/npm/node_modules/npm-lifecycle/node-gyp-bin:/opt/zigbee2mqtt/node_modules/.bin:/usr/loc$
9 verbose lifecycle zigbee2mqtt@0.1.0~start: CWD: /opt/zigbee2mqtt
10 silly lifecycle zigbee2mqtt@0.1.0~start: Args: [ '-c', 'node index.js' ]
11 silly lifecycle zigbee2mqtt@0.1.0~start: Returned: code: 1  signal: null
12 info lifecycle zigbee2mqtt@0.1.0~start: Failed to exec start script
13 verbose stack Error: zigbee2mqtt@0.1.0 start: `node index.js`
13 verbose stack Exit status 1
13 verbose stack     at EventEmitter.<anonymous> (/usr/lib/node_modules/npm/node_modules/npm-lifecycle/index.js:285:16)
13 verbose stack     at emitTwo (events.js:126:13)
13 verbose stack     at EventEmitter.emit (events.js:214:7)
13 verbose stack     at ChildProcess.<anonymous> (/usr/lib/node_modules/npm/node_modules/npm-lifecycle/lib/spawn.js:55:14)
13 verbose stack     at emitTwo (events.js:126:13)
13 verbose stack     at ChildProcess.emit (events.js:214:7)
13 verbose stack     at maybeClose (internal/child_process.js:925:16)
13 verbose stack     at Process.ChildProcess._handle.onexit (internal/child_process.js:209:5)
14 verbose pkgid zigbee2mqtt@0.1.0
15 verbose cwd /opt/zigbee2mqtt/node_modules
16 verbose Linux 4.14.34-v7+
17 verbose argv "/usr/bin/node" "/usr/bin/npm" "start"
18 verbose node v8.11.2
19 verbose npm  v5.6.0
20 error code ELIFECYCLE
21 error errno 1
22 error zigbee2mqtt@0.1.0 start: `node index.js`
22 error Exit status 1
23 error Failed at the zigbee2mqtt@0.1.0 start script.
23 error This is probably not a problem with npm. There is likely additional logging output above.
24 verbose exit [ 1, true ]
```

Disposaber, 2018-06-15:
Get it working manually copying the files. This is what the `DEBUG=zigbee2mqtt:controller npm start` says

```
> zigbee2mqtt@0.1.0 start /opt/zigbee2mqtt
> node index.js

2018-6-15 18:27:55 DEBUG Using zigbee-shepherd with settings: '{"net":{"panId":6754,"channelList":[11]},"dbPath":"/opt/zigbee2mqtt/data/database.db","sp":{"baudrate":115200,"rtscts":true}}'
2018-6-15 18:27:55 INFO Starting zigbee-shepherd
2018-6-15 18:27:56 INFO zigbee-shepherd started
2018-6-15 18:27:56 INFO Currently 3 devices are joined:
2018-6-15 18:27:56 INFO motion_hall (0x00158d0001e5fec8): RTCGQ11LM - Xiaomi Aqara human body movement and illuminance sensor (EndDevice)
2018-6-15 18:27:56 INFO wall_switch_bathroom (0x00158d00022a8278): QBKG12LM - Xiaomi Aqara double key wired wall switch (Router)
2018-6-15 18:27:56 INFO wall_switch_hall (0x00158d0002289982): QBKG11LM - Xiaomi Aqara single key wired wall switch (Router)
2018-6-15 18:27:56 WARN `permit_join` set to  `true` in configuration.yaml.
2018-6-15 18:27:56 WARN Allowing new devices to join.
2018-6-15 18:27:56 WARN Set `permit_join` to `false` once you joined all devices.
2018-6-15 18:27:56 INFO Zigbee: allowing new devices to join.
2018-6-15 18:27:56 INFO Connecting to MQTT server at mqtt://127.0.0.1
2018-6-15 18:27:57 INFO zigbee-shepherd ready
2018-6-15 18:27:57 INFO Connected to MQTT server
2018-6-15 18:27:57 INFO MQTT publish, topic: 'zigbee2mqtt/bridge/state', payload: 'online'
2018-6-15 18:28:05 DEBUG Recieved zigbee message with data
        {"cid":"genOnOff","data":{"61440":65555200,"onOff":1}}
2018-6-15 18:28:05 INFO MQTT publish, topic: 'zigbee2mqtt/wall_switch_hall', payload: '{"state":"ON"}'
2018-6-15 18:28:05 DEBUG Recieved zigbee message with data
        {"cid":"genOnOff","data":{"61440":65555200,"onOff":1}}
2018-6-15 18:28:05 DEBUG Recieved zigbee message with data
        {"cid":"genMultistateInput","data":{"presentValue":1}}
2018-6-15 18:28:05 DEBUG Recieved zigbee message with data
        {"cid":"genMultistateInput","data":{"presentValue":1}}
2018-6-15 18:28:06 DEBUG Recieved zigbee message with data
        {"cid":"genOnOff","data":{"61440":65555200,"onOff":0}}
2018-6-15 18:28:06 INFO MQTT publish, topic: 'zigbee2mqtt/wall_switch_hall', payload: '{"state":"OFF"}'
2018-6-15 18:28:06 DEBUG Recieved zigbee message with data
        {"cid":"genOnOff","data":{"onOff":0}}
2018-6-15 18:28:07 DEBUG Recieved zigbee message with data
        {"cid":"genMultistateInput","data":{"presentValue":1}}
2018-6-15 18:28:17 DEBUG Recieved zigbee message with data
        {"cid":"genOnOff","data":{"61440":63009792,"onOff":1}}
2018-6-15 18:28:17 INFO MQTT publish, topic: 'zigbee2mqtt/wall_switch_bathroom', payload: '{"state_left":"ON"}'
2018-6-15 18:28:17 DEBUG Recieved zigbee message with data
        {"cid":"genOnOff","data":{"61440":63009792,"onOff":1}}
2018-6-15 18:28:17 WARN No converter available for 'QBKG12LM' with cid 'genOnOff' and type 'devChange'
2018-6-15 18:28:17 WARN Please see: https://github.com/Koenkk/zigbee2mqtt/wiki/How-to-support-new-devices.
2018-6-15 18:28:17 DEBUG Recieved zigbee message with data
        {"cid":"genMultistateInput","data":{"presentValue":1}}
2018-6-15 18:28:17 DEBUG Recieved zigbee message with data
        {"cid":"genMultistateInput","data":{"presentValue":1}}
2018-6-15 18:28:19 DEBUG Recieved zigbee message with data
        {"cid":"genOnOff","data":{"61440":63009792,"onOff":1}}
2018-6-15 18:28:19 INFO MQTT publish, topic: 'zigbee2mqtt/wall_switch_bathroom', payload: '{"state_left":"ON","state_right":"ON"}'
2018-6-15 18:28:19 DEBUG Recieved zigbee message with data
        {"cid":"genOnOff","data":{"61440":63009792,"onOff":1}}
2018-6-15 18:28:19 WARN No converter available for 'QBKG12LM' with cid 'genOnOff' and type 'devChange'
2018-6-15 18:28:19 WARN Please see: https://github.com/Koenkk/zigbee2mqtt/wiki/How-to-support-new-devices.
2018-6-15 18:28:19 DEBUG Recieved zigbee message with data
        {"cid":"genMultistateInput","data":{"presentValue":1}}
2018-6-15 18:28:19 DEBUG Recieved zigbee message with data
        {"cid":"genMultistateInput","data":{"presentValue":1}}
2018-6-15 18:28:20 DEBUG Recieved zigbee message with data
        {"cid":"genAnalogInput","data":{"presentValue":10.600000381469727}}
2018-6-15 18:28:20 INFO MQTT publish, topic: 'zigbee2mqtt/wall_switch_bathroom', payload: '{"state_left":"ON","state_right":"ON","power_undefined":10.6}'
2018-6-15 18:28:20 DEBUG Recieved zigbee message with data
        {"cid":"genAnalogInput","data":{"presentValue":10.600000381469727}}
2018-6-15 18:28:24 DEBUG Recieved zigbee message with data
        {"cid":"genAnalogInput","data":{"presentValue":15.380000114440918}}
2018-6-15 18:28:24 INFO MQTT publish, topic: 'zigbee2mqtt/wall_switch_bathroom', payload: '{"state_left":"ON","state_right":"ON","power_undefined":15.38}'
2018-6-15 18:28:24 DEBUG Recieved zigbee message with data
        {"cid":"genAnalogInput","data":{"presentValue":15.380000114440918}}
2018-6-15 18:28:32 DEBUG Recieved zigbee message with data
        {"cid":"genOnOff","data":{"61440":63009792,"onOff":0}}
2018-6-15 18:28:32 INFO MQTT publish, topic: 'zigbee2mqtt/wall_switch_bathroom', payload: '{"state_left":"ON","state_right":"OFF","power_undefined":15.38}'
2018-6-15 18:28:32 DEBUG Recieved zigbee message with data
        {"cid":"genOnOff","data":{"onOff":0}}
2018-6-15 18:28:32 WARN No converter available for 'QBKG12LM' with cid 'genOnOff' and type 'devChange'
[... cut ...]
2018-6-15 18:28:33 DEBUG Recieved zigbee message with data
        {"cid":"genAnalogInput","data":{"presentValue":0}}
2018-6-15 18:28:33 INFO MQTT publish, topic: 'zigbee2mqtt/wall_switch_bathroom', payload: '{"state_left":"OFF","state_right":"OFF","power_undefined":0}'
2018-6-15 18:28:33 DEBUG Recieved zigbee message with data
        {"cid":"genAnalogInput","data":{"presentValue":0}}
2018-6-15 18:28:33 DEBUG Recieved zigbee message with data
        {"cid":"genMultistateInput","data":{"presentValue":1}}
```

Koenkk, 2018-06-15:
Updated, can you check again?

Disposaber, 2018-06-16:
Updated to the latest zigbee2mqtt and replaced the converters. This is what the default log looks like for both devices
```
> zigbee2mqtt@0.0.0 start /opt/zigbee2mqtt
> node index.js

2018-6-16 12:50:37 INFO Starting zigbee2mqtt version 0.0.0 (commit #95db22e)
2018-6-16 12:50:37 INFO Starting zigbee-shepherd
2018-6-16 12:50:38 INFO zigbee-shepherd started
2018-6-16 12:50:38 INFO Currently 3 devices are joined:
2018-6-16 12:50:38 INFO motion_hall (0x00158d0001e5fec8): RTCGQ11LM - Xiaomi Aqara human body movement and illuminance sensor (EndDevice)
2018-6-16 12:50:38 INFO wall_switch_bathroom (0x00158d00022a8278): QBKG12LM - Xiaomi Aqara double key wired wall switch (Router)
2018-6-16 12:50:38 INFO wall_switch_hall (0x00158d0002289982): QBKG11LM - Xiaomi Aqara single key wired wall switch (Router)
2018-6-16 12:50:38 WARN `permit_join` set to  `true` in configuration.yaml.
2018-6-16 12:50:38 WARN Allowing new devices to join.
2018-6-16 12:50:38 WARN Set `permit_join` to `false` once you joined all devices.
2018-6-16 12:50:38 INFO Zigbee: allowing new devices to join.
2018-6-16 12:50:38 INFO Connecting to MQTT server at mqtt://127.0.0.1
2018-6-16 12:50:39 INFO zigbee-shepherd ready
2018-6-16 12:50:39 INFO Connected to MQTT server
2018-6-16 12:50:39 INFO MQTT publish, topic: 'zigbee2mqtt/bridge/state', payload: 'online'
2018-6-16 12:50:48 INFO MQTT publish, topic: 'zigbee2mqtt/wall_switch_bathroom', payload: '{"state_left":"ON"}'
2018-6-16 12:50:49 INFO MQTT publish, topic: 'zigbee2mqtt/wall_switch_bathroom', payload: '{"state_left":"ON","state_right":"ON"}'
EP 1
2018-6-16 12:52:43 INFO MQTT publish, topic: 'zigbee2mqtt/wall_switch_bathroom', payload: '{"state_left":"ON","state_right":"ON","power_undefined":15.38,"consumption_undefined":1.58,"temperature_undefined":32}'
2018-6-16 12:53:09 INFO MQTT publish, topic: 'zigbee2mqtt/wall_switch_bathroom', payload: '{"state_left":"OFF","state_right":"ON","power_undefined":15.38,"consumption_undefined":1.58,"temperature_undefined":32}'
2018-6-16 12:53:09 INFO MQTT publish, topic: 'zigbee2mqtt/wall_switch_bathroom', payload: '{"state_left":"OFF","state_right":"OFF","power_undefined":15.38,"consumption_undefined":1.58,"temperature_undefined":32}'
2018-6-16 12:53:24 INFO MQTT publish, topic: 'zigbee2mqtt/wall_switch_hall', payload: '{"state":"ON"}'
2018-6-16 12:53:30 WARN No converter available for 'QBKG11LM' with cid 'genAnalogInput' and type 'attReport'
2018-6-16 12:53:30 WARN Please see: https://github.com/Koenkk/zigbee2mqtt/wiki/How-to-support-new-devices.
2018-6-16 12:53:30 WARN No converter available for 'QBKG11LM' with cid 'genAnalogInput' and type 'devChange'
2018-6-16 12:53:30 WARN Please see: https://github.com/Koenkk/zigbee2mqtt/wiki/How-to-support-new-devices.
^C2018-6-16 12:54:08 INFO MQTT publish, topic: 'zigbee2mqtt/bridge/state', payload: 'offline'
2018-6-16 12:54:08 INFO Disconnecting from MQTT server
2018-6-16 12:54:08 INFO zigbee-shepherd stopped
```

Debug log for QBKG12LM

```
> zigbee2mqtt@0.0.0 start /opt/zigbee2mqtt
> node index.js

2018-6-16 12:55:02 DEBUG Using zigbee-shepherd with settings: '{"net":{"panId":6754,"channelList":[11]},"dbPath":"/opt/zigbee2mqtt/data/database.db","sp":{"baudrate":115200,"rtscts":true}}'
2018-6-16 12:55:02 INFO Starting zigbee2mqtt version 0.0.0 (commit #95db22e)
2018-6-16 12:55:02 INFO Starting zigbee-shepherd
2018-6-16 12:55:03 INFO zigbee-shepherd started
2018-6-16 12:55:03 INFO Currently 3 devices are joined:
2018-6-16 12:55:03 INFO motion_hall (0x00158d0001e5fec8): RTCGQ11LM - Xiaomi Aqara human body movement and illuminance sensor (EndDevice)
2018-6-16 12:55:03 INFO wall_switch_bathroom (0x00158d00022a8278): QBKG12LM - Xiaomi Aqara double key wired wall switch (Router)
2018-6-16 12:55:03 INFO wall_switch_hall (0x00158d0002289982): QBKG11LM - Xiaomi Aqara single key wired wall switch (Router)
2018-6-16 12:55:03 WARN `permit_join` set to  `true` in configuration.yaml.
2018-6-16 12:55:03 WARN Allowing new devices to join.
2018-6-16 12:55:03 WARN Set `permit_join` to `false` once you joined all devices.
2018-6-16 12:55:03 INFO Zigbee: allowing new devices to join.
2018-6-16 12:55:03 INFO Connecting to MQTT server at mqtt://127.0.0.1
2018-6-16 12:55:04 INFO zigbee-shepherd ready
2018-6-16 12:55:04 INFO Connected to MQTT server
2018-6-16 12:55:04 INFO MQTT publish, topic: 'zigbee2mqtt/bridge/state', payload: 'online'
2018-6-16 12:55:17 DEBUG Recieved zigbee message with data {"cid":"genOnOff","data":{"61440":63009792,"onOff":1}}
2018-6-16 12:55:17 INFO MQTT publish, topic: 'zigbee2mqtt/wall_switch_bathroom', payload: '{"state_left":"ON"}'
2018-6-16 12:55:17 DEBUG Recieved zigbee message with data {"cid":"genOnOff","data":{"61440":63009792,"onOff":1}}
2018-6-16 12:55:17 DEBUG Recieved zigbee message with data {"cid":"genMultistateInput","data":{"presentValue":1}}
2018-6-16 12:55:17 DEBUG Recieved zigbee message with data {"cid":"genMultistateInput","data":{"presentValue":1}}
2018-6-16 12:55:18 DEBUG Recieved zigbee message with data {"cid":"genOnOff","data":{"61440":63009792,"onOff":1}}
2018-6-16 12:55:18 INFO MQTT publish, topic: 'zigbee2mqtt/wall_switch_bathroom', payload: '{"state_left":"ON","state_right":"ON"}'
2018-6-16 12:55:18 DEBUG Recieved zigbee message with data {"cid":"genOnOff","data":{"61440":63009792,"onOff":1}}
2018-6-16 12:55:18 DEBUG Recieved zigbee message with data {"cid":"genMultistateInput","data":{"presentValue":1}}
[... cut ...]
2018-6-16 13:10:16 DEBUG Recieved zigbee message with data {"cid":"genMultistateInput","data":{"presentValue":1}}
2018-6-16 13:10:17 DEBUG Recieved zigbee message with data {"cid":"genBasic","data":{"65281":{"3":32,"5":14,"7":"0x0000000000000000","8":4383,"9":512,"100":0,"101":0,"149":1.5808990001678467,"152":0,"154":16}}}
^C2018-6-16 13:10:31 INFO MQTT publish, topic: 'zigbee2mqtt/bridge/state', payload: 'offline'
2018-6-16 13:10:31 INFO Disconnecting from MQTT server
2018-6-16 13:10:31 INFO zigbee-shepherd stopped
```

Koenkk, 2018-06-16:
Updated, can you check if turning on/off also works?

By sending e.g. for QBKG11LM `zigbee2mqtt/wall_switch_hall/set` `{"state": "ON"}` and for QBKG12LM `zigbee2mqtt/wall_switch_hall/left/set` `{"state": "ON"}`

Disposaber, 2018-06-16:
With the last update the log doesn't show any errors.

### QBKG11LM:
By sending  `zigbee2mqtt/wall_switch_hall/set` `{"state": "ON"}` doesn't work. It has to be `zigbee2mqtt/wall_switch_hall/<random letters>/set` `{"state": "ON"}` for it to work.

```
2018-6-16 18:51:27 DEBUG Recieved mqtt message on topic 'zigbee2mqtt/wall_switch_hall/set' with data '{"state": "ON"}'
2018-6-16 18:51:27 INFO Zigbee publish to '0x00158d0002289982', genOnOff - on - {} - 2
2018-6-16 18:51:33 DEBUG Recieved mqtt message on topic 'zigbee2mqtt/wall_switch_hall/asd/set' with data '{"state": "ON"}'
2018-6-16 18:51:33 INFO Zigbee publish to '0x00158d0002289982', genOnOff - on - {} - null
2018-6-16 18:51:33 INFO MQTT publish, topic: 'zigbee2mqtt/wall_switch_hall', payload: '{"state":"ON"}'
2018-6-16 18:51:33 DEBUG Recieved zigbee message with data {"cid":"genOnOff","data":{"61440":117440525,"onOff":1}}
2018-6-16 18:51:33 INFO MQTT publish, topic: 'zigbee2mqtt/wall_switch_hall', payload: '{"state":"ON"}'
2018-6-16 18:51:33 DEBUG Recieved zigbee message with data {"cid":"genOnOff","data":{"61440":117440525,"onOff":1}}
2018-6-16 18:51:33 DEBUG Recieved zigbee message with data {"cid":"genOnOff","data":{"onOff":1}}
2018-6-16 18:51:35 DEBUG Recieved zigbee message with data {"cid":"genAnalogInput","data":{"presentValue":10.34000015258789}}
```

### QBKG12LM:
It does work
```
2018-6-16 18:53:49 DEBUG Recieved mqtt message on topic 'zigbee2mqtt/wall_switch_bathroom/left/set' with data '{"state": "ON"}'
2018-6-16 18:53:49 INFO Zigbee publish to '0x00158d00022a8278', genOnOff - on - {} - 1
2018-6-16 18:53:49 INFO MQTT publish, topic: 'zigbee2mqtt/wall_switch_bathroom', payload: '{"state":"ON"}'
2018-6-16 18:53:49 DEBUG Recieved zigbee message with data {"cid":"genOnOff","data":{"onOff":1}}
2018-6-16 18:53:54 DEBUG Recieved zigbee message with data {"cid":"genAnalogInput","data":{"presentValue":10.600000381469727}}
2018-6-16 18:53:54 DEBUG Recieved zigbee message with data {"cid":"genAnalogInput","data":{"presentValue":10.600000381469727}}
2018-6-16 18:54:04 DEBUG Recieved mqtt message on topic 'zigbee2mqtt/wall_switch_bathroom/right/set' with data '{"state": "ON"}'
2018-6-16 18:54:04 INFO Zigbee publish to '0x00158d00022a8278', genOnOff - on - {} - 2
2018-6-16 18:54:05 INFO MQTT publish, topic: 'zigbee2mqtt/wall_switch_bathroom', payload: '{"state":"ON"}'
2018-6-16 18:54:05 DEBUG Recieved zigbee message with data {"cid":"genOnOff","data":{"onOff":1}}
2018-6-16 18:54:08 DEBUG Recieved zigbee message with data {"cid":"genAnalogInput","data":{"presentValue":15.890000343322754}}
2018-6-16 18:54:08 DEBUG Recieved zigbee message with data {"cid":"genAnalogInput","data":{"presentValue":15.890000343322754}}
```
But reports weird things such as `"power_left":15.38,"consumption_left":1.59,"temperature_left"` (_left?) and it has multiple states `"state":"ON","state_right":"OFF","state_left":"OFF"`
```
2018-6-16 18:55:53 INFO MQTT publish, topic: 'zigbee2mqtt/wall_switch_bathroom', payload: '{"state":"ON","power_left":15.38,"consumption_left":1.59,"temperature_left":32,"state_right":"OFF"}'
2018-6-16 18:55:53 DEBUG Recieved zigbee message with data {"cid":"genOnOff","data":{"61440":63009792,"onOff":0}}
2018-6-16 18:55:54 DEBUG Recieved zigbee message with data {"cid":"genMultistateInput","data":{"presentValue":1}}
2018-6-16 18:55:54 DEBUG Recieved zigbee message with data {"cid":"genMultistateInput","data":{"presentValue":1}}
2018-6-16 18:55:54 DEBUG Recieved zigbee message with data {"cid":"genOnOff","data":{"61440":63009792,"onOff":0}}
2018-6-16 18:55:54 INFO MQTT publish, topic: 'zigbee2mqtt/wall_switch_bathroom', payload: '{"state":"ON","power_left":15.38,"consumption_left":1.59,"temperature_left":32,"state_right":"OFF","state_left":"OFF"}'
2018-6-16 18:55:54 DEBUG Recieved zigbee message with data {"cid":"genOnOff","data":{"61440":63009792,"onOff":0}}
2018-6-16 18:55:54 DEBUG Recieved zigbee message with data {"cid":"genMultistateInput","data":{"presentValue":1}}
2018-6-16 18:55:54 DEBUG Recieved zigbee message with data {"cid":"genMultistateInput","data":{"presentValue":1}}
2018-6-16 18:55:53 INFO MQTT publish, topic: 'zigbee2mqtt/wall_switch_bathroom', payload: '{"state":"ON","power_left":15.38,"consumption_left":1.59,"temperature_left":32,"state_right":"OFF"}'
2018-6-16 18:55:53 DEBUG Recieved zigbee message with data {"cid":"genOnOff","data":{"61440":63009792,"onOff":0}}
2018-6-16 18:55:54 DEBUG Recieved zigbee message with data {"cid":"genMultistateInput","data":{"presentValue":1}}
2018-6-16 18:55:54 DEBUG Recieved zigbee message with data {"cid":"genMultistateInput","data":{"presentValue":1}}
2018-6-16 18:55:54 DEBUG Recieved zigbee message with data {"cid":"genOnOff","data":{"61440":63009792,"onOff":0}}
2018-6-16 18:55:54 INFO MQTT publish, topic: 'zigbee2mqtt/wall_switch_bathroom', payload: '{"state":"ON","power_left":15.38,"consumption_left":1.59,"temperature_left":32,"state_right":"OFF","state_left":"OFF"}'
2018-6-16 18:55:54 DEBUG Recieved zigbee message with data {"cid":"genOnOff","data":{"61440":63009792,"onOff":0}}
2018-6-16 18:55:54 DEBUG Recieved zigbee message with data {"cid":"genMultistateInput","data":{"presentValue":1}}
2018-6-16 18:55:54 DEBUG Recieved zigbee message with data {"cid":"genMultistateInput","data":{"presentValue":1}}
```

Koenkk, 2018-06-16:
**QBKG11LM**: can you check if it also works without the random letters? (so `zigbee2mqtt/wall_switch_hall/set`.

`power_left`, etc. are expected, these are power/temperature measurements from the right and left switch.

The `state` for QBKG12LM was a bug in zigbee2mqtt, can you check if this has been fixed?

Note: you need to update both zigbee2mqtt as zigbee-shepherd-converters.

Disposaber, 2018-06-16:
QBKKG11LM works fine without the random letters.

QBKG12LM seems to be fixed as well.

I'll test more in depth in a few hours. Thank you for your hard work!

Disposaber, 2018-06-16:
The QBKG11LM seems to be working fine, this is what the log looks like now
```
2018-6-16 23:47:34 DEBUG Recieved mqtt message on topic 'zigbee2mqtt/wall_switch_hall/set' with data 'ON'
2018-6-16 23:47:34 INFO Zigbee publish to '0x00158d0002289982', genOnOff - on - {} - null
2018-6-16 23:47:34 INFO MQTT publish, topic: 'zigbee2mqtt/wall_switch_hall', payload: '{"state":"ON"}'
2018-6-16 23:47:34 DEBUG Recieved zigbee message with data {"cid":"genOnOff","data":{"61440":117440545,"onOff":1}}
2018-6-16 23:47:34 INFO MQTT publish, topic: 'zigbee2mqtt/wall_switch_hall', payload: '{"state":"ON"}'
2018-6-16 23:47:34 DEBUG Recieved zigbee message with data {"cid":"genOnOff","data":{"61440":117440545,"onOff":1}}
2018-6-16 23:47:35 DEBUG Recieved zigbee message with data {"cid":"genOnOff","data":{"onOff":1}}
2018-6-16 23:47:37 DEBUG Recieved zigbee message with data {"cid":"genAnalogInput","data":{"presentValue":10.119999885559082}}
2018-6-16 23:47:37 DEBUG Recieved zigbee message with data {"cid":"genAnalogInput","data":{"presentValue":10.119999885559082}}
```

And for the QBKG12LM seems to work fine if a publish the message "manually" `zigbee2mqtt/wall_switch_bathroom/left/set` `{"state": "ON"}` and `zigbee2mqtt/wall_switch_bathroom/right/set` `{"state": "ON"}`
```
2018-6-16 23:51:28 DEBUG Recieved mqtt message on topic 'zigbee2mqtt/wall_switch_bathroom/left/set' with data '{"state": "ON"}'
2018-6-16 23:51:28 INFO Zigbee publish to '0x00158d00022a8278', genOnOff - on - {} - 1
2018-6-16 23:51:28 INFO MQTT publish, topic: 'zigbee2mqtt/wall_switch_bathroom', payload: '{"state_left":"ON"}'
2018-6-16 23:51:28 DEBUG Recieved zigbee message with data {"cid":"genOnOff","data":{"onOff":1}}
2018-6-16 23:51:32 DEBUG Recieved zigbee message with data {"cid":"genAnalogInput","data":{"presentValue":10.359999656677246}}
2018-6-16 23:51:32 DEBUG Recieved zigbee message with data {"cid":"genAnalogInput","data":{"presentValue":10.359999656677246}}
2018-6-16 23:51:40 DEBUG Recieved mqtt message on topic 'zigbee2mqtt/wall_switch_bathroom/right/set' with data '{"state": "ON"}'
2018-6-16 23:51:40 INFO Zigbee publish to '0x00158d00022a8278', genOnOff - on - {} - 2
2018-6-16 23:51:40 INFO MQTT publish, topic: 'zigbee2mqtt/wall_switch_bathroom', payload: '{"state_left":"ON","state_right":"ON"}'
2018-6-16 23:51:40 DEBUG Recieved zigbee message with data {"cid":"genOnOff","data":{"onOff":1}}
2018-6-16 23:51:42 DEBUG Recieved zigbee message with data {"cid":"genAnalogInput","data":{"presentValue":15.380000114440918}}
2018-6-16 23:51:42 DEBUG Recieved zigbee message with data {"cid":"genAnalogInput","data":{"presentValue":15.380000114440918}}
```

but if I integrate it in Home Assistant using the template for the QBKG03LM I get the triple status bug and only the left one is triggered but I can't control it.
```
  # Left
  - platform: "mqtt"
    name: "wall_switch_bathroom_left"
    state_topic: "zigbee2mqtt/wall_switch_bathroom"
    availability_topic: "zigbee2mqtt/bridge/state"
    payload_off: "OFF"
    payload_on: "ON"
    value_template: "{{ value_json.state_left }}"
    command_topic: "zigbee2mqtt/wall_switch_bathroom/set"
    command_topic_prefix: "left"

  # Right
  - platform: "mqtt"
    name: "wall_switch_bathroom_right"
    state_topic: "zigbee2mqtt/wall_switch_bathroom"
    availability_topic: "zigbee2mqtt/bridge/state"
    payload_off: "OFF"
    payload_on: "ON"
    value_template: "{{ value_json.state_right }}"
    command_topic: "zigbee2mqtt/wall_switch_bathroom/set"
    command_topic_prefix: "right"
```

```
2018-6-16 23:54:10 DEBUG Recieved mqtt message on topic 'zigbee2mqtt/wall_switch_bathroom/set' with data 'ON'
2018-6-16 23:54:10 INFO Zigbee publish to '0x00158d00022a8278', genOnOff - on - {} - null
2018-6-16 23:54:11 INFO MQTT publish, topic: 'zigbee2mqtt/wall_switch_bathroom', payload: '{"state_left":"OFF","state_right":"OFF","state":"ON"}'
2018-6-16 23:54:11 DEBUG Recieved zigbee message with data {"cid":"genOnOff","data":{"onOff":1}}
2018-6-16 23:54:12 DEBUG Recieved zigbee message with data {"cid":"genAnalogInput","data":{"presentValue":15.380000114440918}}
2018-6-16 23:54:13 DEBUG Recieved mqtt message on topic 'zigbee2mqtt/wall_switch_bathroom/set' with data 'ON'
2018-6-16 23:54:14 INFO Zigbee publish to '0x00158d00022a8278', genOnOff - on - {} - null
2018-6-16 23:54:14 INFO MQTT publish, topic: 'zigbee2mqtt/wall_switch_bathroom', payload: '{"state_left":"OFF","state_right":"OFF","state":"ON"}'
2018-6-16 23:54:18 DEBUG Recieved zigbee message with data {"cid":"genAnalogInput","data":{"presentValue":10.140000343322754}}
2018-6-16 23:54:18 DEBUG Recieved zigbee message with data {"cid":"genAnalogInput","data":{"presentValue":10.140000343322754}}
2018-6-16 23:54:19 DEBUG Recieved mqtt message on topic 'zigbee2mqtt/wall_switch_bathroom/set' with data 'ON'
2018-6-16 23:54:19 INFO Zigbee publish to '0x00158d00022a8278', genOnOff - on - {} - null
2018-6-16 23:54:19 INFO MQTT publish, topic: 'zigbee2mqtt/wall_switch_bathroom', payload: '{"state_left":"OFF","state_right":"OFF","state":"ON"}'
2018-6-16 23:54:20 DEBUG Recieved mqtt message on topic 'zigbee2mqtt/wall_switch_bathroom/set' with data 'ON'
2018-6-16 23:54:20 INFO Zigbee publish to '0x00158d00022a8278', genOnOff - on - {} - null
2018-6-16 23:54:20 INFO MQTT publish, topic: 'zigbee2mqtt/wall_switch_bathroom', payload: '{"state_left":"OFF","state_right":"OFF","state":"ON"}'
[... cut ...] null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,1,0,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,1.599717140197754,null,null,10.140000343322754,null,16]}}
```

Disposaber, 2018-06-17:
Now it's working good!

> `power_left`, etc. are expected, these are power/temperature measurements from the right and left switch.

I only get the reports of the left switch. I've never got `power_right`, etc. If I toggle the left switch it reports its power
`2018-6-17 12:27:26 INFO MQTT publish, topic: 'zigbee2mqtt/wall_switch_bathroom', payload: '{"state_right":"OFF","state_left":"ON","power_left":10.14,"consumption_left":1.61,"temperature_left":32}'`
but if I toggle both switches `power_left` reports both powers.
`2018-6-17 12:32:26 INFO MQTT publish, topic: 'zigbee2mqtt/wall_switch_bathroom', payload: '{"state_right":"ON","state_left":"ON","power_left":15.38,"consumption_left":1.61,"temperature_left":32}'`

The `consumption_left` is the "power consumption in current month" for both switches so it shouldn't have the `_left`. And the same goes for the temperature I guess.

Koenkk, 2018-06-17:
Fixed, if there is nothing else, I think it's ready for merge.

Disposaber, 2018-06-17:
It's working fine. I'm going to close the issue. Thanks @Koenkk !

closed 2018-06-17 by Disposaber

phoxy84, 2018-12-06:
Looks like I have the exact same problemL

zigbee2mqtt:info 2018-12-6 19:43:18 0x00158d00014ab06b (0x00158d00014ab06b): unkown - undefined unknown (Router)

  zigbee2mqtt:warn 2018-12-6 19:50:11 Device with modelID 'lumi.ctrl_ln1' is not supported.
  zigbee2mqtt:warn 2018-12-6 19:50:11 Please see: https://github.com/Koenkk/zigbee2mqtt/wiki/How-to-support-new-devices
  zigbee2mqtt:warn 2018-12-6 19:50:11 Device with modelID 'lumi.ctrl_ln1' is not supported.
  zigbee2mqtt:warn 2018-12-6 19:50:11 Please see: https://github.com/Koenkk/zigbee2mqtt/wiki/How-to-support-new-devices

I am on Hassio 0.83.3, use the CC2531 and zigbee2mqtt-edge addon

Any help on getting this fixed?

Thanks

Koenkk, 2018-12-06:
@phoxy84 What is the model number of this device?

Koenkk, 2018-12-06:
@phoxy84 should be supported in the latest edge (wait 1 hour before the image is build).

phoxy84, 2018-12-06:
Hi, no difference. Did a restart of the addon and still unknown

tenkaibr, 2019-06-07:
I cant find the QBKG12LM to buy... could you help me?