Koenkk/zigbee2mqtt #120: Crash after updating & using my database.db

kimonm, 2018-06-13:
I just updated zigbee2mqtt (but I see same problem w/ "clean" install and restoring my data/database.db). It crashes on startup -see output below. If I don't use my database.db, it starts fine.  npm-debug.log below the output.

> zigbee2mqtt@0.1.0 start /home/kimon/Services/test
> node index.js

6/13/2018, 2:51:15 PM INFO Starting zigbee-shepherd
6/13/2018, 2:51:16 PM INFO zigbee-shepherd started
6/13/2018, 2:51:16 PM INFO Currently 5 devices are joined:
6/13/2018, 2:51:16 PM INFO xiaomi_sw1 (0x00158d0001f43047): WXKG01LM - Xiaomi MiJia wireless switch (EndDevice)
6/13/2018, 2:51:16 PM INFO office_lamp (0xb0ce18140310b553): E11-G13 - Sengled Element Classic (A19) (EndDevice)
6/13/2018, 2:51:16 PM INFO xiaomi_cube (0x00158d00024a477d): MFKZQ01LM - Xiaomi Mi smart home cube (EndDevice)
6/13/2018, 2:51:16 PM INFO ge_plug1 (0x0022a300001af090): 45852GE - GE ZigBee plug-in smart dimmer (Router)
6/13/2018, 2:51:16 PM INFO xiaomi_sw2 (0x00158d0002133e41): WXKG11LM - Xiaomi Aqara wireless switch (EndDevice)
6/13/2018, 2:51:16 PM INFO Connecting to MQTT server at mqtt://localhost
6/13/2018, 2:51:16 PM INFO zigbee-shepherd ready
6/13/2018, 2:51:16 PM INFO Connected to MQTT server
6/13/2018, 2:51:16 PM INFO MQTT publish, topic: 'zigbee2mqtt/bridge/state', payload: 'online'
spinlock: false []
path.js:26
function assertPath(path) {
                   ^

RangeError: Maximum call stack size exceeded
    at assertPath (path.js:26:20)
    at Object.isAbsolute (path.js:1225:5)
    at Function.Module._findPath (module.js:183:12)
    at Function.Module._resolveFilename (module.js:545:25)
    at Function.Module._load (module.js:474:25)
    at Module.require (module.js:596:17)
    at require (internal/module.js:11:18)
    at new ReadableState (/home/kimon/Services/test/node_modules/dissolve/node_modules/readable-stream/lib/_stream_readable.js:66:16)
    at Dissolve.Readable (/home/kimon/Services/test/node_modules/dissolve/node_modules/readable-stream/lib/_stream_readable.js:139:25)
    at Dissolve.Duplex (/home/kimon/Services/test/node_modules/dissolve/node_modules/readable-stream/lib/_stream_duplex.js:57:12)

npm ERR! Linux 4.15.0-23-generic
npm ERR! argv "/usr/bin/node" "/usr/bin/npm" "start"
npm ERR! node v8.10.0
npm ERR! npm  v3.5.2
npm ERR! code ELIFECYCLE
npm ERR! zigbee2mqtt@0.1.0 start: `node index.js`
npm ERR! Exit status 1
npm ERR! 
npm ERR! Failed at the zigbee2mqtt@0.1.0 start script 'node index.js'.
npm ERR! Make sure you have the latest version of node.js and npm installed.
npm ERR! If you do, this is most likely a problem with the zigbee2mqtt package,
npm ERR! not with npm itself.
npm ERR! Tell the author that this fails on your system:
npm ERR!     node index.js
npm ERR! You can get information on how to open an issue for this project with:
npm ERR!     npm bugs zigbee2mqtt
npm ERR! Or if that isn't available, you can get their info via:
npm ERR!     npm owner ls zigbee2mqtt
npm ERR! There is likely additional logging output above.

npm ERR! Please include the following file with any support request:
npm ERR!     /home/kimon/Services/test/npm-debug.log

**npm-debug.log:**

0 info it worked if it ends with ok
1 verbose cli [ '/usr/bin/node', '/usr/bin/npm', 'start' ]
2 info using npm@3.5.2
3 info using node@v8.10.0
4 verbose run-script [ 'prestart', 'start', 'poststart' ]
5 info lifecycle zigbee2mqtt@0.1.0~prestart: zigbee2mqtt@0.1.0
6 silly lifecycle zigbee2mqtt@0.1.0~prestart: no script for prestart, continuing
7 info lifecycle zigbee2mqtt@0.1.0~start: zigbee2mqtt@0.1.0
8 verbose lifecycle zigbee2mqtt@0.1.0~start: unsafe-perm in lifecycle true
9 verbose lifecycle zigbee2mqtt@0.1.0~start: PATH: /usr/share/npm/bin/node-gyp-bin:/home/kimon/Services/test/node_modules/.bin:/home/kimon/bin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/usr/games:/usr/local/games:/snap/bin:/home/kimon/Go/bin
10 verbose lifecycle zigbee2mqtt@0.1.0~start: CWD: /home/kimon/Services/test
11 silly lifecycle zigbee2mqtt@0.1.0~start: Args: [ '-c', 'node index.js' ]
12 silly lifecycle zigbee2mqtt@0.1.0~start: Returned: code: 1  signal: null
13 info lifecycle zigbee2mqtt@0.1.0~start: Failed to exec start script
14 verbose stack Error: zigbee2mqtt@0.1.0 start: `node index.js`
14 verbose stack Exit status 1
14 verbose stack     at EventEmitter.<anonymous> (/usr/share/npm/lib/utils/lifecycle.js:232:16)
14 verbose stack     at emitTwo (events.js:126:13)
14 verbose stack     at EventEmitter.emit (events.js:214:7)
14 verbose stack     at ChildProcess.<anonymous> (/usr/share/npm/lib/utils/spawn.js:24:14)
14 verbose stack     at emitTwo (events.js:126:13)
14 verbose stack     at ChildProcess.emit (events.js:214:7)
14 verbose stack     at maybeClose (internal/child_process.js:925:16)
14 verbose stack     at Process.ChildProcess._handle.onexit (internal/child_process.js:209:5)
15 verbose pkgid zigbee2mqtt@0.1.0
16 verbose cwd /home/kimon/Services/test
17 error Linux 4.15.0-23-generic
18 error argv "/usr/bin/node" "/usr/bin/npm" "start"
19 error node v8.10.0
20 error npm  v3.5.2
21 error code ELIFECYCLE
22 error zigbee2mqtt@0.1.0 start: `node index.js`
22 error Exit status 1
23 error Failed at the zigbee2mqtt@0.1.0 start script 'node index.js'.
23 error Make sure you have the latest version of node.js and npm installed.
23 error If you do, this is most likely a problem with the zigbee2mqtt package,
23 error not with npm itself.
23 error Tell the author that this fails on your system:
23 error     node index.js
23 error You can get information on how to open an issue for this project with:
23 error     npm bugs zigbee2mqtt
23 error Or if that isn't available, you can get their info via:
23 error     npm owner ls zigbee2mqtt
23 error There is likely additional logging output above.
24 verbose exit [ 1, true ]

ciotlosm, 2018-06-14:
Probably you have something corrupted in your database.db. Have you done any manual updates in there?

kimonm, 2018-06-14:
No manual updates. zigbee2mqtt still works with the older version of the code. (I made a backup.)  Do you suggest just starting from scratch and re-pairing all the devices? Thanks.

Koenkk, 2018-06-14:
I think that's the best thing to do for now.

closed 2018-06-14 by kimonm