Koenkk/zigbee2mqtt #121: Xiaomi plug reporting voltage: null

subzero79, 2018-06-14:
Is there any way of enabling more verbose log than current debug? I am getting voltage:null published into the mqtt topic but there is no way of seeing the incoming zigbee data to see what's going on.

This is my second plug that i add to zigbee2mqtt, the first one reported fine the voltage.

```
  zigbee-shepherd:msgHdlr IND <-- ZDO:nodeDescRsp +43ms
  zigbee-shepherd:msgHdlr IND <-- ZDO:nodeDescRsp +23ms
  zigbee-shepherd:msgHdlr IND <-- AF:incomingMsg, transId: 0 +2s
2018-6-14 10:45:35 INFO MQTT publish, topic: 'zigbee2mqtt/fan_masterbedroom_dd92', payload: '{"state":"ON","power":11.52,"voltage":null,"consumption":0,"temperature":31}'
  zigbee-shepherd:msgHdlr IND <-- AF:incomingMsg, transId: 0 +3s
2018-6-14 10:45:38 INFO MQTT publish, topic: 'zigbee2mqtt/fan_masterbedroom_dd92', payload: '{"state":"ON","power":16.46,"voltage":null,"consumption":0,"temperature":31}'
  zigbee-shepherd:msgHdlr IND <-- AF:incomingMsg, transId: 0 +16s
2018-6-14 10:45:54 INFO MQTT publish, topic: 'zigbee2mqtt/fan_masterbedroom_dd92', payload: '{"state":"ON","power":13.17,"voltage":null,"consumption":0,"temperature":31}'
  zigbee-shepherd:msgHdlr IND <-- AF:incomingMsg, transId: 0 +1s
  zigbee-shepherd:msgHdlr IND <-- AF:incomingMsg, transId: 0 +7s
2018-6-14 10:46:02 INFO MQTT publish, topic: 'zigbee2mqtt/fan_masterbedroom_dd92', payload: '{"state":"ON","power":13.17,"voltage":null,"consumption":0,"temperature":32}'
  zigbee-shepherd:msgHdlr IND <-- AF:incomingMsg, transId: 0 +17s
  zigbee-shepherd:request REQ --> ZDO:nodeDescReq +13s
  zigbee-shepherd:request REQ --> ZDO:nodeDescReq +0ms
  zigbee-shepherd:msgHdlr IND <-- ZDO:nodeDescRsp +58ms
  zigbee-shepherd:msgHdlr IND <-- ZDO:nodeDescRsp +1ms
  zigbee-shepherd:msgHdlr IND <-- AF:incomingMsg, transId: 0 +12s
  zigbee-shepherd:msgHdlr IND <-- AF:incomingMsg, transId: 0 +4s
2018-6-14 10:46:49 INFO MQTT publish, topic: 'zigbee2mqtt/fan_javier_ddae', payload: '{"state":"ON","power":8.07,"voltage":241.2,"consumption":3.56,"temperature":33}'
  zigbee-shepherd:msgHdlr IND <-- AF:incomingMsg, transId: 0 +19s
2018-6-14 10:47:08 INFO MQTT publish, topic: 'zigbee2mqtt/fan_javier_ddae', payload: '{"state":"ON","power":11.8,"voltage":241.2,"consumption":3.56,"temperature":33}'
  zigbee-shepherd:msgHdlr IND <-- AF:incomingMsg, transId: 0 +3s
2018-6-14 10:47:12 INFO MQTT publish, topic: 'zigbee2mqtt/fan_javier_ddae', payload: '{"state":"ON","power":17.92,"voltage":241.2,"consumption":3.56,"temperature":33}'
  zigbee-shepherd:msgHdlr IND <-- AF:incomingMsg, transId: 0 +4s
2018-6-14 10:47:16 INFO MQTT publish, topic: 'zigbee2mqtt/fan_javier_ddae', payload: '{"state":"ON","power":13.08,"voltage":241.2,"consumption":3.56,"temperature":33}'
  zigbee-shepherd:msgHdlr IND <-- AF:incomingMsg, transId: 0 +4s
  zigbee-shepherd:request REQ --> ZDO:nodeDescReq +13s
  zigbee-shepherd:request REQ --> ZDO:nodeDescReq +0ms
  zigbee-shepherd:msgHdlr IND <-- ZDO:nodeDescRsp +59ms
  zigbee-shepherd:msgHdlr IND <-- ZDO:nodeDescRsp +2ms
  zigbee-shepherd:msgHdlr IND <-- AF:incomingMsg, transId: 0 +2s
  zigbee-shepherd:msgHdlr IND <-- AF:incomingMsg, transId: 0 +44s
  zigbee-shepherd:msgHdlr IND <-- AF:incomingMsg, transId: 0 +7s
  zigbee-shepherd:msgHdlr IND <-- AF:incomingMsg, transId: 0 +5s
2018-6-14 10:48:31 INFO MQTT publish, topic: 'zigbee2mqtt/fan_masterbedroom_dd92', payload: '{"state":"ON","power":21.95,"voltage":null,"consumption":0,"temperature":32}'
  zigbee-shepherd:request REQ --> ZDO:nodeDescReq +2s
  zigbee-shepherd:request REQ --> ZDO:nodeDescReq +1ms
  zigbee-shepherd:msgHdlr IND <-- AF:incomingMsg, transId: 0 +12ms
2018-6-14 10:48:33 INFO MQTT publish, topic: 'zigbee2mqtt/fan_masterbedroom_dd92', payload: '{"state":"ON","power":25.61,"voltage":null,"consumption":0,"temperature":32}'
  zigbee-shepherd:msgHdlr IND <-- ZDO:nodeDescRsp +32ms
  zigbee-shepherd:msgHdlr IND <-- ZDO:nodeDescRsp +21ms
  zigbee-shepherd:msgHdlr IND <-- AF:incomingMsg, transId: 0 +2s
2018-6-14 10:48:35 INFO MQTT publish, topic: 'zigbee2mqtt/fan_masterbedroom_dd92', payload: '{"state":"ON","power":21.95,"voltage":null,"consumption":0,"temperature":32}'
  zigbee-shepherd:msgHdlr IND <-- AF:incomingMsg, transId: 0 +40s
2018-6-14 10:49:15 INFO MQTT publish, topic: 'zigbee2mqtt/fan_javier_ddae', payload: '{"state":"ON","power":17.28,"voltage":241.2,"consumption":3.56,"temperature":33}'
  zigbee-shepherd:msgHdlr IND <-- AF:incomingMsg, transId: 0 +819ms
  zigbee-shepherd:msgHdlr IND <-- AF:incomingMsg, transId: 0 +2s
2018-6-14 10:49:18 INFO MQTT publish, topic: 'zigbee2mqtt/fan_javier_ddae', payload: '{"state":"ON","power":25.47,"voltage":241.2,"consumption":3.56,"temperature":33}'
  zigbee-shepherd:msgHdlr IND <-- AF:incomingMsg, transId: 0 +2s
  zigbee-shepherd:msgHdlr IND <-- AF:incomingMsg, transId: 0 +3s
2018-6-14 10:49:23 INFO MQTT publish, topic: 'zigbee2mqtt/fan_javier_ddae', payload: '{"state":"ON","power":21.04,"voltage":241.2,"consumption":3.56,"temperature":33}'
  zigbee-shepherd:request REQ --> ZDO:nodeDescReq +10s
  zigbee-shepherd:request REQ --> ZDO:nodeDescReq +0ms
  zigbee-shepherd:msgHdlr IND <-- ZDO:nodeDescRsp +63ms
  zigbee-shepherd:msgHdlr IND <-- ZDO:nodeDescRsp +1ms
  zigbee-shepherd:msgHdlr IND <-- AF:incomingMsg, transId: 0 +38s
  zigbee-shepherd:msgHdlr IND <-- AF:incomingMsg, transId: 0 +8s
  zigbee-shepherd:request REQ --> ZDO:nodeDescReq +13s
  zigbee-shepherd:request REQ --> ZDO:nodeDescReq +1ms
  zigbee-shepherd:msgHdlr IND <-- ZDO:nodeDescRsp +43ms
  zigbee-shepherd:msgHdlr IND <-- ZDO:nodeDescRsp +14ms
  zigbee-shepherd:msgHdlr IND <-- AF:incomingMsg, transId: 0 +36s
  zigbee-shepherd:msgHdlr IND <-- AF:incomingMsg, transId: 0 +10s
  zigbee-shepherd:request REQ --> ZDO:nodeDescReq +13s
  zigbee-shepherd:request REQ --> ZDO:nodeDescReq +1ms
  zigbee-shepherd:msgHdlr IND <-- ZDO:nodeDescRsp +48ms
  zigbee-shepherd:msgHdlr IND <-- ZDO:nodeDescRsp +17ms
  zigbee-shepherd:msgHdlr IND <-- AF:incomingMsg, transId: 0 +28s
  zigbee-shepherd:msgHdlr IND <-- AF:incomingMsg, transId: 0 +18s
  zigbee-shepherd:request REQ --> ZDO:nodeDescReq +13s
  zigbee-shepherd:request REQ --> ZDO:nodeDescReq +1ms
  zigbee-shepherd:msgHdlr IND <-- ZDO:nodeDescRsp +57ms
  zigbee-shepherd:msgHdlr IND <-- ZDO:nodeDescRsp +2ms
  zigbee-shepherd:msgHdlr IND <-- AF:incomingMsg, transId: 0 +13s
2018-6-14 10:52:46 INFO MQTT publish, topic: 'zigbee2mqtt/fan_masterbedroom_dd92', payload: '{"state":"ON","power":20.95,"voltage":null,"consumption":0,"temperature":33}'
```

Or could it just be a faulty plug?

ciotlosm, 2018-06-14:
Probably a faulty plug, unless some weird caching bug. Can you try to add consumers to the plug to force a bigger fluctuation?

subzero79, 2018-06-14:
I have a 6 speed+light ceiling fan, attached to it. It fluctuates in between 3 and 50W. The other plug is attached to an identical fan and works fine. I wanted to compare both voltages because both fans report different power consumption at same speed.

ciotlosm, 2018-06-14:
@subzero79 can you try to pair again the plug that reports "null"?

subzero79, 2018-06-14:
Yes, i've tried several times. Unfortunately i won't be able to re-pair it until three days for now as the plugs are in the attic.
Is there any way of seeing the raw incoming zigbee data from the plugs?

ciotlosm, 2018-06-14:
The highest "debug" level can be achieved by using `DEBUG=* npm start` as specified in the documentation. Nothing more above that can be activated without manually editing files to add more logging.

ciotlosm, 2018-06-15:
@subzero79 I'll close this issue for now. Feel free to open a new one if more plugs start displaying the same behaviour. Considering one is working properly, and repairing didn't fix the "null" one highest probability is a faulty plug.

closed 2018-06-15 by ciotlosm

subzero79, 2018-07-05:
Just an update, on this got two more plugs both report voltage. Now that we can see the incoming zigbee msg in debug, the voltage attribute (id 150) does simply not arrive.