Koenkk/zigbee2mqtt #122: Received (battery?) message of Xiaomi Button without pressing it

klein0r, 2018-06-14:
The last night some of my Philips Hue Devices turned on, because they are mapped to Xiaomi Buttons by using FHEM. After checking the log files I found this line:

2018.06.14 02:03:57 5: publish received for zigbee2mqtt/0x00158d000...., {"battery":"100.00","voltage":3062,"click":"single"}

Seems like I've received a battery event(?) which also contains a click event. In this case my home automation thought the button was pressed and turned the light on (in the middle of the night).

Have you ever had such an issue?

ciotlosm, 2018-06-14:
Do you have pets?

<sub>Sent with <a href="http://githawk.com">GitHawk</a></sub>

klein0r, 2018-06-14:
Yes, but the button is unreachable for our dog 😄

Koenkk, 2018-06-14:
Can you update to the last version, probably related to #82

klein0r, 2018-06-14:
@Koenkk I am using the latest docker image (which is not up to date): https://hub.docker.com/r/koenkk/zigbee2mqtt/builds/

Any plans to update the image?

ciotlosm, 2018-06-14:
@klein0r it was last pushed 14 hours ago: https://hub.docker.com/r/koenkk/zigbee2mqtt/tags/

Koenkk, 2018-06-14:
It is up-to-date, but this is done by travis and not by docker hub. See https://hub.docker.com/r/koenkk/zigbee2mqtt/tags/.

As you are running the last version, this looks like a new bug. Can you run with debug enabled?

`docker run -e "DEBUG=zigbee2mqtt:controller ..."`

klein0r, 2018-06-14:
Whops, wrong tab on docker hub. Sorry for that :)
Just pulled the latest image again and I will check if the problem persists.

Thanks for your fast response and the great repo!

closed 2018-06-14 by klein0r