Koenkk/zigbee2mqtt #126: Maximum call stack size exceeded

Disposaber, 2018-06-17:
I've tried pairing my devices (aqara wall switch with neutral) but at the 9th device zigbee2mqtt stops responding and this is what I get
```
^C2018-6-17 19:16:45 INFO MQTT publish, topic: 'zigbee2mqtt/bridge/state', payload: 'offline'
2018-6-17 19:16:45 INFO Disconnecting from MQTT server
events.js:1
(function (exports, require, module, __filename, __dirname) { // Copyright Joyent, Inc. and other Node contributors.
^

RangeError: Maximum call stack size exceeded
    at SerialPort.emit (events.js:1:1)
    at SerialPort._error (/opt/zigbee2mqtt/node_modules/serialport/lib/serialport.js:150:10)
    at SerialPort.close (/opt/zigbee2mqtt/node_modules/serialport/lib/serialport.js:370:17)
    at SerialPort.spErr (/opt/zigbee2mqtt/node_modules/cc-znp/lib/ccznp.js:79:22)
    at emitOne (events.js:116:13)
    at SerialPort.emit (events.js:211:7)
    at SerialPort._error (/opt/zigbee2mqtt/node_modules/serialport/lib/serialport.js:150:10)
    at SerialPort.close (/opt/zigbee2mqtt/node_modules/serialport/lib/serialport.js:370:17)
    at SerialPort.spErr (/opt/zigbee2mqtt/node_modules/cc-znp/lib/ccznp.js:79:22)
    at emitOne (events.js:116:13)
```
I've tried several times but it ends up dying and if I try to start zigbee2mqtt again I get this
```
> zigbee2mqtt@0.0.0 start /opt/zigbee2mqtt
> node index.js

2018-6-17 19:21:56 DEBUG Using zigbee-shepherd with settings: '{"net":{"panId":6754,"channelList":[11]},"dbPath":"/opt/zigbee2mqtt/data/database.db","sp":{"baudrate":115200,"rtscts":true}}'
2018-6-17 19:21:56 INFO Starting zigbee2mqtt version 0.0.0 (commit #4c4fcd7)
2018-6-17 19:21:56 INFO Starting zigbee-shepherd
  zigbee-shepherd:init zigbee-shepherd booting... +0ms
  zigbee-shepherd:request REQ --> SYS:osalNvRead +54ms
  zigbee-shepherd:request RSP <-- SYS:osalNvRead +3s
  zigbee-shepherd:init Coordinator initialize had an error: Error: request timeout
    at CcZnp.<anonymous> (/opt/zigbee2mqtt/node_modules/cc-znp/lib/ccznp.js:254:22)
    at Object.onceWrapper (events.js:315:30)
    at emitOne (events.js:116:13)
    at CcZnp.emit (events.js:211:7)
    at Timeout.<anonymous> (/opt/zigbee2mqtt/node_modules/cc-znp/lib/ccznp.js:233:18)
    at ontimeout (timers.js:498:11)
    at tryOnTimeout (timers.js:323:5)
    at Timer.listOnTimeout (timers.js:290:5) +15ms
2018-6-17 19:21:59 WARN Error while starting zigbee-shepherd, attemping to fix... (takes 60 seconds)
2018-6-17 19:22:59 INFO Starting zigbee-shepherd
  zigbee-shepherd:init zigbee-shepherd booting... +1m
  zigbee-shepherd:request REQ --> SYS:osalNvRead +13ms
  zigbee-shepherd:request RSP <-- SYS:osalNvRead +3s
  zigbee-shepherd:init Coordinator initialize had an error: Error: request timeout
    at CcZnp.<anonymous> (/opt/zigbee2mqtt/node_modules/cc-znp/lib/ccznp.js:254:22)
    at Object.onceWrapper (events.js:315:30)
    at emitOne (events.js:116:13)
    at CcZnp.emit (events.js:211:7)
    at Timeout.<anonymous> (/opt/zigbee2mqtt/node_modules/cc-znp/lib/ccznp.js:233:18)
    at ontimeout (timers.js:498:11)
    at tryOnTimeout (timers.js:323:5)
    at Timer.listOnTimeout (timers.js:290:5) +4ms
2018-6-17 19:23:02 ERROR Error while starting zigbee-shepherd!
2018-6-17 19:23:02 ERROR Failed to start
```

If I restart the raspberry I get this
```
2018-6-17 19:29:35 INFO MQTT publish, topic: 'zigbee2mqtt/bridge/state', payload: 'online'
2018-6-17 19:29:41 INFO 
        {"message":"request timeout","stack":"Error: request timeout\n    at CcZnp.<anonymous> (/opt/zigbee2mqtt/node_modules/cc-znp/lib/ccznp.js:254:22)\n    at Object.onceWrapper (events.js:315:30)\n    at emitOne (events.js:116:13)\n    at CcZnp.emit (events.js:211:7)\n    at Timeout.<anonymous> (/opt/zigbee2mqtt/node_modules/cc-znp/lib/ccznp.js:233:18)\n    at ontimeout (timers.js:498:11)\n    at tryOnTimeout (timers.js:323:5)\n    at Timer.listOnTimeout (timers.js:290:5)"}
```

Koenkk, 2018-06-18:
Can you check if this firmware solves the problem? https://github.com/Koenkk/Z-Stack-firmware/tree/dev/bin/coordinator/CC2531

Disposaber, 2018-06-19:
That firmware seems to have fixed the problem. I have paired 23 devices successfully, 12 routers (wall swiches) and 11 end devices (aqara sensors).

Disposaber, 2018-06-20:
I don't know if this is related to this issue but I still have a problem that makes zigbee2mqtt not usable. Right now I have 24 devices, 13 routers (wall switches and 1 wall socket) and 11 end devices. The cc2531 is the only thing connected to my pi, which is running HA, mqtt broker and zigbee2mqtt. The thing is that works wonderfully good, the response is fast and seems to be reliable. BUT the thing is that it stops working _randomly_. When I say that it stops working is that zigbee2mqtt is running and if I `sudo journalctl -u zigbee2mqtt.service -f` I can see that HA publish the messages just fine but isn't working. No errors. Just dies. I've noticed that when `permit_join: false` stops working ~2 hours later and the light of the stick is off. Today I've tried with `permit_join: true` and took around 8 hours to stop working but the light stayed on.

If I ctrl^c when this happens sometimes I get this [https://ibb.co/g6zjZd](url) . If I restart the zigbee2mqtt service I get

```
> zigbee2mqtt@0.0.0 start /opt/zigbee2mqtt
> node index.js

2018-6-20 20:00:09 DEBUG Using zigbee-shepherd with settings: '{"net":{"panId":6754,"channelList":[11]},"dbPath":"/opt/zigbee2mqtt/data/database.db","sp":{"baudrate":115200,"rtscts":true}}'
2018-6-20 20:00:09 INFO Starting zigbee2mqtt version 0.0.0 (commit #4c4fcd7)
2018-6-20 20:00:09 INFO Starting zigbee-shepherd
  zigbee-shepherd:init zigbee-shepherd booting... +0ms
  cc-znp The serialport /dev/ttyACM0 is opened. +5s
  zigbee-shepherd:request REQ --> SYS:osalNvRead +11ms
  cc-znp:SREQ --> SYS:osalNvRead, { id: 3840, offset: 0, len: 1, value: [ 85 ] } +15ms
  serialport write 8 bytes of data +43ms
  cc-znp:SRSP <-- SYS:osalNvRead, __timeout__ +3s
  zigbee-shepherd:request RSP <-- SYS:osalNvRead +3ms
  zigbee-shepherd:init Coordinator initialize had an error: Error: request timeout
    at CcZnp.<anonymous> (/opt/zigbee2mqtt/node_modules/cc-znp/lib/ccznp.js:254:22)
    at Object.onceWrapper (events.js:315:30)
    at emitOne (events.js:116:13)
    at CcZnp.emit (events.js:211:7)
    at Timeout.<anonymous> (/opt/zigbee2mqtt/node_modules/cc-znp/lib/ccznp.js:233:18)
    at ontimeout (timers.js:498:11)
    at tryOnTimeout (timers.js:323:5)
    at Timer.listOnTimeout (timers.js:290:5) +13ms
2018-6-20 20:00:17 WARN Error while starting zigbee-shepherd, attemping to fix... (takes 60 seconds)
  cc-znp The serialport /dev/ttyACM0 is closed. +35s
2018-6-20 20:01:17 INFO Starting zigbee-shepherd
  zigbee-shepherd:init zigbee-shepherd booting... +25s
  cc-znp The serialport /dev/ttyACM0 is opened. +5s
  zigbee-shepherd:request REQ --> SYS:osalNvRead +3ms
  cc-znp:SREQ --> SYS:osalNvRead, { id: 3840, offset: 0, len: 1, value: [ 85 ] } +2ms
  serialport write 8 bytes of data +4ms
  cc-znp:SRSP <-- SYS:osalNvRead, __timeout__ +3s
  zigbee-shepherd:request RSP <-- SYS:osalNvRead +1ms
  zigbee-shepherd:init Coordinator initialize had an error: Error: request timeout
    at CcZnp.<anonymous> (/opt/zigbee2mqtt/node_modules/cc-znp/lib/ccznp.js:254:22)
    at Object.onceWrapper (events.js:315:30)
    at emitOne (events.js:116:13)
    at CcZnp.emit (events.js:211:7)
    at Timeout.<anonymous> (/opt/zigbee2mqtt/node_modules/cc-znp/lib/ccznp.js:233:18)
    at ontimeout (timers.js:498:11)
    at tryOnTimeout (timers.js:323:5)
    at Timer.listOnTimeout (timers.js:290:5) +3ms
2018-6-20 20:01:26 ERROR Error while starting zigbee-shepherd!
2018-6-20 20:01:26 ERROR Failed to start
```
so that suggest that the stick "disconnects" and zigbee2mqtt lost its connection. But if I `ls -ltr /dev/tty*|tail -n 1` I see this it's connected but not as usually `crw-rw---- 1 root dialout 166,  0 Jun 20 20:23 /dev/ttyACM0` instead it's `crw--w---- 1 root tty 204,  64 Jun 19 18:59 /dev/ttyAMA0`.

If I `sudo reboot` all works fine again but in a matter of hours stops working.

Koenkk, 2018-06-20:
Can you try with this firmware: https://github.com/Koenkk/Z-Stack-firmware/tree/dev/coordinator/CC2531/bin

ciotlosm, 2018-06-20:
@Disposaber can you be a bit more specific here: "When I say that it stops working is that zigbee2mqtt is running and if I sudo journalctl -u zigbee2mqtt.service -f I can see that HA publish the messages just fine but isn't working. No errors. Just dies." ?

Is the service dead at that point? Are you getting normal logs for mqtt publish and you just don't see them in HA? I don't understand what you mean by "HA publish the messages just fine".

Disposaber, 2018-06-20:
@Koenkk sure. It will take me a while.

@ciotlosm sorry for not being clear. The service isn't dead, it's up and running but isn't working. Forget what I said, I was wrong about that.

ciotlosm, 2018-06-20:
@Disposaber if it's working, does it still show mqtt logs when stuff changes or you just don't get state changes recorded at all like the stick isn't sending information to zigbee2mqtt?

Disposaber, 2018-06-20:
@ciotlosm if I toggle a switch (by pressing it physically) it's not reflected in the log so I guess the problem is that the stick isn't communicating with zigbee2mqtt.

@Koenkk when pairing the 10th device I got this and zigbee2mqtt stopped working.
```
2018-6-20 22:47:41 ERROR 
        {"message":"Port is already open","stack":"Error: Port is already open\n    at SerialPort.open (/opt/zigbee2mqtt/node_modules/serialport/lib/serialport.js:156:24)\n    at CcZnp.init (/opt/zigbee2mqtt/node_modules/cc-znp/lib/ccznp.js:140:8)\n    at Promise.post (/opt/zigbee2mqtt/node_modules/q/q.js:1181:36)\n    at Promise.promise.promiseDispatch (/opt/zigbee2mqtt/node_modules/q/q.js:808:41)\n    at /opt/zigbee2mqtt/node_modules/q/q.js:1411:14\n    at runSingle (/opt/zigbee2mqtt/node_modules/q/q.js:137:13)\n    at flush (/opt/zigbee2mqtt/node_modules/q/q.js:125:13)\n    at _combinedTickCallback (internal/process/next_tick.js:131:7)\n    at process._tickCallback (internal/process/next_tick.js:180:9)"}
2018-6-20 22:47:42 ERROR 
        {"message":"request timeout","stack":"Error: request timeout\n    at CcZnp.<anonymous> (/opt/zigbee2mqtt/node_modules/cc-znp/lib/ccznp.js:254:22)\n    at Object.onceWrapper (events.js:315:30)\n    at emitOne (events.js:116:13)\n    at CcZnp.emit (events.js:211:7)\n    at Timeout.<anonymous> (/opt/zigbee2mqtt/node_modules/cc-znp/lib/ccznp.js:233:18)\n    at ontimeout (timers.js:498:11)\n    at tryOnTimeout (timers.js:323:5)\n    at Timer.listOnTimeout (timers.js:290:5)"}
```

Koenkk, 2018-06-21:
Could you try the firmware from: https://github.com/Koenkk/zigbee2mqtt/issues/134 ?

Disposaber, 2018-06-22:
I've paired 24 devices successfully (13 routers and 11 end devices). It's been working fine for 28 hours with that firmware and keeps working

Koenkk, 2018-06-23:
Just to be sure, can you post a md5 hash of the `.hex` file?

Disposaber, 2018-06-23:
Downloaded from https://github.com/Koenkk/Z-Stack-firmware/tree/experimental/coordinator/CC2531

Name: CC2531ZNP-Prod.hex
Size: 697099 bytes (680 KiB)
MD5: 93AFE7CA4A70BFE2B5A9C658E5AD58C5
SHA256: 5E3238A5507531DB151D76CF28036BE41060790F3ECDFA38F8094361CDB1DA4C

Koenkk, 2018-06-23:
Ok, it seems that you are not running the latest (with that firmware I couldn't join anymore enddevices after joining 1 enddevice) 

Last version:
```
koens-mbp:bin Koenkk$ md5 CC2531ZNP-Prod.hex 
MD5 (CC2531ZNP-Prod.hex) = 395ba95b3e80aea21f4a31b54906aa71
koens-mbp:bin Koenkk$ openssl sha -sha256 CC2531ZNP-Prod.hex 
SHA256(CC2531ZNP-Prod.hex)= 3d21e5f385d93f960fff72a5c801d73dfb362612047b19203664f82f8778c4b8
```

Disposaber, 2018-06-23:
After pairing 2 or so devices I had to reboot zigbee2mqtt in order to be able to pair the rest of my devices. But after pairing all of them it's been working flawlessly for 50 hours. I'm going to try the new firmware and this time I'll pair a total of 27 devices. It's going to be a bit of a hassle though.

Disposaber, 2018-06-23:
I've paired 14 routers and 13 end devices without errors. It's working fine for now.

Koenkk, 2018-06-23:
That is with the new firmware? (With the new firmware I expect you don't have to restart zigbee2mqtt after pairing 2 devices).

Disposaber, 2018-06-23:
I used the new firmware. I didn't have to restart zigbee2mqtt, the process was smooth as silk.

Koenkk, 2018-06-23:
Good! I'm also running this firmware with 21 devices paired. No problems yet.

Lets check how it runs for a week, in case of no problems, this will be the default firmware.

Koenkk, 2018-06-26:
@Disposaber is everything still running without problems?

Disposaber, 2018-06-26:
Yes, it's going strong 💪 .

kirovilya, 2018-06-27:
@Koenkk wait a minute... which config of new firmware are you use to compile? how many NWK_MAX_DEVICE_LIST and MAXMEMHEAP ?

Koenkk, 2018-06-27:
@kirovilya https://github.com/Koenkk/Z-Stack-firmware/tree/experimental/coordinator/CC2531

kirovilya, 2018-06-27:
@Koenkk 15? and how it works with 21 devices? or you have routers for mesh?
I see @Disposaber have 14 routers, so other 13 devices connected through routers.

Koenkk, 2018-06-27:
Yes, routers allow you to add more devices to the network.

kirovilya, 2018-06-27:
@Koenkk ok, now understand

Koenkk, 2018-06-30:
As everything is also still stable for me, I will close this issue.

This is the default firmware now.

closed 2018-06-30 by Koenkk