Koenkk/zigbee2mqtt #128: Clarification. Xiaomi sensor buttons cannot be used in automations?

subzero79, 2018-06-18:
Seems like there is an issue with the sensors implementation, either manual mqtt or discovery.

When you press single, you can see the event changed in the log. Another single press and event does not change and so on, unless you change to double or triple. Then the problem repeats if you do double again state won't change, but single or triple will work. Topic get published, as you can see in the screenshot, but state doesn't change. I've played with values expired_after and force_update but doesn't make the difference. Here is a screenshot

[image]

This creates a problem when using node-red and the home assistant plugin, also probably with standard yaml configurations. 

Workaround is to use mqtt topic as trigger, is this always gonna be like this or is suppose to change in the future?

ciotlosm, 2018-06-18:
@subzero79 as stated in the documentation, the way to make buttons work is through mqtt automation (see top of https://github.com/Koenkk/zigbee2mqtt/wiki/Integrating-with-Home-Assistant) 

The reason for this is that MQTT sensor in Home Assistant doesn't have any type of event fire. I have raised this here (for binary, but it's same for sensors): https://github.com/home-assistant/architecture/issues/33 however the official position so far was that an automation to fire events should be added. 

I recommend you close the issue and open an issue in Home Assistant.

subzero79, 2018-06-18:
Thanks for the clarification. I’ll keep an eye on the ha issue.

closed 2018-06-18 by subzero79