Koenkk/zigbee2mqtt #129: Support for eCozy Smart Thermostat

sebastianheierhoff, 2018-06-18:
Did anyone already manage/ try to implement support for the eCozy Smart Thermostat (http://ecozy.de/en)? I can pair the device, but do then get "Device not supported" warnings in the log. If noone else has done so, I will probably try to implement this according to https://github.com/Koenkk/zigbee2mqtt/wiki/How-to-support-new-devices.

I found an existing Hubitat Driver (looks quite helpful) and a SmartThings Device Handler.
Hubitat Driver: https://github.com/ckpt-martin/Hubitat/blob/master/eCozy/eCozy-ZigBee-Thermostat-Driver.groovy
SmartThings Device Handler: https://community.smartthings.com/t/ecozy-zigbee-thermostat-trv-device-handler/107869

Any suggestions/ experiences I could benefit from?

Regards,
Sebastian

tb-killa, 2018-06-18:
I buy one, it will be at home near the end of this week.
Look great and will bring smarthome to my radiator ;)
Lets look what we could do to integrate it to this cool project.

tb-killa, 2018-07-01:
Today I start my Device.
Got them into pairing Mode via push two buttons at the same time if the antenna comes on the display.

@Koenkk  @kirovilya Sadly i got this error:
```
cc-znp:AREQ <-- ZDO:tcDeviceInd, { nwkaddr: 56085, extaddr: '0x70b3d5de00000e2a', parentaddr: 0 } +13s
  zigbee-shepherd:msgHdlr IND <-- ZDO:tcDeviceInd +2ms

  cc-znp:AREQ <-- ZDO:endDeviceAnnceInd, { srcaddr: 56085, nwkaddr: 56085, ieeeaddr: '0x70b3d5de00000e2a', capabilities: 128 } +867ms
spinlock: false []
  zigbee-shepherd:request REQ --> ZDO:nodeDescReq +22ms
  cc-znp:SREQ --> ZDO:nodeDescReq, { dstaddr: 56085, nwkaddrofinterest: 56085 } +2ms
  serialport write 9 bytes of data +3ms
  zigbee-shepherd:msgHdlr IND <-- ZDO:endDeviceAnnceInd +3ms
  cc-znp:SRSP <-- ZDO:nodeDescReq, { status: 0 } +12ms

  cc-znp:AREQ <-- ZDO:nodeDescRsp, { srcaddr: 56085, status: 0, nwkaddr: 56085, logicaltype_cmplxdescavai_userdescavai: 18, apsflags_freqband: 64, maccapflags:128, manufacturercode: 4116, maxbuffersize: 89, maxintransfersize: 61, servermask: 0, maxouttransfersize: 61, descriptorcap: 0 } +2s
  zigbee-shepherd:msgHdlr IND <-- ZDO:nodeDescRsp +3ms
  zigbee-shepherd:request REQ --> ZDO:activeEpReq +3ms
  cc-znp:SREQ --> ZDO:activeEpReq, { dstaddr: 56085, nwkaddrofinterest: 56085 } +8ms
  serialport write 9 bytes of data +3ms
  cc-znp:SRSP <-- ZDO:activeEpReq, { status: 0 } +12ms

  cc-znp:AREQ <-- ZDO:activeEpRsp, { srcaddr: 56085, status: 0, nwkaddr: 56085, activeepcount: 1, activeeplist: <Buffer 03> } +1s
  zigbee-shepherd:msgHdlr IND <-- ZDO:activeEpRsp +2ms
  zigbee-shepherd:request REQ --> ZDO:simpleDescReq +2ms
  cc-znp:SREQ --> ZDO:simpleDescReq, { dstaddr: 56085, nwkaddrofinterest: 56085, endpoint: 3 } +2ms
  serialport write 10 bytes of data +10ms
  cc-znp:SRSP <-- ZDO:simpleDescReq, { status: 0 } +17ms
  cc-znp:AREQ <-- ZDO:simpleDescRsp, { srcaddr: 56085, status: 0, nwkaddr: 56085, len: 24, endpoint: 3, profileid: 260, deviceid: 769, deviceversion: 0, numinclusters: 7, inclusterlist: <Buffer 00 00 01 00 03 00 0a 00 20 00 01 02 04 02>, numoutclusters: 1, outclusterlist: <Buffer 02 04> } +972ms
  zigbee-shepherd:msgHdlr IND <-- ZDO:simpleDescRsp +3ms
  zigbee-shepherd:request REQ --> AF:dataRequest, transId: 1 +92ms
  cc-znp:SREQ --> AF:dataRequest, { dstaddr: 56085, destendpoint: 3, srcendpoint: 1, clusterid: 0, transid: 1, options: 48, radius: 30, len: 9, data: <Buffer 00 01 00 04 00 05 00 07 00> } +2ms
  serialport write 24 bytes of data +4ms
  cc-znp:SRSP <-- AF:dataRequest, { status: 0 } +11ms
  zigbee-shepherd:request RSP <-- AF:dataRequest, status: 0 +2ms
  cc-znp:AREQ <-- AF:dataConfirm, { status: 0, endpoint: 1, transid: 1 } +900ms
  zigbee-shepherd:af dispatchIncomingMsg(): type: dataConfirm, msg: [object Object] +10ms
  zigbee-shepherd:msgHdlr IND <-- AF:dataConfirm, transId: 1 +3ms
  cc-znp:AREQ <-- AF:incomingMsg, { groupid: 0, clusterid: 0, srcaddr: 56085, srcendpoint: 3, dstendpoint: 1, wasbroadcast: 0, linkquality: 0, securityuse: 0, timestamp: 14959676, transseqnumber: 0, len: 33, data: <Buffer 18 01 01 04 00 00 42 05 65 43 6f 7a 79 05 00 00 42 0a 54 68 65 72 6d 6f 73 74 61 74 07 00 00 30 03> } +16ms
  zigbee-shepherd:af dispatchIncomingMsg(): type: incomingMsg, msg: [object Object] +5ms
/opt/zigbee2mqtt/node_modules/zcl-packet/lib/foundation.js:683
        ru[dataType](name)(this);
                    ^

TypeError: ru[dataType] is not a function
    at Dissolve.<anonymous> (/opt/zigbee2mqtt/node_modules/zcl-packet/lib/foundation.js:683:21)
    at Dissolve._transform (/opt/zigbee2mqtt/node_modules/dissolve/index.js:80:16)
    at Dissolve.Transform._read (/opt/zigbee2mqtt/node_modules/dissolve/node_modules/readable-stream/lib/_stream_transform.js:184:10)
    at Dissolve.Transform._write (/opt/zigbee2mqtt/node_modules/dissolve/node_modules/readable-stream/lib/_stream_transform.js:172:12)
    at doWrite (/opt/zigbee2mqtt/node_modules/dissolve/node_modules/readable-stream/lib/_stream_writable.js:279:12)
    at writeOrBuffer (/opt/zigbee2mqtt/node_modules/dissolve/node_modules/readable-stream/lib/_stream_writable.js:266:5)
    at Dissolve.Writable.write (/opt/zigbee2mqtt/node_modules/dissolve/node_modules/readable-stream/lib/_stream_writable.js:211:11)
    at Dissolve.Writable.end (/opt/zigbee2mqtt/node_modules/dissolve/node_modules/readable-stream/lib/_stream_writable.js:426:10)
    at FoundPayload._getObj (/opt/zigbee2mqtt/node_modules/zigbee-shepherd/node_modules/zcl-packet/lib/foundation.js:242:12)
    at getObjCB (/opt/zigbee2mqtt/node_modules/zigbee-shepherd/node_modules/zcl-packet/lib/foundation.js:51:22)
```
This is my database entry:
```
{"id":4,"type":"EndDevice","ieeeAddr":"0x70b3d5de00000e2a","nwkAddr":56085,"manufId":4116,"epList":[3],"status":"online","joinTime":1530476834,"endpoints":{"3":{"profId":260,"epId":3,"devId":769,"inClusterList":[0,1,3,10,32,513,516],"outClusterList":[1026],"clusters":{}}},"_id":"Tt3LDdmyFUU9eZUv"}
```

It seems to be that we have to check in Details what happened.

kirovilya, 2018-07-02:
@tb-killa Add console.log(dataType, name) before /opt/zigbee2mqtt/node_modules/zcl-packet/lib/foundation.js:683 and show log.
I think it is unknown type or cluster

tb-killa, 2018-07-04:
Got it today:
```
2018-07-04T19:00:52.072Z - debug: Recieved zigbee message with data "0x70b3d5de00000e2a"
2018-07-04T19:00:52.077Z - info: Connecting with device...
2018-07-04T19:00:52.081Z - info: MQTT publish, topic: 'zigbee2mqtt/bridge/log', payload: '{"type":"pairing","message":"connecting with device"}'
2018-07-04T19:00:55.175Z - debug: Recieved zigbee message with data "0x70b3d5de00000e2a"
2018-07-04T19:00:55.180Z - info: Connecting with device...
2018-07-04T19:00:55.183Z - info: MQTT publish, topic: 'zigbee2mqtt/bridge/log', payload: '{"type":"pairing","message":"connecting with device"}'
2018-07-04T19:00:57.226Z - debug: Recieved zigbee message with data "0x70b3d5de00000e2a"
2018-07-04T19:00:57.232Z - info: Connecting with device...
2018-07-04T19:00:57.235Z - info: MQTT publish, topic: 'zigbee2mqtt/bridge/log', payload: '{"type":"pairing","message":"connecting with device"}'
2018-07-04T19:01:00.349Z - debug: Recieved zigbee message with data "0x70b3d5de00000e2a"
2018-07-04T19:01:00.351Z - info: Connecting with device...
2018-07-04T19:01:00.354Z - info: MQTT publish, topic: 'zigbee2mqtt/bridge/log', payload: '{"type":"pairing","message":"connecting with device"}'
2018-07-04T19:01:33.343Z - debug: Recieved zigbee message with data "0x70b3d5de00000e2a"
2018-07-04T19:01:33.346Z - info: Connecting with device...
2018-07-04T19:01:33.348Z - info: MQTT publish, topic: 'zigbee2mqtt/bridge/log', payload: '{"type":"pairing","message":"connecting with device"}'
2018-07-04T19:02:03.359Z - debug: Recieved zigbee message with data "0x70b3d5de00000e2a"
2018-07-04T19:02:03.362Z - info: Connecting with device...
2018-07-04T19:02:03.365Z - info: MQTT publish, topic: 'zigbee2mqtt/bridge/log', payload: '{"type":"pairing","message":"connecting with device"}'
2018-07-04T19:02:06.032Z - debug: Recieved zigbee message with data "0x70b3d5de00000e2a"
2018-07-04T19:02:06.036Z - info: Connecting with device...
2018-07-04T19:02:06.039Z - info: MQTT publish, topic: 'zigbee2mqtt/bridge/log', payload: '{"type":"pairing","message":"connecting with device"}'
2018-07-04T19:02:07.030Z - debug: Recieved zigbee message with data "0x70b3d5de00000e2a"
2018-07-04T19:02:07.035Z - info: Connecting with device...
2018-07-04T19:02:07.039Z - info: MQTT publish, topic: 'zigbee2mqtt/bridge/log', payload: '{"type":"pairing","message":"connecting with device"}'
2018-07-04T19:02:07.064Z - debug: Recieved zigbee message with data "0x70b3d5de00000e2a"
2018-07-04T19:02:07.067Z - info: Device incoming...
2018-07-04T19:02:07.069Z - info: MQTT publish, topic: 'zigbee2mqtt/bridge/log', payload: '{"type":"pairing","message":"device incoming"}'
2018-07-04T19:02:07.072Z - info: New device with address 0x70b3d5de00000e2a connected!
2018-07-04T19:02:07.103Z - info: MQTT publish, topic: 'zigbee2mqtt/bridge/log', payload: '{"type":"device_connected","message":"0x70b3d5de00000e2a"}'
2018-07-04T19:02:07.106Z - warn: Device with modelID 'Thermostat' is not supported.
2018-07-04T19:02:07.119Z - warn: Please see: https://github.com/Koenkk/zigbee2mqtt/wiki/How-to-support-new-devices
2018-07-04T19:02:07.123Z - debug: Recieved zigbee message with data "online"
2018-07-04T19:02:07.125Z - warn: Device with modelID 'Thermostat' is not supported.
2018-07-04T19:02:07.127Z - warn: Please see: https://github.com/Koenkk/zigbee2mqtt/wiki/How-to-support-new-devices
```
and my database.db entry:

```
{"id":3,"type":"EndDevice","ieeeAddr":"0x70b3d5de00000e2a","nwkAddr":43483,"manufId":4116,"manufName":"eCozy","powerSource":"Battery","modelId":"Thermostat","epList":[3],"status":"online","joinTime":1530730843,"endpoints":{"3":{"profId":260,"epId":3,"devId":769,"inClusterList":[0,1,3,10,32,513,516],"outClusterList":[1026],"clusters":{"genBasic":{"dir":{"value":1},"attrs":{"zclVersion":1,"appVersion":3,"stackVersion":2,"hwVersion":1,"manufacturerName":"eCozy","modelId":"Thermostat","dateCode":"20160826eCozy428","powerSource":3,"locationDesc":"\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000","deviceEnabled":1,"disableLocalConfig":0}},"genPowerCfg":{"dir":{"value":1},"attrs":{"batterySize":3,"batteryQuantity":4,"batteryAlarmMask":0,"batteryVoltage":29,"batteryVoltMinThres":24,"batteryManufacturer":"\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000","batteryAHrRating":270,"batteryRatedVoltage":15}},"genIdentify":{"dir":{"value":1},"attrs":{"identifyTime":0}},"genTime":{"dir":{"value":1},"attrs":{"time":40,"timeStatus":0,"timeZone":0,"dstStart":4294967295,"dstEnd":4294967295,"dstShift":-1,"standardTime":41,"localTime":41,"lastSetTime":4294967295,"validUntilTime":4294967295}},"genPollCtrl":{"dir":{"value":1},"attrs":{}},"hvacThermostat":{"dir":{"value":1},"attrs":{}},"hvacUserInterfaceCfg":{"dir":{"value":1},"attrs":{"tempDisplayMode":0,"keypadLockout":0}},"msTemperatureMeasurement":{"dir":{"value":2},"attrs":{}}}}},"_id":"RmlNfq1QybMuK9mE"}
```
We could now starting :)

tb-killa, 2018-07-10:
Implemented rudimental Stuff
```
{"id":3,"type":"EndDevice","ieeeAddr":"0x70b3d5de00000e2a","nwkAddr":5140,"manufId":4116,"manufName":"eCozy","powerSource":"Battery","modelId":"Thermostat","epList":[3],"status":"online","joinTime":1531241050,"endpoints":{"3":{"profId":260,"epId":3,"devId":769,"inClusterList":[0,1,3,10,32,513,516],"outClusterList":[1026],"clusters":{"genBasic":{"dir":{"value":1},"attrs":{"zclVersion":1,"appVersion":3,"stackVersion":2,"hwVersion":1,"manufacturerName":"eCozy","modelId":"Thermostat","dateCode":"20160826eCozy428","powerSource":3,"locationDesc":"\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000","deviceEnabled":1,"disableLocalConfig":0}},"genPowerCfg":{"dir":{"value":1},"attrs":{"batterySize":3,"batteryQuantity":4,"batteryAlarmMask":0,"batteryVoltage":29,"batteryVoltMinThres":24,"batteryManufacturer":"\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000","batteryAHrRating":270,"batteryRatedVoltage":15}},"genIdentify":{"dir":{"value":1},"attrs":{"identifyTime":0}},"genTime":{"dir":{"value":1},"attrs":{"time":37,"timeStatus":0,"timeZone":0,"dstStart":4294967295,"dstEnd":4294967295,"dstShift":-1,"standardTime":38,"localTime":38,"lastSetTime":4294967295,"validUntilTime":4294967295}},"genPollCtrl":{"dir":{"value":1},"attrs":{}},"hvacThermostat":{"dir":{"value":1},"attrs":{}},"hvacUserInterfaceCfg":{"dir":{"value":1},"attrs":{"tempDisplayMode":0,"keypadLockout":0}},"msTemperatureMeasurement":{"dir":{"value":2},"attrs":{}}}}},"_id":"4o1tUlZYsE5CCf5i"}
```
the acutally devices.js entry
```
// Ecozy 
	{
        zigbeeModel: ['Thermostat'],
        model: 'ecozy Thermostat',
        vendor: 'ecozy',
        description: '',
        supports: '',
        fromZigbee: [],
        toZigbee: [],
		configure: (ieeeAddr, shepherd, coordinator, callback) => {
            const device = shepherd.find(ieeeAddr, 3);
            const actions = [
                                // from https://github.com/ckpt-martin/Hubitat/blob/master/eCozy/eCozy-ZigBee-Thermostat-Driver.groovy
				(cb) => device.bind('hvacThermostat', coordinator, cb),
				(cb) => device.bind('hvacUserInterfaceCfg', coordinator, cb),
				(cb) => device.bind('msTemperatureMeasurement', coordinator, cb),
				(cb) => device.bind('genPowerCfg', coordinator, cb),
				(cb) => device.report('hvacThermostat', 'pIHeatingDemand', 300, 7200, 5, cb),
				(cb) => device.report('hvacThermostat', 'localTemp', 30, 0, 100, cb),
				(cb) => device.report('hvacThermostat', 'occupiedHeatingSetpoint', 30, 0, 100, cb),
				(cb) => device.report('hvacThermostat', 'systemMode', 1, 0, 1, cb),
				//(cb) => device.report('genPowerCfg', 'batteryVoltage', 600, 21600, 1, cb),
            ];

            execute(device, actions, callback);
        },
    },
```

Some Debug Entrys:
```
2018-7-10 18:52:47 DEBUG Recieved zigbee message with data {"cid":"hvacThermostat","data":{"occupiedHeatingSetpoint":2500,"setpointChangeSource":0,"setpointChangeAmount":500,"setpointChangeSourceTimeStamp":361}}
  zigbee-shepherd:af dispatchIncomingMsg(): type: incomingMsg, msg: [object Object] +41s
  zigbee-shepherd:msgHdlr IND <-- AF:incomingMsg, transId: 0 +14ms
  zigbee-shepherd:af dispatchIncomingMsg(): type: zclIncomingMsg, msg: [object Object] +3ms
2018-7-10 18:53:28 DEBUG Recieved zigbee message with data {"cid":"hvacThermostat","data":{"128":2542,"129":2568,"130":2500}}
2018-7-10 18:53:28 DEBUG Recieved zigbee message with data {"cid":"hvacThermostat","data":{"128":2542}}
  zigbee-shepherd:af dispatchIncomingMsg(): type: incomingMsg, msg: [object Object] +19s
  zigbee-shepherd:msgHdlr IND <-- AF:incomingMsg, transId: 0 +34ms
  zigbee-shepherd:af dispatchIncomingMsg(): type: zclIncomingMsg, msg: [object Object] +1ms
2018-7-10 18:53:47 DEBUG Recieved zigbee message with data {"cid":"hvacThermostat","data":{"occupiedHeatingSetpoint":2500,"setpointChangeSource":0,"setpointChangeAmount":500,"setpointChangeSourceTimeStamp":361}}
  zigbee-shepherd:af dispatchIncomingMsg(): type: incomingMsg, msg: [object Object] +3m
  zigbee-shepherd:msgHdlr IND <-- AF:incomingMsg, transId: 0 +14ms
  zigbee-shepherd:af dispatchIncomingMsg(): type: zclIncomingMsg, msg: [object Object] +1ms
2018-7-10 18:56:27 DEBUG Recieved zigbee message with data {"cid":"hvacThermostat","data":{"128":2509,"129":2532,"130":2500}}
2018-7-10 18:56:27 DEBUG Recieved zigbee message with data {"cid":"hvacThermostat","data":{"128":2509,"129":2532}}
  zigbee-shepherd:af dispatchIncomingMsg(): type: incomingMsg, msg: [object Object] +1m
  zigbee-shepherd:msgHdlr IND <-- AF:incomingMsg, transId: 0 +27ms
  zigbee-shepherd:af dispatchIncomingMsg(): type: zclIncomingMsg, msg: [object Object] +2ms
2018-7-10 18:57:47 DEBUG Recieved zigbee message with data {"cid":"hvacThermostat","data":{"occupiedHeatingSetpoint":2500,"setpointChangeSource":0,"setpointChangeAmount":500,"setpointChangeSourceTimeStamp":361}}
  zigbee-shepherd:af dispatchIncomingMsg(): type: incomingMsg, msg: [object Object] +60s
  zigbee-shepherd:msgHdlr IND <-- AF:incomingMsg, transId: 0 +27ms
  zigbee-shepherd:af dispatchIncomingMsg(): type: zclIncomingMsg, msg: [object Object] +1ms
2018-7-10 18:58:47 DEBUG Recieved zigbee message with data {"cid":"hvacThermostat","data":{"occupiedHeatingSetpoint":2500,"setpointChangeSource":0,"setpointChangeAmount":500,"setpointChangeSourceTimeStamp":361}}
```

Koenkk, 2018-07-12:
Looks good, can you make a PR so that we can start working on this?

tb-killa, 2018-07-12:
Yes, will implement to zigbee for testing too

tb-killa, 2018-07-14:
I make some more Test with my Code.
Modded like this:
```
{
        zigbeeModel: ['Thermostat'],
        model: 'ecozy Thermostat',
        vendor: 'ecozy',
        description: '',
        supports: '',
        fromZigbee: [fz.ecozy_hvacThermostat_attReport,fz.ignore_ecozy_devChange],
        toZigbee: [tz.genBasic_resetFactDefault],
		configure: (ieeeAddr, shepherd, coordinator, callback) => {
            const ep3 = shepherd.find(ieeeAddr, 3);
            const actions = [
                (cb) => ep3.bind('hvacThermostat', coordinator, cb),
				(cb) => setTimeout(() => cb(false), 100),
				(cb) => ep3.bind('genPowerCfg', coordinator, cb),
				(cb) => setTimeout(() => cb(false), 100),
				(cb) => ep3.bind('hvacUserInterfaceCfg', coordinator, cb),
				(cb) => setTimeout(() => cb(false), 100),
				(cb) => ep3.bind('genScenes', coordinator, cb),
				(cb) => setTimeout(() => cb(false), 100),
            ];

            execute(ep3, actions, (result) => {
                if (result) {
                    ep3.report('hvacThermostat', 'pIHeatingDemand', 300, 3600, 1, (err) => {
                        console.log('hvacThermostat pIHeatingDemand result', err);		
								
							ep3.report('hvacThermostat', 'localTemp', 60, 300, 1, (err) => { //Works
							console.log('hvacThermostat localTemp result', err);	
								
								ep3.report('hvacThermostat', 'occupiedHeatingSetpoint', 60, 300, 1, (err) => {
								console.log('hvacThermostat occupiedHeatingSetpoint result', err);
								
									ep3.report('hvacThermostat', 'systemMode', 1, 0, 1, (err) => {
									console.log('hvacThermostat systemMode result', err);
									
										ep3.report('hvacUserInterfaceCfg', 'keypadLockout', 1, 0, 1, (err) => {
										console.log('hvacUserInterfaceCfg keypadLockout result', err);
										
											ep3.report('hvacUserInterfaceCfg', 'tempDisplayMode', 1, 0, 1, (err) => {
											console.log('hvacUserInterfaceCfg tempDisplayMode result', err);
											
												ep3.report('genPowerCfg', 'batteryVoltage', 300, 3600, 0, (err) => {
													console.log('genPowerCfg batteryVoltage result', err);

													callback(!err);
												});
											});	
											
										});	
										
									});		
											
								});									
							});		
								
                    });
                } else {
                    callback(result);
                }
            });
        },
	},
```
But sadly, this are my Results:
```
hvacThermostat pIHeatingDemand result { Error: Timed out after 30000 ms
    at Timeout._onTimeout (/opt/zigbee2mqtt/node_modules/q/q.js:1846:21)
    at ontimeout (timers.js:498:11)
    at tryOnTimeout (timers.js:323:5)
    at Timer.listOnTimeout (timers.js:290:5) code: 'ETIMEDOUT' }
hvacThermostat localTemp result null
hvacThermostat occupiedHeatingSetpoint result unreportableAttribute
hvacThermostat systemMode result unreportableAttribute
hvacUserInterfaceCfg keypadLockout result unreportableAttribute
hvacUserInterfaceCfg tempDisplayMode result unreportableAttribute
genPowerCfg batteryVoltage result unreportableAttribute
2018-7-14 20:02:50 ERROR Failed to configure 0x70b3d5de00000e2a
```

tb-killa, 2018-07-21:
Start to implement some rudimental stuff: https://github.com/tb-killa/zigbee-shepherd-converters/tree/ecozy.
Will generate PR like Koenkk mentioned so we could work all together :)

tb-killa, 2018-07-21:
Need some help to figure out what exactly this data means:
```2018-7-21 20:25:54 DEBUG Recieved zigbee message with data {"cid":"hvacThermostat","data":{"128":2574,"129":2532,"130":2500}}```
This comes periodic from EndDevice and doesn´t changed.

sebastianheierhoff, 2018-08-08:
What happend to the ecozy code? I do finally have time to test this, but cannot find your changes in the current version.

sebastianheierhoff, 2018-08-20:
I corrected the indentation and created a new [PR](https://github.com/Koenkk/zigbee-shepherd-converters/pull/57).

tb-killa, 2018-08-21:
Hi actually I'm busy But will work on this if got some more free time. Yes the document is very good i got this too from eCozy directly ;)
I See you create some sort of my old Code as a PR. Thanks a lot. I will work with @Koenkk on a working Integration if He has some more time. The Code Fragment actually only Show global stuff from device pushing. We need All the stuff for manage device like temp or device time and much more.

ciB89, 2018-09-04:
I was thinking about ordering one too. Is there any way I could help once I got it paired?

sebastianheierhoff, 2018-09-24:
@tb-killa: "zigbee message with data {"cid":"hvacThermostat","data":{"128":2321,"129":2328,"130":2100}}" disappears after flashing the newest firmware.

tb-killa, 2018-09-24:
New Firmware?? Device one or zigbee2mqtt??

sebastianheierhoff, 2018-09-24:
I flashed https://github.com/Koenkk/Z-Stack-firmware/tree/master/coordinator/CC2531/bin, which I did not use before, and don't get the messages anymore

mkan1512, 2018-10-01:
I'm also trying to get this device working on my domotica system, I have version 0.1.3 running. Device is detected, but I do not see any messages ?
Any way to get device supported ?

sebastianheierhoff, 2018-10-01:
@mkan1512: Have a look at https://github.com/sebastianheierhoff/zigbee-shepherd-converters/tree/dev; together with https://github.com/sebastianheierhoff/zigbee2mqtt/tree/dev setting the temperature works fine. I will try to continue working on this asap.

Imperial-Guard, 2018-10-02:
> @mkan1512: Have a look at https://github.com/sebastianheierhoff/zigbee-shepherd-converters/tree/dev; together with https://github.com/sebastianheierhoff/zigbee2mqtt/tree/dev setting the temperature works fine. I will try to continue working on this asap.

Today we did some tests and it seems to be working! Setpoint temperature is working in both ways. We don't receive a measured temperature, any thoughts on this?

sebastianheierhoff, 2018-10-04:
@mkan1512: This is due to a bug in my current devices.js. Try changing the device.report for localTemp to: `(cb) => device.report('hvacThermostat', 'localTemp', 5, 30, 0, cb),`

mkan1512, 2018-10-10:
@sebastianheierhoff 
I try to find this line of code?
I found the devices.js in the ./node_modules/zigbee-shepherd-converters/ folder, but cannot find anything for eCozy ?

My steps : 
sudo git clone https://github.com/sebastianheierhoff/zigbee2mqtt.git /opt/zigbee2mqttDEV
sudo chown -R pi:pi /opt/zigbee2mqttDEV
cd /opt/zigbee2mqttDEV

Missing something ?

dortmund50, 2018-10-12:
I was able to connect the thermostat to my zigbee2mqtt, but how can i set the temperature?
This is the part of the logfile:
`Oct 12 16:55:39 hassbian npm[1575]:   zigbee2mqtt:error 2018-10-12 16:55:39 No converter available for 'state' ({
Oct 12 16:55:39 hassbian npm[1575]:   "occupiedHeatingSetpoint": "23",
`

nerdosity, 2018-10-24:
Hi, I can help too, just bought 5 of them.

Edzilla2000, 2018-11-06:
I just got one of those today, I'm very interested in getting it working.
Is there anything I can do to help?

Edzilla2000, 2018-11-06:
> @sebastianheierhoff
> I try to find this line of code?
> I found the devices.js in the ./node_modules/zigbee-shepherd-converters/ folder, but cannot find anything for eCozy ?
>
> My steps :
> sudo git clone [...]
You should change branch to dev:
`git checkout dev`

sebastianheierhoff, 2018-11-06:
> You should change branch to dev:
>
> `git checkout dev`
I think you need to checkout my version of zigbee-shepherd-converters (https://github.com/sebastianheierhoff/zigbee-shepherd-converters/tree/dev).

Edzilla2000, 2018-11-06:
> 
> 
> @mkan1512: This is due to a bug in my current devices.js. Try changing the device.report for localTemp to: `(cb) => device.report('hvacThermostat', 'localTemp', 5, 30, 0, cb),` and commenting out all other device.report. You should then receive a "Succesfully configured DEVICE ID" message when starting zigbee2mqtt and the temperature should be reported all 30 seconds.

This is working quite well. What would be needed for your changes to be included in the official zigbee-shepherd-converters ?

@alyberty can you share the MQTT HVAC definition that you have? I would be very interested!

sebastianheierhoff, 2018-11-06:
There is a bunch of other commands that I haven't finished implementing nor tested (see https://github.com/sebastianheierhoff/zigbee-shepherd-converters/blob/dev/converters/toZigbee.js). Furthermore, I think we should have a look at @Koenkk's changes in https://github.com/Koenkk/zigbee2mqtt/issues/518 and use this to read attributes.

sebastianheierhoff, 2018-11-08:
> Would it be worth doing a minimal commit of what you have working now, and expanding out further in a second pull request?
> 
> I've had trouble trying to get your changes to apply for testing (I have a problem with debounce not being found when I try), i'd really like to start playing with my ecozy :)

I don't think that's the correct way. Could you rather send us your error message?

nerdosity, 2018-11-14:
Hello, any official update on this? I couldn't see any sensor or other stuff in home assistant, even after using the dev from the fork and editing JS files.
I will glad test since I did not started using them, so I can mess around as you like.

sebastianheierhoff, 2018-11-14:
> Hello, any official update on this? I couldn't see any sensor or other stuff in home assistant, even after using the dev from the fork and editing JS files.
> I will glad test since I did not started using them, so I can mess around as you like.

Before thinking about home assistant: What do you see in the zigbee2mqtt debug? Did you manage to set the temperature using mqtt messages?

Edzilla2000, 2018-11-14:
> Would it be worth doing a minimal commit of what you have working now, and expanding out further in a second pull request?
>
> I've had trouble trying to get your changes to apply for testing (I have a [...]

I had the same issue, I solved it by going into /opt/zigbee2mqtt/node_modules/zigbee-shepherd-converters and running: 
`npm install`

nerdosity, 2018-11-16:
> Before thinking about home assistant: What do you see in the zigbee2mqtt debug? Did you manage to set the temperature using mqtt messages?

Can you please update your fork to latest dev? It's very difficult to use shell for me, I just had a surgery to my right bycep's tendon so I can only write with my left arm.
If someone make an updated git ready to pull I can gladly help but right now I just don' understand what I am missing.
I tried to:
- git cloning dev the fork
- inserting manually all the modification regarding ecozy
- commenting out the js lines

But after that, I just got the pairing message.

Koenkk, 2018-11-16:
FYI: I would accept any PR that partially support this device.

sebastianheierhoff, 2018-11-16:
Ok, if everyone feels that this would be easier, I will create a PR right now

allofmex, 2019-01-24:
Thanks for the good work here. I'm about to add Eurotronic Spirit thermostat too, so your work here helped a lot.
Some question remain: What is the point of occupied and unoccupied temperature settings. I am asking because the Eurotronic has both temperature settings, but it seems there is no support for any occupancy effecting attributes (occupancy, numberOfWeeklyTrans -> unsupAttribute, setWeeklySchedule -> unsupClusterCmd)!? It just uses unoccupied value. The manual value set by devices button is affecting a third (manufacture specific) temperature attribute. **Occupied value seems never to be used.**

How is it working for ecozy? How do you switch between occupied and unoccupied?

sebastianheierhoff, 2019-01-27:
To be honest, I didn't really have time to test this for the ecozy. I just tried to implement as much of the HVAC cluster (see http://www.zigbee.org/~zigbeeor/wp-content/uploads/2014/10/07-5123-06-zigbee-cluster-library-specification.pdf) as possible. What does 'remote_sensing' tell you (see pdf, p. 374/376)?

closed 2019-04-04 by stale[bot]

nerdosity, 2019-04-04:
Hi, 
I can't pair mine with my CC2531 with Zigbee2Mqtt.
After I push the blue antenna button, the antenna picture soon becomes red.
Even if I put the ecozy smart thermostat really near my CC2531 stick.

Any suggestion? I can upload a video.
Everything else is working fine with my configuration, HA and Zigbee2Mqtt latest version, firmware latest version.

simmsdk, 2019-11-29:
I have the same problem with one of my ecozy.
"After I push the blue antenna button, the antenna picture soon becomes red."
nerdosity, did you find a solution ?

Heronimonimo, 2020-06-12:
Just layed my hands one a few off these units. And while they I'm able to connect them to my Zigbee network, they don't seem to accept some of the commands described in the documentation. For example sending {"system_mode": "cool"} seems to have no effect.

Furthermore the battery level needs a conversion. Luckely there seems to be someone who wrote an adapter for the SmartThings hub: https://github.com/ckpt-martin/SmartThings/blob/master/eCozy/DeviceHandler/eCozy-ZigBee-Thermostat-DTH.groovy . He also figured out the mapping for the battery values.

@sebastianheierhoff @Koenkk @tb-killa Would one of you be able to have look?

simmsdk, 2020-06-12:
i contactede eCozy about the "After I push the blue antenna button, the antenna picture soon becomes red.", and they didnt care one bit... it means the firmware is corrupt and you now have a brick... Yes eCozy is pretty, but dont buy them, there is NO support.

Heronimonimo, 2020-06-12:
> i contactede eCozy about the "After I push the blue antenna button, the antenna picture soon becomes red.", and they didnt care one bit... it means the firmware is corrupt and you now have a brick... Yes eCozy is pretty, but dont buy them, there is NO support.

That's really unfortunate. Did they ship the unit as such or did it get bricked after a specific action?