Koenkk/zigbee2mqtt #130: Would this CC debugger work?

nphil, 2018-06-18:
Sorry about opening an issue just for this, but [would this thing on alibaba ](https://www.aliexpress.com/item/Emulator-CC-Debugger-Zigbee-CC1110-CC2530-CC2531-CC2540-Bluetooth-4-0-Emulator-smartrf04eb/32852012979.html?spm=2114.search0604.3.8.27de5f32M5kz68&ws_ab_test=searchweb0_0,searchweb201602_2_10152_10151_10065_10344_10130_10068_10324_10547_10342_10325_10546_10343_10340_10548_10341_10545_10696_10084_10083_10618_10307_10059_100031_10103_10624_10623_10622_10621_10620,searchweb201603_1,ppcSwitch_5&algo_expid=81bd6261-1c3f-4392-a8c4-06c8cbcf9807-1&algo_pvid=81bd6261-1c3f-4392-a8c4-06c8cbcf9807&priceBeautifyAB=0)effectively replace the CC debugger + downloaded cable on the Wiki?

Just trying to be a cheapskate.

tb-killa, 2018-06-18:
@nphil Yes i buy two of them and it work great. you also need the "adapter" or "downloading cable" to connect them to your stick!

nphil, 2018-06-18:
Thanks, so it's just a replacement for the expensive debugger.

closed 2018-06-18 by ciotlosm

xYellx, 2018-07-27:
@nphil @tb-killa  I also brought [this Debugger](https://de.aliexpress.com/item/Emulator-CC-Debugger-Zigbee-CC1110-CC2530-CC2531-CC2540-Bluetooth-4-0-Emulator-smartrf04eb/32852012979.html?spm=a2g0s.9042311.0.0.1c714c4dKW8M3v), the [download cable](https://de.aliexpress.com/item/Downloader-Kabel-Bluetooth-4-0-CC2540-zigbee-CC2531-Sniffer-software-protokoll-analyse-USB-dongle-BTool-Programmierer/32852546368.html?spm=a2g0s.9042311.0.0.1c714c4dKW8M3v) and the [CC2531](https://de.aliexpress.com/item/Drahtlose-Zigbee-CC2531-Sniffer-software-protokoll-analyse-Bare-Board-Paket-Protokoll-Analyzer-Modul-Usb-schnittstelle-Dongle/32865678088.html?spm=a2g0s.9042311.0.0.1c714c4dKW8M3v)  
But for the love of god the Led stays red, is there anything other required like updating the debugger? (no Windows PC here :/ )

tb-killa, 2018-07-27:
@xYellx Do you follow https://github.com/Koenkk/zigbee2mqtt/wiki/Getting-started#for-linuxmacos ??

tb-killa, 2018-07-27:
Maybe someone else could help ? I´m use it on Windows like the Wiki Article and there its running.

xYellx, 2018-08-01:
FYI @tb-killa  and everyone finding this from google/git

 I got my hands on a Windows 10 PC with administrative privileges and follow the wiki... 
except the Debugger LED stayed red.

The CC2531 is working fine for my [Aquara Switch](https://www.gearbest.com/access-control/pp_626695.html?wid=1433363)  on my RP 3b+ with hass.io and mosquitto

mwweissmann, 2020-03-01:
The cc-tool expects a CC-Debugger device with Device-Id 451:16a**0**, not 451:16a**e**; I also have a CC-Debugger with 451:16ae and I cannot get it to work with cc-tool:
Patching the Device-Id detection in the code only leads to a new error:
```libusb_control_transfer (out) failed, Pipe error```
I suspect that TI changed something in the CC-Debugger hardware's interface and therefore had to use a new Device-Id. And cc-tool cannot talk to that later generation. :(

tetti-de, 2020-03-17:
Maybe this helps: https://github.com/dashesy/cc-tool/commit/3ebc61763ff5d0dadbdc2f7163e85ca5d002bb0f

mwweissmann, 2020-03-18:
@tetti-de I tried that, too: Didn't help. I resorted to flashing via TI's Windows tool.