Koenkk/zigbee2mqtt #131: Attempting to add dimmer light switch

newzealandpaul, 2018-06-20:
HI all, I am trying to add a smart dimmer light switch. 

I paired it with zigbee2mqtt.

This was entry that was added to the database.db file:

{"id":6,"type":"Router","ieeeAddr":"0x00124b0014afea16","nwkAddr":49888,"manufId":0,"manufName":"Feibit Inc co.  ","powerSource":"Mains (single phase)","modelId":"FB56+ZSC05HG1.0","epList":[11,13],"status":"online","joinTime":1529473724,"endpoints":{"11":{"profId":260,"epId":11,"devId":257,"inClusterList":[0,3,4,5,6,8],"outClusterList":[],"clusters":{"genBasic":{"dir":{"value":1},"attrs":{}},"genIdentify":{"dir":{"value":1},"attrs":{}},"genGroups":{"dir":{"value":1},"attrs":{}},"genScenes":{"dir":{"value":1},"attrs":{}},"genOnOff":{"dir":{"value":1},"attrs":{}},"genLevelCtrl":{"dir":{"value":1},"attrs":{}}}},"13":{"profId":49246,"epId":13,"devId":57694,"inClusterList":[4096],"outClusterList":[4096],"clusters":{"lightLink":{"dir":{"value":3},"attrs":{}}}}},"_id":"PbFWQoU8HDxluWCY"}

This is the code I added to node_modules/zigbee-shepherd-converters/devices.js:

    'FB56+ZSC05HG1.0': {
        zigbeeModel: ['FB56+ZSC05HG1.0'],
        model: 'FB56+ZSC05HG1.0',
        vendor: 'XYZ',
        description: 'XYZ Switch',
        supports: 'on/off, brightness',
        fromZigbee: [],
        toZigbee: [],
    },

I would expect to see some Warnings ("No converter available for...") in the log when I use it, but I get nothing.

Any ideas what I am doing wrong

ciotlosm, 2018-06-20:
@newzealandpaul have you tried to use `DEBUG=* npm start` ?

newzealandpaul, 2018-06-20:
@ciotlosm Thanks, I didn't know about that.

If I hold down the reset pairing button, I get this:

>   cc-znp:AREQ <-- ZDO:leaveInd, { srcaddr: 49888, extaddr: '0x00124b0014afea16', request: 0, removechildren: 0, rejoin: 1 } +20s
>   zigbee-shepherd Device: 0x00124b0014afea16 leave the network. +20s
>   zigbee2mqtt:controller Recieved zigbee message with data 0x00124b0014afea16 +20s
> 2018-6-20 07:04:01 WARN Message without device!
>   zigbee-shepherd:msgHdlr IND <-- ZDO:leaveInd +6ms
>   cc-znp:AREQ <-- ZDO:leaveInd, { srcaddr: 49888, extaddr: '0x00124b0014afea16', request: 0, removechildren: 0, rejoin: 1 } +101ms
>   zigbee-shepherd:msgHdlr IND <-- ZDO:leaveInd +92ms
>   cc-znp:AREQ <-- ZDO:leaveInd, { srcaddr: 49888, extaddr: '0x00124b0014afea16', request: 0, removechildren: 0, rejoin: 1 } +100ms
>   zigbee-shepherd:msgHdlr IND <-- ZDO:leaveInd +99ms
>   cc-znp:AREQ <-- ZDO:leaveInd, { srcaddr: 49888, extaddr: '0x00124b0014afea16', request: 0, removechildren: 0, rejoin: 1 } +112ms
>   zigbee-shepherd:msgHdlr IND <-- ZDO:leaveInd +112ms
>   cc-znp:AREQ <-- ZDO:leaveInd, { srcaddr: 49888, extaddr: '0x00124b0014afea16', request: 0, removechildren: 0, rejoin: 1 } +90ms
>   zigbee-shepherd:msgHdlr IND <-- ZDO:leaveInd +90ms
>   cc-znp:AREQ <-- ZDO:leaveInd, { srcaddr: 49888, extaddr: '0x00124b0014afea16', request: 0, removechildren: 0, rejoin: 1 } +98ms
>   zigbee-shepherd:msgHdlr IND <-- ZDO:leaveInd +98ms
>   cc-znp:AREQ <-- ZDO:leaveInd, { srcaddr: 49888, extaddr: '0x00124b0014afea16', request: 0, removechildren: 0, rejoin: 1 } +101ms
>   zigbee-shepherd:msgHdlr IND <-- ZDO:leaveInd +101ms
>   cc-znp:AREQ <-- ZDO:leaveInd, { srcaddr: 49888, extaddr: '0x00124b0014afea16', request: 0, removechildren: 0, rejoin: 1 } +757ms
>   zigbee-shepherd:msgHdlr IND <-- ZDO:leaveInd +758ms
>   cc-znp:AREQ <-- ZDO:leaveInd, { srcaddr: 49888, extaddr: '0x00124b0014afea16', request: 0, removechildren: 0, rejoin: 1 } +11ms
>   zigbee-shepherd:msgHdlr IND <-- ZDO:leaveInd +10ms
> 

Then it appears to readd itself:

>   cc-znp:AREQ <-- ZDO:tcDeviceInd, { nwkaddr: 30490, extaddr: '0x00124b0014afea16', parentaddr: 0 } +9s
>   zigbee-shepherd:msgHdlr IND <-- ZDO:tcDeviceInd +9s
>   cc-znp:AREQ <-- ZDO:endDeviceAnnceInd, { srcaddr: 30490, nwkaddr: 30490, ieeeaddr: '0x00124b0014afea16', capabilities: 142 } +3s
> spinlock: false []
>   zigbee-shepherd:request REQ --> ZDO:nodeDescReq +3s
>   cc-znp:SREQ --> ZDO:nodeDescReq, { dstaddr: 30490, nwkaddrofinterest: 30490 } +4ms
>   serialport write 9 bytes of data +1m
>   zigbee-shepherd:msgHdlr IND <-- ZDO:endDeviceAnnceInd +5ms
>   cc-znp:AREQ <-- ZDO:endDeviceAnnceInd, { srcaddr: 30490, nwkaddr: 30490, ieeeaddr: '0x00124b0014afea16', capabilities: 142 } +18ms
> spinlock: true []
>   zigbee-shepherd:msgHdlr IND <-- ZDO:endDeviceAnnceInd +16ms
>   cc-znp:SRSP <-- ZDO:nodeDescReq, { status: 0 } +5ms
>   cc-znp:AREQ <-- ZDO:nodeDescRsp, { srcaddr: 30490, status: 0, nwkaddr: 30490, logicaltype_cmplxdescavai_userdescavai: 1, apsflags_freqband: 64, maccapflags: 142, manufacturercode: 0, maxbuffersize: 80, maxintransfersize: 160, servermask: 0, maxouttransfersize: 160, descriptorcap: 0 } +33ms
>   zigbee-shepherd:msgHdlr IND <-- ZDO:nodeDescRsp +38ms
>   zigbee-shepherd:request REQ --> ZDO:activeEpReq +1ms
>   cc-znp:SREQ --> ZDO:activeEpReq, { dstaddr: 30490, nwkaddrofinterest: 30490 } +3ms
>   serialport write 9 bytes of data +60ms
>   cc-znp:SRSP <-- ZDO:activeEpReq, { status: 0 } +16ms
>   cc-znp:AREQ <-- ZDO:activeEpRsp, { srcaddr: 30490, status: 0, nwkaddr: 30490, activeepcount: 2, activeeplist: <Buffer 0b 0d> } +28ms
>   zigbee-shepherd:msgHdlr IND <-- ZDO:activeEpRsp +46ms
>   zigbee-shepherd:request REQ --> ZDO:simpleDescReq +2ms
>   cc-znp:SREQ --> ZDO:simpleDescReq, { dstaddr: 30490, nwkaddrofinterest: 30490, endpoint: 11 } +5ms
>   serialport write 10 bytes of data +48ms
>   cc-znp:SRSP <-- ZDO:simpleDescReq, { status: 0 } +15ms
>   cc-znp:AREQ <-- ZDO:simpleDescRsp, { srcaddr: 30490, status: 0, nwkaddr: 30490, len: 20, endpoint: 11, profileid: 260, deviceid: 257, deviceversion: 2, numinclusters: 6, inclusterlist: <Buffer 00 00 03 00 04 00 05 00 06 00 08 00>, numoutclusters: 0, outclusterlist: <Buffer > } +31ms
>   zigbee-shepherd:msgHdlr IND <-- ZDO:simpleDescRsp +49ms
>   zigbee-shepherd:request REQ --> ZDO:simpleDescReq +2ms
>   cc-znp:SREQ --> ZDO:simpleDescReq, { dstaddr: 30490, nwkaddrofinterest: 30490, endpoint: 13 } +4ms
>   serialport write 10 bytes of data +51ms
>   cc-znp:SRSP <-- ZDO:simpleDescReq, { status: 0 } +17ms
>   cc-znp:AREQ <-- ZDO:simpleDescRsp, { srcaddr: 30490, status: 0, nwkaddr: 30490, len: 12, endpoint: 13, profileid: 49246, deviceid: 57694, deviceversion: 2, numinclusters: 1, inclusterlist: <Buffer 00 10>, numoutclusters: 1, outclusterlist: <Buffer 00 10> } +29ms
>   zigbee-shepherd:msgHdlr IND <-- ZDO:simpleDescRsp +48ms
>   zigbee-shepherd:request REQ --> AF:dataRequest, transId: 17 +18ms
>   cc-znp:SREQ --> AF:dataRequest, { dstaddr: 30490, destendpoint: 11, srcendpoint: 1, clusterid: 0, transid: 17, options: 48, radius: 30, len: 9, data: <Buffer 00 11 00 04 00 05 00 07 00> } +21ms
>   serialport write 24 bytes of data +66ms
>   cc-znp:SRSP <-- AF:dataRequest, { status: 0 } +15ms
>   zigbee-shepherd:request RSP <-- AF:dataRequest, status: 0 +17ms
>   cc-znp:AREQ <-- AF:dataConfirm, { status: 0, endpoint: 1, transid: 17 } +26ms
>   zigbee-shepherd:msgHdlr IND <-- AF:dataConfirm, transId: 17 +27ms
[... cut ...]
>   cc-znp:SREQ --> ZDO:nodeDescReq, { dstaddr: 30490, nwkaddrofinterest: 30490 } +60s
>   serialport write 9 bytes of data +1m
>   cc-znp:SRSP <-- ZDO:nodeDescReq, { status: 0 } +15ms
>   cc-znp:AREQ <-- ZDO:nodeDescRsp, { srcaddr: 30490, status: 0, nwkaddr: 30490, logicaltype_cmplxdescavai_userdescavai: 1, apsflags_freqband: 64, maccapflags: 142, manufacturercode: 0, maxbuffersize: 80, maxintransfersize: 160, servermask: 0, maxouttransfersize: 160, descriptorcap: 0 } +29ms
>   zigbee-shepherd:msgHdlr IND <-- ZDO:nodeDescRsp +46ms

Koenkk, 2018-06-20:
Can you check with this `devices.js` https://github.com/Koenkk/zigbee-shepherd-converters/blob/FB56ZSC05HG1.0/devices.js#L736 ?

newzealandpaul, 2018-06-21:
Thanks @Koenkk  I just updated zigbee-shepherd-converters and replaced the devices file.

I then reset the switch and in theory it will automatically find a new router.I then tried pressing the on/off and dummer buttons a number of time, but nothing happened. Here is the output 

2018-6-21 04:03:45 DEBUG Recieved zigbee message with data "0x00124b0014afea16"
2018-6-21 04:03:45 INFO Connecting with device...
2018-6-21 04:03:45 INFO MQTT publish, topic: 'zigbee2mqtt/bridge/log', payload: '{"type":"pairing","message":"connecting with device"}'
  zigbee-shepherd Device: 0x00124b0014afea16 join the network. +328ms
2018-6-21 04:03:45 DEBUG Recieved zigbee message with data "0x00124b0014afea16"
2018-6-21 04:03:45 INFO Device incoming...
2018-6-21 04:03:45 INFO MQTT publish, topic: 'zigbee2mqtt/bridge/log', payload: '{"type":"pairing","message":"device incoming"}'
2018-6-21 04:03:45 DEBUG Recieved zigbee message with data "online"
  cc-znp:AREQ <-- ZDO:tcDeviceInd, { nwkaddr: 55762, extaddr: '0x00124b0014afea16', parentaddr: 0 } +10s
  zigbee-shepherd:msgHdlr IND <-- ZDO:tcDeviceInd +1ms
  cc-znp:AREQ <-- ZDO:endDeviceAnnceInd, { srcaddr: 55762, nwkaddr: 55762, ieeeaddr: '0x00124b0014afea16', capabilities: 142 } +3s
spinlock: false []
  zigbee-shepherd:request REQ --> ZDO:nodeDescReq +5ms
  cc-znp:SREQ --> ZDO:nodeDescReq, { dstaddr: 55762, nwkaddrofinterest: 55762 } +2ms
  serialport write 9 bytes of data +3ms
  zigbee-shepherd:msgHdlr IND <-- ZDO:endDeviceAnnceInd +3ms
  cc-znp:SRSP <-- ZDO:nodeDescReq, { status: 0 } +16ms
  cc-znp:AREQ <-- ZDO:nodeDescRsp, { srcaddr: 55762, status: 0, nwkaddr: 55762, logicaltype_cmplxdescavai_userdescavai: 1, apsflags_freqband: 64, maccapflags: 142, manufacturercode: 0, maxbuffersize: 80, maxintransfersize: 160, servermask: 0, maxouttransfersize: 160, descriptorcap: 0 } +30ms
  zigbee-shepherd:msgHdlr IND <-- ZDO:nodeDescRsp +1ms
  zigbee-shepherd:request REQ --> ZDO:activeEpReq +3ms
  cc-znp:SREQ --> ZDO:activeEpReq, { dstaddr: 55762, nwkaddrofinterest: 55762 } +2ms
  serialport write 9 bytes of data +3ms
  cc-znp:SRSP <-- ZDO:activeEpReq, { status: 0 } +16ms
  cc-znp:AREQ <-- ZDO:activeEpRsp, { srcaddr: 55762, status: 0, nwkaddr: 55762, activeepcount: 2, activeeplist: <Buffer 0b 0d> } +29ms
  zigbee-shepherd:msgHdlr IND <-- ZDO:activeEpRsp +2ms
  zigbee-shepherd:request REQ --> ZDO:simpleDescReq +4ms
  cc-znp:SREQ --> ZDO:simpleDescReq, { dstaddr: 55762, nwkaddrofinterest: 55762, endpoint: 11 } +2ms
  serialport write 10 bytes of data +3ms
  cc-znp:SRSP <-- ZDO:simpleDescReq, { status: 0 } +16ms
  cc-znp:AREQ <-- ZDO:simpleDescRsp, { srcaddr: 55762, status: 0, nwkaddr: 55762, len: 20, endpoint: 11, profileid: 260, deviceid: 257, deviceversion: 2, numinclusters: 6, inclusterlist: <Buffer 00 00 03 00 04 00 05 00 06 00 08 00>, numoutclusters: 0, outclusterlist: <Buffer > } +32ms
  zigbee-shepherd:msgHdlr IND <-- ZDO:simpleDescRsp +2ms
  zigbee-shepherd:request REQ --> ZDO:simpleDescReq +3ms
  cc-znp:SREQ --> ZDO:simpleDescReq, { dstaddr: 55762, nwkaddrofinterest: 55762, endpoint: 13 } +3ms
  serialport write 10 bytes of data +2ms
  cc-znp:SRSP <-- ZDO:simpleDescReq, { status: 0 } +16ms
  cc-znp:AREQ <-- ZDO:simpleDescRsp, { srcaddr: 55762, status: 0, nwkaddr: 55762, len: 12, endpoint: 13, profileid: 49246, deviceid: 57694, deviceversion: 2, numinclusters: 1, inclusterlist: <Buffer 00 10>, numoutclusters: 1, outclusterlist: <Buffer 00 10> } +33ms
  zigbee-shepherd:msgHdlr IND <-- ZDO:simpleDescRsp +2ms
  zigbee-shepherd:request REQ --> AF:dataRequest, transId: 10 +23ms
  cc-znp:SREQ --> AF:dataRequest, { dstaddr: 55762, destendpoint: 11, srcendpoint: 1, clusterid: 0, transid: 10, options: 48, radius: 30, len: 9, data: <Buffer 00 0a 00 04 00 05 00 07 00> } +3ms
  serialport write 24 bytes of data +4ms
  cc-znp:SRSP <-- AF:dataRequest, { status: 0 } +17ms
  zigbee-shepherd:request RSP <-- AF:dataRequest, status: 0 +1ms
  cc-znp:AREQ <-- AF:dataConfirm, { status: 0, endpoint: 1, transid: 10 } +24ms
  zigbee-shepherd:msgHdlr IND <-- AF:dataConfirm, transId: 10 +2ms
  cc-znp:AREQ <-- AF:incomingMsg, { groupid: 0, clusterid: 0, srcaddr: 55762, srcendpoint: 11, dstendpoint: 1, wasbroadcast: 0, linkquality: 99, securityuse: 0, timestamp: 4591743, transseqnumber: 0, len: 49, data: <Buffer 18 0a 01 04 00 00 42 10 46 65 69 62 69 74 20 49 6e 63 20 63 6f 2e 20 20 05 00 00 42 0f 46 42 35 36 2b 5a 53 43 30 35 48 47 31 2e 30 07 00 00 30 01> } +20ms
  zigbee-shepherd:msgHdlr IND <-- AF:incomingMsg, transId: 0 +28ms
  zigbee-shepherd Identified Device: { manufacturer: Feibit Inc co.  , product: FB56+ZSC05HG1.0 } +4ms
  zigbee-shepherd:request REQ --> AF:dataRequest, transId: 11 +16ms
  cc-znp:SREQ --> AF:dataRequest, { dstaddr: 55762, destendpoint: 11, srcendpoint: 1, clusterid: 0, transid: 11, options: 48, radius: 30, len: 6, data: <Buffer 00 0b 0c 00 00 f0> } +6ms
  serialport write 21 bytes of data +2ms
  cc-znp:SRSP <-- AF:dataRequest, { status: 0 } +14ms
  zigbee-shepherd:request RSP <-- AF:dataRequest, status: 0 +1ms
  cc-znp:AREQ <-- AF:dataConfirm, { status: 0, endpoint: 1, transid: 11 } +23ms
  zigbee-shepherd:msgHdlr IND <-- AF:dataConfirm, transId: 11 +1ms
  cc-znp:AREQ <-- AF:incomingMsg, { groupid: 0, clusterid: 0, srcaddr: 55762, srcendpoint: 11, dstendpoint: 1, wasbroadcast: 0, linkquality: 97, securityuse: 0, timestamp: 4592110, transseqnumber: 0, len: 5, data: <Buffer 18 0b 0b 0c 82> } +13ms
  zigbee-shepherd:msgHdlr IND <-- AF:incomingMsg, transId: 0 +4ms
[... cut ...]
  cc-znp:SRSP <-- ZDO:nodeDescReq, { status: 0 } +17ms
  cc-znp:AREQ <-- ZDO:nodeDescRsp, { srcaddr: 55762, status: 0, nwkaddr: 55762, logicaltype_cmplxdescavai_userdescavai: 1, apsflags_freqband: 64, maccapflags: 142, manufacturercode: 0, maxbuffersize: 80, maxintransfersize: 160, servermask: 0, maxouttransfersize: 160, descriptorcap: 0 } +30ms
  zigbee-shepherd:msgHdlr IND <-- ZDO:nodeDescRsp +2ms

And here is a photo of the device (it has a TI CC2530 chip:

[image]

Thanks so much for your help.

ciotlosm, 2018-06-21:
@newzealandpaul please use code blocks instead of blockquotes when formatting logs or code. If it's easier, for longer logs please use hastebin.

Koenkk, 2018-06-21:
Could you try with the updated `devices.js` https://github.com/Koenkk/zigbee-shepherd-converters/blob/FB56ZSC05HG1.0/devices.js ?

newzealandpaul, 2018-06-22:
@Koenkk I added the file, restarted the server, held down the reset button so it repaired, then tried pressing buttons (on/off, dimmer+ dimmer-):

https://gist.github.com/newzealandpaul/13680d0d58410c336addfae11bbbac11

Pressing buttons did not appear to send any messages. Weird! 

(I tried hastebin but it kept giving me errors and not saving the file)

newzealandpaul, 2018-06-23:
@Koenkk Thanks so much for your help. I don't think the dimmer is working. I am going to return it, and order a range of their other devices and try get them to work. I will let you know how I go, hopefully I can support the entire range.

closed 2018-06-23 by newzealandpaul