Koenkk/zigbee2mqtt #133: CC2531 LED control

haskimo, 2018-06-20:
In normal working state CC2531 LED is always on in very bright green color

If it's possible would be great to make it more usable like flashing when sending/receiving data, making it less bright, or turn it off

tb-killa, 2018-06-20:
This need modification into the firmware themselve.
Maybe we could bring in some sort of "AT-Command" ?

aymeric106, 2018-07-14:
Yes definitely turn it off ... it lights the whole room ... thank you

okueng, 2018-07-19:
I did a bit of research on this one, as the green light is indeed a bit too bright.
I have found that cc-znp is already able to turn on/off the LED without needing to tinker with the firmware. I tested it, and it works on a dummy project :)

Now I will try to see how this binds with zigbee-shepherd and zigbee2mqttt to have a command to turn it off.

here is the code that turns off : 
`var ccznp = require('cc-znp');

var spCfg = {
    path: "/dev/ttyACM0",
    options: {
        baudrate: 115200,
        rtscts: true
    }
};

ccznp.on('ready', function () {
    ccznp.utilRequest('ledControl', {ledid:3,mode:0}, function (err, result) {
        if (err)
            console.log(err);
        else
            console.log(result);
    });

});

ccznp.init(spCfg, function (err) {
    if (err)
        console.log(err);
});`

tb-killa, 2018-07-20:
Nice, will do some Tests with your Code. Would you Show us the sources of your Information please.? :)

ryanbeaton, 2018-07-20:
With cc-znp approach is it call once and stays off forever (like happens with the Aeotec Z-Stick) or do you have to call it every startup?

<sub>Sent with <a href="http://githawk.com">GitHawk</a></sub>

okueng, 2018-07-20:
@tb-killa : I first started to parse the firmware for LED references, found that some functions were already written. Following that trail, I stumbled upon some code in the cc-znp that was related to LED and documented here: https://github.com/zigbeer/cc-znp/wiki/6.-Z-Stack-MT-API-Reference-Tables#66-util-table

@ryanbeaton : looks like it stays off until you unplug/unpower the stick, where you have to send again the command to turn the LED off. However if you restart the server without unpowering the stick, it stays off.

okueng, 2018-07-20:
As an experiment, I have successfully silenced the green LED when running zigbee2mqtt by adding a line in the zigbee-shepherd repo: https://github.com/okueng/zigbee-shepherd/blob/e33e221c8020f1b95a94bd56614a23df22c5d82c/lib/initializers/init_shepherd.js#L23

Do you have any suggestion where else should I put this function? Ideally I guess it could be an optional param in the configuration.yaml

Koenkk, 2018-07-21:
@okueng, thanks, implemented! (in dev branch for now)

closed 2018-07-21 by Koenkk

osmartjr, 2018-07-30:
Is this also working for the home assistant plug-in?
I put disable_led: true in configuration.yaml and restarted the service, but the LED did not go off.
Sorry for my English, it's not my native language.

Koenkk, 2018-07-31:
Are you running the latest firmware and zigbee2mqtt version 0.1.1?

Koenkk, 2018-07-31:
Can you post your `configuration.yaml` (without urls and passwords)

osmartjr, 2018-07-31:
```
devices:
  '0x00158d0002045fab':
    friendly_name: garagem
    retain: true
homeassistant: true
mqtt:
  base_topic: zigbee2mqtt
  password: xxxxxxxxxx
  server: mqtt://xxx.xxx.x.xxx
  user: xxxxxxx
permit_join: false
serial:
  disable_led: true
  port: /dev/ttyACM0
```

Koenkk, 2018-08-01:
Can you post the log of `DEBUG=* npm start`

jwillmer, 2018-09-05:
This option needs to be highlighted! What a relieve :heart:

imperialjoy, 2018-10-01:
This would be wonderful to control programmatically!  This could be a useful status light for me, since its near the bathroom which has a motion sensor.  The light could come on when the bathroom was occupied, off when empty!  Is there any way to integrate this light with home assistant?