Koenkk/zigbee2mqtt #135: Request unsuccess

kaelsaes, 2018-06-20:
10]:         {"message":"request unsuc
cess: 16","stack":"Error: request unsuccess: 16\n    at /opt/zigbee2mqtt/n
ode_modules/zigbee-shepherd/lib/components/zdo.js:46:21\n    at CcZnp.<ano
nymous> (/opt/zigbee2mqtt/node_modules/cc-znp/lib/ccznp.js:257:13)\n    at
 Object.onceWrapper (events.js:315:30)\n    at emitOne (events.js:116:13)\
n    at CcZnp.emit (events.js:211:7)\n    at CcZnp._mtIncomingDataHdlr (/o
pt/zigbee2mqtt/node_modules/cc-znp/lib/ccznp.js:357:14)\n    at /opt/zigbe
e2mqtt/node_modules/cc-znp/lib/ccznp.js:333:22\n    at _combinedTickCallba
ck (internal/process/next_tick.js:131:7)\n    at process._tickCallback (in
ternal/process/next_tick.js:180:9)"}

ciotlosm, 2018-06-20:
@kaelsaes which versions of node and npm are you using?

kaelsaes, 2018-06-20:
node 8.11.3
npm 5.6.0

Thanks

ciotlosm, 2018-06-20:
Can you please post 1st line from your zigbee2mqtt logs? Need to get the actual version (including hash).

kaelsaes, 2018-06-20:
2018-06-20T12:39:32.361Z - info: Starting zigbee2mqtt version 0.0.0 (commit #4c4fcd7)

ciotlosm, 2018-06-21:
Can you try a fresh install?
Were you able to start zigbee2mqtt at any time or it was like this since you tried to start using it?
What OS are you using? Did you double check that serial port is mapped correctly in configuration?
Do you have correct permissions for zigbee2mqtt to write to database.db?

Koenkk, 2018-06-21:
- What device have you paired?
- Is zigbee2mqtt still working after this error?

kaelsaes, 2018-06-21:
I'm using 9 Xiaomi Aqara devices and all are working fine.

RPi 3 with Raspbian Stretch.

kaelsaes, 2018-06-21:
I deleted the log.txt and rebootet the RPi3

After restart I have noticed that the time stamp in the log file is 2 hours too early. 

The error still occurs.

How can I delete the actual installation and reinstall again?

ciotlosm, 2018-06-21:
@kaelsaes how did you install? Hassio addon or [using these instructions](https://github.com/Koenkk/zigbee2mqtt/wiki/Running-the-bridge#2-installing)?

- If the addon, then just uninstall and reinstall it as documented: [Updating the Add-on and zigbee2mqtt Library](https://github.com/danielwelch/hassio-zigbee2mqtt#updating-the-add-on-and-zigbee2mqtt-library)
- If using curl and npm install just use this documentation: [Update zigbee2mqtt to the latest version](https://github.com/Koenkk/zigbee2mqtt/wiki/Running-the-bridge#6-for-later-update-zigbee2mqtt-to-the-latest-version)

kaelsaes, 2018-06-21:
@ciotlosm 
I already updated with this documentation Update zigbee2mqtt to the latest version

ciotlosm, 2018-06-21:
@kaelsaes is it working now?

Koenkk, 2018-06-21:
But does this error crash your zigbee2mqtt? (I also experience this error sometimes but it does not cause a crash).

kaelsaes, 2018-06-21:
I flashed the firmware from Oskar Neumann. Maybe that's the problem.

zigbee2mqtt works fine despite the error.

kaelsaes, 2018-06-21:
Thanks @ciotlosm  and @Koenkk 

With your zigbee2mqtt firmware for the C22531 the problem is solved!

Koenkk, 2018-06-21:
@kaelsaes  thanks!

closed 2018-06-21 by Koenkk