Koenkk/zigbee2mqtt #136: Xiaomi MiJia Honeywell smoke detector , how to pair ?

kroon040, 2018-06-21:
Hi

I'm trying to connect to Xiaomi MiJia Honeywell smoke detector. It works a little. It says connected, and the it's paring but it response not every x time. 

Does de xiaomi smoke report every 50 minutes also, or ?? It I look at documentation is should give battery en voltage??

I know pressed the button 3x times short is that the procedure for pairing?? And that wait? Should it also be unpaired?

When I know press the button it reports but I don't get every x times a message that smoke is false with battery and voltage.

This is how the log looks like

2018-6-21 06:39:21 INFO Starting zigbee2mqtt version 0.0.0 (commit #9dbdfc0)
2018-6-21 06:39:21 INFO Starting zigbee-shepherd
2018-6-21 06:39:22 INFO zigbee-shepherd started
2018-6-21 06:39:22 INFO Currently 0 devices are joined:
2018-6-21 06:39:22 WARN `permit_join` set to  `true` in configuration.yaml.
2018-6-21 06:39:22 WARN Allowing new devices to join.
2018-6-21 06:39:22 WARN Set `permit_join` to `false` once you joined all devices.
2018-6-21 06:39:22 INFO Zigbee: allowing new devices to join.
2018-6-21 06:39:22 INFO Connecting to MQTT server at mqtt://192.168.180.10
2018-6-21 06:39:22 INFO zigbee-shepherd ready
2018-6-21 06:39:22 INFO Connected to MQTT server
2018-6-21 06:39:22 INFO MQTT publish, topic: 'zigbee2mqtt/bridge/state', payload: 'online'
spinlock: false []
2018-6-21 06:39:57 INFO New device with address 0x00158d0001fd66e2 connected!
2018-6-21 06:39:57 INFO MQTT publish, topic: 'zigbee2mqtt/bridge/log', payload: '{"type":"device_connected","message":"0x00158d0001fd66e2"}'
2018-6-21 06:40:03 INFO Connecting with device...
2018-6-21 06:40:03 INFO MQTT publish, topic: 'zigbee2mqtt/bridge/log', payload: '{"type":"pairing","message":"connecting with device"}'
2018-6-21 06:40:06 INFO Connecting with device...
2018-6-21 06:40:06 INFO MQTT publish, topic: 'zigbee2mqtt/bridge/log', payload: '{"type":"pairing","message":"connecting with device"}'
2018-6-21 06:40:10 INFO Connecting with device...
2018-6-21 06:40:10 INFO MQTT publish, topic: 'zigbee2mqtt/bridge/log', payload: '{"type":"pairing","message":"connecting with device"}'
2018-6-21 06:40:13 INFO Connecting with device...
2018-6-21 06:40:13 INFO MQTT publish, topic: 'zigbee2mqtt/bridge/log', payload: '{"type":"pairing","message":"connecting with device"}'
2018-6-21 06:40:16 INFO Connecting with device...
2018-6-21 06:40:16 INFO MQTT publish, topic: 'zigbee2mqtt/bridge/log', payload: '{"type":"pairing","message":"connecting with device"}'
2018-6-21 06:40:46 INFO Connecting with device...
2018-6-21 06:40:46 INFO MQTT publish, topic: 'zigbee2mqtt/bridge/log', payload: '{"type":"pairing","message":"connecting with device"}'
2018-6-21 06:40:50 INFO Connecting with device...
2018-6-21 06:40:50 INFO MQTT publish, topic: 'zigbee2mqtt/bridge/log', payload: '{"type":"pairing","message":"connecting with device"}'
2018-6-21 06:40:50 INFO Device incoming...
2018-6-21 06:40:50 INFO MQTT publish, topic: 'zigbee2mqtt/bridge/log', payload: '{"type":"pairing","message":"device incoming"}'
2018-6-21 06:41:45 INFO MQTT publish, topic: 'zigbee2mqtt/0x00158d0001fd66e2', payload: '{"smoke":false}'

ciotlosm, 2018-06-21:
@kroon040 Your sensor is working correctly. You would not get updates unless your sensor changes state. I think it's a binary sensor. 

You will still get updates for battery around every hour (not for state), but unfortunately I think as the sensor is currently a binary sensor you won't see these. There is a missing feature for mqtt binary_sensors in home assistant not supporting json attributes. 

`zigbee2mqtt` does support battery for most sensors (haven't checked the Xiaomi MiJia Honeywell), but these won't get processed yet by Home Assistant. 

There is a pull request on Home Assistant side here: https://github.com/home-assistant/home-assistant/pull/14957 to cover this, but it's currently a bit controversial.

ciotlosm, 2018-06-29:
@kroon040 do you need more information or are you happy to close this issue?

closed 2018-06-29 by kroon040

diogopms, 2018-12-17:
I'm having issues, paring the  Xiaomi MiJia Honeywell smoke detector using zigbee2mqtt.
I have only one button, that always triggers a very noise sound.

plantoschka, 2019-02-09:
So what's the correct process to pair MiJia Honeywell Smoke Detector? How long or how often do I need to click the button?

hipohop, 2019-02-21:
pressing the button 3 times worked for me

Stefan300381, 2019-02-25:
actually it didn't work for me, at least not reliable. what helped was a long press. unfortunately the smoke detector will then play a test tone, which is a bad side effect. anyway, it worked for me. the same way I added the xiaomi aqara body sensors (looooong press for about 5 seconds, plus short press every second).

PhilippZ, 2021-02-06:
I finally got it working. You have to press the front button 3 times quickly, but not too quickly. It takes a few tries.

RualdOrdelman, 2022-02-20:
Hmm got the same problem as some others. Can't pair that Honeywell with my Zigbee network. 

Before I used the sensor with an Aquara Gateway. So first I unpaired and removed the device from the gateway and Mi Home app. After that I reset the sensor by holding the front button for 10 seconds and with a lot of beeps. Now I wanted to pair to Zigbee2mqtt, so I put the pairing mode on and pushed the front button 3 times. With no succes. :)

I tried with several intervals. Push tree times very quick and with an second between the pushes. All with no luck. after that I tried to give the unit a powercycle, reset again en try to add but no luck. 

I can't find any traffic in the log when pushing te front button. Anyone any suggestion?