Koenkk/zigbee2mqtt #138: Xiaomi Aqara Temperature Sensor - Friendly Name in Entity Name?

jarrah31, 2018-06-21:
I'm starting to test/integrate my other Xiaomi devices with zigbee2mqtt and noticed that the Aqara temperature sensor (WSDCGQ11LM) entity name doesn't change to include the friendly name like the door sensors do. Is this normal/expected behaviour? 

e.g. Z2M configuration.yaml:
```
devices:
  '0x00158d0001b149eb':
    friendly_name: window_ensuite_velux
  '0x00158d000245b389':
    friendly_name: window_bed2_mainleft
  '0x00158d0001b9560f':
    friendly_name: temperature_test
```
After a Z2M and HA restart, the temp sensor looks like this:
[image]

Whereas the door sensor looks like this:
[image]

Thanks in advance!

Koenkk, 2018-06-21:
After you change the name you need to reset zigbee2mqtt first and than home assistant.

jarrah31, 2018-06-21:
Hi @Koenkk - just to confirm I'm doing this right, I do restart zigbee2mqtt first with `sudo systemctl restart zigbee2mqtt`, wait for that to load and then restart HA. 

I've restarted both a couple of times now and am just gathering logs in case it helps, especially with the missing temp values. :(

Koenkk, 2018-06-21:
It could be that the old entity is not cleared, could you:
- shutdown ha, zigbee2mqtt and mqtt broker
- start mqtt broker
- start zigbee2mqtt
- start ha

jarrah31, 2018-06-21:
Unfortunately that didn't work either - both the sensor values and entity names remain as seen in the screenshot above. 

Hastebin isn't working/saving for me atm, so apologies for the logs being included below.
```
Jun 21 21:04:21 zigbeepi npm[19366]: 2018-6-21 21:04:21 INFO Starting zigbee-shepherd
Jun 21 21:04:21 zigbeepi npm[19366]: 2018-6-21 21:04:21 INFO zigbee-shepherd started
Jun 21 21:04:21 zigbeepi npm[19366]: 2018-6-21 21:04:21 INFO Currently 3 devices are joined:
Jun 21 21:04:21 zigbeepi npm[19366]: 2018-6-21 21:04:21 INFO window_ensuite_velux (0x00158d0001b149eb): MCCGQ11LM - Xiaomi Aqara door & window contact sensor (EndDevice)
Jun 21 21:04:21 zigbeepi npm[19366]: 2018-6-21 21:04:21 INFO window_bed2_mainleft (0x00158d000245b389): MCCGQ11LM - Xiaomi Aqara door & window contact sensor (EndDevice)
Jun 21 21:04:21 zigbeepi npm[19366]: 2018-6-21 21:04:21 INFO temperature_test (0x00158d0001b9560f): WSDCGQ11LM - Xiaomi Aqara temperature, humidity and pressure sensor (EndDevice)
Jun 21 21:04:21 zigbeepi npm[19366]: 2018-6-21 21:04:21 WARN `permit_join` set to  `true` in configuration.yaml.
Jun 21 21:04:21 zigbeepi npm[19366]: 2018-6-21 21:04:21 WARN Allowing new devices to join.
Jun 21 21:04:21 zigbeepi npm[19366]: 2018-6-21 21:04:21 WARN Set `permit_join` to `false` once you joined all devices.
Jun 21 21:04:21 zigbeepi npm[19366]: 2018-6-21 21:04:21 INFO Zigbee: allowing new devices to join.
Jun 21 21:04:21 zigbeepi npm[19366]: 2018-6-21 21:04:21 INFO Connecting to MQTT server at mqtt://192.168.1.50
Jun 21 21:04:22 zigbeepi npm[19366]: 2018-6-21 21:04:22 INFO zigbee-shepherd ready
Jun 21 21:04:32 zigbeepi npm[19366]: 2018-6-21 21:04:32 ERROR Not connected to MQTT server!

Jun 21 21:04:42 zigbeepi npm[19366]: 2018-6-21 21:04:42 ERROR Not connected to MQTT server!
Jun 21 21:04:52 zigbeepi npm[19366]: 2018-6-21 21:04:52 ERROR Not connected to MQTT server!
Jun 21 21:05:02 zigbeepi npm[19366]: 2018-6-21 21:05:02 ERROR Not connected to MQTT server!
Jun 21 21:05:12 zigbeepi npm[19366]: 2018-6-21 21:05:12 ERROR Not connected to MQTT server!
Jun 21 21:05:22 zigbeepi npm[19366]: 2018-6-21 21:05:22 ERROR Not connected to MQTT server!
Jun 21 21:05:25 zigbeepi npm[19366]: 2018-6-21 21:05:25 INFO Connected to MQTT server
Jun 21 21:05:25 zigbeepi npm[19366]: 2018-6-21 21:05:25 INFO MQTT publish, topic: 'zigbee2mqtt/bridge/state', payload: 'online'
Jun 21 21:05:25 zigbeepi npm[19366]: 2018-6-21 21:05:25 INFO MQTT publish, topic: 'homeassistant/binary_sensor/0x00158d0001b149eb/contact/config', payload: '{"payload_on":false,"payload_off":true,"value_template":"{{ value_json.contact }}","device_class":"door","json_attributes":["battery","voltage"],"state_topic":"zigbee2mqtt/window_ensuite_velux","availability_topic":"zigbee2mqtt/bridge/state","name":"window_ensuite_velux","unique_id":"0x00158d0001b149eb_contact_zigbee2mqtt"}'
Jun 21 21:05:25 zigbeepi npm[19366]: 2018-6-21 21:05:25 INFO MQTT publish, topic: 'homeassistant/binary_sensor/0x00158d000245b389/contact/config', payload: '{"payload_on":false,"payload_off":true,"value_template":"{{ value_json.contact }}","device_class":"door","json_attributes":["battery","voltage"],"state_topic":"zigbee2mqtt/window_bed2_mainleft","availability_topic":"zigbee2mqtt/bridge/state","name":"window_bed2_mainleft","unique_id":"0x00158d000245b389_contact_zigbee2mqtt"}'
Jun 21 21:05:25 zigbeepi npm[19366]: 2018-6-21 21:05:25 INFO MQTT publish, topic: 'homeassistant/sensor/0x00158d0001b9560f/temperature/config', payload: '{"unit_of_measurement":"°C","icon":"mdi:temperature-celsius","value_template":"{{ value_json.temperature }}","json_attributes":["battery","voltage"],"state_topic":"zigbee2mqtt/temperature_test","availability_topic":"zigbee2mqtt/bridge/state","name":"temperature_test_temperature","unique_id":"0x00158d0001b9560f_temperature_zigbee2mqtt"}'
Jun 21 21:05:25 zigbeepi npm[19366]: 2018-6-21 21:05:25 INFO MQTT publish, topic: 'homeassistant/sensor/0x00158d0001b9560f/humidity/config', payload: '{"unit_of_measurement":"%","icon":"mdi:water-percent","value_template":"{{ value_json.humidity }}","json_attributes":["battery","voltage"],"state_topic":"zigbee2mqtt/temperature_test","availability_topic":"zigbee2mqtt/bridge/state","name":"temperature_test_humidity","unique_id":"0x00158d0001b9560f_humidity_zigbee2mqtt"}'
Jun 21 21:05:25 zigbeepi npm[19366]: 2018-6-21 21:05:25 INFO MQTT publish, topic: 'homeassistant/sensor/0x00158d0001b9560f/pressure/config', payload: '{"unit_of_measurement":"Pa","icon":"mdi:speedometer","value_template":"{{ value_json.pressure }}","json_attributes":["battery","voltage"],"state_topic":"zigbee2mqtt/temperature_test","availability_topic":"zigbee2mqtt/bridge/state","name":"temperature_test_pressure","unique_id":"0x00158d0001b9560f_pressure_zigbee2mqtt"}'

HA Logs

2018-06-21 21:05:08 INFO (MainThread) [homeassistant.setup] Setting up mqtt
2018-06-21 21:05:08 INFO (MainThread) [homeassistant.loader] Loaded mqtt.server from homeassistant.components.mqtt.server
2018-06-21 21:05:17 INFO (MainThread) [hbmqtt.broker] Listener 'default' bind to 0.0.0.0:1883 (max_connections=-1)
2018-06-21 21:05:17 INFO (MainThread) [hbmqtt.broker] Listener 'ws-1' bind to 0.0.0.0:8080 (max_connections=-1)
2018-06-21 21:05:17 INFO (MainThread) [homeassistant.loader] Loaded mqtt.discovery from homeassistant.components.mqtt.discovery
2018-06-21 21:05:17 DEBUG (MainThread) [homeassistant.components.mqtt] Subscribing to homeassistant/#
[... cut ...]
2018-06-21 21:05:44 DEBUG (MainThread) [homeassistant.components.mqtt] Subscribing to zigbee2mqtt/bridge/state
2018-06-21 21:05:44 DEBUG (MainThread) [homeassistant.components.mqtt] Subscribing to zigbee2mqtt/window_ensuite_velux
2018-06-21 21:05:44 DEBUG (MainThread) [homeassistant.components.mqtt] Subscribing to zigbee2mqtt/window_bed2_mainleft
2018-06-21 21:05:44 DEBUG (MainThread) [homeassistant.components.mqtt] Received message on zigbee2mqtt/bridge/state: b'online'
2018-06-21 21:05:44 DEBUG (MainThread) [homeassistant.components.mqtt] Received message on zigbee2mqtt/bridge/state: b'online'
```

jarrah31, 2018-06-21:
Looks like the sensor values are appearing now - I should be more patient!! 

Entity names remain the same though.

Koenkk, 2018-06-21:
The states should already be reposted.

Can you post your zigbee2mqtt `configuration.yaml` and the entity names in HA?

Koenkk, 2018-06-21:
Yes that of course!

Are you sure there isn't a e.g. `sensor. temperature_test_temperature` in HA?

If so, can you post your startup log of zigbee2mqtt.

jarrah31, 2018-06-21:
Yep, note the find count of 3/3 in the screenshot below:
[image]

Startup log:
```
Jun 21 21:28:19 zigbeepi systemd[1]: Started zigbee2mqtt.
Jun 21 21:28:21 zigbeepi npm[19428]: > zigbee2mqtt@0.1.0 start /opt/zigbee2mqtt
Jun 21 21:28:21 zigbeepi npm[19428]: > node index.js
Jun 21 21:28:22 zigbeepi npm[19428]: 2018-6-21 21:28:22 INFO Starting zigbee-shepherd
Jun 21 21:28:23 zigbeepi npm[19428]: 2018-6-21 21:28:23 INFO zigbee-shepherd started
Jun 21 21:28:23 zigbeepi npm[19428]: 2018-6-21 21:28:23 INFO Currently 3 devices are joined:
Jun 21 21:28:23 zigbeepi npm[19428]: 2018-6-21 21:28:23 INFO window_ensuite_velux (0x00158d0001b149eb): MCCGQ11LM - Xiaomi Aqara door & window contact sensor (EndDevice)
Jun 21 21:28:23 zigbeepi npm[19428]: 2018-6-21 21:28:23 INFO window_bed2_mainleft (0x00158d000245b389): MCCGQ11LM - Xiaomi Aqara door & window contact sensor (EndDevice)
Jun 21 21:28:23 zigbeepi npm[19428]: 2018-6-21 21:28:23 INFO temperature_test (0x00158d0001b9560f): WSDCGQ11LM - Xiaomi Aqara temperature, humidity and pressure sensor (EndDevice)
Jun 21 21:28:23 zigbeepi npm[19428]: 2018-6-21 21:28:23 WARN `permit_join` set to  `true` in configuration.yaml.
Jun 21 21:28:23 zigbeepi npm[19428]: 2018-6-21 21:28:23 WARN Allowing new devices to join.
Jun 21 21:28:23 zigbeepi npm[19428]: 2018-6-21 21:28:23 WARN Set `permit_join` to `false` once you joined all devices.
Jun 21 21:28:23 zigbeepi npm[19428]: 2018-6-21 21:28:23 INFO Zigbee: allowing new devices to join.
Jun 21 21:28:23 zigbeepi npm[19428]: 2018-6-21 21:28:23 INFO Connecting to MQTT server at mqtt://192.168.1.50
Jun 21 21:28:23 zigbeepi npm[19428]: 2018-6-21 21:28:23 INFO zigbee-shepherd ready
Jun 21 21:28:27 zigbeepi npm[19428]: 2018-6-21 21:28:27 INFO Connected to MQTT server
Jun 21 21:28:27 zigbeepi npm[19428]: 2018-6-21 21:28:27 INFO MQTT publish, topic: 'zigbee2mqtt/bridge/state', payload: 'online'
Jun 21 21:28:27 zigbeepi npm[19428]: 2018-6-21 21:28:27 INFO MQTT publish, topic: 'homeassistant/binary_sensor/0x00158d0001b149eb/contact/config', payload: '{"payload_on":false,"payload_off":true,"value_template":"{{ value_json.contact }}","device_class":"door","json_attributes":["battery","voltage"],"state_topic":"zigbee2mqtt/window_ensuite_velux","availability_topic":"zigbee2mqtt/bridge/state","name":"window_ensuite_velux","unique_id":"0x00158d0001b149eb_contact_zigbee2mqtt"}'
Jun 21 21:28:27 zigbeepi npm[19428]: 2018-6-21 21:28:27 INFO MQTT publish, topic: 'homeassistant/binary_sensor/0x00158d000245b389/contact/config', payload: '{"payload_on":false,"payload_off":true,"value_template":"{{ value_json.contact }}","device_class":"door","json_attributes":["battery","voltage"],"state_topic":"zigbee2mqtt/window_bed2_mainleft","availability_topic":"zigbee2mqtt/bridge/state","name":"window_bed2_mainleft","unique_id":"0x00158d000245b389_contact_zigbee2mqtt"}'
Jun 21 21:28:27 zigbeepi npm[19428]: 2018-6-21 21:28:27 INFO MQTT publish, topic: 'homeassistant/sensor/0x00158d0001b9560f/temperature/config', payload: '{"unit_of_measurement":"°C","icon":"mdi:temperature-celsius","value_template":"{{ value_json.temperature }}","json_attributes":["battery","voltage"],"state_topic":"zigbee2mqtt/temperature_test","availability_topic":"zigbee2mqtt/bridge/state","name":"temperature_test_temperature","unique_id":"0x00158d0001b9560f_temperature_zigbee2mqtt"}'
Jun 21 21:28:27 zigbeepi npm[19428]: 2018-6-21 21:28:27 INFO MQTT publish, topic: 'homeassistant/sensor/0x00158d0001b9560f/humidity/config', payload: '{"unit_of_measurement":"%","icon":"mdi:water-percent","value_template":"{{ value_json.humidity }}","json_attributes":["battery","voltage"],"state_topic":"zigbee2mqtt/temperature_test","availability_topic":"zigbee2mqtt/bridge/state","name":"temperature_test_humidity","unique_id":"0x00158d0001b9560f_humidity_zigbee2mqtt"}'
Jun 21 21:28:27 zigbeepi npm[19428]: 2018-6-21 21:28:27 INFO MQTT publish, topic: 'homeassistant/sensor/0x00158d0001b9560f/pressure/config', payload: '{"unit_of_measurement":"Pa","icon":"mdi:speedometer","value_template":"{{ value_json.pressure }}","json_attributes":["battery","voltage"],"state_topic":"zigbee2mqtt/temperature_test","availability_topic":"zigbee2mqtt/bridge/state","name":"temperature_test_pressure","unique_id":"0x00158d0001b9560f_pressure_zigbee2mqtt"}'
Jun 21 21:28:27 zigbeepi npm[19428]: 2018-6-21 21:28:27 DEBUG Recieved mqtt message on topic 'hass/status' with data 'online'
```

Koenkk, 2018-06-22:
@ciotlosm Could it be the the unique_id is used as name instead of the `name` here?

Koenkk, 2018-06-23:
What if we go for option 2 (Allow unique_id every time, but set it to friendly_name in case the user set it) and only allow **unique** `friendly_name` in `configuration.yaml` (checked on start of zigbee2mqtt)?

ciotlosm, 2018-06-23:
That would work, but if the user still sets something very generic like "bedroom" it will create errors in home assistant if he has another entity setup by the same name. I know it's up to the users to make sure they don't setup duplicates, but would be also less issues reported that stuff doesn't work.

jarrah31, 2018-06-24:
I've just tested your latest commit to see if it would help but I'm afraid HA still shows the unique id for the entity name. I did a full shutdown as suggested earlier in the thread.

I've added another Xiaomi device (Xiaomi Aqara wireless switch - WXKG11LM) and this also doesn't have the friendly name set as the entity id. 

One thing I've noticed is that a sensor.* entity doesn't use the friendly name, whereas binary_sensor.* does use the friendly name.

Koenkk, 2018-06-25:
Can you provide your zigbee2mqtt startup log?

I've tested myself and when a `friendly_name` is set, `unique_id` is not included in the homeassistant configure message: https://hastebin.com/aditoqapet.scala (1 with and 1 without `friendly_name`).

jarrah31, 2018-06-25:
Sure: https://hastebin.com/adosiweveb.scala

Just curious to see if could set your sensor to a friendly name instead please to see if it still works? I found that sensors were the ones not picking up the friendly name in my setup.

Koenkk, 2018-06-26:
Are you sure you are running with 19f6d5f72a8ef0f608479ee1b9114ff6c2afe9bc? It's is in a different branch: https://github.com/Koenkk/zigbee2mqtt/commits/issue_138. Can you provide the commit hash which zigbee2mqtt prints on start?

Koenkk, 2018-06-26:
Ok, so this issue can be closed?

jarrah31, 2018-06-26:
Yep, happy for it to be closed, thanks for your help!

closed 2018-06-26 by jarrah31

Koenkk, 2018-06-26:
Great, thanks!

mihalski, 2018-09-01:
I've been butting my head against the issue created by this pull request and was wondering, why not decouple the unique_id and friendly_name?

The unique_id could be calculated as if the user had not set the friendly_name for every occasion. Seems like this would be the perfect solution.