Koenkk/zigbee2mqtt #139: Xiaomi Aqara Temperature Sensor - State Values Initially Show as Unknown

jarrah31, 2018-06-21:
It appears that the Aqara temperature sensor (WSDCGQ11LM) temperature, humidity, and pressure values aren't sent to HA after a zigbee2mqtt restart until ~10 mins later. They remain as unknown until the sensor data is sent through. 
[image]

Is there perhaps a way to request information from the sensor (pull) after a zigbee2mqtt restart rather than waiting for a poll?

Apologies for the logs below, hastebin isn't working...

Restart zigbee2mqtt
```
Jun 21 21:28:19 zigbeepi systemd[1]: Started zigbee2mqtt.
Jun 21 21:28:21 zigbeepi npm[19428]: > zigbee2mqtt@0.1.0 start /opt/zigbee2mqtt
Jun 21 21:28:21 zigbeepi npm[19428]: > node index.js
Jun 21 21:28:22 zigbeepi npm[19428]: 2018-6-21 21:28:22 INFO Starting zigbee-shepherd
Jun 21 21:28:23 zigbeepi npm[19428]: 2018-6-21 21:28:23 INFO zigbee-shepherd started
Jun 21 21:28:23 zigbeepi npm[19428]: 2018-6-21 21:28:23 INFO Currently 3 devices are joined:
Jun 21 21:28:23 zigbeepi npm[19428]: 2018-6-21 21:28:23 INFO window_ensuite_velux (0x00158d0001b149eb): MCCGQ11LM - Xiaomi Aqara door & window contact sensor (EndDevice)
Jun 21 21:28:23 zigbeepi npm[19428]: 2018-6-21 21:28:23 INFO window_bed2_mainleft (0x00158d000245b389): MCCGQ11LM - Xiaomi Aqara door & window contact sensor (EndDevice)
Jun 21 21:28:23 zigbeepi npm[19428]: 2018-6-21 21:28:23 INFO temperature_test (0x00158d0001b9560f): WSDCGQ11LM - Xiaomi Aqara temperature, humidity and pressure sensor (EndDevice)
Jun 21 21:28:23 zigbeepi npm[19428]: 2018-6-21 21:28:23 WARN `permit_join` set to  `true` in configuration.yaml.
Jun 21 21:28:23 zigbeepi npm[19428]: 2018-6-21 21:28:23 WARN Allowing new devices to join.
Jun 21 21:28:23 zigbeepi npm[19428]: 2018-6-21 21:28:23 WARN Set `permit_join` to `false` once you joined all devices.
Jun 21 21:28:23 zigbeepi npm[19428]: 2018-6-21 21:28:23 INFO Zigbee: allowing new devices to join.
Jun 21 21:28:23 zigbeepi npm[19428]: 2018-6-21 21:28:23 INFO Connecting to MQTT server at mqtt://192.168.1.50
Jun 21 21:28:23 zigbeepi npm[19428]: 2018-6-21 21:28:23 INFO zigbee-shepherd ready
Jun 21 21:28:27 zigbeepi npm[19428]: 2018-6-21 21:28:27 INFO Connected to MQTT server
Jun 21 21:28:27 zigbeepi npm[19428]: 2018-6-21 21:28:27 INFO MQTT publish, topic: 'zigbee2mqtt/bridge/state', payload: 'online'
Jun 21 21:28:27 zigbeepi npm[19428]: 2018-6-21 21:28:27 INFO MQTT publish, topic: 'homeassistant/binary_sensor/0x00158d0001b149eb/contact/config', payload: '{"payload_on":false,"payload_off":true,"value_template":"{{ value_json.contact }}","device_class":"door","json_attributes":["battery","voltage"],"state_topic":"zigbee2mqtt/window_ensuite_velux","availability_topic":"zigbee2mqtt/bridge/state","name":"window_ensuite_velux","unique_id":"0x00158d0001b149eb_contact_zigbee2mqtt"}'
Jun 21 21:28:27 zigbeepi npm[19428]: 2018-6-21 21:28:27 INFO MQTT publish, topic: 'homeassistant/binary_sensor/0x00158d000245b389/contact/config', payload: '{"payload_on":false,"payload_off":true,"value_template":"{{ value_json.contact }}","device_class":"door","json_attributes":["battery","voltage"],"state_topic":"zigbee2mqtt/window_bed2_mainleft","availability_topic":"zigbee2mqtt/bridge/state","name":"window_bed2_mainleft","unique_id":"0x00158d000245b389_contact_zigbee2mqtt"}'
Jun 21 21:28:27 zigbeepi npm[19428]: 2018-6-21 21:28:27 INFO MQTT publish, topic: 'homeassistant/sensor/0x00158d0001b9560f/temperature/config', payload: '{"unit_of_measurement":"°C","icon":"mdi:temperature-celsius","value_template":"{{ value_json.temperature }}","json_attributes":["battery","voltage"],"state_topic":"zigbee2mqtt/temperature_test","availability_topic":"zigbee2mqtt/bridge/state","name":"temperature_test_temperature","unique_id":"0x00158d0001b9560f_temperature_zigbee2mqtt"}'
Jun 21 21:28:27 zigbeepi npm[19428]: 2018-6-21 21:28:27 INFO MQTT publish, topic: 'homeassistant/sensor/0x00158d0001b9560f/humidity/config', payload: '{"unit_of_measurement":"%","icon":"mdi:water-percent","value_template":"{{ value_json.humidity }}","json_attributes":["battery","voltage"],"state_topic":"zigbee2mqtt/temperature_test","availability_topic":"zigbee2mqtt/bridge/state","name":"temperature_test_humidity","unique_id":"0x00158d0001b9560f_humidity_zigbee2mqtt"}'
Jun 21 21:28:27 zigbeepi npm[19428]: 2018-6-21 21:28:27 INFO MQTT publish, topic: 'homeassistant/sensor/0x00158d0001b9560f/pressure/config', payload: '{"unit_of_measurement":"Pa","icon":"mdi:speedometer","value_template":"{{ value_json.pressure }}","json_attributes":["battery","voltage"],"state_topic":"zigbee2mqtt/temperature_test","availability_topic":"zigbee2mqtt/bridge/state","name":"temperature_test_pressure","unique_id":"0x00158d0001b9560f_pressure_zigbee2mqtt"}'
Jun 21 21:28:27 zigbeepi npm[19428]: 2018-6-21 21:28:27 DEBUG Recieved mqtt message on topic 'hass/status' with data 'online'
```
Restart HA...
```
Jun 21 21:28:53 zigbeepi npm[19428]: 2018-6-21 21:28:53 ERROR Not connected to MQTT server!
Jun 21 21:29:03 zigbeepi npm[19428]: 2018-6-21 21:29:03 ERROR Not connected to MQTT server!
Jun 21 21:29:13 zigbeepi npm[19428]: 2018-6-21 21:29:13 ERROR Not connected to MQTT server!
Jun 21 21:29:23 zigbeepi npm[19428]: 2018-6-21 21:29:23 ERROR Not connected to MQTT server!
Jun 21 21:29:24 zigbeepi npm[19428]: 2018-6-21 21:29:24 INFO Connected to MQTT server
Jun 21 21:29:24 zigbeepi npm[19428]: 2018-6-21 21:29:24 INFO MQTT publish, topic: 'zigbee2mqtt/bridge/state', payload: 'online'
Jun 21 21:29:24 zigbeepi npm[19428]: 2018-6-21 21:29:24 INFO MQTT publish, topic: 'homeassistant/binary_sensor/0x00158d0001b149eb/contact/config', payload: '{"payload_on":false,"payload_off":true,"value_template":"{{ value_json.contact }}","device_class":"door","json_attributes":["battery","voltage"],"state_topic":"zigbee2mqtt/window_ensuite_velux","availability_topic":"zigbee2mqtt/bridge/state","name":"window_ensuite_velux","unique_id":"0x00158d0001b149eb_contact_zigbee2mqtt"}'
Jun 21 21:29:24 zigbeepi npm[19428]: 2018-6-21 21:29:24 INFO MQTT publish, topic: 'homeassistant/binary_sensor/0x00158d000245b389/contact/config', payload: '{"payload_on":false,"payload_off":true,"value_template":"{{ value_json.contact }}","device_class":"door","json_attributes":["battery","voltage"],"state_topic":"zigbee2mqtt/window_bed2_mainleft","availability_topic":"zigbee2mqtt/bridge/state","name":"window_bed2_mainleft","unique_id":"0x00158d000245b389_contact_zigbee2mqtt"}'
Jun 21 21:29:24 zigbeepi npm[19428]: 2018-6-21 21:29:24 INFO MQTT publish, topic: 'homeassistant/sensor/0x00158d0001b9560f/temperature/config', payload: '{"unit_of_measurement":"°C","icon":"mdi:temperature-celsius","value_template":"{{ value_json.temperature }}","json_attributes":["battery","voltage"],"state_topic":"zigbee2mqtt/temperature_test","availability_topic":"zigbee2mqtt/bridge/state","name":"temperature_test_temperature","unique_id":"0x00158d0001b9560f_temperature_zigbee2mqtt"}'
Jun 21 21:29:24 zigbeepi npm[19428]: 2018-6-21 21:29:24 INFO MQTT publish, topic: 'homeassistant/sensor/0x00158d0001b9560f/humidity/config', payload: '{"unit_of_measurement":"%","icon":"mdi:water-percent","value_template":"{{ value_json.humidity }}","json_attributes":["battery","voltage"],"state_topic":"zigbee2mqtt/temperature_test","availability_topic":"zigbee2mqtt/bridge/state","name":"temperature_test_humidity","unique_id":"0x00158d0001b9560f_humidity_zigbee2mqtt"}'
Jun 21 21:29:24 zigbeepi npm[19428]: 2018-6-21 21:29:24 INFO MQTT publish, topic: 'homeassistant/sensor/0x00158d0001b9560f/pressure/config', payload: '{"unit_of_measurement":"Pa","icon":"mdi:speedometer","value_template":"{{ value_json.pressure }}","json_attributes":["battery","voltage"],"state_topic":"zigbee2mqtt/temperature_test","availability_topic":"zigbee2mqtt/bridge/state","name":"temperature_test_pressure","unique_id":"0x00158d0001b9560f_pressure_zigbee2mqtt"}'
Jun 21 21:29:24 zigbeepi npm[19428]: 2018-6-21 21:29:24 DEBUG Recieved mqtt message on topic 'hass/status' with data 'online'
```
No sensor values posted in the above logs after a zigbee2mqtt restart and then HA restart

Restart HA again:
```
Jun 21 21:33:13 zigbeepi npm[19428]: 2018-6-21 21:33:13 ERROR Not connected to MQTT server!
Jun 21 21:33:23 zigbeepi npm[19428]: 2018-6-21 21:33:23 ERROR Not connected to MQTT server!
Jun 21 21:33:33 zigbeepi npm[19428]: 2018-6-21 21:33:33 ERROR Not connected to MQTT server!
Jun 21 21:33:43 zigbeepi npm[19428]: 2018-6-21 21:33:43 ERROR Not connected to MQTT server!
Jun 21 21:33:47 zigbeepi npm[19428]: 2018-6-21 21:33:47 INFO Connected to MQTT server
[... cut ...]
2018-06-21 21:34:06 DEBUG (MainThread) [homeassistant.components.mqtt] Received message on zigbee2mqtt/bridge/state: b'online'
2018-06-21 21:41:54 DEBUG (MainThread) [homeassistant.components.mqtt] Received message on zigbee2mqtt/temperature_test: b'{"temperature":25.38}'
2018-06-21 21:41:54 DEBUG (MainThread) [homeassistant.components.mqtt] Received message on zigbee2mqtt/temperature_test: b'{"temperature":25.38,"humidity":44.2}'
2018-06-21 21:41:54 DEBUG (MainThread) [homeassistant.components.mqtt] Received message on zigbee2mqtt/temperature_test: b'{"temperature":25.38,"humidity":44.2,"pressure":1021}'
```

ciotlosm, 2018-06-22:
@jarrah31 sensor cache sits in memory for zigbee2mqtt, so a restart will reset the value. If you have recorder enabled, at least on home assistant restart you should get back the last value of the sensor instead of unknown, but I'm not sure exactly for which sensor types that works. 

If you want to somehow get around this, if in your restarts MQTT is still up, is by setting `retain: true` for zigbee2mqtt messages.

haskimo, 2018-06-22:
This also happens with other Xiaomi sensors (round temp sensors and motion sensor)

Until **major** change happens it is not updating after restart, if you blow into your temperature sensor it will update immediately and continue updating normally, but if you wait it can take 30 minutes until it updates by itself for the first time and only then it will work normally, like the problem is with the first trigger

One difference is Motion sensor which has two states: Motion and Illuminance, after restart Illuminance state is "Unknown" but Motion state is "Clear" so it does keep the Motion state, if Motion state is triggered it updates immediately and also updates Illuminance state

Door sensor working normal, it keeps state after restart

Koenkk, 2018-06-23:
I think setting `retain: true` is indeed the best solution for this problem. See: https://github.com/Koenkk/zigbee2mqtt/wiki/Device-specific-configuration

haskimo, 2018-06-23:
Yes `retain: true` fixed it but I'm not sure which one, I set `retain: true` both in zigbee2mqtt configuration.yaml for each device, and in Home Assistant configuration.yaml:

```
mqtt:
  birth_message:
    topic: 'hass/status'
    payload: 'online'
    retain: true
  will_message:
    topic: 'hass/status'
    payload: 'offline'
    retain: true
```

Now it's working, keeping data after restart

Koenkk, 2018-06-23:
@jarrah31 does this solve your problem?

jarrah31, 2018-06-24:
Yes, I can confirm this works. Thanks!!

closed 2018-06-24 by ciotlosm