Koenkk/zigbee2mqtt #141: Docker on Synology NAS

19Homer73, 2018-06-22:
Hello,

how can I run this docker container on an Synology NAS?
How does it finds the zwave-stick on the usb-slot?

Many thanks in advance!

Greetings,
Homer

ciotlosm, 2018-06-22:
@19Homer73 you need to use docker to pass the device inside the container and reference it there through the configuration. 

- On your host do ls /dev/ and find the device you want to map inside the container. 
- Map the device inside the container (see: https://www.safaribooksonline.com/library/view/docker-cookbook/9781783984862/ch02s13.html)
- In your configuration, just point to the device inside the container.

19Homer73, 2018-06-23:
Ok, thank you, it works!!! GREAT!!!

One little thing:

The output in iobroker looks like this:
**{"temperature":23.2,"humidity":39.46}**

But I would like to have this:
**temperature:      23.2 °C**
**humidity:            39.46 %**

What do I have to change for that?
Many thanks in advance!

19Homer73, 2018-08-27:
Everything ist working fine now! Ticket can be closed!

closed 2018-08-27 by Koenkk