Koenkk/zigbee2mqtt #142: Lowbat

kaelsaes, 2018-06-22:
My Smart Home FHEM alway shows Lowbat warning with my new Xiaomi devices although they are 1 week old.
Could you please check if the battery level is calculated correct?
Thx

ciotlosm, 2018-06-22:
@kaelsaes the battery is calculated using:

```
const battery3V = {
    min: 2800,
    max: 3300,
};
```
```
battery: toPercentage(voltage, battery3V.min, battery3V.max),
```

```
const toPercentage = (value, min, max) => {
    if (value > max) {
        value = max;
    } else if (value < min) {
        value = min;
    }

    const normalised = (value - min) / (max - min);
    return (normalised * 100).toFixed(2);
};
```

You can find details in `https://github.com/Koenkk/zigbee-shepherd-converters/blob/master/converters/fromZigbee.js`

This is the current setup also in home assistant. What % does FHEM report low battery?

kaelsaes, 2018-06-22:
Version 0.0.0

ciotlosm, 2018-06-22:
@kaelsaes besides 0.0.0 there should have been also a hash.

69% should not be low, so some other attribute is triggering this. Zigbee2MQTT doesn't publish battery: low

ryanbeaton, 2018-06-23:
@kaelsaes The two sensors I have added did not start at 100% and after one week the human body sensor is at 45% and the temperature is at 35%. As the % lines up with the voltage I tend to believe it.  I think there might be some old stock getting sold!

kaelsaes, 2018-06-24:
Okay, thanks.

kaelsaes, 2018-06-24:
I inserted a completely new CR2032 battery and it started with voltage 2995 and 39% battery level.

But a 3 V battery can never have 3,3 V. Why do you calculate with max 3,3 V?

Koenkk, 2018-06-24:
We kept this the same as the xiaomi aqara component for home assistant (https://github.com/Koenkk/zigbee-shepherd-converters/commit/6101411c111cf49372e4e1c20317cdfb4e008a6c). But this indeed seems unlogical.

@ciotlosm what do you think about reverting to the 2500-3000 range?

ciotlosm, 2018-06-24:
@Koenkk I tried to do a little research, and there are various assumptions out there.

Things to take into consideration:
- Many of my batteries report above value of 3000, and the datasheet [here](http://www.farnell.com/datasheets/1496885.pdf) reports initial value between 3.0 and 3.4 at room temperature. 
- Some sites say under 2700 you might not get a working electronic
- Discharge is not linear, it discharges slower when full and faster when more empty

Happy to go back to maximum being 3000 (as that is the spec most sites agree on as initial) >= 3000 to be 100%, but maybe consider minimum to be 2700.

Koenkk, 2018-06-24:
@ciotlosm Thanks!

@kaelsaes can you close this issue if everything is OK?

kaelsaes, 2018-06-25:
Thanks for your quick help.

closed 2018-06-25 by kaelsaes

Sonsi79, 2018-06-25:
I had the same error while using new batteries.
Is it fixed by update to newest Zigbee2MQTT by the following commands?
```
git checkout HEAD -- npm-shrinkwrap.json
git pull
rm -rf node_modules
npm install
```

Koenkk, 2018-06-26:
@Sonsi79 here you can find how to update: https://github.com/Koenkk/zigbee2mqtt/wiki/Running-the-bridge#6-for-later-update-zigbee2mqtt-to-the-latest-version (that should fix your problem).

Sonsi79, 2018-06-26:
I works fine for me. Thanks a lot

cyablo, 2018-12-27:
Sorry to reopen, but my WSDCGQ11LM shows 2,995v and calculated 55% of battery. This seems to be a little bit odd if the battery range is 2700 - 3000.

Koenkk, 2018-12-27:
@cyable how do you know the range is 2700-3000?

Sonsi79, 2018-12-27:
I also get the Low bat error on Batteries which are ok :(

Koenkk, 2018-12-27:
Fixed now in the dev branch.

Offtopic: Could somebody make a PR on how to integrate with FHEM? (https://github.com/Koenkk/zigbee2mqtt/blob/master/docs/index.md#integration)

cyablo, 2018-12-28:
Switched to dev branch, 2,985v, 95%, great!

Offtopic: I have only one Xiaomi Temp.-Sensor to test with FHEM, but second CC2531, a Xiaomi motion sensor and spare Ikea & Osram light bulb are on the way to test FHEM integration further. So, I have very little experience with zigbee2mqtt <-> FHEM at the moment.