Koenkk/zigbee2mqtt #143: Trying to add Hive Motion sensors (will be adding many more tihngs)

matchett808, 2018-06-23:
This is the first lot of things I'm trying to add - will also need to add Hive plugs, thermostat and other zigbee devices.

I get this message in my logs on startup (I added it to devices.js, think this part is ok):
```
2018-6-23 14:00:13 INFO 0x00124b000fa0779e (0x00124b000fa0779e): MOT003 - Hive Hive Motion Sensor (EndDevice)
```
I have this in the devices.js:
```
    {
        zigbeeModel: ['MOT003'],
        model: 'MOT003',
        vendor: 'Hive',
        description: 'Hive Motion Sensor',
        supports: 'occupancy',
        fromZigbee: [fz.hive_ias, fz.hive_occupancy, fz.hive_illum, fz.hive_temp],
        toZigbee: [],
    },
```
and here are the entries in fromZigbee.js:
```
  hive_occupancy: {
        cid: 'msOccupancySensing',
        type: 'attReport',
        convert: (model, msg, publish, options) => {
            console.log(msg);
        },
    },
    hive_illum: {
        cid: 'msIlluminanceMeasurement',
        type: 'attReport',
        convert: (model, msg, publish, options) => {
            console.log(msg);
        },
    },
    hive_temp: {
        cid: 'msTemperatureMeasurement',
        type: 'attReport',
        convert: (model, msg, publish, options) => {
            console.log(msg);
        },
    },

    hive_ias: {
        cid: 'ssIasZone',
        type: 'statusChange',
        convert: (model, msg, publish, options) => {
            return {motion: msg.data.zoneStatus === 1};
        },
    },
```

Thing is that I don't get any messages in the logs or over MQTT - here is the entry in database.db:
```
{"id":2,"type":"EndDevice","ieeeAddr":"0x00124b000fa0779e","nwkAddr":39651,"manufId":4153,"manufName":"HiveHome.com","powerSource":"Battery","modelId":"MOT003","epList":[6],"status":"offline","joinTime":null,"endpoints":{"6":{"profId":260,"epId":6,"devId":1026,"inClusterList":[0,1,3,32,1024,1026,1030,1280],"outClusterList":[25],"clusters":{"genBasic":{"dir":{"value":1},"attrs":{}},"genPowerCfg":{"dir":{"value":1},"attrs":{}},"genIdentify":{"dir":{"value":1},"attrs":{}},"genOta":{"dir":{"value":2},"attrs":{}},"genPollCtrl":{"dir":{"value":1},"attrs":{}},"msIlluminanceMeasurement":{"dir":{"value":1},"attrs":{}},"msTemperatureMeasurement":{"dir":{"value":1},"attrs":{}},"msOccupancySensing":{"dir":{"value":1},"attrs":{}},"ssIasZone":{"dir":{"value":1},"attrs":{}}}}},"_id":"ZSCUNpjs70nw1k5V"}
```
am I missing something? The device _should_ report battery, illumaninance, motion and tempreature - but I never get anything out of it?

matchett808, 2018-06-23:
Looks like there's something up at part 2 of the supporting new devices tutorial - before adding it to devices.js I get the following in the logs:
```
2018-6-23 14:46:49 INFO MQTT publish, topic: 'zb/bridge/log', payload: '{"type":"pairing","message":"device incoming"}'
2018-6-23 14:46:49 WARN Device with modelID 'MOT003' is not supported.
2018-6-23 14:46:49 WARN Please see: https://github.com/Koenkk/zigbee2mqtt/wiki/How-to-support-new-devices
```

I then add this to devices.js:
```
    {
        zigbeeModel: ['MOT003'],
        model: 'MOT003',
        vendor: 'Hive',
        description: 'Hive Motion Sensor',
        supports: 'occupancy',
        fromZigbee: [],
        toZigbee: [],
    },
```

but I never get any of the messages like this:
```
WARN No converter available
```

Koenkk, 2018-06-23:
We probably need to bind it.

Go to zigbee2mqtt directory:

```bash
cd node_modules
rm -rf zigbee-shepherd-converters
git clone https://github.com/Koenkk/zigbee-shepherd-converters.git -b hive_motion
cd zigbee-shepherd-converters
npm install
```

Now start zigbee2mqtt and trigger events on the motion sensor, do you see any `WARN No converter available`. ?

matchett808, 2018-06-23:
I'm running with docker and have a volume mapped there:

```
      - ./zigbee_conv:/app/node_modules/zigbee-shepherd-converters/
```

that's where I was adding it to devices.js and fromZigbee.js

matchett808, 2018-06-23:
Just re-did it, still not seeing any, I have 2 hive bulbs in as well (to test) and changed their descriptors:

```
2018-6-23 19:09:24 INFO hallway_hive_motion_mqtt (0x00124b000fa0779e): MOT003 - Hive Hive Motion Sensor (EndDevice)
2018-6-23 19:09:24 INFO hallway_light (0x00158d00016766fc): HALIGHTDIMWWE27 - Hive Active light dimmable - I WIZ ERE (Router)
2018-6-23 19:09:24 INFO stair_light (0x00158d000166bcf7): HALIGHTDIMWWE27 - Hive Active light dimmable - I WIZ ERE (Router)
```
so my docker volume is definately hitting the right place (I added 'I WIZ ERE' to the descriptors)

Still not seeing anything in the logs when I trigger motion

matchett808, 2018-06-23:
I'm now trying it with your commit (missed that lol)

matchett808, 2018-06-23:
I now get 
```
2018-06-23T19:16:21.810Z - error: Failed to configure 0x00124b000fa0779e
```
(0x00124b000fa0779e is the motion sensor) - this appears right after 'device is incoming'

Just as I started to reset and repair it I got this message:

```
2018-06-23T19:18:19.081Z - warn: Message without device!
```
so that's possibly done the trick - will reset and repair and let you know

matchett808, 2018-06-23:
after reset/repairing a couple of times - still not seeing any messages in reponse to stimulii 

one thing to mention is that it keeps attempting to pair (on every restart) - not sure if this is an acceptable behaviour

matchett808, 2018-06-23:
Just to ensure we're getting all the info in here:
```
2018-6-23 19:42:04 INFO Starting zigbee2mqtt version 0.0.0 (commit #01a26ad)
2018-6-23 19:42:04 INFO Starting zigbee-shepherd
2018-6-23 19:42:04 INFO zigbee-shepherd started
2018-6-23 19:42:04 INFO Currently 3 devices are joined:
2018-6-23 19:42:04 INFO hallway_light (0x00158d00016766fc): HALIGHTDIMWWE27 - Hive Active light dimmable (Router)
2018-6-23 19:42:04 INFO stair_light (0x00158d000166bcf7): HALIGHTDIMWWE27 - Hive Active light dimmable (Router)
2018-6-23 19:42:04 INFO hallway_hive_motion_mqtt (0x00124b000fa0779e): MOT003 - Hive Hive Motion Sensor (EndDevice)
2018-6-23 19:42:04 WARN `permit_join` set to  `true` in configuration.yaml.
2018-6-23 19:42:04 WARN Allowing new devices to join.
2018-6-23 19:42:04 WARN Set `permit_join` to `false` once you joined all devices.
2018-6-23 19:42:04 INFO Zigbee: allowing new devices to join.
2018-6-23 19:42:04 INFO Connecting to MQTT server at mqtt://192.168.0.8
2018-6-23 19:42:04 INFO zigbee-shepherd ready
spinlock: false []
2018-6-23 19:42:04 INFO Connected to MQTT server
2018-6-23 19:42:04 INFO MQTT publish, topic: 'zb/bridge/state', payload: 'online'
2018-6-23 19:42:04 INFO MQTT publish, topic: 'homeassistant/light/0x00158d00016766fc/light/config', payload: '{"brightness":true,"platform":"mqtt_json","command_topic":"zb/hallway_light/set","state_topic":"zb/hallway_light","availability_topic":"zb/bridge/state","name":"hallway_light","unique_id":"0x00158d00016766fc_light_zb"}'
2018-6-23 19:42:04 INFO MQTT publish, topic: 'homeassistant/light/0x00158d000166bcf7/light/config', payload: '{"brightness":true,"platform":"mqtt_json","command_topic":"zb/stair_light/set","state_topic":"zb/stair_light","availability_topic":"zb/bridge/state","name":"stair_light","unique_id":"0x00158d000166bcf7_light_zb"}'
spinlock: true []
2018-6-23 19:42:05 INFO Connecting with device...
2018-6-23 19:42:05 INFO MQTT publish, topic: 'zb/bridge/log', payload: '{"type":"pairing","message":"connecting with device"}'
2018-6-23 19:42:05 INFO Connecting with device...
2018-6-23 19:42:05 INFO MQTT publish, topic: 'zb/bridge/log', payload: '{"type":"pairing","message":"connecting with device"}'
2018-6-23 19:42:12 INFO Connecting with device...
2018-6-23 19:42:12 INFO MQTT publish, topic: 'zb/bridge/log', payload: '{"type":"pairing","message":"connecting with device"}'
2018-6-23 19:42:13 INFO Connecting with device...
2018-6-23 19:42:13 INFO MQTT publish, topic: 'zb/bridge/log', payload: '{"type":"pairing","message":"connecting with device"}'
2018-6-23 19:42:14 INFO Connecting with device...
2018-6-23 19:42:14 INFO MQTT publish, topic: 'zb/bridge/log', payload: '{"type":"pairing","message":"connecting with device"}'
2018-6-23 19:42:15 INFO Connecting with device...
2018-6-23 19:42:15 INFO MQTT publish, topic: 'zb/bridge/log', payload: '{"type":"pairing","message":"connecting with device"}'
2018-6-23 19:42:16 INFO Connecting with device...
2018-6-23 19:42:16 INFO MQTT publish, topic: 'zb/bridge/log', payload: '{"type":"pairing","message":"connecting with device"}'
2018-6-23 19:42:16 INFO Device incoming...
2018-6-23 19:42:16 INFO MQTT publish, topic: 'zb/bridge/log', payload: '{"type":"pairing","message":"device incoming"}'
2018-6-23 19:42:16 INFO Connecting with device...
2018-6-23 19:42:16 INFO MQTT publish, topic: 'zb/bridge/log', payload: '{"type":"pairing","message":"connecting with device"}'
2018-6-23 19:42:16 INFO Connecting with device...
2018-6-23 19:42:16 INFO MQTT publish, topic: 'zb/bridge/log', payload: '{"type":"pairing","message":"connecting with device"}'
2018-6-23 19:42:16 INFO Connecting with device...
2018-6-23 19:42:16 INFO MQTT publish, topic: 'zb/bridge/log', payload: '{"type":"pairing","message":"connecting with device"}'
2018-6-23 19:42:18 INFO Connecting with device...
2018-6-23 19:42:18 INFO MQTT publish, topic: 'zb/bridge/log', payload: '{"type":"pairing","message":"connecting with device"}'
2018-6-23 19:42:18 INFO Connecting with device...
2018-6-23 19:42:18 INFO MQTT publish, topic: 'zb/bridge/log', payload: '{"type":"pairing","message":"connecting with device"}'
2018-6-23 19:42:18 INFO Connecting with device...
2018-6-23 19:42:18 INFO MQTT publish, topic: 'zb/bridge/log', payload: '{"type":"pairing","message":"connecting with device"}'
2018-6-23 19:42:18 INFO Connecting with device...
2018-6-23 19:42:18 INFO MQTT publish, topic: 'zb/bridge/log', payload: '{"type":"pairing","message":"connecting with device"}'
2018-6-23 19:42:18 INFO Device incoming...
2018-6-23 19:42:18 INFO MQTT publish, topic: 'zb/bridge/log', payload: '{"type":"pairing","message":"device incoming"}'
2018-6-23 19:42:54 INFO Succesfully configured 0x00124b000fa0779e
```

matchett808, 2018-06-23:
Used the Debug=* method - now logs are v. noisy but I can't pinpoint anything for this device

matchett808, 2018-06-23:
Tried that, I did get this - but only because the device was sitting for a while (and woke up I reckon):

```

Sat, 23 Jun 2018 20:49:53 GMT cc-znp:AREQ <-- ZDO:endDeviceAnnceInd, { srcaddr: 11322, nwkaddr: 11322, ieeeaddr: '0x00124b000fa0779e', capabilities: 128 }
spinlock: false []
device already in network
Sat, 23 Jun 2018 20:49:53 GMT zigbee-shepherd:msgHdlr IND <-- ZDO:endDeviceAnnceInd

Sat, 23 Jun 2018 20:49:53 GMT cc-znp:AREQ <-- ZDO:tcDeviceInd, { nwkaddr: 11322, extaddr: '0x00124b000fa0779e', parentaddr: 0 }
Sat, 23 Jun 2018 20:49:53 GMT zigbee-shepherd:msgHdlr IND <-- ZDO:tcDeviceInd
```

It didn't report the 2nd or 3rd time I tried it (there's an LED on it for when it detects motion)

Don't know if it will help much but there are smartthings device handlers: https://community.smartthings.com/t/hive-motion-contact-sensors/57452 this doesn't work for motion with this version - but the orvibo one apparently does https://github.com/Kriskit/SmartThingsPublic/blob/master/devicetypes/kriskit/orvibo/orvibo-motion-sensor.groovy#L68

The orvibo DTH looks at IASzone 19?

matchett808, 2018-06-23:
Pulling the battery also produces this:

```
spinlock: false []
device already in network
Sat, 23 Jun 2018 20:54:54 GMT cc-znp:AREQ <-- ZDO:endDeviceAnnceInd, { srcaddr: 11322, nwkaddr: 11322, ieeeaddr: '0x00124b000fa0779e', capabilities: 128 }
Sat, 23 Jun 2018 20:54:54 GMT zigbee-shepherd:msgHdlr IND <-- ZDO:endDeviceAnnceInd
```

Koenkk, 2018-06-23:
Did it also confgure correctly with my new change? e.g. `2018-6-23 19:42:54 INFO Succesfully configured 0x00124b000fa0779e`

matchett808, 2018-06-23:
🎉 getting something now, needed to completely strip it out completely in order to get it (didn't get a configured correctly/incorrectly message)

```
2018-06-23T21:40:04.274Z - debug: Recieved zigbee message with data {"cid":"msOccupancySensing","data":{"occupancy":0}}
2018-06-23T21:40:04.275Z - warn: No converter available for 'MOT003' with cid 'msOccupancySensing' and type 'attReport'
2018-06-23T21:40:04.275Z - warn: Please see: https://github.com/Koenkk/zigbee2mqtt/wiki/How-to-support-new-devices.
2018-06-23T21:40:04.277Z - debug: Recieved zigbee message with data {"cid":"msOccupancySensing","data":{"occupancy":0}}
2018-06-23T21:40:04.279Z - warn: No converter available for 'MOT003' with cid 'msOccupancySensing' and type 'devChange'
2018-06-23T21:40:04.282Z - warn: Please see: https://github.com/Koenkk/zigbee2mqtt/wiki/How-to-support-new-devices.
```

However - even with debug=* as an environement variable - I'm still not seeing anything happening (attaching to the docker container or in the logs - but is that the right behaviour?) I'll add a converter and see what happens

Koenkk, 2018-06-23:
Yes this looks good! How does your `configure()` now look?

matchett808, 2018-06-23:
I get this now:
```
2018-06-23T21:54:15.409Z - info: Succesfully configured 0x00124b000fa0779e
2018-06-23T21:54:15.793Z - debug: Recieved zigbee message with data {"cid":"msOccupancySensing","data":{"occupancy":1}}
```
but no _actual_ reaction to the device LED telling me that it's sensing motion

```
  {
        zigbeeModel: ['MOT003'],
        model: 'MOT003',
        vendor: 'Hive',
        description: 'Hive Motion Sensor',
        supports: 'occupancy',
        fromZigbee: [fz.hive_occupancy],
        toZigbee: [],
        configure: (ieeeAddr, shepherd, coordinator, callback) => {
            const cfgRptRec = {
                direction: 0, attrId: 0, dataType: 24, minRepIntval: 0, maxRepIntval: 1000, repChange: 0,
            };

            const device = shepherd.find(ieeeAddr, 6);
            if (device) {
                device.bind('msOccupancySensing', coordinator, (error) => {
                    if (error) {
                        callback(error);
                    } else {
                        device.foundation('msOccupancySensing', 'configReport', [cfgRptRec]).then((rsp) => {
                            callback(rsp[0].status === 0);
                        });
                    }
                });
            }
        },
    },
```

Koenkk, 2018-06-23:
But does it send 

```
2018-06-23T21:54:15.793Z - debug: Recieved zigbee message with data {"cid":"msOccupancySensing","data":{"occupancy":1}}
```

when you trigger the sensor?

matchett808, 2018-06-23:
Not when I trigger the sensor - that happens when the configure happens

matchett808, 2018-06-23:
By attaching to the docker container I get logs like this for other devices:
```
2018-6-23 23:09:10 DEBUG Check online 0x00158d00016766fc
Sat, 23 Jun 2018 22:09:10 GMT zigbee-shepherd:request REQ --> ZDO:nodeDescReq
Sat, 23 Jun 2018 22:09:10 GMT cc-znp:SREQ --> ZDO:nodeDescReq, { dstaddr: 16507, nwkaddrofinterest: 16507 }
Sat, 23 Jun 2018 22:09:10 GMT serialport write 9 bytes of data
```

Koenkk, 2018-06-23:
Could you try to change the `maxRepIntval` to 5?

matchett808, 2018-06-23:
what about repChange? It's 0 atm which says false to me?

Koenkk, 2018-06-23:
Yes, you could also try with repChange 1

matchett808, 2018-06-23:
Hmm - still don't seem to be getting anything - the sensor was working with a Hive hub ( and has previously worked with the smartthings hub) - I'm a bit lost with the zigbee stuff but is the configure step subscribing to the IAS alarms? (based on reading that looks liek it's how it this model works)

matchett808, 2018-06-24:
IT's quite weird - woke up to logs like this:

```
2018-06-24T08:44:47.396Z - debug: Check online 0x00158d00016766fc
2018-06-24T08:44:47.398Z - debug: Check online 0x00158d000166bcf7
2018-06-24T08:44:48.883Z - debug: Recieved zigbee message with data {"cid":"msOccupancySensing","data":{"occupancy":1}}
2018-06-24T08:44:48.883Z - info: MQTT publish, topic: 'zb/hallway_hive_motion_mqtt', payload: '{"cid":"msOccupancySensing","data":{"occupancy":1}}'
2018-06-24T08:44:54.881Z - debug: Recieved zigbee message with data {"cid":"msOccupancySensing","data":{"occupancy":1}}
2018-06-24T08:44:54.882Z - info: MQTT publish, topic: 'zb/hallway_hive_motion_mqtt', payload: '{"cid":"msOccupancySensing","data":{"occupancy":1}}'
2018-06-24T08:45:00.877Z - debug: Recieved zigbee message with data {"cid":"msOccupancySensing","data":{"occupancy":1}}
2018-06-24T08:45:00.878Z - info: MQTT publish, topic: 'zb/hallway_hive_motion_mqtt', payload: '{"cid":"msOccupancySensing","data":{"occupancy":1}}'
2018-06-24T08:45:06.876Z - debug: Recieved zigbee message with data {"cid":"msOccupancySensing","data":{"occupancy":1}}
2018-06-24T08:45:06.877Z - info: MQTT publish, topic: 'zb/hallway_hive_motion_mqtt', payload: '{"cid":"msOccupancySensing","data":{"occupancy":1}}'
2018-06-24T08:45:12.874Z - debug: Recieved zigbee message with data {"cid":"msOccupancySensing","data":{"occupancy":1}}
2018-06-24T08:45:12.874Z - info: MQTT publish, topic: 'zb/hallway_hive_motion_mqtt', payload: '{"cid":"msOccupancySensing","data":{"occupancy":1}}'
2018-06-24T08:45:18.870Z - debug: Recieved zigbee message with data {"cid":"msOccupancySensing","data":{"occupancy":1}}
2018-06-24T08:45:18.870Z - info: MQTT publish, topic: 'zb/hallway_hive_motion_mqtt', payload: '{"cid":"msOccupancySensing","data":{"occupancy":1}}'
2018-06-24T08:45:24.986Z - debug: Recieved zigbee message with data {"cid":"msOccupancySensing","data":{"occupancy":1}}
2018-06-24T08:45:24.986Z - info: MQTT publish, topic: 'zb/hallway_hive_motion_mqtt', payload: '{"cid":"msOccupancySensing","data":{"occupancy":1}}'
2018-06-24T08:45:30.985Z - debug: Recieved zigbee message with data {"cid":"msOccupancySensing","data":{"occupancy":1}}
2018-06-24T08:45:30.986Z - info: MQTT publish, topic: 'zb/hallway_hive_motion_mqtt', payload: '{"cid":"msOccupancySensing","data":{"occupancy":1}}'
2018-06-24T08:45:36.976Z - debug: Recieved zigbee message with data {"cid":"msOccupancySensing","data":{"occupancy":1}}
2018-06-24T08:45:36.979Z - info: MQTT publish, topic: 'zb/hallway_hive_motion_mqtt', payload: '{"cid":"msOccupancySensing","data":{"occupancy":1}}'
2018-06-24T08:45:42.975Z - debug: Recieved zigbee message with data {"cid":"msOccupancySensing","data":{"occupancy":1}}
2018-06-24T08:45:42.976Z - info: MQTT publish, topic: 'zb/hallway_hive_motion_mqtt', payload: '{"cid":"msOccupancySensing","data":{"occupancy":1}}'
2018-06-24T08:45:47.403Z - debug: Check online 0x00158d00016766fc
2018-06-24T08:45:47.405Z - debug: Check online 0x00158d000166bcf7
2018-06-24T08:45:48.974Z - debug: Recieved zigbee message with data {"cid":"msOccupancySensing","data":{"occupancy":1}}
2018-06-24T08:45:48.974Z - info: MQTT publish, topic: 'zb/hallway_hive_motion_mqtt', payload: '{"cid":"msOccupancySensing","data":{"occupancy":1}}'
2018-06-24T08:45:54.967Z - debug: Recieved zigbee message with data {"cid":"msOccupancySensing","data":{"occupancy":1}}
2018-06-24T08:45:54.968Z - info: MQTT publish, topic: 'zb/hallway_hive_motion_mqtt', payload: '{"cid":"msOccupancySensing","data":{"occupancy":1}}'
2018-06-24T08:46:00.978Z - debug: Recieved zigbee message with data {"cid":"msOccupancySensing","data":{"occupancy":1}}
2018-06-24T08:46:00.979Z - info: MQTT publish, topic: 'zb/hallway_hive_motion_mqtt', payload: '{"cid":"msOccupancySensing","data":{"occupancy":1}}'
2018-06-24T08:46:06.974Z - debug: Recieved zigbee message with data {"cid":"msOccupancySensing","data":{"occupancy":1}}
2018-06-24T08:46:06.974Z - info: MQTT publish, topic: 'zb/hallway_hive_motion_mqtt', payload: '{"cid":"msOccupancySensing","data":{"occupancy":1}}'
2018-06-24T08:46:12.958Z - debug: Recieved zigbee message with data {"cid":"msOccupancySensing","data":{"occupancy":1}}
2018-06-24T08:46:12.959Z - info: MQTT publish, topic: 'zb/hallway_hive_motion_mqtt', payload: '{"cid":"msOccupancySensing","data":{"occupancy":1}}'
2018-06-24T08:46:18.953Z - debug: Recieved zigbee message with data {"cid":"msOccupancySensing","data":{"occupancy":1}}
2018-06-24T08:46:18.954Z - info: MQTT publish, topic: 'zb/hallway_hive_motion_mqtt', payload: '{"cid":"msOccupancySensing","data":{"occupancy":1}}'
2018-06-24T08:46:24.950Z - debug: Recieved zigbee message with data {"cid":"msOccupancySensing","data":{"occupancy":1}}
2018-06-24T08:46:24.951Z - info: MQTT publish, topic: 'zb/hallway_hive_motion_mqtt', payload: '{"cid":"msOccupancySensing","data":{"occupancy":1}}'
2018-06-24T08:46:30.948Z - debug: Recieved zigbee message with data {"cid":"msOccupancySensing","data":{"occupancy":1}}
2018-06-24T08:46:30.948Z - info: MQTT publish, topic: 'zb/hallway_hive_motion_mqtt', payload: '{"cid":"msOccupancySensing","data":{"occupancy":1}}'
2018-06-24T08:46:36.965Z - debug: Recieved zigbee message with data {"cid":"msOccupancySensing","data":{"occupancy":1}}
2018-06-24T08:46:36.966Z - info: MQTT publish, topic: 'zb/hallway_hive_motion_mqtt', payload: '{"cid":"msOccupancySensing","data":{"occupancy":1}}'
```
The 'check online' entries are for the bulbs, but the other entries aren't in response to motion (motion sensor is sitting face down on a table atm)

Koenkk, 2018-06-24:
Yes, it's just reporting the occupancy on a 6 seconds interval.

I the configure, can you try replacing `msOccupancySensing` with `ssIasZone`.

matchett808, 2018-06-24:
HM.....this is interesting - pulling the battery gives 
```
2018-06-24T09:04:49.520Z - debug: Recieved zigbee message with data {"cid":"msOccupancySensing","data":{"occupancy":0}}
2018-06-24T09:04:49.521Z - info: MQTT publish, topic: 'zb/hallway_hive_motion_mqtt', payload: '{"cid":"msOccupancySensing","data":{"occupancy":0}}'
```
until motion is triggered - might it need to write a 0 after 30 seconds or reset the IASzonealarm at some point?

Chaning it to ssIasZone doesn't seem to have made much of a difference to it

matchett808, 2018-06-24:
Actually - as an update - the IAszone change (after re-pairing) removes the /6 seconds report - without replacing ti with anything at all

Koenkk, 2018-06-24:
So to clarify:

- remove battery and putting it again
- trigger motion on device -> motion is reported
- no motion is reported anymore

matchett808, 2018-06-24:
Sorry - when battery is pulled/it's a clean reset/repair (no motion has been detected since reset/repair) the following messages are received as per the maxreinterval (roughly):

```
2018-06-24T09:46:37.328Z - debug: Recieved zigbee message with data {"cid":"msOccupancySensing","data":{"occupancy":0}}
2018-06-24T09:46:37.329Z - info: MQTT publish, topic: 'zb/hallway_hive_motion_mqtt', payload: '{"cid":"msOccupancySensing","data":{"occupancy":0}}'
```

Then, if motion is detected, it switches over to reporting this (again as per the maxrepinterval):
```
2018-06-24T09:48:49.484Z - debug: Recieved zigbee message with data {"cid":"msOccupancySensing","data":{"occupancy":1}}
2018-06-24T09:48:49.485Z - info: MQTT publish, topic: 'zb/hallway_hive_motion_mqtt', payload: '{"cid":"msOccupancySensing","data":{"occupancy":1}}'
```
it does this (as far as I can tell - based on it doing that all night) until infinity - it's like the sensor itself isn't clearing the motion detected flag, might this me needed from the software side?

matchett808, 2018-06-24:
*This is with the msOccupancy configuration

matchett808, 2018-06-24:
Here is a diff (I just added the msTemperatureMeasurement to see if that'd give us additional logging):
```
diff --git a/converters/fromZigbee.js b/converters/fromZigbee.js
index cc61c5a..57b1989 100644
--- a/converters/fromZigbee.js
+++ b/converters/fromZigbee.js
@@ -78,6 +78,21 @@ const ictcg1 = (model, msg, publish, options, action) => {
 
 
 const converters = {
+    hive_occupancy: {
+        cid: 'msOccupancySensing',
+        type: 'devChange',
+        convert: (model, msg, publish, options) => {
+            return msg.data;
+        },
+    },
+
+    hive_occ: {
+        cid: 'msOccupancySensing',
+        type: 'attReport',
+        convert: (model, msg, publish, options) => {
+            return msg.data;
+        },
+    },
     xiaomi_battery_3v: {
         cid: 'genBasic',
         type: 'attReport',
diff --git a/devices.js b/devices.js
index 095c7dd..60c1f63 100644
--- a/devices.js
+++ b/devices.js
@@ -539,11 +539,11 @@ const devices = [
         vendor: 'Hive',
         description: 'Hive Motion Sensor',
         supports: 'occupancy',
-        fromZigbee: [],
+        fromZigbee: [fz.hive_occupancy, fz.hive_occ],
         toZigbee: [],
         configure: (ieeeAddr, shepherd, coordinator, callback) => {
             const cfgRptRec = {
-                direction: 0, attrId: 0, dataType: 24, minRepIntval: 0, maxRepIntval: 1000, repChange: 0,
+                direction: 0, attrId: 0, dataType: 24, minRepIntval: 0, maxRepIntval: 10, repChange: 0,
             };
 
             const device = shepherd.find(ieeeAddr, 6);
@@ -557,6 +557,16 @@ const devices = [
                         });
                     }
                 });
+               
+                device.bind('msTemperatureMeasurement', coordinator, (error) => {
+                    if (error) {
+                        callback(error);
+                    } else {
+                        device.foundation('msTemperatureMeasurement', 'configReport', [cfgRptRec]).then((rsp) => {
+                            callback(rsp[0].status === 0);
+                        });
+                    }
+                });
             }
         },
     },
```

matchett808, 2018-06-24:
Adding the msTemp part doesn't seem to have done anything - still getting the msocc message every interval (0 atm because it's been face down since I did the reset)

matchett808, 2018-06-24:
Don't think this is within my abilities (this is my first foray into zigbee outside of hive/tradfri/smarttthings hubs) - I'm going to move on from trying to get these motion sensors to work (for the moment)

Koenkk, 2018-06-25:
Ok, no problem, lets keep this issue open for future reference.

matchett808, 2018-06-25:
Just a slight furtherance - I paired a smartthings motion sensor and the dB entry looked very, very similar

matchett808, 2018-06-26:
https://stackoverflow.com/questions/39599822/ias-zone-device-enroll-steps

Looks like the issue might be the cie address being 0x0..0 - will check this out tonight

Koenkk, 2018-06-26:
That looks promising, please let me know if you need more help.

Koenkk, 2018-06-28:
If I read you response from #153 correctly, this didn't work?

matchett808, 2018-06-28:
Yeah, configure is as follows(includes writing  the CIE address):

```
            const cfgRptRec = {
                direction: 0, attrId: 0, dataType: 32, minRepIntval: 10, maxRepIntval: 20, repChange: 0,
            };
            const cfgRptRec2 = {
                direction: 0, attrId: 0, dataType: 33, minRepIntval: 10, maxRepIntval: 20, repChange: 0,
            };

            const device = shepherd.find(ieeeAddr, 6);
            if (device) {
              device.write('ssIasZone' , 'iasCieAddr' , coordinator.device.getIeeeAddr(), (err) => {
                        console.log(err);
                });
                device.bind('msOccupancySensing', coordinator, (error) => {
                    if (error) {
                        callback(error);
                    } else {
                        device.foundation('msOccupancySensing', 'configReport', [cfgRptRec]).then((rsp) => {
                            callback(rsp[0].status === 0);
                        });
                    }
                });

                device.bind('msTemperatureMeasurement', coordinator, (error) => {
                    if (error) {
                        callback(error);
                    } else {
                        device.foundation('msTemperatureMeasurement', 'configReport', [cfgRptRec2]).then((rsp) => {
                            callback(rsp[0].status === 0);
                        });
                    }
                });

```

But we never receive an IAS alarm packet (as far as I can tell)

matchett808, 2018-06-28:
it's like we need to clear the alarm | reset the attrib ourself - again, if I pull the batter it starts up by reporting no motion (until it trips, then it reports motion forever)

Still can't get Temp. readings though...

matchett808, 2018-06-28:
Makes me wonder - is it possible to bind 2 clusters?

Koenkk, 2018-06-28:
I've found some interesting code example here https://github.com/zigbeer/ziee/wiki Initialize a 'ssIasZone' cluster

Koenkk, 2018-06-28:
After the cieeadress write we need to send the enroll request.

matchett808, 2018-06-28:
tried this - no effect what so ever:
```
device.foundation('ssIasZone', 'enrollReq', [ ], function (err, rsp) { console.log(err); console.log(rsp); });
```

I'm not sure what to do with this here 🤷‍♂️

Koenkk, 2018-06-29:
The flow is as follows

1. Write `iasCieAddr`
2. Receive enroll request from motion sensor
3. Respond to enroll request.

Let's first check if the enroll request is send from the motion sensor. Can you try with:

```

matchett808, 2018-06-29:
Thanks, I'll give all this a try tonight and let you know what I get - I'm still having issues with my cc2531 going into a bad state (put in a new issue with this (#161), going to add suport for Nanoleaf bulbs and hive plugs this weekend (assuming they go relatively easily)

matchett808, 2018-06-29:
Haven't managed to test this tonight, will tomorrow 👍 

PR will be incoming for Hive plug and Nanoleaf bulb as well

matchett808, 2018-06-30:
Howdy, 

Got the following log:

```
2018-6-30 09:52:14 INFO MQTT publish, topic: 'zb/bridge/log', payload: '{"type":"device_connected","message":"0x00124b000fa0779e"}'
Configuring Hive motion sensor after 10 seconds
2018-6-30 09:52:14 DEBUG Recieved zigbee message with data "online"
  zigbee-shepherd:request REQ --> AF:dataRequest, transId: 146 +10s
  cc-znp:SREQ --> AF:dataRequest, { dstaddr: 11918, destendpoint: 6, srcendpoint: 1, clusterid: 1280, transid: 146, options: 48, radius: 30, len: 14, data: <Buffer 00 92 02 10 00 f0 da eb bd 14 00 4b 12 00> } +1ms
  serialport write 29 bytes of data +1ms
  cc-znp:SRSP <-- AF:dataRequest, { status: 0 } +11ms
  zigbee-shepherd:request RSP <-- AF:dataRequest, status: 0 +1ms
  cc-znp:AREQ <-- AF:dataConfirm, { status: 0, endpoint: 1, transid: 146 } +211ms
  zigbee-shepherd:msgHdlr IND <-- AF:dataConfirm, transId: 146 +0ms
  cc-znp:AREQ <-- AF:incomingMsg, { groupid: 0, clusterid: 1280, srcaddr: 11918, srcendpoint: 6, dstendpoint: 1, wasbroadcast: 0, linkquality: 110, securityuse: 0, timestamp: 11489702, transseqnumber: 0, len: 4, data: <Buffer 18 92 04 00> } +9ms
  zigbee-shepherd:msgHdlr IND <-- AF:incomingMsg, transId: 0 +0ms
  zigbee-shepherd:request REQ --> AF:dataRequest, transId: 147 +1ms
  cc-znp:SREQ --> AF:dataRequest, { dstaddr: 11918, destendpoint: 6, srcendpoint: 1, clusterid: 1280, transid: 147, options: 48, radius: 30, len: 6, data: <Buffer 00 93 0c 00 00 f0> } +1ms
  serialport write 21 bytes of data +0ms
Failed to write iasCieAddr
  cc-znp:SRSP <-- AF:dataRequest, { status: 0 } +11ms
  zigbee-shepherd:request RSP <-- AF:dataRequest, status: 0 +0ms
  cc-znp:AREQ <-- AF:dataConfirm, { status: 0, endpoint: 1, transid: 147 } +119ms
  zigbee-shepherd:msgHdlr IND <-- AF:dataConfirm, transId: 147 +1ms
  cc-znp:AREQ <-- AF:incomingMsg, { groupid: 0, clusterid: 1280, srcaddr: 11918, srcendpoint: 6, dstendpoint: 1, wasbroadcast: 0, linkquality: 110, securityuse: 0, timestamp: 11490181, transseqnumber: 0, len: 19, data: <Buffer 18 93 0d 01 00 00 30 01 00 31 02 00 19 10 00 f0 11 00 20> } +22ms
  zigbee-shepherd:msgHdlr IND <-- AF:incomingMsg, transId: 0 +3ms
  zigbee-shepherd:request REQ --> AF:dataRequest, transId: 148 +0ms
  cc-znp:SREQ --> AF:dataRequest, { dstaddr: 11918, destendpoint: 6, srcendpoint: 1, clusterid: 1280, transid: 148, options: 48, radius: 30, len: 13, data: <Buffer 00 94 00 00 00 01 00 02 00 10 00 11 00> } +1ms
  serialport write 28 bytes of data +0ms
  cc-znp:SRSP <-- AF:dataRequest, { status: 0 } +11ms
  zigbee-shepherd:request RSP <-- AF:dataRequest, status: 0 +0ms
  cc-znp:AREQ <-- AF:dataConfirm, { status: 0, endpoint: 1, transid: 148 } +100ms
  zigbee-shepherd:msgHdlr IND <-- AF:dataConfirm, transId: 148 +0ms
  cc-znp:AREQ <-- AF:incomingMsg, { groupid: 0, clusterid: 1280, srcaddr: 11918, srcendpoint: 6, dstendpoint: 1, wasbroadcast: 0, linkquality: 110, securityuse: 0, timestamp: 11490581, transseqnumber: 0, len: 37, data: <Buffer 18 94 01 00 00 00 30 00 01 00 00 31 0d 00 02 00 00 19 00 00 10 00 00 f0 da eb bd 14 00 4b 12 00 11 00 00 20 ff> } +14ms
  zigbee-shepherd:msgHdlr IND <-- AF:incomingMsg, transId: 0 +5ms
2018-6-30 09:52:25 DEBUG Recieved zigbee message with data {"cid":"ssIasZone","data":{"iasCieAddr":"0x00124b0014bdebda"}}
2018-6-30 09:52:25 WARN No converter available for 'MOT003' with cid 'ssIasZone' and type 'devChange'
2018-6-30 09:52:25 WARN Please see: https://github.com/Koenkk/zigbee2mqtt/wiki/How-to-support-new-devices.
```

However - I notice something in the code you posted - are the console.log statements the right way round? (err will evaluate to true if it's got a value - so if there is and error - and if(true) - print out success else print out fail is the wrong way round then?)

Koenkk, 2018-06-30:
Yes the `if` is indeed wrong.

What is logged after no converter available? We need to check if the enrollRequest is send.

matchett808, 2018-06-30:
Here's a pastebin - I can't ID an enrollRequest: https://pastebin.com/r5AmZWNS (I'm trying again to try and ID it)

Koenkk, 2018-06-30:
Can you try the following code:

```js
    {
        zigbeeModel: ['MOT003'],
        model: 'MOT003',
        vendor: 'Hive',
        description: 'Hive Motion Sensor',
        supports: 'occupancy',
        fromZigbee: [],
        toZigbee: [],
        configure: (ieeeAddr, shepherd, coordinator, callback) => {
            console.log('Configuring Hive motion sensor after 10 seconds');

            setTimeout(() => {
                const device = shepherd.find(ieeeAddr, 6);

                if (device) {
                    device.write('ssIasZone', 'iasCieAddr', coordinator.device.getIeeeAddr(), (err) => {
                        if (err) {
                            console.log('Failed to write iasCieAddr')
                        } else {
                            console.log('Sucesfully wrote iasCieAddr, with DEBUG=* you should now see an enroll request');
                            setTimeout(() => {
                                device.functional('ssIasZone', 'enrollRsp', {enrollrspcode: 0x00, zoneid : 255}, (err) => {
                                    if (err) {
                                        console.log('Error on enrollRsp');
                                    } else {
                                        console.log('Sucesfully enrolled!!! :DDDDD');
                                    }
                                });
                            }, 5000);
                        }
                    });
                }
            }, 10 * 1000);
        },
    },
```

matchett808, 2018-07-01:
Howdy,

I get the 'successfully enrolled' message in the logs - but motion doesn't trigger any messages

Here's another pastebin: https://pastebin.com/YkEMr3L9  - I'm not sure how to make sense of some of those messages - there seems to be a fair bit of traffic after that message (but motion doesn't trigger any later on)

Koenkk, 2018-07-01:
The `zoneState` is still `0` which means that the device is not enrolled yet (should be `1`).

Can you try with the following code: (also write `zoneId` now).

```js
    {
        zigbeeModel: ['MOT003'],
        model: 'MOT003',
        vendor: 'Hive',
        description: 'Hive Motion Sensor',
        supports: 'occupancy',
        fromZigbee: [],
        toZigbee: [],
        configure: (ieeeAddr, shepherd, coordinator, callback) => {
            console.log('Configuring Hive motion sensor after 10 seconds');

            setTimeout(() => {
                const device = shepherd.find(ieeeAddr, 6);
                const zoneid = 5;
                if (device) {
                    device.write('ssIasZone', 'iasCieAddr', coordinator.device.getIeeeAddr(), (err) => {
                        if (err) {
                            console.log('Failed to write iasCieAddr');
                        } else {
                            console.log('Sucesfully wrote iasCieAddr');

                            device.write('ssIasZone', 'zoneId', zoneid, (err) => {
                                if (err) {
                                    console.log('failed to write zoneid');
                                } else {
                                    console.log('wrote zone id :D');
                                    setTimeout(() => {
                                        device.functional('ssIasZone', 'enrollRsp', {enrollrspcode: 0x00, zoneid: zoneid}, (err) => {
                                            if (err) {
                                                console.log('Error on enrollRsp');
                                            } else {
                                                console.log('Sucesfully enrolled!!! :DDDDD');
                                            }
                                        });
                                    }, 3000);
                                }
                            });
                        }
                    });
                }
            }, 10 * 1000);
        },
    },
```

matchett808, 2018-07-01:
Howdy,

It failed to write the zoneId - here's a pastebin: https://pastebin.com/3Eu7Ueqp

Did the same trying with a couple of different zoneid's - not sure where to go from here - guessing that this is writing in the write format? (zb data type 240)?

ciotlosm, 2018-07-01:
@matchett808 I've updated your comment to point to hastebin as it's more clean

Koenkk, 2018-07-02:
I've found this [Smartthings handler for the Hive motion sensor](https://github.com/simonjgreen/Smartthings-Hive-Sensors/blob/master/handler-hive-motion-sensor.groovy). That means that theoretically we should be able to support this.

Can you try with:

```js
    {
        zigbeeModel: ['MOT003'],
        model: 'MOT003',
        vendor: 'Hive',
        description: 'Hive Motion Sensor',
        supports: 'occupancy',
        fromZigbee: [],
        toZigbee: [],
        configure: (ieeeAddr, shepherd, coordinator, callback) => {
            const device = shepherd.find(ieeeAddr, 6);
            const actions = [
                (cb) => device.write('ssIasZone', 'iasCieAddr', coordinator.device.getIeeeAddr(), cb),
                (cb) => device.report('msOccupancySensing', 'occupancy', 0, 600, null, cb),
                (cb) => device.report('msTemperatureMeasurement', 'measuredValue', 30, 600, 1, cb),
                (cb) => device.report('genPowerCfg', 'batteryVoltage', 10, 600, 1, cb),
                (cb) => device.report('ssIasZone', 'zoneStatus', 0, 30, null, cb),
                (cb) => device.functional('ssIasZone', 'enrollRsp', {enrollrspcode: 1, zoneid: 23}, cb),
            ];

            execute(device, actions, callback);
        },
    },
```

iain-matchett, 2018-07-02:
I'll do that tonight - unfortunately I've fell back into the issue where I can't control the devices (similar to #172 and #155) - still trying to chase down if I've got a hardware issue

I have a ST hub, I will also test that ST handler (there are a couple of versions of the motion sensor - and that DTH looks to be for V1) - I linked that DTH: 

https://community.smartthings.com/t/hive-motion-contact-sensors/57452

This comment on that thread:

https://community.smartthings.com/t/hive-motion-contact-sensors/57452/26

discusses the difference between v1 and v2 - it's a v2 I've got (MOT003), which makes me realise something: 

> Don't know if it will help much but there are smartthings device handlers: https://community.smartthings.com/t/hive-motion-contact-sensors/57452 this doesn't work for motion with this version - but [...]

> The orvibo DTH looks at IASzone 19?

Wich means your code:

```
 {
        zigbeeModel: ['MOT003'],
        model: 'MOT003',
        vendor: 'Hive',
        description: 'Hive Motion Sensor',
        supports: 'occupancy',
        fromZigbee: [],
        toZigbee: [],
        configure: (ieeeAddr, shepherd, coordinator, callback) => {
            console.log('Configuring Hive motion sensor after 10 seconds');

            setTimeout(() => {
                const device = shepherd.find(ieeeAddr, 6);
                const zoneid = 5; // Wrong zone ID?
                if (device) {
                    device.write('ssIasZone', 'iasCieAddr', coordinator.device.getIeeeAddr(), (err) => {
                        if (err) {
                            console.log('Failed to write iasCieAddr');
                        } else {
                            console.log('Sucesfully wrote iasCieAddr');

                            device.write('ssIasZone', 'zoneId', zoneid, (err) => {
                                if (err) {
                                    console.log('failed to write zoneid');
                                } else {
                                    console.log('wrote zone id :D');
                                    setTimeout(() => {
                                        device.functional('ssIasZone', 'enrollRsp', {enrollrspcode: 0x00, zoneid: zoneid}, (err) => {
                                            if (err) {
                                                console.log('Error on enrollRsp');
                                            } else {
                                                console.log('Sucesfully enrolled!!! :DDDDD');
                                            }
                                        });
                                    }, 3000);
                                }
                            });
                        }
                    });
                }
            }, 10 * 1000);
        },
    },
```

Could be modified `zoneid = 19` rather than `zoneid = 5` ?

matchett808, 2018-07-02:
Also a note - on the link:

https://community.smartthings.com/t/hive-motion-contact-sensors/57452

I tested that DTH in October 2017, no changes since 👍

matchett808, 2018-07-02:
None of the above work - tried both the code you posted and the code I posted, each with different zoneids (5/19/23), none of them seem to have done much 🤷‍♂️

matchett808, 2018-07-02:
Scratch that, your code brings back temperature reporting - not saw battery (might not have ran for long enough) - but still no motion...

matchett808, 2018-07-02:
oh wait, got battery in logs - still can't get any motion

[8 more replies]

closed 2018-12-07 by stale[bot]