Koenkk/zigbee2mqtt #144: Running the bridge on a different machine as the USB sniffer

1of16, 2018-06-24:
hi,
I hope I haven't overlooked this info:
but is it possible to plug in the USB sniffer in on machine A, make it remotely accessible e.g. via ser2net and run the bridge on machine B?
how do I need to change the configuration?
`# Required: serial settings
serial:
  # Required: location of CC2531 USB sniffer
  port: tcp://192.168.1.1:1234`
something like that?
Thank you!

Koenkk, 2018-06-24:
I've never tried this, but `serial` should be the path to your sniffer. So probably `serial: tcp://192.168.1.1:1234` (but I doubt if it works).

indrora, 2018-06-24:
Do you have software that makes serial ports from one machine magically appear on another?

1of16, 2018-06-24:
`serial: tcp://192.168.1.1:1234` seems not to be working:
`npm start

> zigbee2mqtt@0.0.0 start /opt/zigbee2mqtt
> node index.js

2018-6-24 15:56:08 INFO Starting zigbee2mqtt version 0.0.0 (commit #01a26ad)
2018-6-24 15:56:08 INFO Starting zigbee-shepherd
/opt/zigbee2mqtt/node_modules/proving/index.js:5
        throw new TypeError(msg || 'Input value should be a string.');
        ^

TypeError: path should be a string.
    at Object.proving.string (/opt/zigbee2mqtt/node_modules/proving/index.js:5:15)
    at new ZShepherd (/opt/zigbee2mqtt/node_modules/zigbee-shepherd/lib/shepherd.js:38:13)
    at Zigbee.start (/opt/zigbee2mqtt/lib/zigbee.js:33:25)
    at startupLogVersion (/opt/zigbee2mqtt/lib/controller.js:43:25)
    at git.getLastCommit (/opt/zigbee2mqtt/lib/controller.js:547:13)
    at /opt/zigbee2mqtt/node_modules/git-last-commit/source/index.js:55:4
    at /opt/zigbee2mqtt/node_modules/git-last-commit/source/index.js:23:3
    at ChildProcess.exithandler (child_process.js:267:7)
    at emitTwo (events.js:126:13)
    at ChildProcess.emit (events.js:214:7)
npm ERR! code ELIFECYCLE
npm ERR! errno 1
npm ERR! zigbee2mqtt@0.0.0 start: `node index.js`
npm ERR! Exit status 1
npm ERR!
npm ERR! Failed at the zigbee2mqtt@0.0.0 start script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.

npm ERR! A complete log of this run can be found in:
npm ERR!     /.npm/_logs/2018-06-24T13_56_08_848Z-debug.log`

cat .npm/_logs/2018-06-24T13_56_08_848Z-debug.log
`
0 info it worked if it ends with ok
1 verbose cli [ '/usr/bin/node', '/usr/bin/npm', 'start' ]
2 info using npm@5.6.0
3 info using node@v8.11.3
4 verbose run-script [ 'prestart', 'start', 'poststart' ]
5 info lifecycle zigbee2mqtt@0.0.0~prestart: zigbee2mqtt@0.0.0
6 info lifecycle zigbee2mqtt@0.0.0~start: zigbee2mqtt@0.0.0
7 verbose lifecycle zigbee2mqtt@0.0.0~start: unsafe-perm in lifecycle true
8 verbose lifecycle zigbee2mqtt@0.0.0~start: PATH: /usr/lib/node_modules/npm/node_modules/npm-lifecycle/node-gyp-bin:/opt/zigbee2mqtt/node_modules/.bin:/home/philip/bin:/home/philip/.local/bin:/home/philip/platform-tools:/home/philip/bin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/usr/games:/usr/local/games:/snap/bin
9 verbose lifecycle zigbee2mqtt@0.0.0~start: CWD: /opt/zigbee2mqtt
10 silly lifecycle zigbee2mqtt@0.0.0~start: Args: [ '-c', 'node index.js' ]
11 silly lifecycle zigbee2mqtt@0.0.0~start: Returned: code: 1  signal: null
12 info lifecycle zigbee2mqtt@0.0.0~start: Failed to exec start script
13 verbose stack Error: zigbee2mqtt@0.0.0 start: `node index.js`
13 verbose stack Exit status 1
13 verbose stack     at EventEmitter.<anonymous> (/usr/lib/node_modules/npm/node_modules/npm-lifecycle/index.js:285:16)
13 verbose stack     at emitTwo (events.js:126:13)
13 verbose stack     at EventEmitter.emit (events.js:214:7)
13 verbose stack     at ChildProcess.<anonymous> (/usr/lib/node_modules/npm/node_modules/npm-lifecycle/lib/spawn.js:55:14)
13 verbose stack     at emitTwo (events.js:126:13)
13 verbose stack     at ChildProcess.emit (events.js:214:7)
13 verbose stack     at maybeClose (internal/child_process.js:925:16)
13 verbose stack     at Process.ChildProcess._handle.onexit (internal/child_process.js:209:5)
14 verbose pkgid zigbee2mqtt@0.0.0
15 verbose cwd /opt/zigbee2mqtt
16 verbose Linux 4.4.0-124-generic
17 verbose argv "/usr/bin/node" "/usr/bin/npm" "start"
18 verbose node v8.11.3
19 verbose npm  v5.6.0
20 error code ELIFECYCLE
21 error errno 1
22 error zigbee2mqtt@0.0.0 start: `node index.js`
22 error Exit status 1
23 error Failed at the zigbee2mqtt@0.0.0 start script.
23 error This is probably not a problem with npm. There is likely additional logging output above.
24 verbose exit [ 1, true ]`

Koenkk, 2018-06-24:
I dont think that the [serial library](https://www.npmjs.com/package/serialport#opening-a-port) supports this. Would it be possible to make it look like a local device (`/dev/mydevice`)?

1of16, 2018-06-24:
yes, a solution would be socat: https://github.com/asaif/socat-init
I just kinda hoped, it would be possible to support serial and tcp connections :)

ptvoinfo, 2018-07-01:
I'm using this connection frequently:
A - BananaPi M2U with ser2net and CC2530 (or CC2531)
B - my Windows development computer with http://www.aggsoft.com/tcp-com/. nodejs and zigbee2mqtt perfectly works with a virtual com ports (on Windows).

ciotlosm, 2018-07-01:
@ptvoinfo maybe write another of your awesome posts to show such a setup?

1of16, 2018-07-02:
some information about my setup (both debain/ubuntu):
ser2net config:
`<a-port-you-like>:raw:0:/dev/serial/by-id/usb-Texas_Instruments_TI_CC2531_USB_CDC___<your-id>-if00:38400 NONE 1STOPBIT 8DATABITS HANGUP_WHEN_DONE`

used this as socat-deamon: https://github.com/docksal/service-socat
`OPTIONS="pty,link=/dev/tty<your-usb-tty-name>,raw,user=<root-or-your-user>,group=dialout,mode=777 tcp:<ser2net-ip>:<ser2net-port>"`

lolorc, 2018-07-02:
it looks like this question has already been raised
https://github.com/node-serialport/node-serialport/issues/825

closed 2018-09-07 by stale[bot]