Koenkk/zigbee2mqtt #145: Docker Container stopped ervery couple of hours

19Homer73, 2018-06-25:
Hello,

my docker-container stoppes every couple hours with this log-message:

{"message":"request timeout","stack":"Error: request timeout\n    at CcZnp.<anonymous> (/app/node_modules/cc-znp/lib/ccznp.js:254:22)\n    at Object.onceWrapper (events.js:315:30)\n    at emitOne (events.js:116:13)\n    at CcZnp.emit (events.js:211:7)\n    at Timeout.<anonymous> (/app/node_modules/cc-znp/lib/ccznp.js:233:18)\n    at ontimeout (timers.js:475:11)\n    at tryOnTimeout (timers.js:310:5)\n    at Timer.listOnTimeout (timers.js:270:5)"}

Automatic restart is not possible.
I have to stop and start the container manually, then it works fine again.

Can anyone help me with this problem, please?

ciotlosm, 2018-06-25:
Please update stick firmware to latest provided

<sub>Sent with <a href="http://githawk.com">GitHawk</a></sub>

ciotlosm, 2018-06-25:
@19Homer73 Can you please post the zigbee2mqtt version (it's shown when started) and if you have the firmware version md5 hash or URL that you used to install + date when you flashed the stick?

19Homer73, 2018-06-25:
The log says:

2018-6-25 07:39:07 INFO Starting zigbee2mqtt version 0.0.0 (commit #094c288)

I just downloaded the "RAW" file, doing a right mouseclick and save as **CC2531ZNP-Prod.hex**

I flashed the stick last week friday, 22nd June

ciotlosm, 2018-06-25:
@19Homer73 it might be that you flashed latest "stable" or "master. The fixes so far are in the "experimental" branch. See my link above. 

I'd recommend you flash this and help us out with a test. So far it's been reported to be far better stability.

ciotlosm, 2018-06-25:
yes

19Homer73, 2018-06-25:
OK, I will try later and give you response on it!
Thanks!!!

ciotlosm, 2018-06-28:
@19Homer73 any update?

19Homer73, 2018-06-28:
Sorry, very busy days...
Yes, I did an update on the docker-container on sunday evening.
Because that was easier for me than flashing the USB-stick.
Since then, there are no problems anymore...
It's running for 4 days now.

Koenkk, 2018-06-30:
Looks good, I will close this issue for now, in case you experience any problems please reopen.

closed 2018-06-30 by Koenkk