Koenkk/zigbee2mqtt #147: Change mqtt output of body movement sensor

mpuff, 2018-06-25:
Hello,

i want to change the mqtt message of a Xiaomi MiJia door & window contact sensor to OPEN and CLOSED, is this possible?

Kind regards

Koenkk, 2018-06-25:
You would have manually modify https://github.com/Koenkk/zigbee-shepherd-converters/blob/master/converters/fromZigbee.js#L249. As this is a binary sensor it uses `true` and `false`.

jarrah31, 2018-06-25:
Something like this that I use with my Xiaomi sensors should work:
```
sensor:
  - platform: template
    sensors:
      win_study_lrg:
        friendly_name: "Window - Study Large"
        value_template: >-
          {% if is_state("binary_sensor.door_window_sensor_158d0001d68293", "on") %}Open
          {% elif is_state("binary_sensor.door_window_sensor_158d0001d68293", "off") %}Closed
          {% else %}unknown
          {% endif %}
        icon_template: >
          {% set win_study_lrg_state = states.binary_sensor.door_window_sensor_158d0001d68293.state %}
          {% if win_study_lrg_state == 'on' %}mdi:window-open
          {% elif win_study_lrg_state == 'off' %}mdi:window-closed
          {% else %}mdi:alert-outline
          {% endif %}
```

Koenkk, 2018-06-25:
Yes, that should work!

mpuff, 2018-06-26:
i have changed the code like this:

xiaomi_contact: {
        cid: 'genOnOff',
        type: 'attReport',
        convert: (model, msg, publish, options) => {
            return {contact: msg.data.data['onOff'] === 1 ? 'OPEN' : 'CLOSED'};
        },

Now it works and I can add it in openhab as it should.

Thank you very much :)

closed 2018-06-26 by mpuff