Koenkk/zigbee2mqtt #148: How to configure MQTT Client Id?

aaamitsingh, 2018-06-25:
Some MQTT broker requires "client id" for authentication. In "data/configuration.yaml" there is no any option to enter the Client Id. Can anyone tell me how to pass the Client Id?

# Required: MQTT settings
mqtt:
  # Required: MQTT base topic for zigbee2mqtt MQTT messages
  base_topic: zigbee2mqtt
  # Required: MQTT server URL
  server: 'mqtt://localhost:1883'
  # Optional: MQTT server authentication user
  user: my_user
  # Optional: MQTT server authentication password
  password: my_password

Koenkk, 2018-06-25:
Isn't this just the `user:`?

ryanbeaton, 2018-06-26:
I did a quick bit of research:

https://stackoverflow.com/questions/27174271/what-is-the-clientid-needed-for 
Every MQTT Connection needs a unique client ID. With this unique client ID the broker can recognize when a client reconnects and can close an old potentially half-open TCP connection for the client. Also, the MQTT broker can hold a persistent session for MQTT clients if they desire to do so (cleanSession=false on CONNECT). If you disconnect and reconnect with the same client ID, you will receive all messages for your subscriptions you may have missed and the broker knows your old subscriptions for that client ID.

https://stackoverflow.com/questions/49796461/how-do-i-get-clientid-with-nodered
When the client connects, it sends the clientId, I guess this info is kept somewhere and could be extracted. In IBM bluemix, for example, they manage con add to msg.payload two parameters, DeviceID and deviceType, that are part of the clientId string when connecting.

<sub>Sent with <a href="http://githawk.com">GitHawk</a></sub>

Koenkk, 2018-06-26:
@ryanbeaton thanks!

I've added support for the MQTT client in branch: https://github.com/Koenkk/zigbee2mqtt/tree/issue_148

You can configure it via:

```yaml
mqtt:
  base_topic: zigbee2mqtt
  server: 'mqtt://localhost'
  client_id: 'my_client_id'
```

@aaamitsingh  can you test if it works?

aaamitsingh, 2018-06-27:
@ryanbeaton 👍 : Yes you are right and without client_id an MQTT connection cannot be established. Normally broker generates a random client_id for every connection. But some cases where the user needs to validate the client_id or only allow a connection with the prefix client_id there it is mandatory to pass the client_id.

@Koenkk 👍 : Thank you so much for the quick enhancement. Still now I havent used this library I was just checking the configuration that it will support my system or not. So give me some time I will test and let you know the status.

tb-killa, 2018-07-24:
@Koenkk 
Work great. I can confirm this work as it should.
Please merge it to master or dev and close this branch and Issue.

Koenkk, 2018-07-24:
Merged and documented: https://github.com/Koenkk/zigbee2mqtt/wiki/Configuration

closed 2018-07-24 by Koenkk