Koenkk/zigbee2mqtt #149: Trying to add Philips Hue E27 lightbulb 9290011370

wilfredsmit, 2018-06-25:
Hello, I'm trying to add support for the Philips Hue white A60 9.5W 50mA 806 Lumen 9290011370 LWB006.

The Philips configuration (LWB0006) inside devices.js looks like this:

```
    // Philips
    {
        zigbeeModel: ['LLC020'],
        model: '7146060PH',
        vendor: 'Philips',
        description: 'Hue Go',
        supports: generic.light_onoff_brightness_colortemp_colorxy().supports,
        fromZigbee: generic.light_onoff_brightness_colortemp_colorxy().fromZigbee,
        toZigbee: generic.light_onoff_brightness_colortemp_colorxy().toZigbee,
    },
    {
        zigbeeModel: ['LWB0006'],
        model: '9290011370',
        vendor: 'Philips',
        description: 'Hue white Hue white A60 E27',
        supports: generic.light_onoff_brightness().supports,
        fromZigbee: generic.light_onoff_brightness().fromZigbee,
        toZigbee: generic.light_onoff_brightness().toZigbee,
    },
    {
        zigbeeModel: ['LWB010'],
        model: '8718696449691',
        vendor: 'Philips',
        description: 'Hue White Single bulb B22',
        supports: generic.light_onoff_brightness().supports,
        fromZigbee: generic.light_onoff_brightness().fromZigbee,
        toZigbee: generic.light_onoff_brightness().toZigbee,
    },
    {
        zigbeeModel: ['LCT015'],
        model: '9290012573A',
        vendor: 'Philips',
        description: 'Hue white and color ambiance E27',
        supports: generic.light_onoff_brightness_colortemp_colorxy().supports,
        fromZigbee: generic.light_onoff_brightness_colortemp_colorxy().fromZigbee,
        toZigbee: generic.light_onoff_brightness_colortemp_colorxy().toZigbee,
    },
```

I use zigbee2mqtt in a docker container. After a restart the docker container. It's still not recognized. 
Any idea what I'm doing wrong?

Koenkk, 2018-06-25:
What do you mean with not recognized? 

Can you post your zigbee2mqtt startup log on hastebin.com and link here?

wilfredsmit, 2018-06-25:
Nevermind I made an initial error: I thought I used zigbeeModel: ['LWB006'] but instead I used zigbeeModel: ['LWB0006']. One zero to much. It works now, perfectly. I edited devices.js and homeassistant.js. 
How can I make this available to everyone?

Koenkk, 2018-06-25:
You can submit a pull request here: https://github.com/koenkk/zigbee-shepherd-converters

Koenkk, 2018-06-25:
I will update `homeassistant.js`. 

I don't see a pull request yet? https://github.com/koenkk/zigbee-shepherd-converters/pulls

wilfredsmit, 2018-06-25:
Oké, done!

Koenkk, 2018-06-25:
Great, thanks!

closed 2018-06-25 by Koenkk