Koenkk/zigbee2mqtt #150: No Battery State for Xiaomi Smokedetector JTYJ-GD-01LM/BW

Sonsi79, 2018-06-26:
I have add my Xiaomi Honeywell smoke detector JTYJ-GD-01LM/BW and it works fine and publish the "smoke" state.
But the smoke detector is the didn't publish any state of the battery in MQTT.

Is it possible to get this state also as a publish?

Sonsi79, 2018-06-26:
I have run 4 devices for few days and only get the smoke state as publish. All of my other Xiaomi devices publish the battery state only the smoke detector didn't.

Hope you can help

Sonsi79, 2018-06-27:
I have try it and get this error by pressing the Button on smoke detector:
`WARN No converter available for 'JTYJ-GD-01LM/BW' with cid 'genBasic' and type 'attReport'`

Koenkk, 2018-06-28:
Good! Can you now set `advanced` -> `log_level: debug` in `data/configuration.yaml`. https://github.com/Koenkk/zigbee2mqtt/wiki/Configuration

This will print the message.

Sonsi79, 2018-06-29:
OK, now im getting the message
`2018-6-29 23:35:02 DEBUG Recieved zigbee message with data {"cid":"ssIasZone","zoneStatus":0}
2018-6-29 23:35:02 INFO MQTT publish, topic: 'zigbee2mqtt/0x00158d000202eff6', payload: '{"smoke":false}'`

Koenkk, 2018-06-30:
Could you post the entry of this device from `data/database.db`?

Sonsi79, 2018-06-30:
I hope this is the right one.

`{"id":11,"type":"EndDevice","ieeeAddr":"0x00158d000202eff6","nwkAddr":20594,"manufId":4151,"manufName":"LUMI","powerSource":"Battery","modelId":"lumi.sensor_smoke","epL$
`

Sonsi79, 2018-06-30:
I hope this is the right line...

`{"id":11,"type":"EndDevice","ieeeAddr":"0x00158d000202eff6","nwkAddr":20594,"manufId":4151,"manufName":"LUMI","powerSource":"Battery","modelId":"lumi.sensor_smoke","epList":[1],"status":"offline","joinTime":null,"endpoints":{"1":{"profId":260,"epId":1,"devId":1026,"inClusterList":[0,1,3,12,18,1280],"outClusterList":[25],"clusters":{"genBasic":{"dir":{"value":1},"attrs":{}},"genPowerCfg":{"dir":{"value":1},"attrs":{}},"genIdentify":{"dir":{"value":1},"attrs":{}},"genAnalogInput":{"dir":{"value":1},"attrs":{}},"genMultistateInput":{"dir":{"value":1},"attrs":{}},"genOta":{"dir":{"value":2},"attrs":{}},"ssIasZone":{"dir":{"value":1},"attrs":{}}}}},"_id":"pCPL00secZLGvXec"}
`

Koenkk, 2018-06-30:
That's the right line, but I don't see the battery being in.

@kirovilya  can you help with this?

kirovilya, 2018-07-02:
I have not Xiaomi Smoke sensor... I'll ask the users of my application for this issue.
Think, it was like for other xiaomi sensor. Strange that this is not so...

kirovilya, 2018-07-02:
Confirmed. Battery not received :(
Need research events and data... I request a detailed log of work with the sensor.

Found some code here 
https://github.com/veeceeoh/xiaomi-hubitat/blob/0c42285fc9c7cb4554d7b8968087d50ef9cb048b/devicedrivers/xiaomi-honeywell-smoke-detector.src/xiaomi-honeywell-smoke-detector.groovy#L86

Sonsi79, 2018-07-02:
In i can help, let me know...

kirovilya, 2018-07-03:
@Sonsi79 Can you collect detailed log from smoke-sensor like describe here https://github.com/Koenkk/zigbee2mqtt/wiki/How-to-debug#all-debug-logging during short press button on sensor. And put file here.

Sonsi79, 2018-07-03:
@kirovilya ok i will do it later after work and publish the results here...

Sonsi79, 2018-07-03:
Here's the log file from DEBUG run.
During this run i have first pressed buttons on my Switch and later pressed the button on smoke detector. It seems my thermometer also publish something during thins time.

I hope this file helps...
[log.txt](https://github.com/Koenkk/zigbee2mqtt/files/2160848/log.txt)

kirovilya, 2018-07-04:
@Sonsi79 that is not all... you need start `DEBUG=* npm start` and save all from console to file. So, command will be: `DEBUG=zigbee-shepherd* npm start > log.txt`

Sonsi79, 2018-07-04:
Ok. I will retry it as soon as possible. My fault

Sonsi79, 2018-07-04:
Ok, done. Here the file...
[log.txt](https://github.com/Koenkk/zigbee2mqtt/files/2162502/log.txt)

Sonsi79, 2018-07-04:
Hm it seems that most of Debug Infos wasn't Safe in this file. 
Here the Output of Console during Debug start.

`DEBUG=* npm start > log.txt
2018-7-4 10:57:17 DEBUG Using zigbee-shepherd with settings: '{"net":{"panId":6754,"channelList":[11]},"dbPath":"/opt/zigbee2mqtt/data/database.db","sp":{"baudrate":115200,"rtscts":true}}'
  zigbee-shepherd:init zigbee-shepherd booting... +0ms
  cc-znp The serialport /dev/ttyACM0 is opened. +25ms
  zigbee-shepherd:request REQ --> SYS:osalNvRead +3ms
  cc-znp:SREQ --> SYS:osalNvRead, { id: 3840, offset: 0, len: 1, value: [ 85 ] } +5ms
  serialport write 8 bytes of data +16ms
  cc-znp:SRSP <-- SYS:osalNvRead, { status: 0, len: 1, value: <Buffer 55> } +33ms
  zigbee-shepherd:request RSP <-- SYS:osalNvRead, status: 0 +2ms
  zigbee-shepherd:request REQ --> SAPI:readConfiguration +35ms
  cc-znp:SREQ --> SAPI:readConfiguration, { configid: 131, len: 2, value: [ 98, 26 ] } +1ms
  serialport write 6 bytes of data +1ms
  cc-znp:SRSP <-- SAPI:readConfiguration, { status: 0, configid: 131, len: 2, value: <Buffer 62 1a> } +9ms
  zigbee-shepherd:request RSP <-- SAPI:readConfiguration, status: 0 +1ms
  zigbee-shepherd:request REQ --> SAPI:readConfiguration +12ms
  cc-znp:SREQ --> SAPI:readConfiguration, { configid: 132, len: 4, value: [ 0, 8, 0, 0 ] } +1ms
  serialport write 6 bytes of data +1ms
  cc-znp:SRSP <-- SAPI:readConfiguration, { status: 0, configid: 132, len: 4, value: <Buffer 00 08 00 00> } +12ms
  zigbee-shepherd:request RSP <-- SAPI:readConfiguration, status: 0 +1ms
  zigbee-shepherd:request REQ --> SAPI:readConfiguration +11ms
  cc-znp:SREQ --> SAPI:readConfiguration, { configid: 98, len: 16, value: [ 1, 3, 5, 7, 9, 11, 13, 15, 0, 2, 4, 6, 8, 10, 12, 13 ] } +1ms
  serialport write 6 bytes of data +2ms
  cc-znp:SRSP <-- SAPI:readConfiguration, { status: 0, configid: 98, len: 16, value: <Buffer 01 03 05 07 09 0b 0d 0f 00 02 04 06 08 0a 0c 0d> } +21ms
  zigbee-shepherd:request RSP <-- SAPI:readConfiguration, status: 0 +1ms
  zigbee-shepherd:request REQ --> SAPI:readConfiguration +12ms
  cc-znp:SREQ --> SAPI:readConfiguration, { configid: 99, len: 1, value: [ 0 ] } +1ms
  serialport write 6 bytes of data +1ms
  cc-znp:SRSP <-- SAPI:readConfiguration, { status: 0, configid: 99, len: 1, value: <Buffer 00> } +15ms
  zigbee-shepherd:request RSP <-- SAPI:readConfiguration, status: 0 +1ms
  zigbee-shepherd:request REQ --> SAPI:getDeviceInfo +13ms
  cc-znp:SREQ --> SAPI:getDeviceInfo, { param: 0 } +1ms
  serialport write 6 bytes of data +1ms
  cc-znp:SRSP <-- SAPI:getDeviceInfo, { param: 0, value: <Buffer 00 01 00 10 00 00 00 09> } +7ms
  zigbee-shepherd:request RSP <-- SAPI:getDeviceInfo +1ms
  zigbee-shepherd:init Start the ZNP as a coordinator... +2ms
  zigbee-shepherd:request REQ --> ZDO:startupFromApp +1ms
  cc-znp:SREQ --> ZDO:startupFromApp, { startdelay: 100 } +20ms
  serialport write 7 bytes of data +2ms
  cc-znp:SRSP <-- ZDO:startupFromApp, { status: 0 } +611ms
  cc-znp:AREQ <-- ZDO:stateChangeInd, { state: 9 } +6ms
  zigbee-shepherd:msgHdlr IND <-- ZDO:stateChangeInd +2ms
  zigbee-shepherd:init Now the ZNP is a coordinator. +2ms
  zigbee-shepherd:request REQ --> SAPI:getDeviceInfo +2ms
  cc-znp:SREQ --> SAPI:getDeviceInfo, { param: 0 } +1ms
  serialport write 6 bytes of data +2ms
  cc-znp:SRSP <-- SAPI:getDeviceInfo, { param: 0, value: <Buffer 09 00 00 00 00 00 00 00> } +7ms
  zigbee-shepherd:request RSP <-- SAPI:getDeviceInfo +1ms
  zigbee-shepherd:request REQ --> SAPI:getDeviceInfo +1ms
  cc-znp:SREQ --> SAPI:getDeviceInfo, { param: 1 } +1ms
  serialport write 6 bytes of data +1ms
  cc-znp:SRSP <-- SAPI:getDeviceInfo, { param: 1, value: <Buffer dc eb bd 14 00 4b 12 00> } +7ms
  zigbee-shepherd:request RSP <-- SAPI:getDeviceInfo +1ms
  zigbee-shepherd:request REQ --> SAPI:getDeviceInfo +2ms
  cc-znp:SREQ --> SAPI:getDeviceInfo, { param: 2 } +0ms
  serialport write 6 bytes of data +1ms
  cc-znp:SRSP <-- SAPI:getDeviceInfo, { param: 2, value: <Buffer 00 00 bd 14 00 4b 12 00> } +7ms
  zigbee-shepherd:request RSP <-- SAPI:getDeviceInfo +1ms
  zigbee-shepherd:request REQ --> SAPI:getDeviceInfo +2ms
  cc-znp:SREQ --> SAPI:getDeviceInfo, { param: 5 } +1ms
  serialport write 6 bytes of data +1ms
  cc-znp:SRSP <-- SAPI:getDeviceInfo, { param: 5, value: <Buffer 0b 00 bd 14 00 4b 12 00> } +7ms
  zigbee-shepherd:request RSP <-- SAPI:getDeviceInfo +1ms
  zigbee-shepherd:request REQ --> SAPI:getDeviceInfo +1ms
  cc-znp:SREQ --> SAPI:getDeviceInfo, { param: 6 } +0ms
  serialport write 6 bytes of data +1ms
  cc-znp:SRSP <-- SAPI:getDeviceInfo, { param: 6, value: <Buffer 62 1a bd 14 00 4b 12 00> } +7ms
  zigbee-shepherd:request RSP <-- SAPI:getDeviceInfo +1ms
[... 174 lines cut ...]
  cc-znp:SREQ --> ZDO:nodeDescReq, { dstaddr: 28764, nwkaddrofinterest: 28764 } +2ms
  serialport write 9 bytes of data +1ms
  cc-znp:SRSP <-- ZDO:nodeDescReq, { status: 17 } +9ms
  cc-znp:SREQ --> ZDO:nodeDescReq, { dstaddr: 20594, nwkaddrofinterest: 20594 } +2ms
  serialport write 9 bytes of data +1ms
  cc-znp:SRSP <-- ZDO:nodeDescReq, { status: 17 } +8ms
  cc-znp:SREQ --> ZDO:nodeDescReq, { dstaddr: 55732, nwkaddrofinterest: 55732 } +2ms
  serialport write 9 bytes of data +2ms
  cc-znp:SRSP <-- ZDO:nodeDescReq, { status: 17 } +7ms
  cc-znp:SREQ --> ZDO:nodeDescReq, { dstaddr: 50253, nwkaddrofinterest: 50253 } +2ms
  serialport write 9 bytes of data +2ms
  cc-znp:SRSP <-- ZDO:nodeDescReq, { status: 17 } +8ms
[... cut ...] 1ms
  zigbee-shepherd:msgHdlr IND <-- AF:incomingMsg, transId: 0 +17ms
  cc-znp:AREQ <-- ZDO:tcDeviceInd, { nwkaddr: 20594, extaddr: '0x00158d000202eff6', parentaddr: 0 } +86ms
  zigbee-shepherd:msgHdlr IND <-- ZDO:tcDeviceInd +2ms
^C  zigbee-shepherd:request REQ --> ZDO:mgmtPermitJoinReq +42s
  cc-znp:SREQ --> ZDO:mgmtPermitJoinReq, { addrmode: 15, dstaddr: 65532, duration: 0, tcsignificance: 0 } +1ms
  serialport write 10 bytes of data +2ms
  cc-znp The serialport /dev/ttyACM0 is closed. +9ms`

kirovilya, 2018-07-04:
@Sonsi79 Well! I think we got it:
`cc-znp:AREQ <-- AF:incomingMsg, { groupid: 0, clusterid: 0, srcaddr: 20594, srcendpoint: 1, dstendpoint: 1, wasbroadcast: 0, linkquality: 52, securityuse: 0, timestamp: 94508, transseqnumber: 0, len: 46, data: <Buffer 18 7f 0a 01 ff 42 28 01 21 85 0c 03 28 19 04 21 a8 43 05 21 0e 00 06 24 03 00 00 00 00 0a 21 00 00 08 21 04 10 64 20 00 96 23 00 00 00 00> } +21ms`

Splited version of buffer:
18 7f 0a - header
01 ff - attrId ----- this is xiaomi attribute for heartbeat = 65281
42 - type 'charStr'
28 - length 40 byte
01 - index '1' ----- this is battery value of attribute
21 - type 'uint16'
85 0c - data ----- this is battery voltage = 3205
03 - index '3'
28 - type 'int8'
19 - data
04 - index '4'
21 - type 'uint16'
a8 43 - data
05 - index '5'
21 - type 'uint16'
0e 00 - data
06 - index '6'
24 - type 'uint40'
03 00 00 00 00 - data
0a 21 04 10 - '10' 'uint16' data
64 20 00 - '100' 'int8' data
96 23 00 00 00 00 - '150', 'uint32', data

So, this message is correctly understood by me, but why it not parsed by zigbee-shepherd

kirovilya, 2018-07-04:
Or may be it is not from smoke sensor message :)

Sonsi79, 2018-07-04:
Is this the message from 0x00158d000202eff6?
This is my smoke detector for sure.

You can see it at Logfile
`2018-7-4 10:57:19 [32mINFO[39m 0x00158d000202eff6 (0x00158d000202eff6): JTYJ-GD-01LM/BW - Xiaomi MiJia Honeywell smoke detector (EndDevice)
`

kirovilya, 2018-07-04:
I think I understand the reason.
The specified length of the charStr does not match the length of the data in the buffer.
In the evening I'll try again and suggest how to fix it.

Found this case with FF01 attribute https://github.com/Frans-Willem/AqaraHub/commit/a2cc15fad172ccb9a323f30ea50991e6001c32d0#diff-e45011ea876daa1366f56d10f998db49

Sonsi79, 2018-07-04:
You're great. Thanks much.

kirovilya, 2018-07-04:
@Sonsi79  Ok. let's try.
find file /node_modules/zcl-packet/lib/foundation.js:652.
was:
`if (stopLen <= parsedBufLen ) end();`
change to:
`if (this._buffer.length <= parsedBufLen+1) end();`

after that was error `TypeError: ru[dataType] is not a function` - it fixing in file /node_modules/zcl-packet/lib/foundation.js:681.
was:
```
    this.tap(function () {
        var dataType = getDataType(this.vars[dataTypeParam]);
        ru[dataType](name)(this);
    });
```
change to:
```
    this.tap(function () {
        if (this.vars[dataTypeParam]) {
            var dataType = getDataType(this.vars[dataTypeParam]);
            ru[dataType](name)(this);
        }
    });
```

try it and you will get battery :) I think so.
If ok, then I change in library.

Sonsi79, 2018-07-04:
Im sorry, there is still just the "smoke" state. No "battery" state :-(

kirovilya, 2018-07-04:
Did you short press on sensor? show detailed log please

Sonsi79, 2018-07-04:
Yes i pressed shot on sensor button.
Did you mean the DEBUG log or normal Log?

kirovilya, 2018-07-04:
with DEBUG

Sonsi79, 2018-07-04:
@kirovilya Here the new log
[Log2.txt](https://github.com/Koenkk/zigbee2mqtt/files/2164242/Log2.txt)

Sonsi79, 2018-07-04:
I edited the lines for sure.
Heres the DEBUG Output on Screen

In the Logfile is
`2018-7-4 22:28:54 ^[[32mINFO^[[39m Connecting to MQTT server at mqtt://localhost
2018-7-4 22:28:54 ^[[32mINFO^[[39m zigbee-shepherd ready
2018-7-4 22:28:55 ^[[32mINFO^[[39m Connected to MQTT server
2018-7-4 22:28:55 ^[[32mINFO^[[39m MQTT publish, topic: 'zigbee2mqtt/bridge/state', payload: 'online'
42 43 7
42 43 11
42 43 14
42 43 18
42 43 22
42 43 29
42 43 33
42 43 37
42 43 40
42 43 46
2018-7-4 22:29:16 ^[[32mINFO^[[39m MQTT publish, topic: 'zigbee2mqtt/0x00158d000202eff6', payload: '{"smoke":false}'
42 43 7
42 43 11
42 43 14
42 43 18
42 43 22
42 43 29
42 43 33
42 43 37
42 43 40
42 43 46
2018-7-4 22:29:34 ^[[32mINFO^[[39m MQTT publish, topic: 'zigbee2mqtt/bridge/state', payload: 'offline'
2018-7-4 22:29:34 ^[[32mINFO^[[39m Disconnecting from MQTT server
2018-7-4 22:29:34 ^[[32mINFO^[[39m zigbee-shepherd stopped
[Log3.txt](https://github.com/Koenkk/zigbee2mqtt/files/2164272/Log3.txt)
`

Sonsi79, 2018-07-04:
[Log3.txt](https://github.com/Koenkk/zigbee2mqtt/files/2164274/Log3.txt)

kirovilya, 2018-07-04:
Ok. Replace block from /node_modules/zcl-packet/lib/foundation.js:644:

```
ru.clause('xiaoMiStruct', function (name) {
    var stopLen = parsedBufLen+this.vars.len-2;
    this.tap(function(){
        this.loop('attrData', function (end) {
            console.log(stopLen, this._buffer.length, parsedBufLen+1);
            if (this._buffer.length <= parsedBufLen+1) {
                end();
            } else {
                parsedBufLen += 2;
                this.uint8('index').uint8('dT').tap(function () {
                    ru.variable('data', 'dT')(this);
                });
                console.log(stopLen, this._buffer.length, parsedBufLen+1);
                if (this._buffer.length <= parsedBufLen+1) end();
            }
        }).tap(function () {
            var res = {};
            this.vars.attrData.forEach(function (element) {
                if (element.index != undefined) { 
                    res[element.index.toString()] = element.data;
                }
            });
            console.log(res);
            this.vars.attrData = res;
        });
    });
});
```

kirovilya, 2018-07-04:
In your case, end() function never called and so console.log(res) not called too

Sonsi79, 2018-07-04:
Oh my god. You're best. Im getting now:

- battery
- battery_level
- smoke
- transmission-state
- voltage

And here is the output on screen. It looks strange. There're many "Null" results.
Or is it ok?

[log4.txt](https://github.com/Koenkk/zigbee2mqtt/files/2164313/log4.txt)

kirovilya, 2018-07-04:
:)

Sonsi79, 2018-07-04:
And here is also the new log file...
There are many infos more as before
[log.txt](https://github.com/Koenkk/zigbee2mqtt/files/2164317/log.txt)

kirovilya, 2018-07-04:
Good. Now you can remove console.log lines... I will change zcp-packet library

Sonsi79, 2018-07-04:
Ok. Thanks so much for great support!

kirovilya, 2018-07-04:
@Koenkk Look at https://github.com/Koenkk/zcl-packet/pull/2

Koenkk, 2018-07-04:
@kirovilya thanks!

Koenkk, 2018-07-05:
@Sonsi79 can you check out the zigbee2mqtt dev branch and check if this is fixed?  https://github.com/Koenkk/zigbee2mqtt/tree/dev

Sonsi79, 2018-07-05:
@Koenkk For sure, i will.
Can i also "git clone" the dev branch? Im getting error by using git clone with your link.

ciotlosm, 2018-07-05:
```
git clone git@github.com:Koenkk/zigbee2mqtt.git
cd zigbee2mqtt
git checkout dev
```

Sonsi79, 2018-07-05:
@ciotlosm Thanks for help...

@Koenkk I can confirm. The dev branch fixed the bug, the smoke detector published everything as excepted :-)

Sonsi79, 2018-07-29:
Strange, after new installation of zigbee2mqtt the smoke sensor doesn't get information of battery again.
@kirovilya Did you have any idea about the reason?

kirovilya, 2018-07-29:
Recently my changes cause of error #198  So we rolled back.
I trying to fix https://github.com/kirovilya/zcl-packet/commit/274ac22332394caaf9a8d5ee0028a987dcd248c7 
can you check this commit? You need update zcl-packet module to this commit.

Sonsi79, 2018-07-29:
Sure i will. Can you please tell me how to update to this version?
Thnx lot

kirovilya, 2018-07-29:
Go to `zigbee2mqtt `directory and run:
`npm install git+https://github.com/kirovilya/zcl-packet.git#dev`
Check that `zigbee2mqtt\node_modules\zcl-packet\lib\foundation.js` in 648 line has code:
`  stopLen = this._buffer.length-2;`
Then, go to `zigbee2mqtt\node_modules\zigbee-shepherd\node_modules` and remove or rename zcl-packet directory in this location.

Sonsi79, 2018-07-29:
@kirovilya It works great!

Sonsi79, 2018-07-29:
Should i wait for release of this fix or leave this workaround?
Will it survive a update of zigbee2mqtt?

Thnx for your quick reply and help

Koenkk, 2018-07-29:
@Sonsi79 no, merged into the dev branch.

Sonsi79, 2018-07-30:
@kirovilya and @Koenkk:
Im sorry but i think there is a new bug through this fix.
When i press the button ob my "Xiaomi Water Leak Sensor" (SJCGQ11LM) the zigbee2mqtt crash with the following error.

`/opt/zigbee2mqtt/node_modules/zcl-packet/lib/foundation.js:297
        dataTypeStr = zclId.dataType(dataType).key;
                                              ^

TypeError: Cannot read property 'key' of undefined
    at ensureDataTypeString (/opt/zigbee2mqtt/node_modules/zcl-packet/lib/foundation.js:297:47)
    at getDataType (/opt/zigbee2mqtt/node_modules/zcl-packet/lib/foundation.js:307:16)
    at Dissolve.<anonymous> (/opt/zigbee2mqtt/node_modules/zcl-packet/lib/foundation.js:693:28)
    at Dissolve._transform (/opt/zigbee2mqtt/node_modules/dissolve/index.js:80:16)
    at Dissolve.Transform._read (/opt/zigbee2mqtt/node_modules/dissolve/node_modules/readable-stream/lib/_stream_transform.js:184:10)
    at Dissolve.Transform._write (/opt/zigbee2mqtt/node_modules/dissolve/node_modules/readable-stream/lib/_stream_transform.js:172:12)
    at doWrite (/opt/zigbee2mqtt/node_modules/dissolve/node_modules/readable-stream/lib/_stream_writable.js:279:12)
    at writeOrBuffer (/opt/zigbee2mqtt/node_modules/dissolve/node_modules/readable-stream/lib/_stream_writable.js:266:5)
    at Dissolve.Writable.write (/opt/zigbee2mqtt/node_modules/dissolve/node_modules/readable-stream/lib/_stream_writable.js:211:11)
    at Dissolve.Writable.end (/opt/zigbee2mqtt/node_modules/dissolve/node_modules/readable-stream/lib/_stream_writable.js:426:10)
npm ERR! code ELIFECYCLE
npm ERR! errno 1
npm ERR! zigbee2mqtt@0.1.1 start: `node index.js`
npm ERR! Exit status 1
npm ERR! 
npm ERR! Failed at the zigbee2mqtt@0.1.1 start script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.

npm ERR! A complete log of this run can be found in:
npm ERR!     /home/csonsalla/.npm/_logs/2018-07-30T18_50_30_618Z-debug.log`

Sonsi79, 2018-07-30:
When i simulate a water leak, the sensor works and i get the publish via mqtt. Just pressing the button let the zigbee2mqtt crash. Im not sure about the reason by this fix. Maybe it is a new/other bug?.

kirovilya, 2018-07-31:
@Sonsi79 :( It is on dev version? I will check the leak sensor in the evening...

Sonsi79, 2018-07-31:
Yes ist is on the dev Version by the way you told me.

kirovilya, 2018-08-01:
@Sonsi79 Can you check again `dev` branch of `https://github.com/kirovilya/zcl-packet.git` (as I decribed above)?

kirovilya, 2018-08-01:
It is necessary to check the smoke and leak sensors. And also you can check the addition of new devices.

Sonsi79, 2018-08-01:
@kirovilya check done :-) smoke sensor and the leak sensor works fine again. thnx much!

Sonsi79, 2018-08-01:
I have to check the addition of new devices later. Have to go to work now.

kirovilya, 2018-08-01:
@Sonsi79 battery for smoke sensor works? hmm, strange... :)

[8 more replies]

closed 2018-08-01 by Koenkk