Koenkk/zigbee2mqtt #151: Ikea Tradfri state

jay-jey, 2018-06-26:
Hello, 

would it be possible to get the state of the ikea tradfri bulbs? At the moment I don't know if the bulb is on or off when it is switched on, because not all of my bulbs are constantly powered.

ciotlosm, 2018-06-26:
@jay-jey you could setup on your mqtt `expire_after` to know when a bulb goes into unknown state and setup a templated sensor to mark it as off if unknown.

jay-jey, 2018-06-26:
okay I that would help when I switch the bulb off. But the light also don't give a signal when it is switch on. So it would stand at off the hole time.

ciotlosm, 2018-06-27:
@jay-jey I don't have the device so I can't help much, maybe @Koenkk has more knowledge on this

Koenkk, 2018-06-27:
@ciotlosm what do you think about pinging all devices which don't have a battery? That would also partly solve #13

ciotlosm, 2018-06-27:
I think that would work. Just to make sure we don't ping devices that have a battery but don't report it yet to avoid draining.

ryanbeaton, 2018-06-27:
@ciotlosm I though battery devices sleep and wouldn’t answer anyway

ciotlosm, 2018-06-27:
You might catch a device out of deep sleep and just force it to answer again draining battery unnecessary.

tb-killa, 2018-07-04:
The best Thing would be to check via ```devleave``` event who is actually ignored.

ppnacho, 2018-10-02:
Hi, I like your work. I have a TRADFRI LED1537R6, I've paired the bulb and I can switch on /off  the bulb. This bulb has two types of LEDs, for wharm white and for cold white. With IKEA controller it's possible switch on/off both separately, and change the brigtness individually or join. How can control all functions from HA. I have mqtt2zigbee conected to HA.
Thanks,

Koenkk, 2018-10-02:
@ppnacho this can be done by changing the color temperature via HA

ppnacho, 2018-10-04:
OK, but whay type of device have I configure in configuration.yaml? I have a light with mqtt_json platform. Please, could you tell me about the configuration.yaml? I'm learning about HA and different devices, Wemos, Xiaomi, etc..
Thanks

Koenkk, 2018-10-05:
See https://github.com/Koenkk/zigbee2mqtt/wiki/Integrating-with-Home-Assistant#led1537r6

closed 2018-12-22 by stale[bot]