Koenkk/zigbee2mqtt #153: Hive Window Door Sensors

HACee, 2018-06-27:
Hiya there, Wondering if you can help out at all.  I picked up a USB sniffer, and have it all flashed, have paired some Xaiomi Door/Window sensors fine, and they all work as expected, and also paired a Xaimoi Temperature and Humidity sensor, also working fine, but when I try to add the Hive Door/Window sensor's, this is when I am running into problems.  They aren't a supported device, which comes up in the logs,

```
2018-06-25T23:24:18.378Z - info: MQTT publish, topic: 'zigbee2mqtt/bridge/log', payload: '{"type":"pairing","message":"device incoming"}'
2018-06-25T23:24:18.381Z - info: New device with address 0x000d6f0005a56a8a connected!
2018-06-25T23:24:18.407Z - info: MQTT publish, topic: 'zigbee2mqtt/bridge/log', payload: '{"type":"device_connected","message":"0x000d6f0005a56a8a"}'
2018-06-25T23:24:18.410Z - warn: Device with modelID 'WDS00140002' is not supported.
2018-06-25T23:24:18.412Z - warn: Please see: https://github.com/Koenkk/zigbee2mqtt/wiki/How-to-support-new-devices
```
so I add them as a device with the following code,

```
    { 
        zigbeeModel: ['WDS00140002'], // The model ID from: Device with modelID 'lumi.sens' is not supported.
        model: 'WDS002', // Vendor model number, look on the device for a model number
        vendor: 'Hive', // Vendor of the device (only used for wiki and startup logging)
        description: 'Window and Door sensor ', // Description of the device, copy from vendor site. (only used for wiki and startup logging)
        supports: 'contact', // Actions this device supports (only used for wiki)
        fromZigbee: [], // We will add this later
        toZigbee: [], // Should be empty, unless device can be controlled (e.g. lights, switches).
    },
```

After this is where I am getting problems, its shows as connected in the server

```
2018-06-25T23:38:37.704Z - info: Currently 3 devices are joined:
2018-06-25T23:38:37.708Z - info: 0x00158d00022b764e (0x00158d00022b764e): MCCGQ01LM - Xiaomi MiJia door & window contact sensor (EndDevice)
2018-06-25T23:38:37.711Z - info: 0x00158d00027b3daa (0x00158d00027b3daa): MCCGQ01LM - Xiaomi MiJia door & window contact sensor (EndDevice)
2018-06-25T23:38:37.714Z - info: 0x000d6f0005a56a8a (0x000d6f0005a56a8a): WDS002 - Hive Window and Door sensor  (EndDevice)

```
but when I open and close the sensor to trigger and event, I am not getting anything showing up. 

Unfortunately my linux/programming skills aren't great, but I can follow anything you need me to do to try and help debug it at all.

Many thanks in advance.

Koenkk, 2018-06-27:
Probably similar to #143, can you post your `data/database.db`?

HACee, 2018-06-27:
Seems to be a lot of entries in here,

```
{"id":1,"type":"Coordinator","ieeeAddr":"0x00124b0014bdeb87","nwkAddr":0,"manufId":0,"epList":[1,2,3,4,5,6],"status":"online","joinTime":1530060918,"endpoints":{"1":{"profId":260,"epId":1,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{}},"2":{"profId":257,"epId":2,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{}},"3":{"profId":261,"epId":3,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{}},"4":{"profId":263,"epId":4,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{}},"5":{"profId":264,"epId":5,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{}},"6":{"profId":265,"epId":6,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{}}},"_id":"KDkt9C53HJb3q6Sz"}
{"id":5,"type":"EndDevice","ieeeAddr":"0x00158d00020ee3ca","nwkAddr":36585,"manufId":4151,"manufName":"LUMI","powerSource":"Battery","modelId":"lumi.sens","epList":[1,2,3],"status":"offline","joinTime":null,"endpoints":{"1":{"profId":260,"epId":1,"devId":24321,"inClusterList":[0,3,18,25,65535],"outClusterList":[0,3,4,5,18,25,65535],"clusters":{"genBasic":{"dir":{"value":3},"attrs":{}},"genIdentify":{"dir":{"value":3},"attrs":{}},"genGroups":{"dir":{"value":2},"attrs":{}},"genScenes":{"dir":{"value":2},"attrs":{}},"genMultistateInput":{"dir":{"value":3},"attrs":{}},"genOta":{"dir":{"value":3},"attrs":{}},"manuSpecificCluster":{"dir":{"value":3},"attrs":{}}}},"2":{"profId":260,"epId":2,"devId":24322,"inClusterList":[3,18],"outClusterList":[3,4,5,18],"clusters":{"genIdentify":{"dir":{"value":3},"attrs":{}},"genGroups":{"dir":{"value":2},"attrs":{}},"genScenes":{"dir":{"value":2},"attrs":{}},"genMultistateInput":{"dir":{"value":3},"attrs":{}}}},"3":{"profId":260,"epId":3,"devId":24323,"inClusterList":[3,12],"outClusterList":[3,4,5,12],"clusters":{"genIdentify":{"dir":{"value":3},"attrs":{}},"genGroups":{"dir":{"value":2},"attrs":{}},"genScenes":{"dir":{"value":2},"attrs":{}},"genAnalogInput":{"dir":{"value":3},"attrs":{}}}}},"_id":"StFmfGeiLycP8MnD"}
{"id":2,"type":"EndDevice","ieeeAddr":"0x00158d00022b764e","nwkAddr":52326,"manufId":4151,"manufName":"LUMI","powerSource":"Battery","modelId":"lumi.sensor_magnet","epList":[1],"status":"offline","joinTime":null,"endpoints":{"1":{"profId":260,"epId":1,"devId":260,"inClusterList":[0,3,25,65535],"outClusterList":[0,3,4,5,6,8,25],"clusters":{"genBasic":{"dir":{"value":3},"attrs":{}},"genIdentify":{"dir":{"value":3},"attrs":{}},"genGroups":{"dir":{"value":2},"attrs":{}},"genScenes":{"dir":{"value":2},"attrs":{}},"genOnOff":{"dir":{"value":2},"attrs":{}},"genLevelCtrl":{"dir":{"value":2},"attrs":{}},"genOta":{"dir":{"value":3},"attrs":{}},"manuSpecificCluster":{"dir":{"value":1},"attrs":{}}}}},"_id":"V0ZjeFH0IKkREtk0"}
{"id":4,"type":"EndDevice","ieeeAddr":"0x000d6f0005a56a8a","nwkAddr":30122,"manufId":4153,"manufName":"AlertMe.com","powerSource":"Battery","modelId":"WDS00140002","epList":[2,240],"status":"offline","joinTime":null,"endpoints":{"2":{"profId":260,"epId":2,"devId":1026,"inClusterList":[0,1,3,32,1026,1280],"outClusterList":[],"clusters":{}},"240":{"profId":49686,"epId":240,"devId":5,"inClusterList":[],"outClusterList":[64512],"clusters":{}}},"_id":"YupWKOaVYxX6lUe8"}
{"id":3,"type":"EndDevice","ieeeAddr":"0x00158d00027b3daa","nwkAddr":47981,"manufId":4151,"manufName":"LUMI","powerSource":"Battery","modelId":"lumi.sensor_magnet","epList":[1],"status":"offline","joinTime":null,"endpoints":{"1":{"profId":260,"epId":1,"devId":260,"inClusterList":[0,3,25,65535],"outClusterList":[0,3,4,5,6,8,25],"clusters":{"genBasic":{"dir":{"value":3},"attrs":{}},"genIdentify":{"dir":{"value":3},"attrs":{}},"genGroups":{"dir":{"value":2},"attrs":{}},"genScenes":{"dir":{"value":2},"attrs":{}},"genOnOff":{"dir":{"value":2},"attrs":{}},"genLevelCtrl":{"dir":{"value":2},"attrs":{}},"genOta":{"dir":{"value":3},"attrs":{}},"manuSpecificCluster":{"dir":{"value":1},"attrs":{}}}}},"_id":"bw7Ty9qU3K8wdiby"}
{"$$indexCreated":{"fieldName":"id","unique":true,"sparse":false}}
{"id":1,"type":"Coordinator","ieeeAddr":"0x00124b0014bdeb87","nwkAddr":0,"manufId":0,"epList":[1,2,3,4,5,6],"status":"online","joinTime":1530062977,"endpoints":{"1":{"profId":260,"epId":1,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{}},"2":{"profId":257,"epId":2,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{}},"3":{"profId":261,"epId":3,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{}},"4":{"profId":263,"epId":4,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{}},"5":{"profId":264,"epId":5,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{}},"6":{"profId":265,"epId":6,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{}}},"_id":"KDkt9C53HJb3q6Sz"}
{"id":5,"type":"EndDevice","ieeeAddr":"0x00158d00020ee3ca","nwkAddr":36585,"manufId":4151,"manufName":"LUMI","powerSource":"Battery","modelId":"lumi.sens","epList":[1,2,3],"status":"offline","joinTime":null,"endpoints":{"1":{"profId":260,"epId":1,"devId":24321,"inClusterList":[0,3,18,25,65535],"outClusterList":[0,3,4,5,18,25,65535],"clusters":{"genBasic":{"dir":{"value":3},"attrs":{}},"genIdentify":{"dir":{"value":3},"attrs":{}},"genGroups":{"dir":{"value":2},"attrs":{}},"genScenes":{"dir":{"value":2},"attrs":{}},"genMultistateInput":{"dir":{"value":3},"attrs":{}},"genOta":{"dir":{"value":3},"attrs":{}},"manuSpecificCluster":{"dir":{"value":3},"attrs":{}}}},"2":{"profId":260,"epId":2,"devId":24322,"inClusterList":[3,18],"outClu [... cut ...] :{"profId":260,"epId":1,"devId":260,"inClusterList":[0,3,25,65535],"outClusterList":[0,3,4,5,6,8,25],"clusters":{"genBasic":{"dir":{"value":3},"attrs":{}},"genIdentify":{"dir":{"value":3},"attrs":{}},"genGroups":{"dir":{"value":2},"attrs":{}},"genScenes":{"dir":{"value":2},"attrs":{}},"genOnOff":{"dir":{"value":2},"attrs":{}},"genLevelCtrl":{"dir":{"value":2},"attrs":{}},"genOta":{"dir":{"value":3},"attrs":{}},"manuSpecificCluster":{"dir":{"value":1},"attrs":{}}}}},"_id":"bw7Ty9qU3K8wdiby"}

```

HACee, 2018-06-27:
I have a spare hive door sensor, so happy to post it out if you want something to play with without having to spend out, can just post it back to me when done.

ciotlosm, 2018-06-28:
>I have a spare hive door sensor, so happy to post it out if you want something to play with without having to spend out, can just post it back to me when done.

@HACee you can post info anyway. Maybe we can add it with what you post

<sub>Sent with <a href="http://githawk.com">GitHawk</a></sub>

matchett808, 2018-06-28:
```
{"id":4,"type":"EndDevice","ieeeAddr":"0x000d6f0005a56a8a","nwkAddr":30122,"manufId":4153,"manufName":"AlertMe.com","powerSource":"Battery","modelId":"WDS00140002","epList":[2,240],"status":"offline","joinTime":null,"endpoints":{"2":{"profId":260,"epId":2,"devId":1026,"inClusterList":[0,1,3,32,1026,1280],"outClusterList":[],"clusters":{}},"240":{"profId":49686,"epId":240,"devId":5,"inClusterList":[],"outClusterList":[64512],"clusters":{}}},"_id":"YupWKOaVYxX6lUe8"}
```
from the DB dump above - also see the linked post for the Hive motion sensor - looks like they are both Ias devices (as is the Samsung Smartthings motion sensor, may also be why the Hue motion sensor isn't producing messages)

I've got as far as getting the motion sensor to report msOccupancy every x seconds - but once tripped it stays tripped, I reckon it's something to do IAS, this might require shepherd support?

HACee, 2018-06-30:
As I mentioned before, most of what you typed went pretty much over my head, not a coder/programmer at all unfortunately, so if there is anything you need me to do, just give me a shout, and I'll do my best.

closed 2018-09-05 by stale[bot]

afonsoguerra, 2022-08-29:
Has this been solved somehow in any other way? I would also like to get one of these working.