Koenkk/zigbee2mqtt #156: WARN Error while starting zigbee-shepherd, attemping to fix... (takes 60 seconds)

mpuff, 2018-06-27:
Hello,

i get this error when i reboot the pi (zero), i have installed node v8.11.3.
After the install it is working, but when i do a reboot, zigbee2mqtt doesnt start correctly.
I had running zigbee2mqtt for 2 ore 3 days without problems, than it stops working!
I have installed zigbee2mqtt several times, this error comes every time.
Does someone has the same problem?

Koenkk, 2018-06-27:
Can you provide the log?

matchett808, 2018-06-28:
Rather than creating a new issue - I am hitting this as well (the only fix I've found is to reflash the stick - this allows shepherd to boot, but the network is then flakey)

```

> zigbee2mqtt@0.0.0 start /app
> node index.js

2018-6-28 08:43:39 [34mDEBUG[39m Using zigbee-shepherd with settings: '{"net":{"panId":6754,"channelList":[11]},"dbPath":"/app/data/database.db","sp":{"baudrate":115200,"rtscts":true}}'
2018-6-28 08:43:39 [32mINFO[39m Starting zigbee2mqtt version 0.0.0 (commit #2ba2864)
2018-6-28 08:43:39 [32mINFO[39m Starting zigbee-shepherd
Thu, 28 Jun 2018 07:43:39 GMT zigbee-shepherd:init zigbee-shepherd booting...
Thu, 28 Jun 2018 07:43:44 GMT cc-znp The serialport /dev/zigbee is opened.
Thu, 28 Jun 2018 07:43:44 GMT zigbee-shepherd:request REQ --> SYS:osalNvRead
Thu, 28 Jun 2018 07:43:44 GMT cc-znp:SREQ --> SYS:osalNvRead, { id: 3840, offset: 0, len: 1, value: [ 85 ] }
Thu, 28 Jun 2018 07:43:44 GMT serialport write 8 bytes of data
Thu, 28 Jun 2018 07:43:47 GMT cc-znp:SRSP <-- SYS:osalNvRead, __timeout__
Thu, 28 Jun 2018 07:43:47 GMT zigbee-shepherd:request RSP <-- SYS:osalNvRead
Thu, 28 Jun 2018 07:43:47 GMT zigbee-shepherd:init Coordinator initialize had an error: Error: request timeout
    at CcZnp.<anonymous> (/app/node_modules/cc-znp/lib/ccznp.js:254:22)
    at Object.onceWrapper (events.js:315:30)
    at emitOne (events.js:116:13)
    at CcZnp.emit (events.js:211:7)
    at Timeout.<anonymous> (/app/node_modules/cc-znp/lib/ccznp.js:233:18)
    at ontimeout (timers.js:475:11)
    at tryOnTimeout (timers.js:310:5)
    at Timer.listOnTimeout (timers.js:270:5)
2018-6-28 08:43:47 [33mWARN[39m Error while starting zigbee-shepherd, attemping to fix... (takes 60 seconds)
Thu, 28 Jun 2018 07:43:52 GMT cc-znp The serialport /dev/zigbee is closed.
```

ciotlosm, 2018-06-28:
>Rather than creating a new issue - I am hitting this as well (the only fix I've found is to reflash the stick - this allows shepherd to boot, but the network is then flakey) ...

@matchett808 which firmware are you using?

<sub>Sent with <a href="http://githawk.com">GitHawk</a></sub>

matchett808, 2018-06-28:
I'm on the one on the 'getting started page'

Hash:
```
|||:~/cc-tool$ sha1sum ./CC2531ZNP-Prod.hex                                                                                                                               
7bc45ba4ab7489dc4fc234765eef48c07ab3e448  ./CC2531ZNP-Prod.hex
```

ciotlosm, 2018-06-28:
Try switching to the one mentioned here: https://github.com/Koenkk/zigbee2mqtt/issues/126#issuecomment-399661622

matchett808, 2018-06-28:
Will give that a spin tonight 👍 

Will this require reset/repair of devices or should it be gravy?

ciotlosm, 2018-06-28:
Not sure. You can try without repair first to see if it starts correctly.

Odianosen25, 2018-06-28:
Well I kind of had this same issue this morning, and my system has been working great for the best part of the last 2 weeks. I noticed one of my wireless buttons wasn't responding any longer and wanted to look into starting a paring to fix it, as it sometimes fixes others I have.

I stopped my docker, try to start it again and boom on my face after saying its trying to fix, it didn't work again and came up with errors. What worried me the most was that this is a system that has been working for so long and all I have done was to stop my docker and try to restart it again. re-sited the sniffer and that didn't solve it. restarted the Pi and that still didn't solve it. 

Felt there might be a new docker image, downloaded a new one, still didn't solve it as I kept having `trying to fix zigbee-sheppard...`. So I then looked into re-flashing the sniffer and I felt there hasn't been any change since my last download as ti says 9 days ago. Well downloaded still and re-flashed. When starting the system this time, for the first time I saw the LED light on it change state between Green and RED. Really that was my first time, then it started to work. Maybe its a new thing, know not.

It is all working now, as i then tried to pair the switch, which is already paired and that also solved my issue. Well this is just a report as all is working fine this morning, but might want to look into what is causing this.

Thanks for all your great work and kind regards

Koenkk, 2018-06-28:
Pressing the reset button on the stick (the one closest to the USB) solves this problem. 

This is now being logged when this error appears.

@mpuff does this solve your problem?

matchett808, 2018-06-28:
(That button is labeled DEBUG on mine - doesn't seem to help for me) - what helped was:
- Stop Docker Container
- Unplugging
- Insert stick attached to CC Debugger
- sudo chown the /dev/zigbee (root:root -> root/dialout - might have been part of my issue)

mpuff, 2018-06-28:
@Koenkk, yes, this solved the problem for me :)

mpuff, 2018-06-28:
The zigbee-shepherd startet successful after a reboot of the pi

closed 2018-06-28 by mpuff