Koenkk/zigbee2mqtt #157: Handle devLeaving event

kirovilya, 2018-06-28:
When device already in network and we press setup key for 5 sec on Xiaomi sensor - we get 'devLeaving' event.
We skip this event now... but vain :)
We need inform about it and remove all around this device. Cause we will no longer receive any messages from it.

It may be cause of https://github.com/Koenkk/zigbee2mqtt/issues/155

Koenkk, 2018-06-28:
Yes, I don't know if it's already removed from `database.db`, but it should be removed from `configuration.yaml` for sure.

tb-killa, 2018-07-07:
I removed the ikea remote control via 4 pushes at the backside. The device themselve removed from the database via delete entry which means we have to restart the instance again to get the file clean.
But from the configuration file it doesnt remove automaticly.
Maybe we should check for an specific Event from zigbee shepherd themselve and fire the remove function.?

Koenkk, 2018-07-07:
@tb-killa yes, when this event is triggered it should be removed from `configuration.yaml`.

tb-killa, 2018-08-27:
Hi @Koenkk 

"an itinerary of thoughts"

What about implement in zigbee shepherd ``lib/components/controller.js``

some sort of (prototype)
```
this.on("ZDO:leaveInd", function(msg){
var nwkAddr = msg.srcaddr
}
``` 
function who only return or bring trigger ?

On this we could build inside zigbee2mqtt our sort of logic to verify and work with it... ??

Koenkk, 2018-08-30:
Isn't this message already passed in the `handleZigbeeMessage` (`controller.js`)?

tb-killa, 2018-09-02:
@Koenkk 
Here one snippet with ``DEBUG=zigbee-shepherd* npm start`` and my ikea remote device (button 4 pushed to leave) ...

```
zigbee-shepherd:request REQ --> ZDO:nodeDescReq +5s
  zigbee-shepherd Device: 0x90fd9ffffed17c15 leave the network. +490ms
  zigbee2mqtt:debug 2018-9-2 21:01:03 Recieved zigbee message with data "0x90fd9ffffed17c15"
  zigbee2mqtt:warn 2018-9-2 21:01:03 Message without device!
  zigbee-shepherd:msgHdlr IND <-- ZDO:leaveInd +10ms
```
You see .. zigbee2mqtt only get ``Message without device!`` ... no informations like zigbee-shepherd debug content .. 

Maybe my snippet of prototype code is faulty .. we have to show that this message of what device is leaving passed your zigbee2mqtt controller.js (handleZigbeeMessage) too ..
Only with this informations we could work on building own logic for remove device ...
zigbee shepherd do this with entry on database.db (devbox) .. after reload / restart its removed ..
but we doesn´t work with this informations to cleanup our configuration.yaml file content ..
or maybe some other stuff we would like to do ;)

lolorc, 2018-09-03:
why should it be removed from configuration.yaml ? could it be disable instead ? friendly_name is helpful to keep and sooner or later some devices could have some extra params.

Last time I had to restart my network from scratch, I left it intact, was that a wrong idea ? was that the reason why it was so difficult/long to repair my devices ?

Koenkk, 2018-09-05:
After some thinking, I agree with @lolorc that we should not remove it from the `configuration.yaml`. 

As the device will already be removed from the zigbee-shepherd database, the only way we should handle this message is by logging it to console/MQTT.

tb-killa, 2018-09-06:
This could have a strange effect.
If a device has been removed from the zigbee shepherd database or removes itself, you can't communicate with that device anyway.
This also means that we no longer have to carry the device with us as a corpse.
Of course, the charming side effect is that if the end device reconnects, the data from the friendly_name can be transferred directly.
However, here I clearly see the duty of the user of zigbee2mqtt to possibly update the data.
If it should be desired that the data remain due to the friendly_name, I do not see any advantage in NOT deleting the data, because I can restore the data from my log file in an emergency!

You should always work with so little needed data, that means to remove them in case you don't need them.
The case of the cases can always occur, but how far do we want to continue this ...
A bottomless pit ....

Translated with www.DeepL.com/Translator

tb-killa, 2018-09-06:
This could have a strange effect.
If a device has been removed from the zigbee shepherd database or removes itself, you can't communicate with that device anyway.
This also means that we no longer have to carry the device with us as a corpse.
Of course, the charming side effect is that if the end device reconnects, the data from the friendly_name can be transferred directly.
However, here I clearly see the duty of the user of zigbee2mqtt to possibly update the data.
If it should be desired that the data remain due to the friendly_name, I do not see any advantage in NOT deleting the data, because I can restore the data from my log file in an emergency!

You should always work with so little needed data, that means to remove them in case you don't need them.
The case of the cases can always occur, but how far do we want to continue this ...
A bottomless pit ....

lolorc, 2018-09-06:
I'd add a disabled/gone/unpaired attribute in configuration.yaml.
And I'd say the zigbee sheperd database needs fixing first, it's full of duplicates, I'm not even sure it's correctly removing all entries when a device leaves.

tb-killa, 2018-09-06:
the "zigbee shepherd database multiple entries" thematic is often mentioned ..
one sample: https://github.com/Koenkk/zigbee2mqtt/issues/234
we use zigbee shepherd as our base .. if this is like it works .. we have to use it ...
if somebody know how to fix ... it would be great .. but this is another story of this opened "issue"!

closed 2018-12-22 by stale[bot]