Koenkk/zigbee2mqtt #159: Switch 4x EU-LIGHTIFY

arteck, 2018-06-28:
its switch from Osram with 4 Buttons

Koenkk, 2018-06-28:
Can you follow: https://github.com/Koenkk/zigbee2mqtt/wiki/How-to-support-new-devices?

arteck, 2018-06-30:
{
        zigbeeModel: ['Switch 4x EU-LIGHTIFY'],
        model: 'xxx',
        vendor: 'OSRAM',
        description: 'LIGHTIFY Switch 4x',
        supports: xxxxxxxxxxxx
        fromZigbee: xxxxxxxxxxxxxxxxx
        toZigbee: xxxxxxxxxxxxxxxxxxxxxxxxxxxxx
    },

Koenkk, 2018-06-30:
Can you check if with the following [`devices.js`](https://github.com/Koenkk/zigbee-shepherd-converters/blob/lightify_switch/devices.js) you see any `WARN: no converter available` messages when pressing the buttons?

arteck, 2018-07-01:
its very strange .. i can on/off all lamps with this switch without CC2531pairing.. 
it works buttons only on left side. i dont know why ... any idea ??

Koenkk, 2018-07-01:
Probably this remote can also be paired directly to bulbs.

arteck, 2018-07-01:
hmmm.. I do not think so.. i have here a gateway for this device..

Koenkk, 2018-07-02:
Lets find that out later, but does it work with this devices.js: https://github.com/Koenkk/zigbee-shepherd-converters/blob/lightify_switch/devices.js ?

arteck, 2018-07-04:
I'm waiting for new switch..

arteck, 2018-07-06:
so.. here output
```
zigbee-shepherd Identified Device: { manufacturer: OSRAM, product: Switch 4x EU-LIGHTIFY } +9ms
  zigbee-shepherd:request REQ --> AF:dataRequest, transId: 2 +37ms
  zigbee-shepherd:request RSP <-- AF:dataRequest, status: 0 +16ms
  zigbee-shepherd:msgHdlr IND <-- AF:dataConfirm, transId: 2 +2s
  zigbee-shepherd:msgHdlr IND <-- AF:incomingMsg, transId: 0 +39ms
  zigbee-shepherd:request REQ --> AF:dataRequest, transId: 3 +9ms
  zigbee-shepherd:request RSP <-- AF:dataRequest, status: 0 +18ms
  zigbee-shepherd:msgHdlr IND <-- AF:dataConfirm, transId: 3 +2s
  zigbee-shepherd:msgHdlr IND <-- AF:incomingMsg, transId: 0 +39ms
  zigbee-shepherd:request REQ --> AF:dataRequest, transId: 4 +12ms
  zigbee-shepherd:request RSP <-- AF:dataRequest, status: 0 +20ms
  zigbee-shepherd:msgHdlr IND <-- AF:dataConfirm, transId: 4 +3s
  zigbee-shepherd:msgHdlr IND <-- AF:incomingMsg, transId: 0 +59ms
  zigbee-shepherd:request REQ --> AF:dataRequest, transId: 5 +8ms
  zigbee-shepherd:request RSP <-- AF:dataRequest, status: 0 +18ms
/opt/zigbee2mqtt/node_modules/zcl-packet/lib/functional.js:33
        throw new Error('Unrecognized command');
        ^

Error: Unrecognized command
    at new FuncPayload (/opt/zigbee2mqtt/node_modules/zcl-packet/lib/functional.js:33:15)
    at /opt/zigbee2mqtt/node_modules/zcl-packet/lib/zcl.js:33:22
    at Dissolve.<anonymous> (/opt/zigbee2mqtt/node_modules/zcl-packet/lib/zcl.js:121:9)
    at Object.onceWrapper (events.js:315:30)
    at emitOne (events.js:116:13)
    at Dissolve.emit (events.js:211:7)
    at Dissolve.<anonymous> (/opt/zigbee2mqtt/node_modules/dissolve-chunks/index.js:73:29)
    at emitNone (events.js:106:13)
    at Dissolve.emit (events.js:208:7)
    at emitReadable_ (/opt/zigbee2mqtt/node_modules/dissolve/node_modules/readable-stream/lib/_stream_readable.js:456:10)
npm ERR! code ELIFECYCLE
npm ERR! errno 1
npm ERR! zigbee2mqtt@0.0.0 start: `node index.js`
npm ERR! Exit status 1
npm ERR!
npm ERR! Failed at the zigbee2mqtt@0.0.0 start script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.

npm ERR! A complete log of this run can be found in:

```

Koenkk, 2018-07-07:
Can you put:

```js
console.log(this.cluster, direction, cmd);
```

before https://github.com/Koenkk/zcl-packet/blob/master/lib/functional.js#L30

arteck, 2018-07-08:
```
[zigbee-shepherd:request REQ --> ZDO:nodeDescReq +2ms
lightingColorCtrl 0 76
/opt/zigbee2mqtt/node_modules/zcl-packet/lib/functional.js:35
        throw new Error('Unrecognized command');
        ^

```

Koenkk, 2018-07-09:
Can you add to https://github.com/Koenkk/zcl-id/blob/master/definitions/cluster_defs.json#L557 `"lightifyTest": 76`

Koenkk, 2018-07-10:
Can you checkout the [dev branch](https://github.com/Koenkk/zigbee2mqtt/tree/dev) and add `"lightifyTest": 76`. Do any `WARN: No converted available` messages appear?

arteck, 2018-07-10:
```
2018-7-10 20:11:53 INFO Starting zigbee2mqtt version 0.1.0 (commit #ae41b7b)
2018-7-10 20:11:53 INFO Starting zigbee-shepherd
2018-7-10 20:11:57 INFO zigbee-shepherd started
2018-7-10 20:11:57 INFO Currently 1 devices are joined:
2018-7-10 20:11:57 INFO unknown (0x000d6f000ec8fc3e): unkown - undefined unknown (EndDevice)
2018-7-10 20:11:57 WARN `permit_join` set to  `true` in configuration.yaml.
2018-7-10 20:11:57 WARN Allowing new devices to join.
2018-7-10 20:11:57 WARN Set `permit_join` to `false` once you joined all devices.
2018-7-10 20:11:57 INFO Zigbee: allowing new devices to join.
2018-7-10 20:11:57 INFO Connecting to MQTT server at mqtt://192.168.1.184
2018-7-10 20:11:58 INFO zigbee-shepherd ready
2018-7-10 20:11:58 INFO Connected to MQTT server
2018-7-10 20:11:58 INFO MQTT publish, topic: 'zigbee2mqtt/bridge/state', payload: 'online'
lightingColorCtrl 0 76
/opt/zigbee2mqtt/node_modules/zcl-packet/lib/functional.js:42
        throw new Error('Unrecognized direction');
        ^

Error: Unrecognized direction
    at new FuncPayload (/opt/zigbee2mqtt/node_modules/zcl-packet/lib/functional.js:42:15)
    at /opt/zigbee2mqtt/node_modules/zcl-packet/lib/zcl.js:33:22
    at Dissolve.<anonymous> (/opt/zigbee2mqtt/node_modules/zcl-packet/lib/zcl.js:121:9)
    at Object.onceWrapper (events.js:315:30)
    at emitOne (events.js:116:13)
    at Dissolve.emit (events.js:211:7)
    at Dissolve.<anonymous> (/opt/zigbee2mqtt/node_modules/dissolve-chunks/index.js:73:29)
    at emitNone (events.js:106:13)
    at Dissolve.emit (events.js:208:7)
    at emitReadable_ (/opt/zigbee2mqtt/node_modules/dissolve/node_modules/readable-stream/lib/_stream_readable.js:456:10)
npm ERR! code ELIFECYCLE
npm ERR! errno 1

```

Koenkk, 2018-07-12:
Can you add in: `node_modules/zcl-packet/lib/defs/zcl_meta.json` below line 1200:

```js
            "lightifyTest":{
                "params":[
                ],
                "dir":0},
```

arteck, 2018-07-13:
```
2018-7-13 17:00:28 INFO Starting zigbee2mqtt version 0.1.0 (commit #ae41b7b)
2018-7-13 17:00:28 INFO Starting zigbee-shepherd
2018-7-13 17:00:32 INFO zigbee-shepherd started
2018-7-13 17:00:32 INFO Currently 1 devices are joined:
2018-7-13 17:00:32 INFO unknown (0x000d6f000ec8fc3e): unkown - undefined unknown (EndDevice)
2018-7-13 17:00:32 WARN `permit_join` set to  `true` in configuration.yaml.
2018-7-13 17:00:32 WARN Allowing new devices to join.
2018-7-13 17:00:32 WARN Set `permit_join` to `false` once you joined all devices.
2018-7-13 17:00:32 INFO Zigbee: allowing new devices to join.
2018-7-13 17:00:32 INFO Connecting to MQTT server at mqtt://192.168.1.184
2018-7-13 17:00:33 INFO zigbee-shepherd ready
2018-7-13 17:00:33 INFO Connected to MQTT server
2018-7-13 17:00:33 INFO MQTT publish, topic: 'zigbee2mqtt/bridge/state', payload: 'online'
lightingColorCtrl 0 76
lightingColorCtrl 0 76
lightingColorCtrl 0 76
lightingColorCtrl 0 76
lightingColorCtrl 0 76
genOnOff 0 0
genOnOff 0 1
2018-7-13 17:00:47 INFO New device with address 0x000d6f000ec8fc3e connected!
2018-7-13 17:00:48 INFO MQTT publish, topic: 'zigbee2mqtt/bridge/log', payload: '{"type":"device_connected","message":"0x000d6f000ec8fc3e"}'
2018-7-13 17:00:48 WARN Device with modelID 'Switch 4x EU-LIGHTIFY' is not supported.
2018-7-13 17:00:48 WARN Please see: https://github.com/Koenkk/zigbee2mqtt/wiki/How-to-support-new-devices
genOnOff 0 1
2018-7-13 17:00:50 WARN Device with modelID 'Switch 4x EU-LIGHTIFY' is not supported.
2018-7-13 17:00:50 WARN Please see: https://github.com/Koenkk/zigbee2mqtt/wiki/How-to-support-new-devices
```

arteck, 2018-07-16:
little problem after press button top left

```
  { func: [Function: func], ieeeAddr: '0x0017880102ca68f3' } ]
genOnOff 0 1
/opt/zigbee2mqtt/lib/controller.js:286
                    `type '${message.type}' and message '${JSON.stringify(message)}'`
                                                                ^

TypeError: Converting circular structure to JSON
    at JSON.stringify (<anonymous>)
    at Controller.handleZigbeeMessage (/opt/zigbee2mqtt/lib/controller.js:286:65)
    at Zigbee.handleMessage (/opt/zigbee2mqtt/lib/zigbee.js:162:18)
    at emitOne (events.js:116:13)
    at ZShepherd.emit (events.js:211:7)
    at ZShepherd.<anonymous> (/opt/zigbee2mqtt/node_modules/zigbee-shepherd/lib/shepherd.js:123:14)
    at emitMany (events.js:147:13)
    at ZShepherd.emit (events.js:224:7)
    at dispatchIncomingMsg (/opt/zigbee2mqtt/node_modules/zigbee-shepherd/lib/components/af.js:673:45)
    at Controller.zclIncomingMsgHandler (/opt/zigbee2mqtt/node_modules/zigbee-shepherd/lib/components/af.js:730:12)
npm ERR! code ELIFECYCLE
npm ERR! errno 1
npm ERR! zigbee2mqtt@0.1.0 start: `node index.js`
npm ERR! Exit status 1

```

tb-killa, 2018-07-17:
@arteck https://github.com/Koenkk/zigbee2mqtt/issues/199

Koenkk, 2018-07-18:
@arteck issue has been fixed, can you update and try again?

arteck, 2018-07-24:
```
genOnOff 0 0
genOnOff 0 0
genOnOff 0 1
2018-7-24 12:13:22 WARN No converter available for 'TODO' with cid 'genOnOff', type 'cmdOn' and data '{"cid":"genOnOff","data":{}}'
2018-7-24 12:13:22 WARN Please see: https://github.com/Koenkk/zigbee2mqtt/wiki/How-to-support-new-devices.
lightingColorCtrl 0 76
lightingColorCtrl 0 76
```

Koenkk, 2018-07-25:
@arteck nice, can you provide output for all buttons?

arteck, 2018-07-25:
this is the output from all buttons pressed
as same  here

> here button right top -> lightingColorCtrl 0 76
> here button right botom -> lightingColorCtrl 0 76
> here button left top -> genOnOff 0 1 with the warning message
> here button left botom -> genOnOff 0 0

Koenkk, 2018-07-25:
I see, for those you don't get a `No converter available`?

tb-killa, 2018-07-25:
If i didn´t missread we bind ``genLevelCtrl`` but this doesn´t return inside the debug log .. right ?
What about remove ``lightingColorCtrl`` and let only ``genLevelCtrl`` work ?
Because i doesnt see any Changed Value of ``lightingColorCtrl ``
```
here button right top -> lightingColorCtrl 0 76
here button right botom -> lightingColorCtrl 0 76
```
Unfortunately, I have none of these devices, so I can not test myself, but I would like it :)

tb-killa, 2018-07-25:
@Koenkk  Btw: Read this: https://community.smartthings.com/t/osram-lightify-4-button-switch-dth-binder-alpha/99877/56 ... it seems to be that there are two Versions of this Remote Control available!
@arteck: Where did you buy your device?

arteck, 2018-07-25:
i have here "Switch 4x EU-LIGHTIFY"= European variant bought at amazon.de :-)

tb-killa, 2018-07-25:
Nice so every third message is stopped from push or hold on ?

Please edit your devices.js entry and add this
```
fromZigbee: [fz.generic_state],
```
and test ``button left top click`` and ``button left buttom click`` ... it should now message ``state: ON or OFF``

for ``genLevelCtrl `` we could adopt some stuff from https://github.com/Koenkk/zigbee2mqtt/issues/36
.. @Koenkk could help :)

Doesn´t you got some ``WARN`` Messages for the ``lightingColorCtrl``  ??

tb-killa, 2018-07-25:
Nice so every third message is stopped from push or hold on ?
for ``genLevelCtrl `` we could adopt some stuff from https://github.com/Koenkk/zigbee2mqtt/issues/36
.. @Koenkk could help :)

Doesn´t you got some ``WARN`` Messages for the ``lightingColorCtrl``  ??

We need now for every button function the ``WARN`` Messages ..
For ``genOnOff `` we should ``cmdOn`` and ``cmdOffWithEffect`` ??

arteck, 2018-07-27:
Doesn´t you got some WARN Messages for the lightingColorCtrl ?? 
no nothing

We need now for every button function the WARN Messages ..
For genOnOff we should cmdOn and cmdOffWithEffect ??

i have no more other messages

tb-killa, 2018-07-28:
My ``Switch 4x EU-LIGHTIFY`` arrived today so i could help easier.
If i got some free time i will digging into it deeper to show what we need and will work with @Koenkk to finish the supporting =)

tb-killa, 2018-07-28:
I´v join my Switch to check if functional.
This is my database.db entry
``
{"id":4,"type":"EndDevice","ieeeAddr":"0x000d6f000ec8fb45","nwkAddr":6183,"manufId":4364,"manufName":"OSRAM","powerSource":"Battery","modelId":"Switch 4x EU-LIGHTIFY","epList":[1,2,3,4,5,6],"status":"online","joinTime":1532806133,"endpoints":{"1":{"profId":260,"epId":1,"devId":2064,"inClusterList":[0,1,32,4096,64768],"outClusterList":[3,4,5,6,8,25,768,4096],"clusters":{"64768":{"dir":{"value":1},"attrs":{}},"genBasic":{"dir":{"value":1},"attrs":{"zclVersion":1,"appVersion":1,"stackVersion":0,"hwVersion":1,"manufacturerName":"OSRAM","modelId":"Switch 4x EU-LIGHTIFY","dateCode":"201707051445390","powerSource":3,"swBuildId":"e.1.0.0M"}},"genPowerCfg":{"dir":{"value":1},"attrs":{"batteryVoltage":31,"batteryVoltMinThres":24,"batteryVoltThres1":26,"batteryVoltThres2":27,"batteryVoltThres3":28,"batteryAlarmState":0}},"genIdentify":{"dir":{"value":2},"attrs":{}},"genGroups":{"dir":{"value":2},"attrs":{}},"genScenes":{"dir":{"value":2},"attrs":{}},"genOnOff":{"dir":{"value":2},"attrs":{}},"genLevelCtrl":{"dir":{"value":2},"attrs":{}},"genOta":{"dir":{"value":2},"attrs":{}},"genPollCtrl":{"dir":{"value":1},"attrs":{"checkinInterval":0,"longPollInterval":48000,"shortPollInterval":4,"fastPollTimeout":720}},"lightingColorCtrl":{"dir":{"value":2},"attrs":{}},"lightLink":{"dir":{"value":3},"attrs":{}}}},"2":{"profId":260,"epId":2,"devId":2064,"inClusterList":[0,4096,64768],"outClusterList":[3,4,5,6,8,768,4096],"clusters":{"64768":{"dir":{"value":1},"attrs":{}},"genBasic":{"dir":{"value":1},"attrs":{"zclVersion":1,"appVersion":1,"stackVersion":0,"hwVersion":1,"manufacturerName":"OSRAM","modelId":"Switch 4x EU-LIGHTIFY","dateCode":"201707051445390","powerSource":3,"swBuildId":"e.1.0.0M"}},"genIdentify":{"dir":{"value":2},"attrs":{}},"genGroups":{"dir":{"value":2},"attrs":{}},"genScenes":{"dir":{"value":2},"attrs":{}},"genOnOff":{"dir":{"value":2},"attrs":{}},"genLevelCtrl":{"dir":{"value":2},"attrs":{}},"lightingColorCtrl":{"dir":{"value":2},"attrs":{}},"lightLink":{"dir":{"value":3},"attrs":{}}}},"3":{"profId":260,"epId":3,"devId":2064,"inClusterList":[0,4096,64768],"outClusterList":[3,4,5,6,8,768,4096],"clusters":{"64768":{"dir":{"value":1},"attrs":{}},"genBasic":{"dir":{"value":1},"attrs":{"zclVersion":1,"appVersion":1,"stackVersion":0,"hwVersion":1,"manufacturerName":"OSRAM","modelId":"Switch 4x EU-LIGHTIFY","dateCode":"201707051445390","powerSource":3,"swBuildId":"e.1.0.0M"}},"genIdentify":{"dir":{"value":2},"attrs":{}},"genGroups":{"dir":{"value":2},"attrs":{}},"genScenes":{"dir":{"value":2},"attrs":{}},"genOnOff":{"dir":{"value":2},"attrs":{}},"genLevelCtrl":{"dir":{"value":2},"attrs":{}},"lightingColorCtrl":{"dir":{"value":2},"attrs":{}},"lightLink":{"dir":{"value":3},"attrs":{}}}},"4":{"profId":260,"epId":4,"devId":2064,"inClusterList":[0,4096,64768],"outClusterList":[3,4,5,6,8,768,4096],"clusters":{"64768":{"dir":{"value":1},"attrs":{}},"genBasic":{"dir":{"value":1},"attrs":{}},"genIdentify":{"dir":{"value":2},"attrs":{}},"genGroups":{"dir":{"value":2},"attrs":{}},"genScenes":{"dir":{"value":2},"attrs":{}},"genOnOff":{"dir":{"value":2},"attrs":{}},"genLevelCtrl":{"dir":{"value":2},"attrs":{}},"lightingColorCtrl":{"dir":{"value":2},"attrs":{}},"lightLink":{"dir":{"value":3},"attrs":{}}}},"5":{"profId":260,"epId":5,"devId":2064,"inClusterList":[0,4096,64768],"outClusterList":[3,4,5,6,8,768,4096],"clusters":{"64768":{"dir":{"value":1},"attrs":{}},"genBasic":{"dir":{"value":1},"attrs":{"zclVersion":1,"appVersion":1,"stackVersion":0,"hwVersion":1,"manufacturerName":"OSRAM","modelId":"Switch 4x EU-LIGHTIFY","dateCode":"201707051445390","powerSource":3,"swBuildId":"e.1.0.0M"}},"genIdentify":{"dir":{"value":2},"attrs":{}},"genGroups":{"dir":{"value":2},"attrs":{}},"genScenes":{"dir":{"value":2},"attrs":{}},"genOnOff":{"dir":{"value":2},"attrs":{}},"genLevelCtrl":{"dir":{"value":2},"attrs":{}},"lightingColorCtrl":{"dir":{"value":2},"attrs":{}},"lightLink":{"dir":{"value":3},"attrs":{}}}},"6":{"profId":260,"epId":6,"devId":2064,"inClusterList":[0,4096,64768],"outClusterList":[3,4,5,6,8,768,4096],"clusters":{"64768":{"dir":{"value":1},"attrs":{}},"genBasic":{"dir":{"value":1},"attrs":{"zclVersion":1,"appVersion":1,"stackVersion":0,"hwVersion":1,"manufacturerName":"OSRAM","modelId":"Switch 4x EU-LIGHTIFY","dateCode":"201707051445390","powerSource":3,"swBuildId":"e.1.0.0M"}},"genIdentify":{"dir":{"value":2},"attrs":{}},"genGroups":{"dir":{"value":2},"attrs":{}},"genScenes":{"dir":{"value":2},"attrs":{}},"genOnOff":{"dir":{"value":2},"attrs":{}},"genLevelCtrl":{"dir":{"value":2},"attrs":{}},"lightingColorCtrl":{"dir":{"value":2},"attrs":{}},"lightLink":{"dir":{"value":3},"attrs":{}}}}},"_id":"xwXAlch01d0vMEMF"}
``

arteck, 2018-08-10:
something new ????

Koenkk, 2018-08-10:
@tb-killa can I provide any help?

tb-killa, 2018-08-10:
I´v doesn´t work on this in the last time.
Maybe we should first check the output again @arteck provide some sample.
If i got some freetime i could check again ...

tb-killa, 2018-08-30:
I´v work this evening with this device and got some interesting message if i remove battery and reinsert them:

```

tb-killa, 2018-08-30:
I´v got a new message if device is paired and battery removed and reinsert:
```
zigbee2mqtt:debug 2018-8-30 22:33:40 Recieved zigbee message with data {"cid":"genPowerCfg","data":{"batteryAlarmState":0}}
  zigbee2mqtt:warn 2018-8-30 22:33:40 No converter available for 'TODO' with cid 'genPowerCfg', type 'attReport' and data '{"cid":"genPowerCfg","data":{"batteryAlarmState":0}}'
  zigbee2mqtt:warn 2018-8-30 22:33:40 Please see: https://github.com/Koenkk/zigbee2mqtt/wiki/How-to-support-new-devices.
```

Actually my ``devices.js`` entry look like this:

```
{
        zigbeeModel: ['Switch 4x EU-LIGHTIFY'],
        model: 'TODO',
        vendor: 'OSRAM',
        description: 'LIGHTIFY Switch 4x',
        supports: 'TODO',
        fromZigbee: [],
        toZigbee: [],
        configure: (ieeeAddr, shepherd, coordinator, callback) => {
            const device = shepherd.find(ieeeAddr, 1);
            const actions = [
                (cb) => device.bind('genOnOff', coordinator, cb),
                (cb) => device.bind('lightingColorCtrl', coordinator, cb),
                (cb) => device.bind('genLevelCtrl', coordinator, cb),
                (cb) => device.bind('genPowerCfg', coordinator, cb),
                (cb) => device.report('genPowerCfg', 'batteryVoltage', 300, 3600, 0, cb),
            ];
             execute(device, actions, callback);
        },
    },
```

tb-killa, 2018-09-22:
@Koenkk 
I´m actually working on this to maybe finishing:

First of all i search for ``zcl-id`` / ``zcl-packet definition`` - ``lightingColorCtrl`` to get them correct extended:

```
Cmd-API | CmdId | Direction | Arguments
moveColorTemp | 0x4b (75) | c2s | { ratex, ratey }
stepColorTemp | 0x4c (76) | c2s | { stepx, stepy, transtime }

```

for ``zcl_meta.json`` 

```
"moveColorTemp":{
                "params":[
                    {"ratex":"int16"},
                    {"ratey":"int16"}
                ],
                "dir":0},
"stepColorTemp":{
                "params":[
                    {"stepx":"int16"},
                    {"stepy":"int16"},
                    {"transtime":"uint16"}
                ],
                "dir":0},
```

Will check what we now get on Informations back.

tb-killa, 2018-09-30:
Hi @Koenkk 
it seems to be that we got something not finish or broken.
One Sample:
from zcl-packet -> FuncPayload with debug message extended:

```
Debug for FuncPayload: genOnOff 0 1
```
and this is our converter message output:
```
zigbee2mqtt:warn 2018-9-30 14:41:15 No converter available for '4x EU-LIGHTIFY' with cid 'genOnOff', type 'cmdOn' and data '{"cid":"genOnOff","data":{}}'
```

as you could see .. it seems to use the right ``cid`` but ``data`` is empty!

now we got another button who is also binding to ``genOnOff`` default:

```
Debug for FuncPayload: genOnOff 0 0
```
but we doesnt see any converter messages .. so here it seems to be some broken structure.

the same for another functionality:

```
Debug for FuncPayload: genLevelCtrl 0 5
..
Debug for FuncPayload: genLevelCtrl 0 3
```
and this is our converter message:
```
zigbee2mqtt:warn 2018-9-30 14:45:07 No converter available for '4x EU-LIGHTIFY' with cid 'genLevelCtrl', type 'cmdStop' and data '{"cid":"genLevelCtrl","data":{}}'
```
we doesn´t see the ``data`` messages .. from debug on zcl-packet -> functionality.js we got them.

Do you know what happened in detail ??

Koenkk, 2018-10-02:
Why do we need `data`? We can check the action based on the type, in the example below: `cmdOn`.

> ```
> genOnOff 0 0
> genOnOff 0 0
> genOnOff 0 1
> 2018-7-24 12:13:22 WARN No converter available for 'TODO' with cid 'genOnOff', type 'cmdOn' and data '{"cid":"genOnOff","data":{}}'
> 2018-7-24 12:13:22 WARN [...]

tb-killa, 2018-10-02:
As you could see at this example:
https://github.com/Koenkk/zigbee2mqtt/issues/159#issuecomment-407655117

we see the converter message only for the first button.
as the debug log shows you that pushing the second one we also get ``genOnOff`` but NO converter message.

That the same as i did post above.
As we doesn´t see on every button the right converter message i´m thinking about some missing data values .. but maybe I'm on a complete wrong track and you're showing me how it could work right?

Koenkk, 2018-10-02:
Could you add to: https://github.com/Koenkk/zigbee-shepherd/blob/master/lib/components/af.js#L670

```js
console.log(`-- ${msg.zclMsg}`)
```

tb-killa, 2018-10-02:
Hi @Koenkk 
because of your code and the return of ``-- [object Object]``
i modify them to ``console.log(`-- ${JSON.stringify(msg.zclMsg)}`)`` and get this results:

```
  zigbee-shepherd:msgHdlr IND <-- AF:incomingMsg, transId: 0 +7ms
-- {"frameCntl":{"frameType":1,"manufSpec":0,"direction":0,"disDefaultRsp":1},"manufCode":0,"seqNum":63,"cmdId":"on","payload":{}}
  zigbee2mqtt:debug 2018-10-2 17:06:05 Recieved zigbee message with data {"cid":"genOnOff","data":{}}
  zigbee2mqtt:warn 2018-10-2 17:06:05 No converter available for '4x EU-LIGHTIFY' with cid 'genOnOff', type 'cmdOn' and data '{"cid":"genOnOff","data":{}}'
  zigbee2mqtt:warn 2018-10-2 17:06:05 Please see: https://github.com/Koenkk/zigbee2mqtt/wiki/How-to-support-new-devices.
```

and second button:

```
  zigbee-shepherd:msgHdlr IND <-- AF:incomingMsg, transId: 0 +23ms
-- {"frameCntl":{"frameType":1,"manufSpec":0,"direction":0,"disDefaultRsp":1},"manufCode":0,"seqNum":65,"cmdId":"off","payload":{}}
```

without converter message....
but i see .. after i extend ``const cmdIDs`` with ``off`` 

i got the message now:

```
  zigbee-shepherd:msgHdlr IND <-- AF:incomingMsg, transId: 0 +2ms
-- {"frameCntl":{"frameType":1,"manufSpec":0,"direction":0,"disDefaultRsp":1},"manufCode":0,"seqNum":73,"cmdId":"off","payload":{}}
  zigbee2mqtt:debug 2018-10-2 17:09:33 Recieved zigbee message with data {"cid":"genOnOff","data":{}}
  zigbee2mqtt:warn 2018-10-2 17:09:33 No converter available for '4x EU-LIGHTIFY' with cid 'genOnOff', type 'cmdOff' and data '{"cid":"genOnOff","data":{}}'
  zigbee2mqtt:warn 2018-10-2 17:09:33 Please see: https://github.com/Koenkk/zigbee2mqtt/wiki/How-to-support-new-devices.
```

tb-killa, 2018-10-02:
Okay i got them now:
``const cmdIDs = ['on', 'off', 'offWithEffect', 'step', 'stop', 'stepColorTemp', 'moveWithOnOff', 'move', 'moveHue', 'moveToSaturation'];``

Now we need to build all type of converter, hope you could help me @Koenkk ??

This is my converter.js part to get them all working:

```
{
        zigbeeModel: ['Switch 4x EU-LIGHTIFY'],
        model: '4x EU-LIGHTIFY',
        vendor: 'OSRAM',
        description: 'LIGHTIFY Switch 4x',
        supports: 'on/off',
        fromZigbee: [fz.ignore_power_change, fz.generic_power_battvolt],
        toZigbee: [tz.factory_reset],
        configure: (ieeeAddr, shepherd, coordinator, callback) => {
        const ep1 = shepherd.find(ieeeAddr, 1);
	const ep2 = shepherd.find(ieeeAddr, 2);
        const ep3 = shepherd.find(ieeeAddr, 3);
	const ep4 = shepherd.find(ieeeAddr, 4);
	
        const actions = [
		(cb) => ep1.bind('genOnOff', coordinator, cb),
                (cb) => ep1.bind('lightingColorCtrl', coordinator, cb),
                (cb) => ep1.bind('genLevelCtrl', coordinator, cb),
                (cb) => ep1.bind('genPowerCfg', coordinator, cb),
		(cb) => ep1.report('genPowerCfg', 'batteryVoltage', 300, 3600, 0, cb),
		(cb) => ep2.bind('genOnOff', coordinator, cb),
                (cb) => ep2.bind('lightingColorCtrl', coordinator, cb),
                (cb) => ep2.bind('genLevelCtrl', coordinator, cb),
		(cb) => ep3.bind('genOnOff', coordinator, cb),
                (cb) => ep3.bind('lightingColorCtrl', coordinator, cb),
                (cb) => ep3.bind('genLevelCtrl', coordinator, cb),
		(cb) => ep4.bind('genOnOff', coordinator, cb),
                (cb) => ep4.bind('lightingColorCtrl', coordinator, cb),
                (cb) => ep4.bind('genLevelCtrl', coordinator, cb),
			 ];
            
	execute(ep1, actions, callback);
        },
    },
```
**see also my modification here:** https://github.com/Koenkk/zigbee2mqtt/issues/159?_pjax=%23js-repo-pjax-container#issuecomment-423765975

and here are the sample log for every button functionality like push, push and hold, release:

```
  zigbee-shepherd:msgHdlr IND <-- AF:incomingMsg, transId: 0 +35s
-- {"frameCntl":{"frameType":1,"manufSpec":0,"direction":0,"disDefaultRsp":1},"manufCode":0,"seqNum":41,"cmdId":"on","payload":{}}
  zigbee2mqtt:debug 2018-10-2 17:28:21 Recieved zigbee message with data {"cid":"genOnOff","data":{}}
  zigbee2mqtt:warn 2018-10-2 17:28:21 No converter available for '4x EU-LIGHTIFY' with cid 'genOnOff', type 'cmdOn' and data '{"cid":"genOnOff","data":{}}'
  zigbee2mqtt:warn 2018-10-2 17:28:21 Please see: https://github.com/Koenkk/zigbee2mqtt/wiki/How-to-support-new-devices.

  zigbee-shepherd:msgHdlr IND <-- AF:incomingMsg, transId: 0 +2s
-- {"frameCntl":{"frameType":1,"manufSpec":0,"direction":0,"disDefaultRsp":1},"manufCode":0,"seqNum":43,"cmdId":"off","payload":{}}
  zigbee2mqtt:debug 2018-10-2 17:28:23 Recieved zigbee message with data {"cid":"genOnOff","data":{}}
  zigbee2mqtt:warn 2018-10-2 17:28:23 No converter available for '4x EU-LIGHTIFY' with cid 'genOnOff', type 'cmdOff' and data '{"cid":"genOnOff","data":{}}'
  zigbee2mqtt:warn 2018-10-2 17:28:23 Please see: https://github.com/Koenkk/zigbee2mqtt/wiki/How-to-support-new-devices.

  zigbee-shepherd:msgHdlr IND <-- AF:incomingMsg, transId: 0 +2s
-- {"frameCntl":{"frameType":1,"manufSpec":0,"direction":0,"disDefaultRsp":1},"manufCode":0,"seqNum":45,"cmdId":"moveWithOnOff","payload":{"movemode":0,"rate":38}}
  zigbee2mqtt:debug 2018-10-2 17:28:26 Recieved zigbee message with data {"cid":"genLevelCtrl","data":{"movemode":0,"rate":38}}
  zigbee2mqtt:warn 2018-10-2 17:28:26 No converter available for '4x EU-LIGHTIFY' with cid 'genLevelCtrl', type 'cmdMoveWithOnOff' and data '{"cid":"genLevelCtrl","data":{"movemode":0,"rate":38}}'
  zigbee2mqtt:warn 2018-10-2 17:28:26 Please see: https://github.com/Koenkk/zigbee2mqtt/wiki/How-to-support-new-devices.
  zigbee-shepherd:msgHdlr IND <-- AF:incomingMsg, transId: 0 +420ms
-- {"frameCntl":{"frameType":1,"manufSpec":0,"direction":0,"disDefaultRsp":1},"manufCode":0,"seqNum":47,"cmdId":"stop","payload":{}}
  zigbee2mqtt:debug 2018-10-2 17:28:26 Recieved zigbee message with data {"cid":"genLevelCtrl","data":{}}
  zigbee2mqtt:warn 2018-10-2 17:28:26 No converter available for '4x EU-LIGHTIFY' with cid 'genLevelCtrl', type 'cmdStop' and data '{"cid":"genLevelCtrl","data":{}}'
  zigbee2mqtt:warn 2018-10-2 17:28:26 Please see: https://github.com/Koenkk/zigbee2mqtt/wiki/How-to-support-new-devices.
[... 22 lines cut ...]
  zigbee2mqtt:warn 2018-10-2 17:28:33 Please see: https://github.com/Koenkk/zigbee2mqtt/wiki/How-to-support-new-devices.

  zigbee-shepherd:msgHdlr IND <-- AF:incomingMsg, transId: 0 +2s
-- {"frameCntl":{"frameType":1,"manufSpec":0,"direction":0,"disDefaultRsp":1},"manufCode":0,"seqNum":57,"cmdId":"moveToSaturation","payload":{"saturation":254,"transtime":2}}
  zigbee2mqtt:debug 2018-10-2 17:28:36 Recieved zigbee message with data {"cid":"lightingColorCtrl","data":{"saturation":254,"transtime":2}}
  zigbee2mqtt:warn 2018-10-2 17:28:36 No converter available for '4x EU-LIGHTIFY' with cid 'lightingColorCtrl', type 'cmdMoveToSaturation' and data '{"cid":"lightingColorCtrl","data":{"saturation":254,"transtime":2}}'
  zigbee2mqtt:warn 2018-10-2 17:28:36 Please see: https://github.com/Koenkk/zigbee2mqtt/wiki/How-to-support-new-devices.
  zigbee-shepherd:msgHdlr IND <-- AF:incomingMsg, transId: 0 +949ms
[... cut ...]
-- {"frameCntl":{"frameType":1,"manufSpec":0,"direction":0,"disDefaultRsp":1},"manufCode":0,"seqNum":63,"cmdId":"moveHue","payload":{"movemode":0,"rate":0}}
  zigbee2mqtt:debug 2018-10-2 17:28:39 Recieved zigbee message with data {"cid":"lightingColorCtrl","data":{"movemode":0,"rate":0}}
  zigbee2mqtt:warn 2018-10-2 17:28:39 No converter available for '4x EU-LIGHTIFY' with cid 'lightingColorCtrl', type 'cmdMoveHue' and data '{"cid":"lightingColorCtrl","data":{"movemode":0,"rate":0}}'
  zigbee2mqtt:warn 2018-10-2 17:28:39 Please see: https://github.com/Koenkk/zigbee2mqtt/wiki/How-to-support-new-devices.

```

tb-killa, 2018-10-02:
actually i got the ``genOnOff`` and the ``genLevelCtrl`` converter part finished .. thanks to our Philips-Switch we got a solid template :)

```
 debug: Recieved zigbee message with data {"cid":"genOnOff","data":{}}
 info: MQTT publish, topic: 'zigbee2mqtt/Switch 4x EULIGHTIFY', payload: '{"voltage":2.9,"action":"on"}'
 debug: Recieved zigbee message with data {"cid":"genOnOff","data":{}}
 info: MQTT publish, topic: 'zigbee2mqtt/Switch 4x EULIGHTIFY', payload: '{"voltage":2.9,"action":"off"}'
 debug: Recieved zigbee message with data {"cid":"genPowerCfg","data":{"batteryVoltage":29}}
 info: MQTT publish, topic: 'zigbee2mqtt/Switch 4x EULIGHTIFY', payload: '{"voltage":2.9}'
 debug: Recieved zigbee message with data {"cid":"genLevelCtrl","data":{}}
 debug: Recieved zigbee message with data {"cid":"genLevelCtrl","data":{"movemode":0,"rate":38}}
 info: MQTT publish, topic: 'zigbee2mqtt/Switch 4x EULIGHTIFY', payload: '{"voltage":2.9,"action":"uphold","brightness":255}'
 debug: Recieved zigbee message with data {"cid":"genLevelCtrl","data":{"movemode":1,"rate":38}}
 info: MQTT publish, topic: 'zigbee2mqtt/Switch 4x EULIGHTIFY', payload: '{"voltage":2.9,"action":"downhold","brightness":255}'
 debug: Recieved zigbee message with data {"cid":"genLevelCtrl","data":{}}
 info: MQTT publish, topic: 'zigbee2mqtt/Switch 4x EULIGHTIFY', payload: '{"voltage":2.9,"brightness":219.7,"action":"downholdrelease"}'
 debug: Recieved zigbee message with data {"cid":"genLevelCtrl","data":{"movemode":1,"rate":38}}
 info: MQTT publish, topic: 'zigbee2mqtt/Switch 4x EULIGHTIFY', payload: '{"voltage":2.9,"action":"downhold","brightness":219.7}'
 debug: Recieved zigbee message with data {"cid":"genLevelCtrl","data":{}}
 info: MQTT publish, topic: 'zigbee2mqtt/Switch 4x EULIGHTIFY', payload: '{"voltage":2.9,"brightness":88.29999999999998,"action":"downholdrelease"}'
 debug: Recieved zigbee message with data {"cid":"genLevelCtrl","data":{"movemode":1,"rate":38}}
 info: MQTT publish, topic: 'zigbee2mqtt/Switch 4x EULIGHTIFY', payload: '{"voltage":2.9,"action":"downhold","brightness":88.29999999999998}'
 debug: Recieved zigbee message with data {"cid":"genLevelCtrl","data":{}}
 info: MQTT publish, topic: 'zigbee2mqtt/Switch 4x EULIGHTIFY', payload: '{"voltage":2.9,"brightness":48.499999999999986,"action":"downholdrelease"}'
 debug: Recieved zigbee message with data {"cid":"genLevelCtrl","data":{"movemode":1,"rate":38}}
 info: MQTT publish, topic: 'zigbee2mqtt/Switch 4x EULIGHTIFY', payload: '{"voltage":2.9,"action":"downhold","brightness":48.499999999999986}'
 debug: Recieved zigbee message with data {"cid":"genLevelCtrl","data":{}}
 info: MQTT publish, topic: 'zigbee2mqtt/Switch 4x EULIGHTIFY', payload: '{"voltage":2.9,"brightness":11.999999999999986,"action":"downholdrelease"}'
 debug: Recieved zigbee message with data {"cid":"genLevelCtrl","data":{"movemode":1,"rate":38}}
 info: MQTT publish, topic: 'zigbee2mqtt/Switch 4x EULIGHTIFY', payload: '{"voltage":2.9,"action":"downhold","brightness":11.999999999999986}'
 debug: Recieved zigbee message with data {"cid":"genLevelCtrl","data":{}}
 info: MQTT publish, topic: 'zigbee2mqtt/Switch 4x EULIGHTIFY', payload: '{"voltage":2.9,"brightness":0,"action":"downholdrelease"}'
 debug: Recieved zigbee message with data {"cid":"genLevelCtrl","data":{"movemode":1,"rate":38}}
 info: MQTT publish, topic: 'zigbee2mqtt/Switch 4x EULIGHTIFY', payload: '{"voltage":2.9,"action":"downhold","brightness":0}'
 debug: Recieved zigbee message with data {"cid":"genLevelCtrl","data":{}}
 info: MQTT publish, topic: 'zigbee2mqtt/Switch 4x EULIGHTIFY', payload: '{"voltage":2.9,"brightness":0,"action":"downholdrelease"}'
```

now i´m working on ``lightingColorCtrl`` converter part .. will report if i got something working.

tb-killa, 2018-10-02:
@Koenkk Question: 
if i push the button single stepp for up or down i got this message:
```
warn: No converter available for '4x EU-LIGHTIFY' with cid 'lightingColorCtrl', type 'cmdStepColorTemp' and data '{"cid":"lightingColorCtrl","data":{"stepx":6401,"stepy":2560,"transtime":0}}'
```
and
```
warn: No converter available for '4x EU-LIGHTIFY' with cid 'lightingColorCtrl', type 'cmdStepColorTemp' and data '{"cid":"lightingColorCtrl","data":{"stepx":6403,"stepy":2560,"transtime":0}}'
```
The two values for stepx only changed between this two .. so no higher or lower value by pushing.

for the push and hold i got multiple type and cid messages ..
from the complexibility i think we should check what we exactly need.
if we would like to transport ``lightingColorCtrl`` i need the help from @Koenkk 

so what should we official present ? the ``lightingColorCtrl`` themselve or maybe some sort of own button mapping  like another on/off and brighter/lower function ??

Koenkk, 2018-10-03:
The `lightingColorCtrl` command should send the x/y color to MQTT.

tb-killa, 2018-10-03:
okay thanks.
as you could see by this logfile (hold and release up / hold and release down button), the single step push are the value from the above post so i doesn´t duplicate them here, i doesn´t got easy your x/y color values ..

Maybe you got an Idea so i could implement??

```
2018-10-03T12:56:30.392Z - debug: Recieved zigbee message with data {"cid":"lightingColorCtrl","data":{"saturation":254,"transtime":2}}
2018-10-03T12:56:30.395Z - warn: No converter available for '4x EU-LIGHTIFY' with cid 'lightingColorCtrl', type 'cmdMoveToSaturation' and data '{"cid":"lightingColorCtrl","data":{"saturation":254,"transtime":2}}'
2018-10-03T12:56:30.398Z - warn: Please see: https://github.com/Koenkk/zigbee2mqtt/wiki/How-to-support-new-devices.
2018-10-03T12:56:31.340Z - debug: Recieved zigbee message with data {"cid":"lightingColorCtrl","data":{"movemode":1,"rate":25}}
2018-10-03T12:56:31.343Z - warn: No converter available for '4x EU-LIGHTIFY' with cid 'lightingColorCtrl', type 'cmdMoveHue' and data '{"cid":"lightingColorCtrl","data":{"movemode":1,"rate":25}}'
2018-10-03T12:56:31.345Z - warn: Please see: https://github.com/Koenkk/zigbee2mqtt/wiki/How-to-support-new-devices.
2018-10-03T12:56:33.504Z - debug: Recieved zigbee message with data {"cid":"lightingColorCtrl","data":{"movemode":0,"rate":0}}
2018-10-03T12:56:33.507Z - warn: No converter available for '4x EU-LIGHTIFY' with cid 'lightingColorCtrl', type 'cmdMoveHue' and data '{"cid":"lightingColorCtrl","data":{"movemode":0,"rate":0}}'
2018-10-03T12:56:33.510Z - warn: Please see: https://github.com/Koenkk/zigbee2mqtt/wiki/How-to-support-new-devices.
```
and
```
2018-10-03T12:57:42.249Z - debug: Recieved zigbee message with data {"cid":"lightingColorCtrl","data":{"saturation":254,"transtime":2}}
2018-10-03T12:57:42.253Z - warn: No converter available for '4x EU-LIGHTIFY' with cid 'lightingColorCtrl', type 'cmdMoveToSaturation' and data '{"cid":"lightingColorCtrl","data":{"saturation":254,"transtime":2}}'
2018-10-03T12:57:42.255Z - warn: Please see: https://github.com/Koenkk/zigbee2mqtt/wiki/How-to-support-new-devices.
2018-10-03T12:57:43.207Z - debug: Recieved zigbee message with data {"cid":"lightingColorCtrl","data":{"movemode":3,"rate":25}}
2018-10-03T12:57:43.215Z - warn: No converter available for '4x EU-LIGHTIFY' with cid 'lightingColorCtrl', type 'cmdMoveHue' and data '{"cid":"lightingColorCtrl","data":{"movemode":3,"rate":25}}'
2018-10-03T12:57:43.219Z - warn: Please see: https://github.com/Koenkk/zigbee2mqtt/wiki/How-to-support-new-devices.
2018-10-03T12:57:44.236Z - debug: Recieved zigbee message with data {"cid":"lightingColorCtrl","data":{"saturation":254,"transtime":2}}
2018-10-03T12:57:44.240Z - warn: No converter available for '4x EU-LIGHTIFY' with cid 'lightingColorCtrl', type 'cmdMoveToSaturation' and data '{"cid":"lightingColorCtrl","data":{"saturation":254,"transtime":2}}'
2018-10-03T12:57:44.243Z - warn: Please see: https://github.com/Koenkk/zigbee2mqtt/wiki/How-to-support-new-devices.
2018-10-03T12:57:45.238Z - debug: Recieved zigbee message with data {"cid":"lightingColorCtrl","data":{"movemode":3,"rate":25}}
2018-10-03T12:57:45.242Z - warn: No converter available for '4x EU-LIGHTIFY' with cid 'lightingColorCtrl', type 'cmdMoveHue' and data '{"cid":"lightingColorCtrl","data":{"movemode":3,"rate":25}}'
2018-10-03T12:57:45.244Z - warn: Please see: https://github.com/Koenkk/zigbee2mqtt/wiki/How-to-support-new-devices.
2018-10-03T12:57:47.053Z - debug: Recieved zigbee message with data {"cid":"lightingColorCtrl","data":{"movemode":0,"rate":0}}
2018-10-03T12:57:47.057Z - warn: No converter available for '4x EU-LIGHTIFY' with cid 'lightingColorCtrl', type 'cmdMoveHue' and data '{"cid":"lightingColorCtrl","data":{"movemode":0,"rate":0}}'
2018-10-03T12:57:47.061Z - warn: Please see: https://github.com/Koenkk/zigbee2mqtt/wiki/How-to-support-new-devices.
```

Should we ignore this output and build our own logic like we do with ``genLevelCtrl`` ??

Koenkk, 2018-10-03:
Maybe you could use the TRADFRI wireless dimmer implementation as a reference? https://github.com/Koenkk/zigbee-shepherd-converters/blob/master/converters/fromZigbee.js#L828

tb-killa, 2018-10-03:
Okay thanks that could help :)

arteck, 2018-11-18:
no .. its new zigbee2mqtt installation...

Koenkk, 2018-11-19:
AFAIK zcl-packet has not been changed in 0.2.0.

Can you add to: https://github.com/Koenkk/zcl-packet/blob/master/lib/functional.js#L20

```js
console.log(clusterId, direction, cmd)
```

arteck, 2018-12-05:
with new installation... works..

arteck, 2018-12-05:
sry for close..

with new installation works..

Koenkk, 2018-12-05:
For iobroker issues, please open one at the iobroker issue tracker.

irni, 2019-01-22:
I have the same switch so i followed the provided steps above. 
I had a hard Time to change the lines in my hassio- docker, but i am Not able to get the Switch Running. 
Is there a Chance to implement the changes into the edge Build?

dreimer1986, 2019-01-31:
Same thing here. Did not check here before buying and now tried to make my own new device support. Results:
`/app/node_modules/zcl-packet/lib/functional.js:33
        throw new Error('Unrecognized command');
        ^
Error: Unrecognized command
    at new FuncPayload (/app/node_modules/zcl-packet/lib/functional.js:33:15)
    at /app/node_modules/zcl-packet/lib/zcl.js:33:22
    at Dissolve.<anonymous> (/app/node_modules/zcl-packet/lib/zcl.js:121:9)
    at Object.onceWrapper (events.js:315:30)
    at emitOne (events.js:116:13)
    at Dissolve.emit (events.js:211:7)
    at Dissolve.<anonymous> (/app/node_modules/dissolve-chunks/index.js:73:29)
    at emitNone (events.js:106:13)
    at Dissolve.emit (events.js:208:7)
    at emitReadable_ (/app/node_modules/dissolve/node_modules/readable-stream/lib/_stream_readable.js:456:10)
npm
 ERR! code ELIFECYCLE
npm ERR! errno 1
npm
 ERR! zigbee2mqtt@1.1.0 start: `node index.js`
npm ERR!
 Exit status 1
npm ERR! 
npm ERR! Failed at the zigbee2mqtt@1.1.0 start script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.`

I managed to get the left up and down working and being recognized by stealing all stuff from "Lightify Switch Mini" though. Current results:
`	{
        zigbeeModel: ['Switch 4x EU-LIGHTIFY'],
        model: 'AB371860155',
        vendor: 'OSRAM',
        description: 'LIGHTIFY Switch',
        supports: 'on/off, brightness',
        fromZigbee: [
            fz.AC0251100NJ_on, fz.AC0251100NJ_off, fz.generic_batteryvoltage_3000_2500,
        ],
        toZigbee: [],
		configure: (ieeeAddr, shepherd, coordinator, callback) => {
            const device = shepherd.find(ieeeAddr, 1);
            const actions = [
                (cb) => device.bind('genOnOff', coordinator, cb),
                (cb) => device.bind('lightingColorCtrl', coordinator, cb),
                (cb) => device.bind('genLevelCtrl', coordinator, cb),
                (cb) => device.bind('genPowerCfg', coordinator, cb),
                (cb) => device.report('genPowerCfg', 'batteryVoltage', 900, 3600, 0, cb),
            ];
            execute(device, actions, callback);
        },
    },`

This makes it detect a switching on and off on the left side, but right side results in what I pasted above.

Koenkk, 2019-02-02:
This switch send a broadcast request. Let's continue this discussion here: https://github.com/Koenkk/zigbee2mqtt/issues/962

closed 2019-02-02 by Koenkk

tb-killa, 2019-03-10:
What does change that we doesn´t get them not working ?
We found out that this return on zcl-packet site ``lightingColorCtrl 0 76``.

But after rebuild my old work i doesn´t get them working.
Does we modify something with the ``lightingColorCtrl`` ??

dreimer1986, 2019-03-10:
The problem is that the switch does stuff to lights in the same network by default and that has to stop. If this is not possible, it's useless. Thus koenkk might need some Zigbee sniffs from a Switch and a official gateway and commands of what what button should do. I have all the stuff here, only the zigbee sniffer is a lil problem right now...

tb-killa, 2019-03-10:
Yes that's what I actually read.
But if you follow the complete issue we got them working on an older time. I need to figure out why this isn't work anymore.

[8 more replies]