Koenkk/zigbee2mqtt #161: Issues with USB stick

matchett808, 2018-06-29:
I get the following in my Dmesg log before needing to do the whole reset process as #156:

```
[177315.829555] usb 7-2: USB disconnect, device number 118                                                                                                                                                                                                                                                                                      
[177315.830026] cdc_acm 7-2:1.1: urb 0 failed submission with -19                                                                                                                                                                                                                                                                               
[177315.830705] cdc_acm 7-2:1.0: failed to set dtr/rts                                                                                                                                                                                                                                                                                          
[177315.831399] cdc_acm 7-2:1.1: urb 1 failed submission with -19                                                                                                                                                                                                                                                                               
[177315.832097] cdc_acm 7-2:1.1: urb 2 failed submission with -19                                                                                                                                                                                                                                                                               
[177316.307702] usb 7-2: new full-speed USB device number 119 using ohci-pci                                                                                                                                                                                                                                                                    
[177316.503128] usb 7-2: New USB device found, idVendor=0451, idProduct=16a8                                                                                                                                                                                                                                                                    
[177316.503134] usb 7-2: New USB device strings: Mfr=1, Product=2, SerialNumber=3                                                                                                                                                                                                                                                               
[177316.503138] usb 7-2: Product: TI CC2531 USB CDC                                                                                                                                                                                                                                                                                             
[177316.503141] usb 7-2: Manufacturer: Texas Instruments                                                                                                                                                                                                                                                                                        
[177316.503144] usb 7-2: SerialNumber: __0X00124B0014BDEBDA                                                                                                                                                                                                                                                                                     
[177316.506501] cdc_acm 7-2:1.0: ttyACM1: USB ACM device                                                                                                                                                                                                                                                                                        
[177359.560091] usb 7-2: USB disconnect, device number 119                                                                                                                                                                                                                                                                                      
[177361.887813] usb 7-2: new full-speed USB device number 120 using ohci-pci                                                                                                                                                                                                                                                                    
[177362.083100] usb 7-2: New USB device found, idVendor=0451, idProduct=16a8                                                                                                                                                                                                                                                                    
[177362.083104] usb 7-2: New USB device strings: Mfr=1, Product=2, SerialNumber=3                                                                                                                                                                                                                                                               
[177362.083107] usb 7-2: Product: TI CC2531 USB CDC                                                                                                                                                                                                                                                                                             
[177362.083109] usb 7-2: Manufacturer: Texas Instruments                                                                                                                                                                                                                                                                                        
[177362.083111] usb 7-2: SerialNumber: __0X00124B0014BDEBDA                                                                                                                                                                                                                                                                                     
[177362.086530] cdc_acm 7-2:1.0: ttyACM1: USB ACM device                                
```

I know the reset process works - but it being necessary (for me on a dialy basis) makes this unusable - some info:

- A few bulbs in network
- running in a docker contianer
- passing device as /dev/zigbee -> /dev/zigbee
- device is mapped using a udev rule: ```SUBSYSTEM=="tty", ATTRS{idVendor}=="0451", ATTRS{idProduct}=="16a8",  SYMLINK+="zigbee"```

Things I have tried:
- Re-flashing with regular firmware
- Re-flashing with experimental firmwas
- machine restart
- using /dev/ttyACM1 instead of /dev/zigbee
- fixing permissions

Things I still need to try:
- Another stick (this is a maybe - wondering if it could be a hardware quality issue)
- running in a different USB port (Server is currently in an on HAF-912 case, stick in front USB) - will try board USB
- check out Bios USB power options

matchett808, 2018-06-29:
Doing a bit of further digging (remotely unfortunately) I'm wondering if it could be a power saving issue?

ciotlosm, 2018-06-29:
@matchett808 might be. Except when joining multiple items with older firmware I haven't seen other issues similar to yours on RPI & such

matchett808, 2018-06-29:
hm - don't reckon it should be joining multiple items - powersaving modes seem most likely, I'll look into it 👍

iain-matchett, 2018-06-29:
Just as some additional debug:

```
zigbee-shepherd:request RSP <-- SAPI:getDeviceInfo +0ms                                                                                                                                                                                                                                                                                       
  zigbee-shepherd:request REQ --> SAPI:getDeviceInfo +0ms                                                                                                                                                                                                                                                                                       
  cc-znp:SREQ --> SAPI:getDeviceInfo, { param: 5 } +0ms                                                                                                                                                                                                                                                                                         
  serialport write 6 bytes of data +0ms                                                                                                                                                                                                                                                                                                         
  cc-znp:SRSP <-- SAPI:getDeviceInfo, { param: 5, value: <Buffer 0b 00 bd 14 00 4b 12 00> } +5ms                                                                                                                                                                                                                                                
  zigbee-shepherd:request RSP <-- SAPI:getDeviceInfo +0ms                                                                                                                                                                                                                                                                                       
  zigbee-shepherd:request REQ --> SAPI:getDeviceInfo +0ms                                                                                                                                                                                                                                                                                       
  cc-znp:SREQ --> SAPI:getDeviceInfo, { param: 6 } +0ms                                                                                                                                                                                                                                                                                         
  serialport write 6 bytes of data +0ms                                                                                                                                                                                                                                                                                                         
  cc-znp:SRSP <-- SAPI:getDeviceInfo, { param: 6, value: <Buffer 63 1a bd 14 00 4b 12 00> } +5ms                                                                                                                                                                                                                                                
  zigbee-shepherd:request RSP <-- SAPI:getDeviceInfo +0ms                                                                                                                                                                                                                                                                                       
  zigbee-shepherd:request REQ --> SAPI:getDeviceInfo +0ms                                                                                                                                                                                                                                                                                       
  cc-znp:SREQ --> SAPI:getDeviceInfo, { param: 7 } +0ms                                                                                                                                                                                                                                                                                         
  serialport write 6 bytes of data +0ms                                                                                                                                                                                                                                                                                                         
  cc-znp:SRSP <-- SAPI:getDeviceInfo, { param: 7, value: <Buffer dd dd dd dd dd dd dd dd> } +5ms                                                                                                                                                                                                                                                
  zigbee-shepherd:request RSP <-- SAPI:getDeviceInfo +0ms                                                                                                                                                                                                                                                                                       
[... cut ...]
2018-6-29 15:47:51 INFO Error while starting zigbee-shepherd, attemping to fix... (takes 60 seconds)                                                                                                                                                                                                                                            
  cc-znp The serialport /dev/zigbee is closed. +21ms                                                                                                                                                                                                                                                                                            
```

matchett808, 2018-06-29:
Gah, my god - can't beleive it, narrowed it down to being the docker container?? Weird that it worked and now doesn't work - will look into why this is (see if there's something I can do to fix the docker container, might catch some issues further down the line 👍 )

matchett808, 2018-06-29:
Closing - will PR if neccessary

closed 2018-06-29 by matchett808