Koenkk/zigbee2mqtt #162: Battery Level for sensors

talondnb, 2018-06-29:
I’ve been adding a few sensors to my setup and have created additional MQTT battery sensors for each device using the json parameter, but even after a few restarts and even pair/unpairing, I’m struggling to have the values appear on some devices. I’ve even tried leaving them over a few days to poll but still nothing appears.

Any idea what would be the cause? I’m using discovery, perhaps I should try manual sensor config for each?

talondnb, 2018-06-29:
It seems to only be this way for motion sensors. I'm seeing this error in the logs:

[homeassistant.helpers.template] Error parsing value: 'dict object' has no attribute 'battery' (value: {"occupancy":false}, template: {{ value_json.battery | int }})

And my typical battery sensor config (works on door/window sensors fine):

- platform: mqtt
  name: "Front Door Sensor Battery"
  state_topic: "zigbee2mqtt/0x00158d000239f6ee"
  unit_of_measurement: '%'
  value_template: "{{ value_json.battery | int }}"

ryanbeaton, 2018-06-29:
I only did the temp sensor battery because it was worryingly low. I’ll do the motion tomorrow and report back.
Note: I have the Mi Temperature and the Aqara Body Sensor

<sub>Sent with <a href="http://githawk.com">GitHawk</a></sub>

ciotlosm, 2018-06-29:
@talondnb you don't always get battery on messages. @Koenkk maybe send battery with empty value or null when when no value received yet?

talondnb, 2018-06-29:
@ciotlosm I understand, and I've noticed this on my other working sensors. So far the behaviour is a little odd on the motion sensors but some other odd behaviour on the temp sensors.

Here is the output when I run mosquitto_sub -v -t '#' in hassio:

zigbee2mqtt/0x00158d00022809d8 {"occupancy":false}
zigbee2mqtt/0x00158d000222c719 {"battery":"39.00","voltage":2995,"temperature":18.87,"humidity":60.87}
zigbee2mqtt/0x00158d000222c64a {"battery":"43.00","voltage":3015}
zigbee2mqtt/0x00158d000239f6bd {"contact":true,"battery":"35.00","voltage":2975}
zigbee2mqtt/0x00158d00023dbef6 {"contact":true,"battery":"45.00","voltage":3025}
zigbee2mqtt/0x00158d00022809ec {"occupancy":false}
zigbee2mqtt/0x00158d000239f6ee {"battery":"39.00","voltage":2995}
zigbee2mqtt/0x00124b001202247a {"state":true}
zigbee2mqtt/0x00158d000239cb52 {"contact":true,"battery":"37.00","voltage":2985}

The #2 and #3 lines are both temp sensors that are the same model (MiJia). One shows battery but the other doesn't, yet in my battery sensor list they look fine:

[image]

talondnb, 2018-06-29:
@ciotlosm Yes, I'm aware, which is why I'm stumped as to why it's only just populated overnight. I've been monitoring for at least 6 hours for these levels to appear.

talondnb, 2018-06-30:
I'll close this issue for now, I have more sensors to add but if I face any further issues i'll raise a new one and link. Thanks.

closed 2018-06-30 by talondnb