Koenkk/zigbee2mqtt #163: dont get cc2530 working

1981nick1982, 2018-06-29:
My CC2531 and CC2530 with baseboard are now delivered. The CC2531 works perfectly, but I do not get the CC2530 to work. If I connect this via the baseboard to the USB of my RPI, I get the following message in the log.

`2018-06-29T17:37:07.052Z - info: Starting zigbee2mqtt version 0.0.0 (commit #2b$ 2018-06-29T17:37:07.137Z - info: Starting zigbee-shepherd 2018-06-29T17:37:10.364Z - warn: Error while starting zigbee-shepherd, attempin$ 2018-06-29T17:38:10.408Z - info: Starting zigbee-shepherd 2018-06-29T17:38:13.439Z - error: Error while starting zigbee-shepherd! 2018-06-29T17:38:13.444Z - error: Failed to start Error: request timeout at CcZnp.<anonymous> (/app/node_modules/cc-znp/lib/ccznp.js:254:22) at Object.onceWrapper (events.js:315:30) at emitOne (events.js:116:13) at CcZnp.emit (events.js:211:7) at Timeout.<anonymous> (/app/node_modules/cc-znp/lib/ccznp.js:233:18) at ontimeout (timers.js:475:11) at tryOnTimeout (timers.js:310:5) at Timer.listOnTimeout (timers.js:270:5)`

kirovilya, 2018-06-30:
Checking:
1) P20 pin to GND
2) port speed 115200
3) reboot rpi

1981nick1982, 2018-06-30:
I was too fast. Have the cc2530 flashed again. And now the problem has been solved. Have already been able to link a cube and 2 temperature sensors. With the temperature sensors, there is still a problem that they only transmit the pressure. At the CC2531 I immediately got all the information.

closed 2018-06-30 by 1981nick1982