Koenkk/zigbee2mqtt #164: Can not connect WXKG01LM

paskyecijano, 2018-06-29:
This is the log when i try to pairing the device.

2018-06-29T18:06:52.400Z - error: Cannot get the Node Descriptor of the Device: 0x00158d0001abcf90 (Error: Timed out after 10000 ms)
2018-06-29T18:07:39.995Z - warn: No converter available for 'QBKG04LM' with cid 'genBasic' and type 'attReport'
2018-06-29T18:07:39.998Z - warn: Please see: https://github.com/Koenkk/zigbee2mqtt/wiki/How-to-support-new-devices.
2018-06-29T18:07:40.009Z - warn: No converter available for 'QBKG04LM' with cid 'genBasic' and type 'devChange'

ciotlosm, 2018-06-29:
@paskyecijano keep pressing the button when pairing every couple of seconds to avoid it going to sleep and interrupting the pairing. Make sure you read in more detail https://github.com/Koenkk/zigbee2mqtt/wiki/Pairing-devices#xiaomi

talondnb, 2018-06-30:
@paskyecijano I receive that error when I'm not close enough to the co-ordinator (CC2531). I literally have to perform the pairing standing next to it.

Also, while pairing, as @ciotlosm mentioned, keep triggering the device every second or so as outlined in the wiki. I tend to also have the logs up and refreshing (from my phone) after each trigger.

Here is what I typically see for a pairing:

2018-06-22T10:56:08.428Z - info: Zigbee: allowing new devices to join.
2018-06-22T10:56:50.308Z - warn: Message without device!
2018-06-22T10:56:50.325Z - info: **Connecting with device...**
2018-06-22T10:56:50.328Z - info: MQTT publish, topic: 'zigbee2mqtt/bridge/log', payload: '{"type":"pairing","message":"connecting with device"}'
2018-06-22T10:56:50.986Z - info: Connecting with device...
2018-06-22T10:56:50.989Z - info: MQTT publish, topic: 'zigbee2mqtt/bridge/log', payload: '{"type":"pairing","message":"connecting with device"}'
2018-06-22T10:56:52.676Z - info: Connecting with device...
2018-06-22T10:56:52.679Z - info: MQTT publish, topic: 'zigbee2mqtt/bridge/log', payload: '{"type":"pairing","message":"connecting with device"}'
2018-06-22T10:56:53.561Z - info: Connecting with device...
2018-06-22T10:56:53.564Z - info: MQTT publish, topic: 'zigbee2mqtt/bridge/log', payload: '{"type":"pairing","message":"connecting with device"}'
2018-06-22T10:56:54.867Z - info: Connecting with device...
2018-06-22T10:56:54.870Z - info: MQTT publish, topic: 'zigbee2mqtt/bridge/log', payload: '{"type":"pairing","message":"connecting with device"}'
2018-06-22T10:56:55.521Z - info: Connecting with device...
2018-06-22T10:56:55.524Z - info: MQTT publish, topic: 'zigbee2mqtt/bridge/log', payload: '{"type":"pairing","message":"connecting with device"}'
2018-06-22T10:56:56.050Z - info: Connecting with device...
2018-06-22T10:56:56.053Z - info: MQTT publish, topic: 'zigbee2mqtt/bridge/log', payload: '{"type":"pairing","message":"connecting with device"}'
2018-06-22T10:56:56.791Z - info: Connecting with device...
2018-06-22T10:56:56.793Z - info: MQTT publish, topic: 'zigbee2mqtt/bridge/log', payload: '{"type":"pairing","message":"connecting with device"}'
2018-06-22T10:56:57.427Z - info: Connecting with device...
2018-06-22T10:56:57.430Z - info: MQTT publish, topic: 'zigbee2mqtt/bridge/log', payload: '{"type":"pairing","message":"connecting with device"}'
2018-06-22T10:56:57.929Z - info: Connecting with device...
2018-06-22T10:56:57.932Z - info: MQTT publish, topic: 'zigbee2mqtt/bridge/log', payload: '{"type":"pairing","message":"connecting with device"}'
2018-06-22T10:56:58.708Z - info: Connecting with device...
2018-06-22T10:56:58.711Z - info: MQTT publish, topic: 'zigbee2mqtt/bridge/log', payload: '{"type":"pairing","message":"connecting with device"}'
2018-06-22T10:56:59.548Z - info: Connecting with device...
2018-06-22T10:56:59.551Z - info: MQTT publish, topic: 'zigbee2mqtt/bridge/log', payload: '{"type":"pairing","message":"connecting with device"}'
2018-06-22T10:57:00.068Z - info: Connecting with device...
2018-06-22T10:57:00.071Z - info: MQTT publish, topic: 'zigbee2mqtt/bridge/log', payload: '{"type":"pairing","message":"connecting with device"}'
2018-06-22T10:57:00.567Z - info: Connecting with device...
2018-06-22T10:57:00.569Z - info: MQTT publish, topic: 'zigbee2mqtt/bridge/log', payload: '{"type":"pairing","message":"connecting with device"}'
2018-06-22T10:57:01.043Z - info: Connecting with device...
2018-06-22T10:57:01.046Z - info: MQTT publish, topic: 'zigbee2mqtt/bridge/log', payload: '{"type":"pairing","message":"connecting with device"}'
2018-06-22T10:57:01.060Z - info: **Device incoming...** - _at this point the device has joined and I can stop triggering_
2018-06-22T10:57:01.062Z - info: MQTT publish, topic: 'zigbee2mqtt/bridge/log', payload: '{"type":"pairing","message":"device incoming"}'

Hope this helps. YMMV.