Koenkk/zigbee2mqtt #165: Aqara temperature sensor dont give any info beside preasure

1981nick1982, 2018-06-30:
Since I switched from the CC2531 to the CC2530. My 2 temperature sensors only the give pressure. Very occasionally I do get a report of the temperature but it disappears again. I also get no notification of the humidity. The log (debug) can be seen via the link below. Have been trying for hours but not success.

https://docs.google.com/document/d/1Jgn3LXeBwNKABXBDFnQvoFN8ekpvQV0st2eZ9j84PyY/edit?usp=sharing

The CC2530 i use is:
https://nl.aliexpress.com/item/CC2530-Zigbee-UART-Wireless-Core-Board-Development-Board-CC2530F256-Serial-Port-Wireless-Module-24MHz/32767470382.html

With:
https://nl.aliexpress.com/item/ZigBee-CC2530-Sensor-Node-Baseboard-Functional-Module-Expansion-Board-USB-Port-24MHz-256KB/32767498002.html

Koenkk, 2018-06-30:
Can you try repairing the device?

1981nick1982, 2018-06-30:
@Koenkk I have reinstalled home assistant with a backup before is used zigbee2mqtt and installed everything from scatch without old data from the cc2531 that i have used earlier. I have repaired one of the temperature senors but still the only info i got is the pressure.

1981nick1982, 2018-06-30:
Here is the new log https://docs.google.com/document/d/1mgcvilOgsXE3a7FNrizDup8033rRrAj4LUdifogASgg/edit?usp=sharing

Koenkk, 2018-06-30:
Are you sure these are the correct temperature and humidity values? Can you do some more tests with this? (e.g. blowing into the sensor should increase humidity to 90+ %)

The cluster is `genBasic` while for humidity/temperature `msRelativeHumidity` and `msTemperatureMeasurement` should be used.

Koenkk, 2018-07-01:
Can you try with the following firmware: https://drive.google.com/open?id=1EXQu6JQIUkamt-1bIsCmOV5FmjCFLQvl

1981nick1982, 2018-07-01:
This version of the firmware is not working. When i flash the board only 1 of the 4 leds on de baseboard light up. This is the error i got:

`[Info] Configuration file found. Will overwrite configurable               fields with values from add-on configuration
[Info] Configuration written to /share/zigbee2mqtt/configuration.yaml
[2018-07-01 13:03:59] PM2 log: Launching in no daemon mode
[2018-07-01 13:03:59] PM2 log: Starting execution sequence in -fork mode- for app name:npm id:0
[2018-07-01 13:03:59] PM2 log: App name:npm id:0 online
> zigbee2mqtt@0.0.0 start /app
> node index.js
2018-7-1 13:04:07 DEBUG Using zigbee-shepherd with settings: '{"net":{"panId":6754,"channelList":[11]},"dbPath":"/share/zigbee2mqtt/database.db","sp":{"baudrate":115200,"rtscts":true}}'
2018-7-1 13:04:07 INFO Starting zigbee2mqtt version 0.0.0 (commit #7cec5f2)
2018-7-1 13:04:07 INFO Starting zigbee-shepherd
2018-7-1 13:04:10 INFO Error while starting zigbee-shepherd, attemping to fix... (takes 60 seconds)
2018-7-1 13:05:10 INFO Starting zigbee-shepherd
2018-7-1 13:05:13 ERROR Error while starting zigbee-shepherd!
2018-7-1 13:05:13 ERROR Press the reset button on the stick (the one closest to the USB) and start again
2018-7-1 13:05:13 ERROR Failed to start
	{"message":"request timeout","stack":"Error: request timeout\n    at CcZnp.<anonymous> (/app/node_modules/cc-znp/lib/ccznp.js:254:22)\n    at Object.onceWrapper (events.js:315:30)\n    at emitOne (events.js:116:13)\n    at CcZnp.emit (events.js:211:7)\n    at Timeout.<anonymous> (/app/node_modules/cc-znp/lib/ccznp.js:233:18)\n    at ontimeout (timers.js:475:11)\n    at tryOnTimeout (timers.js:310:5)\n    at Timer.listOnTimeout (timers.js:270:5)"}`

Koenkk, 2018-07-01:
@ptvoinfo do you know how to compile the ZNP firmware for CC2530?

Koenkk, 2018-07-03:
@1981nick1982 can you try with this firmware: https://github.com/Koenkk/Z-Stack-firmware/tree/master/coordinator/CC2530/bin

1981nick1982, 2018-07-03:
@koenkk  I am at work now but when i'm home I will try it.

Koenkk, 2018-07-04:
@kirovilya does this firmware work for you?

kirovilya, 2018-07-04:
@Koenkk I can not check this firmware (not have cc2530 now). Early I used this firmware https://github.com/kirovilya/files/blob/master/CC2530ZNP-Pro-Secure_LinkKeyJoin.hex

@1981nick1982 On my baseboard with cc2530+cc2591 1 red led is on too. And it works!

lolorc, 2018-07-07:
been moving from cc2531 to cc2530+cc2590, I've beenencountering the same kind of issues.

1981nick1982, 2018-07-09:
I'm on vacation at the moment. When i'm back home this saterday i will give it another try.

closed 2018-08-22 by 1981nick1982