Koenkk/zigbee2mqtt #166: Support for other Zigbee adapters

trsqr, 2018-06-30:
Instead of CC2530, would it be possible to use another Zigbee adapter, such as the Telegesis ETRX357?

I'm not sure how standard the interfaces are here - is the amount of work significant or is it just a small adaptation?

Some references:
https://github.com/vpetrigo/etrx35x-utils

http://www.wless.ru/files/ZigBee/Firmware/TG-ETRXn-R309-Commands.pdf

https://github.com/zsmartsystems/com.zsmartsystems.zigbee/blob/master/com.zsmartsystems.zigbee.dongle.telegesis/src/main/java/com/zsmartsystems/zigbee/dongle/telegesis/ZigBeeDongleTelegesis.java

Koenkk, 2018-06-30:
Current only the texas instruments z-stack 1.2 is supported. 

It will require a lot of work to support more devices. This needs to be rewritten: https://github.com/zigbeer/cc-znp

trsqr, 2018-06-30:
Ok, thanks for the answer!

closed 2018-06-30 by trsqr