Koenkk/zigbee2mqtt #167: WXKG12LM and Home Assistant - button state not expiring

tomusher, 2018-06-30:
I've just set up a few Xiaomi devices with this project - I'm really impressed with it so far!

I am currently having an issue with the Xiaomi WXKG12LM square button and Home Assistant. Everything seems to be working fine, I'm seeing the button state update to 'single', 'double', etc. in HA, and I would expect the state to reset to being blank after a short period - but it remains in the last known state.

This means I can't use a single press to turn something on, then another single press to turn it off, as the second press won't trigger a HA `state_changed` event.

Happy to provide any more information on this - I may just be missing something!

Koenkk, 2018-06-30:
You need to listen to the MQTT events instead of the state change. See https://github.com/Koenkk/zigbee2mqtt/wiki/Integrating-with-Home-Assistant *To respond to button clicks (e.g. WXKG01LM) you can use the following Home Assistant configuration:*

closed 2018-06-30 by Koenkk

tomusher, 2018-06-30:
Excellent thanks. I was building the automation with Node-RED but can't see a way to use the HA package to trigger on an MQTT message.

For anyone else looking, I've just skipped HA and connected Node-RED directly to the mqtt broker:

[image]

ciotlosm, 2018-06-30:
>Excellent thanks. I was building the automation with Node-RED but can't see a way to use the HA package to trigger on an MQTT message. ...

@tomusher there is a need for a PR to allow events to trigger on mqtt messages 

<sub>Sent with <a href="http://githawk.com">GitHawk</a></sub>