Koenkk/zigbee2mqtt #169: Add "ban" option to Devices if we Remove them.

tb-killa, 2018-06-30:
We should extend zigbee2mqtt to allow "device ban" because if you allow joining (default one) and then there is a device near you like something from your neighbour who is in joining mode too, its paired.

Unfortunately, I have just such a device and have to check if it connects flawed i have manually removed them and disable the "allowed joining".

tb-killa, 2018-07-01:
We could use the https://github.com/zigbeer/zigbee-shepherd/wiki#API_remove function.

tb-killa, 2018-07-05:
@Koenkk We should add json encode and parse ```friendly_name``` and the optional state ```reJoin``` with true or false (allow rejoin / ban).

closed 2018-10-22 by tb-killa

ryanbeaton, 2019-01-21:
Just noting here that blacklist has now been implemented.

https://github.com/Koenkk/zigbee2mqtt/issues/816#issuecomment-452394088