Koenkk/zigbee2mqtt #170: Total newbie - outbound mqtt message to turn on/off switch

eflemingtra, 2018-07-01:
Hi guys,

I'm using OpenHAB and have successfully got zigbee2mqtt working. First was a Xiaomi temperature sensor which successfully gives me temperature and humidty readings via MQTT. I've also paired a Xiaomi smart plug.

I'm struggling to understand the outbound message in JSON format required to turn the plug on and off. Here's what I use in OpenHAB:

> Switch Xiaomi_Switch_State "In Line Fan Switch" {mqtt=">[broker:zigbee2mqtt/0x00158d000215fcfb/set:state:ON]"}

I'm pretty sure the /set:state:ON is incorrect. I'm using MQTT Lens to observe the output of my commands and the message is being sent, but nothing is happening.

Koenkk, 2018-07-01:
The topic should be `zigbee2mqtt/0x00158d000215fcfb/set` with payload `{"state": "ON"}`

tb-killa, 2018-07-24:
@eflemingtra Do you test ? Does it works now ? Could we close this issue ?

jboeddeker, 2018-07-29:
This is my openHAB item definition for the on/off switches. Works like a charm.

`Switch AZLJ_TOGGLE "AZ Laserjet"   <poweroutlet> (gDevices){mqtt=">[broker:zigbee/AZLJ/set:command:*:MAP(json_onoff.map)],<[broker:zigbee/AZLJ:state:JSONPATH($.state)]"}`

This is the file "json_onoff.map":
`ON={ "state" : "ON" }
OFF={ "state" : "OFF" }
1={ "state" : "ON" }
0={ "state" : "OFF" }`

mpuff, 2018-08-01:
@jboeddeker  i have tried what you did, but for me it doesnt work.
my item in zigbee2mqtt called Tratfri_GU10_Gaestezimmer_1
so i changed the item to:
Switch Tratfri_GU10_Gaestezimmer_1                          "Spot Gästezimmer"                              <light>         (FF_Child, Lights, Lampe)                   {mqtt=">[mosquitto:zigbee2mqtt/Tratfri_GU10_Gaestezimmer_1/set:command:*:MAP(json_onoff.map)],<[mosquitto:zigbee2mqtt/Tratfri_GU10_Gaestezimmer_1:state:JSONPATH($.state)]"}

but it doesnt work.
i also created the map file with the same name.

jboeddeker, 2018-08-01:
This is my Ikea Tradfri E27 1000 lm bulb. Works as expected. 

`Switch SZWL_TOGGLE "Wand [%s]"         <light> (glightSwitches) {mqtt=">[broker:zigbee/SZWL/set:command:*:MAP(json_onoff.map)],<[broker:zigbee/SZWL:state:JSONPATH($.state)]", expire="120m,command=OFF"}`

I'm sorry, but i don't own any GU10 bulbs. But if you see the correct message in mqtt, the issue is surely not in the openHAB definition.

Did you restart zigbee2mqtt after setting the friendly name? Could you try a different friendly name? X perhaps. Just to make sure nothing like character encoding or unprintable characters play with you.

closed 2018-10-18 by stale[bot]