Koenkk/zigbee2mqtt #172: reflash CC2531 after not work, now pairing problem with xiaomi

arteck, 2018-07-01:
i have reflash the usb stick.. and now i cant repairing all my xiaomi devices.. 
i used the new flash file https://github.com/Koenkk/Z-Stack-firmware/tree/master/coordinator/CC2531/bin
its 10 days old....

ciotlosm, 2018-07-01:
@arteck have you reset the devices? What errors do you get?

arteck, 2018-07-01:
My devices were not available.. the cc2531 did not respond...

ciotlosm, 2018-07-01:
Try pressing the reset button on the CC2531

arteck, 2018-07-01:
which one is it ??? 
left or right from usb port

ciotlosm, 2018-07-01:
Maybe try complete host reboot as well (let it without power for a minute).

arteck, 2018-07-01:
hue and osram works..the problem is only xiaomi..

ciotlosm, 2018-07-01:
Can you post your version on startup of zigbee2mqtt?

arteck, 2018-07-01:
2018-7-1 19:05:02 INFO Starting zigbee2mqtt version 0.0.0 (commit #7cec5f2)
2018-7-1 19:05:02 INFO Starting zigbee-shepherd
2018-7-1 19:05:03 INFO zigbee-shepherd started

ciotlosm, 2018-07-01:
Your zigbee2mqtt is latest version. Probably something changed in zcl dependency. @Koenkk maybe latest release broke the xiaomi patch.

Koenkk, 2018-07-01:
Could you try removing the battery from the Xiaomi devices for a few seconds, and then repeat the pairing procedure?

arteck, 2018-07-02:
I check that .. but i think the flash file ist buggy
i found other files here https://github.com/kirovilya/files 
any ideas ??

Koenkk, 2018-07-02:
@arteck can you try the CC2531ZNP-Pro-Secure_LinkKeyJoin.hex from here: https://github.com/mtornblad/zstack-1.2.2a.44539/tree/master/CC2531

arteck, 2018-07-02:
I'll contact you

arteck, 2018-07-02:
CC2531ZNP-Pro-Secure_LinkKeyJoin.hex work... 
I look for how long .....

closed 2018-07-02 by arteck