Koenkk/zigbee2mqtt #175: Unable to pair with CC2531 Router

aidbish, 2018-07-03:
Hi
i have bought some off the CC2531 devices as listed on the supported devices, flashing them with the coordinator firmware they work flawlessly, but if i flash with the router firmware they are never paired with the main coordinator device, nothing is listed in the logs other than the endpoint that are already paired

is there any extra v0odoo magic required to get the router to pair?

aidbish, 2018-07-03:
Resolved
i had channel set in configuration.yaml to 20 to try and overcome interference with wifi, the router must have a default channel set. removing from config allowed  it to join

closed 2018-07-03 by aidbish

wilfredsmit, 2018-07-07:
@aidbish When you change the channel of the coordinator, does the sensors, lights, etc follow these change? Or do you have to make each sensor, lights, etc. powerless (restart)? I find it very tricky to play with channels and the already paired devices.

aidbish, 2018-07-08:
@wilfredsmit you have to restart the server and then re-pair the devices to the coordinator if the channel is changed