Koenkk/zigbee2mqtt #176: Device support: 'Nue' (Feibit) smart switch range

highground88, 2018-07-03:
Hi all,

I think I've hit a snag and would appreciate some tips if someone could? It's been quite a mission just to get this far, I'm not very experienced in linux and even getting a machine up and running wasn't much fun.

I've taken two " 'Nue' 2 gang smart switches " off the supplied 'Nue' hub, and believe I have updated the 'devices.js' appropriately as they are being detected (copy of terminal below). Here's the model info lines from both entries in the 'devices' file:

`zigbeeModel: ['FB56+ZSW1HKJ2.5']`
and
`zigbeeModel: ['FB56+ZSW1HKJ1.7']`

Funnily enough I think these are the same model 2 switch devices.. Perhaps one is a later revision? Regardless, this is the two switch model. I am yet to test on the 1 switch, 3 switch, dimmer and also the 1 and four button 'scene' switches I have just installed. [PRODUCT LINK](http://www.3asmarthome.com/products)

Once the 'devices.js' file is updated, and zigbee2mqtt is restarted, I'm struggling to see any immediate messages when the switches are turned on and off - there doesn't seem to be anything coming through, and no sort of pattern reflecting how I've switched them on/off.

Where should I go next? 

I don't know if this is helpful or not, but others seem to have been attaching these switches to the SmartThigns hub, via a similar detection method which is supplied by the distributor - would this be useful? [LINK TO PDF](https://storage.googleapis.com/wzukusers/user-22742145/documents/5a04d3798019bAhKCYof/How%20to%20add%20Nue%20ZigBee%20Switches%20to%20SmartThings%20Hub.pdf)

Thanks!

```
user@debian:/opt/zigbee2mqtt$ sudo DEBUG=* npm start

> zigbee2mqtt@0.0.0 start /opt/zigbee2mqtt
> node index.js

2018-7-3 11:49:30 DEBUG Using zigbee-shepherd with settings: '{"net":{"panId":6754,"channelList":[11]},"dbPath":"/opt/zigbee2mqtt/data/database.db","sp":{"baudrate":115200,"rtscts":true}}'
2018-7-3 11:49:30 INFO Starting zigbee2mqtt version 0.0.0 (commit #7cec5f2)
2018-7-3 11:49:30 INFO Starting zigbee-shepherd
  zigbee-shepherd:init zigbee-shepherd booting... +0ms
  cc-znp The serialport /dev/ttyACM0 is opened. +5ms
  zigbee-shepherd:request REQ --> SYS:osalNvRead +0ms
  cc-znp:SREQ --> SYS:osalNvRead, { id: 3840, offset: 0, len: 1, value: [ 85 ] } +1ms
  serialport write 8 bytes of data +3ms
  cc-znp:SRSP <-- SYS:osalNvRead, { status: 0, len: 1, value: <Buffer 55> } +13ms
  zigbee-shepherd:request RSP <-- SYS:osalNvRead, status: 0 +1ms
  zigbee-shepherd:request REQ --> SAPI:readConfiguration +11ms
  cc-znp:SREQ --> SAPI:readConfiguration, { configid: 131, len: 2, value: [ 98, 26 ] } +0ms
  serialport write 6 bytes of data +1ms
  cc-znp:SRSP <-- SAPI:readConfiguration, { status: 0, configid: 131, len: 2, value: <Buffer 62 1a> } +9ms
  zigbee-shepherd:request RSP <-- SAPI:readConfiguration, status: 0 +0ms
  zigbee-shepherd:request REQ --> SAPI:readConfiguration +11ms
  cc-znp:SREQ --> SAPI:readConfiguration, { configid: 132, len: 4, value: [ 0, 8, 0, 0 ] } +0ms
  serialport write 6 bytes of data +0ms
  cc-znp:SRSP <-- SAPI:readConfiguration, { status: 0, configid: 132, len: 4, value: <Buffer 00 08 00 00> } +16ms
  zigbee-shepherd:request RSP <-- SAPI:readConfiguration, status: 0 +0ms
  zigbee-shepherd:request REQ --> SAPI:readConfiguration +10ms
  cc-znp:SREQ --> SAPI:readConfiguration, { configid: 98, len: 16, value: [ 1, 3, 5, 7, 9, 11, 13, 15, 0, 2, 4, 6, 8, 10, 12, 13 ] } +0ms
  serialport write 6 bytes of data +1ms
  cc-znp:SRSP <-- SAPI:readConfiguration, { status: 0, configid: 98, len: 16, value: <Buffer 01 03 05 07 09 0b 0d 0f 00 02 04 06 08 0a 0c 0d> } +26ms
  zigbee-shepherd:request RSP <-- SAPI:readConfiguration, status: 0 +1ms
  zigbee-shepherd:request REQ --> SAPI:readConfiguration +11ms
  cc-znp:SREQ --> SAPI:readConfiguration, { configid: 99, len: 1, value: [ 0 ] } +1ms
  serialport write 6 bytes of data +1ms
  cc-znp:SRSP <-- SAPI:readConfiguration, { status: 0, configid: 99, len: 1, value: <Buffer 00> } +20ms
  zigbee-shepherd:request RSP <-- SAPI:readConfiguration, status: 0 +0ms
  zigbee-shepherd:request REQ --> SAPI:getDeviceInfo +14ms
  cc-znp:SREQ --> SAPI:getDeviceInfo, { param: 0 } +0ms
  serialport write 6 bytes of data +1ms
  cc-znp:SRSP <-- SAPI:getDeviceInfo, { param: 0, value: <Buffer 09 04 00 ff fd 00 00 3b> } +8ms
  zigbee-shepherd:request RSP <-- SAPI:getDeviceInfo +1ms
  zigbee-shepherd:init Now the ZNP is a coordinator. +1ms
  zigbee-shepherd:request REQ --> SAPI:getDeviceInfo +1ms
  cc-znp:SREQ --> SAPI:getDeviceInfo, { param: 0 } +0ms
  serialport write 6 bytes of data +1ms
  cc-znp:SRSP <-- SAPI:getDeviceInfo, { param: 0, value: <Buffer 09 04 00 ff fd 00 00 3b> } +7ms
  zigbee-shepherd:request RSP <-- SAPI:getDeviceInfo +1ms
  zigbee-shepherd:request REQ --> SAPI:getDeviceInfo +1ms
  cc-znp:SREQ --> SAPI:getDeviceInfo, { param: 1 } +1ms
  serialport write 6 bytes of data +1ms
[... 399 lines cut ...]
2018-7-3 11:50:02 DEBUG Recieved zigbee message with data "0x00124b0014afe1fb"
2018-7-3 11:50:02 INFO Device incoming...
2018-7-3 11:50:02 INFO MQTT publish, topic: 'zigbee2mqtt/bridge/log', payload: '{"type":"pairing","message":"device incoming"}'
2018-7-3 11:50:02 DEBUG Recieved zigbee message with data "online"
2018-7-3 11:50:30 DEBUG Check online 0x00124b0019e14fa6
  zigbee-shepherd:request REQ --> ZDO:nodeDescReq +28s
  cc-znp:SREQ --> ZDO:nodeDescReq, { dstaddr: 25438, nwkaddrofinterest: 25438 } +0ms
  serialport write 9 bytes of data +2ms
2018-7-3 11:50:30 DEBUG Check online 0x00124b0014afe1fb
  zigbee-shepherd:request REQ --> ZDO:nodeDescReq +2ms
  cc-znp:SRSP <-- ZDO:nodeDescReq, { status: 0 } +10ms
  cc-znp:SREQ --> ZDO:nodeDescReq, { dstaddr: 49994, nwkaddrofinterest: 49994 } +0ms
  serialport write 9 bytes of data +1ms
  cc-znp:SRSP <-- ZDO:nodeDescReq, { status: 0 } +11ms
  cc-znp:AREQ <-- ZDO:nodeDescRsp, { srcaddr: 25438, status: 0, nwkaddr: 25438, logicaltype_cmplxdescavai_userdescavai: 1, apsflags_freqband: 64, maccapflags: 142, manufacturercode: 0, maxbuffersize: 80, maxintransfersize: 160, servermask: 0, maxouttransfersize: 160, descriptorcap: 0 } +32ms
  zigbee-shepherd:msgHdlr IND <-- ZDO:nodeDescRsp +0ms
  cc-znp:AREQ <-- ZDO:nodeDescRsp, { srcaddr: 49994, status: 0, nwkaddr: 49994, logicaltype_cmplxdescavai_userdescavai: 1, apsflags_freqband: 64, maccapflags: 142, manufacturercode: 0, maxbuffersize: 80, maxintransfersize: 160, servermask: 0, maxouttransfersize: 160, descriptorcap: 0 } +1ms
  zigbee-shepherd:msgHdlr IND <-- ZDO:nodeDescRsp +0ms
^C2018-7-3 11:51:29 INFO MQTT publish, topic: 'zigbee2mqtt/bridge/state', payload: 'offline'
2018-7-3 11:51:29 INFO Disconnecting from MQTT server
  zigbee-shepherd:request REQ --> ZDO:mgmtPermitJoinReq +59s
  cc-znp:SREQ --> ZDO:mgmtPermitJoinReq, { addrmode: 15, dstaddr: 65532, duration: 0, tcsignificance: 0 } +1ms
  serialport write 10 bytes of data +1ms

```

Koenkk, 2018-07-04:
Can you:

- Provide the entry of this device from `data/database.db` (there are probably multiple)
- Put the following in your `devices.js` and check if you see any `No converter available` messages when pressing one of the buttons on the switch:

```js
    // Nue
    {
        zigbeeModel: ['FB56+ZSW1HKJ2.5', 'FB56+ZSW1HKJ1.7'],
        model: 'TODO',
        vendor: 'Nue',
        description: 'Smart light switch - 2 gang',
        supports: 'TODO',
        fromZigbee: [],
        toZigbee: [],
        configure: (ieeeAddr, shepherd, coordinator, callback) => {
            const device = shepherd.find(ieeeAddr, 1);
            execute(device, [(cb) => device.bind('genOnOff', coordinator, cb)], callback);
        },
    },
```

highground88, 2018-07-04:
Thanks @Koenkk

I will try this tonight and let you know how I get on, and will gather all the other model details I can for the other devices

highground88, 2018-07-05:
@Koenkk ,

This is the database.db contents. Below I will post the shepherd results, but there were not any "no converter available" messages. I hope this helps and appreciate your time.
```
{"id":1,"type":"Coordinator","ieeeAddr":"0x00124b0014bdea20","nwkAddr":0,"manufId":0,"epList":[1,2,3,4,5,6],"status":"online","joinTime":1530775671,"endpoints":{"1":{"profId":260,"epId":1,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{}},"2":{"profId":257,"epId":2,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{}},"3":{"profId":261,"epId":3,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{}},"4":{"profId":263,"epId":4,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{}},"5":{"profId":264,"epId":5,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{}},"6":{"profId":265,"epId":6,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{}}},"_id":"Iir7FWKHwlHAGayt"}
{"id":2,"type":"Router","ieeeAddr":"0x00124b0014afe1fb","nwkAddr":1772,"manufId":0,"manufName":"Feibit Inc co.  ","powerSource":"Mains (single phase)","modelId":"FB56+ZSW1HKJ1.7","epList":[16,17],"status":"online","joinTime":1530775707,"endpoints":{"16":{"profId":260,"epId":16,"devId":2,"inClusterList":[0,4,5,6],"outClusterList":[0],"clusters":{"genBasic":{"dir":{"value":3},"attrs":{}},"genGroups":{"dir":{"value":1},"attrs":{}},"genScenes":{"dir":{"value":1},"attrs":{}},"genOnOff":{"dir":{"value":1},"attrs":{"onOff":0}}}},"17":{"profId":260,"epId":17,"devId":2,"inClusterList":[0,4,5,6],"outClusterList":[0],"clusters":{"genBasic":{"dir":{"value":3},"attrs":{"16393":1287651328,"hwVersion":0,"manufacturerName":"Feibit Inc co.  ","modelId":"FB56+ZSW1HKJ1.7","dateCode":"20161030        ","powerSource":1,"locationDesc":"                ","physicalEnv":0,"deviceEnabled":1}},"genGroups":{"dir":{"value":1},"attrs":{}},"genScenes":{"dir":{"value":1},"attrs":{}},"genOnOff":{"dir":{"value":1},"attrs":{"onOff":0}}}}},"_id":"QvXnFnosVrO4lSis"}
{"$$indexCreated":{"fieldName":"id","unique":true,"sparse":false}}
{"id":1,"type":"Coordinator","ieeeAddr":"0x00124b0014bdea20","nwkAddr":0,"manufId":0,"epList":[1,2,3,4,5,6],"status":"online","joinTime":1530777270,"endpoints":{"1":{"profId":260,"epId":1,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{}},"2":{"profId":257,"epId":2,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{}},"3":{"profId":261,"epId":3,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{}},"4":{"profId":263,"epId":4,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{}},"5":{"profId":264,"epId":5,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{}},"6":{"profId":265,"epId":6,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{}}},"_id":"Iir7FWKHwlHAGayt"}
{"id":2,"type":"Router","ieeeAddr":"0x00124b0014afe1fb","nwkAddr":1772,"manufId":0,"manufName":"Feibit Inc co.  ","powerSource":"Mains (single phase)","modelId":"FB56+ZSW1HKJ1.7","epList":[16,17],"status":"offline","joinTime":null,"endpoints":{"16":{"profId":260,"epId":16,"devId":2,"inClusterList":[0,4,5,6],"outClusterList":[0],"clusters":{"genBasic":{"dir":{"value":3},"attrs":{}},"genGroups":{"dir":{"value":1},"attrs":{}},"genScenes":{"dir":{"value":1},"attrs":{}},"genOnOff":{"dir":{"value":1},"attrs":{"onOff":0}}}},"17":{"profId":260,"epId":17,"devId":2,"inClusterList":[0,4,5,6],"outClusterList":[0],"clusters":{"genBasic":{"dir":{"value":3},"attrs":{"16393":1287651328,"hwVersion":0,"manufacturerName":"Feibit Inc co.  ","modelId":"FB56+ZSW1HKJ1.7","dateCode":"20161030        ","powerSource":1,"locationDesc":"                ","physicalEnv":0,"deviceEnabled":1}},"genGroups":{"dir":{"value":1},"attrs":{}},"genScenes":{"dir":{"value":1},"attrs":{}},"genOnOff":{"dir":{"value":1},"attrs":{"onOff":0}}}}},"_id":"QvXnFnosVrO4lSis"}
{"id":2,"type":"Router","ieeeAddr":"0x00124b0014afe1fb","nwkAddr":1772,"manufId":0,"manufName":"Feibit Inc co.  ","powerSource":"Mains (single phase)","modelId":"FB56+ZSW1HKJ1.7","epList":[16,17],"status":"online","joinTime":1530777271,"endpoints":{"16":{"profId":260,"epId":16,"devId":2,"inClusterList":[0,4,5,6],"outClusterList":[0],"clusters":{"genBasic":{"dir":{"value":3},"attrs":{}},"genGroups":{"dir":{"value":1},"attrs":{}},"genScenes":{"dir":{"value":1},"attrs":{}},"genOnOff":{"dir":{"value":1},"attrs":{"onOff":0}}}},"17":{"profId":260,"epId":17,"devId":2,"inClusterList":[0,4,5,6],"outClusterList":[0],"clusters":{"genBasic":{"dir":{"value":3},"attrs":{"16393":1287651328,"hwVersion":0,"manufacturerName":"Feibit Inc co.  ","modelId":"FB56+ZSW1HKJ1.7","dateCode":"20161030        ","powerSource":1,"locationDesc":"                ","physicalEnv":0,"deviceEnabled":1}},"genGroups":{"dir":{"value":1},"attrs":{}},"genScenes":{"dir":{"value":1},"attrs":{}},"genOnOff":{"dir":{"value":1},"attrs":{"onOff":0}}}}},"_id":"QvXnFnosVrO4lSis"}
{"id":2,"type":"Router","ieeeAddr":"0x00124b0014afe1fb","nwkAddr":1772,"manufId":0,"manufName":"Feibit Inc co.  ","powerSource":"Mains (single phase)","modelId":"FB56+ZSW1HKJ1.7","epList":[16,17],"status":"online","joinTime":1530777271,"endpoints":{"16":{"profId":260,"epId":16,"devId":2,"inClusterList":[0,4,5,6],"outClusterList":[0],"clusters":{"genBasic":{"dir":{"value":3},"attrs":{}},"genGroups":{"dir":{"value":1},"attrs":{}},"genScene{"id":1,"type":"Coordinator","ieeeAddr":"0x00124b0014bdea20","nwkAddr":0,"manufId":0,"epList":[1,2,3,4,5,6],"status":"online","joinTime":1530775671,"endpoints":{"1":{"profId":260,"epId":1,"devId":5,"i [... cut ...]
  zigbee-shepherd Device: 0x00124b0014afe1fb join the network. +5ms
2018-7-5 07:28:57 DEBUG Recieved zigbee message with data "0x00124b0014afe1fb"
2018-7-5 07:28:57 INFO Device incoming...
2018-7-5 07:28:57 INFO MQTT publish, topic: 'zigbee2mqtt/bridge/log', payload: '{"type":"pairing","message":"device incoming"}'
2018-7-5 07:28:57 DEBUG Recieved zigbee message with data "online"
2018-7-5 07:29:51 DEBUG Check online 0x00124b0014afe1fb
  zigbee-shepherd:request REQ --> ZDO:nodeDescReq +54s
  zigbee-shepherd:msgHdlr IND <-- ZDO:nodeDescRsp +40ms

```

Koenkk, 2018-07-05:
Thanks, can you check with this code:

```js
    // Nue
    {
        zigbeeModel: ['FB56+ZSW1HKJ2.5', 'FB56+ZSW1HKJ1.7'],
        model: 'TODO',
        vendor: 'Nue',
        description: 'Smart light switch - 2 gang',
        supports: 'TODO',
        fromZigbee: [],
        toZigbee: [],
        configure: (ieeeAddr, shepherd, coordinator, callback) => {
            const ep16 = shepherd.find(ieeeAddr, 16);
            execute(ep16, [(cb) => ep16.bind('genOnOff', coordinator, cb)], () => {
                const ep17 = shepherd.find(ieeeAddr, 17);
                execute(ep17, [(cb) => ep17.bind('genOnOff', coordinator, cb)], callback);
            });
        },
    },
```

Koenkk, 2018-07-06:
Can you try with this code: 

```js
    // Nue
    {
        zigbeeModel: ['FB56+ZSW1HKJ2.5', 'FB56+ZSW1HKJ1.7'],
        model: 'TODO',
        vendor: 'Nue',
        description: 'Smart light switch - 2 gang',
        supports: 'TODO',
        fromZigbee: [],
        toZigbee: [],
        configure: (ieeeAddr, shepherd, coordinator, callback) => {
            const ep16 = shepherd.find(ieeeAddr, 16);
            const ep17 = shepherd.find(ieeeAddr, 17);

            const actions = [
                (cb) => ep16.bind('genOnOff', coordinator, cb),
                (cb) => ep17.bind('genOnOff', coordinator, cb),
                (cb) => ep16.report('genOnOff', 'onOff', 0, 1000, 0, cb),
                (cb) => ep17.report('genOnOff', 'onOff', 0, 1000, 0, cb),
            ];

            execute(ep16, actions, callback);
        },
    },
```

Koenkk, 2018-07-07:
Can you try with:

```js
    // Nue
    {
        zigbeeModel: ['FB56+ZSW1HKJ2.5', 'FB56+ZSW1HKJ1.7'],
        model: 'TODO',
        vendor: 'Nue',
        description: 'Smart light switch - 2 gang',
        supports: 'TODO',
        fromZigbee: [],
        toZigbee: [],
        configure: (ieeeAddr, shepherd, coordinator, callback) => {
            const ep16 = shepherd.find(ieeeAddr, 16);
            const ep17 = shepherd.find(ieeeAddr, 17);
            const cfg = {direction: 0, attrId: 0, dataType: 16, minRepIntval: 0, maxRepIntval: 10, repChange: 0};

            const actions = [
                (cb) => ep16.bind('genOnOff', coordinator, cb),
                (cb) => ep16.foundation('genOnOff', 'configReport', [cfg], foundationCfg, cb),
                (cb) => ep17.bind('genOnOff', coordinator, cb),
                (cb) => ep17.foundation('genOnOff', 'configReport', [cfg], foundationCfg, cb),
            ];

            execute(ep16, actions, callback);
        },
    },
```

Koenkk, 2018-07-10:
Strange that still nothing is reported.

In the meantime somebody got a different Nue model working: https://github.com/Koenkk/zigbee-shepherd-converters/pull/29#issuecomment-403803514

highground88, 2018-08-12:
Hi @Koenkk ,

I haven't given up... yet! But I'm getting close I'm afraid, as this is all a bit beyond my skill level getting this far. I would really like this to work however as a lot of my plans for the smart side of our renovation rely on this, and I mistakenly thought this would be easy (hah!). I figured that if Alexa can do it then I could too!

I'm wondering though if this will be of benefit:

https://community.smartthings.com/t/help-with-new-dh-for-aust-3-gang-light-switch-catchall-included/101434/86?

The SmartThings guys seem to have most of this figured out for HomeAssistant/MQTT and utilising the Smart Things hub - I'm just hoping that the device handlers for that hub may in fact be ported over to Zigbee2MQTT easily in this case? 

As always, your help and great interface is very much appreciated!

Koenkk, 2018-08-13:
Could you try the following code? According to the smartthings docs these are the commands that are executed in the configure method:

```js
    // Nue
    {
        zigbeeModel: ['FB56+ZSW1HKJ2.5', 'FB56+ZSW1HKJ1.7'],
        model: 'TODO',
        vendor: 'Nue',
        description: 'Smart light switch - 2 gang',
        supports: 'TODO',
        fromZigbee: [],
        toZigbee: [],
        configure: (ieeeAddr, shepherd, coordinator, callback) => {
            const ep16 = shepherd.find(ieeeAddr, 16);
            const cfg = {direction: 0, attrId: 0, dataType: 16, minRepIntval: 0, maxRepIntval: 10, repChange: 0};

            const actions = [
                (cb) => ep16.read('genOnOff', 'onOff', cb),
                (cb) => ep16.foundation('genOnOff', 'configReport', [cfg], foundationCfg, cb),
            ];

            execute(ep16, actions, callback);
        },
    },
```

highground88, 2018-08-16:
Hi @Koenkk , sorry for the delay in trying this, and thanks again for continuing to help. I decided to try reflashing the firmware on the CC2531 first to see if that made a difference (I'm currently using your latest with the large buffer - let me know if that's a bad idea!)

I have used the above code you provided for the devices.js file.

The good news is that I've noticed some new lines in the log - but at this time cycling the switches does not result in immediate messages, however I have seen them appear once seemingly out of the blue:

**Log.txt** file:
```
2018-08-16T11:47:31.668Z - debug: Using zigbee-shepherd with settings: '{"net":{"panId":6754,"channelList":[11]},"dbPath":"/opt/zigbee2mqtt/data/database.db","sp":{"baudrate":115200,"rtscts":true}}'
2018-08-16T11:47:31.690Z - info: Starting zigbee2mqtt version 0.1.2 (commit #7dd1c75)
2018-08-16T11:47:31.690Z - info: Starting zigbee-shepherd
2018-08-16T11:47:32.079Z - info: zigbee-shepherd started
2018-08-16T11:47:32.080Z - debug: zigbee-shepherd info: {"enabled":true,"net":{"state":"Coordinator","channel":11,"panId":"0x1a62","extPanId":"0xdddddddddddddddd","ieeeAddr":"0x00124b0014bdea20","nwkAddr":0},"firmware":{"transportrev":2,"product":0,"version":"2.6.3","revision":20180815},"startTime":1534420052,"joinTimeLeft":0}
2018-08-16T11:47:32.082Z - info: Currently 1 devices are joined:
2018-08-16T11:47:32.082Z - info: 0x00124b0014afe1fb (0x00124b0014afe1fb): TODO - Nue Smart light switch - 2 gang (Router)
2018-08-16T11:47:32.084Z - warn: `permit_join` set to  `true` in configuration.yaml.
2018-08-16T11:47:32.084Z - warn: Allowing new devices to join.
2018-08-16T11:47:32.085Z - warn: Set `permit_join` to `false` once you joined all devices.
2018-08-16T11:47:32.085Z - info: Zigbee: allowing new devices to join.
2018-08-16T11:47:32.086Z - info: Connecting to MQTT server at mqtt://192.168.0.101
2018-08-16T11:47:32.144Z - info: zigbee-shepherd ready
2018-08-16T11:47:32.151Z - info: Connected to MQTT server
2018-08-16T11:47:32.151Z - info: MQTT publish, topic: 'zigbee2mqtt/bridge/state', payload: 'online'
**2018-08-16T11:47:32.447Z - debug: Recieved zigbee message with data {"cid":"genOnOff","data":{"onOff":1}}
2018-08-16T11:47:32.447Z - warn: No converter available for 'TODO' with cid 'genOnOff', type 'devChange' and data '{"cid":"genOnOff","data":{"onOff":1}}'**
2018-08-16T11:47:32.447Z - warn: Please see: https://github.com/Koenkk/zigbee2mqtt/wiki/How-to-support-new-devices.
2018-08-16T11:47:32.810Z - info: Succesfully configured 0x00124b0014afe1fb
2018-08-16T11:48:02.426Z - debug: Recieved zigbee message with data "0x00124b0014afe1fb"
2018-08-16T11:48:02.483Z - debug: Recieved zigbee message with data "0x00124b0014afe1fb"
2018-08-16T11:48:02.538Z - debug: Recieved zigbee message with data "0x00124b0014afe1fb"
2018-08-16T11:48:02.654Z - debug: Recieved zigbee message with data "0x00124b0014afe1fb"
2018-08-16T11:48:02.867Z - debug: Recieved zigbee message with data "0x00124b0014afe1fb"
2018-08-16T11:48:02.922Z - debug: Recieved zigbee message with data "0x00124b0014afe1fb"
2018-08-16T11:48:02.980Z - debug: Recieved zigbee message with data "0x00124b0014afe1fb"
2018-08-16T11:48:03.096Z - debug: Recieved zigbee message with data "0x00124b0014afe1fb"
2018-08-16T11:48:03.102Z - debug: Recieved zigbee message with data "0x00124b0014afe1fb"
2018-08-16T11:48:03.103Z - info: Device incoming...
2018-08-16T11:48:03.104Z - info: MQTT publish, topic: 'zigbee2mqtt/bridge/log', payload: '{"type":"pairing","message":"device incoming"}'
2018-08-16T11:48:03.106Z - debug: Recieved zigbee message with data "online"
2018-08-16T11:48:32.098Z - debug: Check online 0x00124b0014afe1fb
2018-08-16T11:49:32.113Z - debug: Check online 0x00124b0014afe1fb
2018-08-16T11:50:32.127Z - debug: Check online 0x00124b0014afe1fb
2018-08-16T11:51:32.141Z - debug: Check online 0x00124b0014afe1fb
2018-08-16T11:52:32.154Z - debug: Check online 0x00124b0014afe1fb
2018-08-16T11:53:32.169Z - debug: Check online 0x00124b0014afe1fb
2018-08-16T11:54:32.181Z - debug: Check online 0x00124b0014afe1fb
2018-08-16T11:55:32.194Z - debug: Check online 0x00124b0014afe1fb

```
This is a segment from the **debug terminal**:

```
2018-8-16 11:47:32 INFO zigbee-shepherd started
2018-8-16 11:47:32 DEBUG zigbee-shepherd info: {"enabled":true,"net":{"state":"Coordinator","channel":11,"panId":"0x1a62","extPanId":"0xdddddddddddddddd","ieeeAddr":"0x00124b0014bdea20","nwkAddr":0},"firmware":{"transportrev":2,"product":0,"version":"2.6.3","revision":20180815},"startTime":1534420052,"joinTimeLeft":0}
2018-8-16 11:47:32 INFO Currently 1 devices are joined:
2018-8-16 11:47:32 INFO 0x00124b0014afe1fb (0x00124b0014afe1fb): TODO - Nue Smart light switch - 2 gang (Router)
2018-8-16 11:47:32 WARN `permit_join` set to  `true` in configuration.yaml.
2018-8-16 11:47:32 WARN Allowing new devices to join.
2018-8-16 11:47:32 WARN Set `permit_join` to `false` once you joined all devices.
2018-8-16 11:47:32 INFO Zigbee: allowing new devices to join.
2018-8-16 11:47:32 INFO Connecting to MQTT server at mqtt://192.168.0.101
  zigbee-shepherd:request REQ --> ZDO:mgmtPermitJoinReq +53ms
2018-8-16 11:47:32 INFO zigbee-shepherd ready
  zigbee-shepherd:msgHdlr IND <-- ZDO:nodeDescRsp +14ms
spinlock: false []
  zigbee-shepherd:request REQ --> ZDO:nodeDescReq +2ms
2018-8-16 11:47:32 INFO Connected to MQTT server
2018-8-16 11:47:32 INFO MQTT publish, topic: 'zigbee2mqtt/bridge/state', payload: 'online'
[... cut ...]
I feel like I'm getting close!!! Does this look like it's going in the right direction now?
I'm just still using the one device (for reference the 2 gang switch model number is "HGZB-042") and I'm leaving the manufacturers hub switched off, and re-pairing the switch on each reboot.

Once I get one working, I'll be happy to find and properly format the settings data for the 5 or so related devices I have and create a PR, so hopefully this will help others also.

Any further steps forward would be great!! Thank you again.

Koenkk, 2018-08-16:
After `2018-08-16T11:47:32.810Z - info: Succesfully configured 0x00124b0014afe1fb`. Do you still see any `No converter available` messages when toggling the switch?

highground88, 2018-08-16:
No, that was the first time I've ever seen the "no converter available" message, and the associated payloads, eg.: 
2018-8-16 11:47:32 DEBUG Recieved zigbee message with data {"cid":"genOnOff","data":{"onOff":1}}
2018-8-16 11:47:32 WARN No converter available for 'TODO' with cid 'genOnOff', type 'devChange' and data '{"cid":"genOnOff","data":{"onOff":1}}'

butggling the switch does not produce any messages that I can see on the debug output or the log.

Koenkk, 2018-08-17:
The one report is because of the read in the configure method. Somehow, the device just refuses to report it's state change :(.

Could you try pairing your `FB56+ZSW1HKJ2.5` ?

highground88, 2018-08-18:
I believe this is the same FB56+ZSW1HKJ2.5, 

**Debug output:**
```
  zigbee-shepherd:af dispatchIncomingMsg(): type: dataConfirm, msg: [object Object] +25ms
  zigbee-shepherd:msgHdlr IND <-- AF:dataConfirm, transId: 22 +0ms
  zigbee-shepherd:af dispatchIncomingMsg(): type: incomingMsg, msg: [object Object] +20ms
  zigbee-shepherd:msgHdlr IND <-- AF:incomingMsg, transId: 0 +7ms
  zigbee-shepherd:af dispatchIncomingMsg(): type: zclIncomingMsg, msg: [object Object] +1ms
  zigbee-shepherd Identified Device: { manufacturer: Feibit Inc co.  , product: FB56+ZSW1HKJ2.5 } +0ms
  zigbee-shepherd:request REQ --> AF:dataRequest, transId: 23 +5ms
  zigbee-shepherd:request RSP <-- AF:dataRequest, status: 0 +14ms
  zigbee-shepherd:af dispatchIncomingMsg(): type: dataConfirm, msg: [object Object] +23ms
  zigbee-shepherd:msgHdlr IND <-- AF:dataConfirm, transId: 23 +0ms
  zigbee-shepherd:af dispatchIncomingMsg(): type: incomingMsg, msg: [object Object] +12ms
  zigbee-shepherd:msgHdlr IND <-- AF:incomingMsg, transId: 0 +2ms
  zigbee-shepherd:af dispatchIncomingMsg(): type: zclIncomingMsg, msg: [object Object] +1ms
2018-8-18 12:30:05 DEBUG Recieved zigbee message with data "0x00124b0019e14fa6"
2018-8-18 12:30:05 INFO Connecting with device...
2018-8-18 12:30:05 INFO MQTT publish, topic: 'zigbee2mqtt/bridge/log', payload: '{"type":"pairing","message":"connecting with device"}'
  zigbee-shepherd:request REQ --> AF:dataRequest, transId: 24 +7ms
  zigbee-shepherd:request RSP <-- AF:dataRequest, status: 0 +12ms
  zigbee-shepherd:af dispatchIncomingMsg(): type: dataConfirm, msg: [object Object] +26ms
  zigbee-shepherd:msgHdlr IND <-- AF:dataConfirm, transId: 24 +0ms
  zigbee-shepherd:af dispatchIncomingMsg(): type: incomingMsg, msg: [object Object] +12ms
  zigbee-shepherd:msgHdlr IND <-- AF:incomingMsg, transId: 0 +3ms
  zigbee-shepherd:af dispatchIncomingMsg(): type: zclIncomingMsg, msg: [object Object] +0ms
2018-8-18 12:30:05 DEBUG Recieved zigbee message with data "0x00124b0019e14fa6"
2018-8-18 12:30:05 INFO Connecting with device...
2018-8-18 12:30:05 INFO MQTT publish, topic: 'zigbee2mqtt/bridge/log', payload: '{"type":"pairing","message":"connecting with device"}'
  zigbee-shepherd:request REQ --> AF:dataRequest, transId: 25 +7ms
  zigbee-shepherd:request RSP <-- AF:dataRequest, status: 0 +13ms
  zigbee-shepherd:af dispatchIncomingMsg(): type: dataConfirm, msg: [object Object] +24ms
  zigbee-shepherd:msgHdlr IND <-- AF:dataConfirm, transId: 25 +1ms
  zigbee-shepherd:af dispatchIncomingMsg(): type: incomingMsg, msg: [object Object] +10ms
  zigbee-shepherd:msgHdlr IND <-- AF:incomingMsg, transId: 0 +4ms
  zigbee-shepherd:af dispatchIncomingMsg(): type: zclIncomingMsg, msg: [object Object] +1ms
2018-8-18 12:30:05 DEBUG Recieved zigbee message with data "0x00124b0019e14fa6"
2018-8-18 12:30:05 INFO Connecting with device...
2018-8-18 12:30:05 INFO MQTT publish, topic: 'zigbee2mqtt/bridge/log', payload: '{"type":"pairing","message":"connecting with device"}'
  zigbee-shepherd:request REQ --> AF:dataRequest, transId: 26 +9ms
  zigbee-shepherd:request RSP <-- AF:dataRequest, status: 0 +17ms
  zigbee-shepherd:af dispatchIncomingMsg(): type: dataConfirm, msg: [object Object] +336ms
  zigbee-shepherd:msgHdlr IND <-- AF:dataConfirm, transId: 26 +1ms
  zigbee-shepherd:af dispatchIncomingMsg(): type: incomingMsg, msg: [object Object] +11ms
  zigbee-shepherd:msgHdlr IND <-- AF:incomingMsg, transId: 0 +3ms
  zigbee-shepherd:af dispatchIncomingMsg(): type: zclIncomingMsg, msg: [object Object] +0ms
2018-8-18 12:30:06 DEBUG Recieved zigbee message with data "0x00124b0019e14fa6"
2018-8-18 12:30:06 INFO Connecting with device...
2018-8-18 12:30:06 INFO MQTT publish, topic: 'zigbee2mqtt/bridge/log', payload: '{"type":"pairing","message":"connecting with device"}'
  zigbee-shepherd:request REQ --> AF:dataRequest, transId: 27 +14ms
  zigbee-shepherd:request RSP <-- AF:dataRequest, status: 0 +15ms
  zigbee-shepherd:af dispatchIncomingMsg(): type: dataConfirm, msg: [object Object] +348ms
  zigbee-shepherd:msgHdlr IND <-- AF:dataConfirm, transId: 27 +0ms
  zigbee-shepherd:af dispatchIncomingMsg(): type: incomingMsg, msg: [object Object] +12ms
  zigbee-shepherd:msgHdlr IND <-- AF:incomingMsg, transId: 0 +1ms
  zigbee-shepherd:af dispatchIncomingMsg(): type: zclIncomingMsg, msg: [object Object] +0ms
2018-8-18 12:30:06 DEBUG Recieved zigbee message with data "0x00124b0019e14fa6"
2018-8-18 12:30:06 INFO Connecting with device...
2018-8-18 12:30:06 INFO MQTT publish, topic: 'zigbee2mqtt/bridge/log', payload: '{"type":"pairing","message":"connecting with device"}'
  zigbee-shepherd:request REQ --> AF:dataRequest, transId: 28 +2ms
  zigbee-shepherd:request RSP <-- AF:dataRequest, status: 0 +13ms
  zigbee-shepherd:af dispatchIncomingMsg(): type: dataConfirm, msg: [object Object] +171ms
  zigbee-shepherd:msgHdlr IND <-- AF:dataConfirm, transId: 28 +0ms
  zigbee-shepherd:af dispatchIncomingMsg(): type: incomingMsg, msg: [object Object] +13ms
  zigbee-shepherd:msgHdlr IND <-- AF:incomingMsg, transId: 0 +3ms
  zigbee-shepherd:af dispatchIncomingMsg(): type: zclIncomingMsg, msg: [object Object] +1ms
2018-8-18 12:30:06 DEBUG Recieved zigbee message with data "0x00124b0019e14fa6"
2018-8-18 12:30:06 INFO Connecting with device...
2018-8-18 12:30:06 INFO MQTT publish, topic: 'zigbee2mqtt/bridge/log', payload: '{"type":"pairing","message":"connecting with device"}'
[... 82 lines cut ...]
2018-08-18T12:30:06.726Z - info: Connecting with device...
[... cut ...]
2018-08-18T12:30:08.035Z - warn: Please see: https://github.com/Koenkk/zigbee2mqtt/wiki/How-to-support-new-devices.
2018-08-18T12:30:14.641Z - debug: Check online 0x00124b0014afe1fb
2018-08-18T12:30:14.644Z - debug: Check online 0x00124b000bd6d293
2018-08-18T12:30:14.644Z - debug: Check online 0x00124b0019e14fa6
2018-08-18T12:30:38.341Z - error: Failed to configure 0x00124b0019e14fa6
2018-08-18T12:31:14.655Z - debug: Check online 0x00124b0014afe1fb
2018-08-18T12:31:14.657Z - debug: Check online 0x00124b000bd6d293
2018-08-18T12:31:14.657Z - debug: Check online 0x00124b0019e14fa6
```

Koenkk, 2018-08-20:
The configure failed:

```
  zigbee-shepherd-converters:devices Configured '(cb) => ep16.foundation('genOnOff', 'configReport', [cfg], foundationCfg, cb)' with result 'Error: Timed out after 30000 ms' +30s
2018-8-18 12:30:38 ERROR Failed to configure 0x00124b0019e14fa6
```

Can you try repairing?

highground88, 2018-08-27:
Sorry I'm currently away with work - the device might be difficult to get to now but I can try.  

Is there any chance this is related to the sniffer, and/or firmware? I'm holding out for some changes to Zigbee2MQTT that allow modification of the .js files, as this will make the process much easier to debug.

Something else (a little off topic) that I've been wondering about for a while now is if/when I get this up and running, is how to continue to utilise Alexa integration. 

The supplied usb hub I assume will become redundant, but therefore so will the links to the hub via the app, which is the Alexa link. Is it possible to utilise the CC2531 as a repeater, rather than a coordinator, but still to pass/receive messages to HASS.io?

Koenkk, 2018-08-28:
You could try updating to: https://github.com/Koenkk/Z-Stack-firmware/tree/increased_buffer/coordinator/CC2531/bin

No this is not possible, as zigbee2mqtt has to be the coordinator. (and you cannot have more than one coordinator in a zigbee network).

ryanbeaton, 2018-10-27:
@highground88 how did you go in the end. As, seemingly, the only AU approved switches there should be interest.

highground88, 2018-10-27:
Hi @ryanbeaton, I haven't got any further myself, but am still planning to try. There seems to be some great new features added in the past week which will make sniffing easier - unfortunately when I moved from ResinOS to HassOS, I E had issues getting Zigbee2MQTT up and running as before. I think this may have resolved itself now and hopefully I'll have a look at my system again soon. If/when I do get anything difinitive, I'll make sure I push it through. Still keen to hear from anyone else who's made more progress - it seems these have been selling pretty well, so I'm surprised  really.

highground88, 2018-11-26:
Still not having much luck but will kepp trying. I notcied that in addition to SmartThings, the Hubitat users have also managed to import the handlers etc into their ecosystem and can utilise this with HA (although there seems to be problems getting the 3rd switch from the 3 gang units working). Unfortunately WAF too low to purchase a new hub right now....

I have also found this information on the Feibet site which I'm hoping might help those smarter than me figure out more of the problem perhaps? (translated into Chinglish from here: http://www.feibit.com/changjianwenti.html ). Unsure if this is referencing the app control or communication to the gateway unit.

5. What is the gateway API interface protocol?

Interface Protocol: TCP Socket Interface

Port number: 8001

The illuminance data is parsed high in the previous low position and converted to decimal in D554 . The illuminance intensity is 54612.

6. Flying than cloud server interface protocol

Interface protocol: socket and web socket

Port number: socket 8090 , web socket 8092

Format: packet length (2 bytes )  data type (2 bytes )  data length (2 bytes )  data ( variable length )

highground88, 2018-12-29:
Hi @Koenkk & @ryanbeaton ,

My apologies for the delay in getting back to this - just getting a bit of time now to re-visit, and see that there's been some more revisions. 

I've updated to a fresh version (1.0.1) of the add-on and have had a clean up, and I think I'm finally starting to see some of the messages based on switch activation which I wasn't able to see previously. 

Unfortunately I've tried a few variants of the handler that you originally kindly provided me with Koen, but the device isn't being accepted fully, e.g.  _(Message without device / Device with modelID 'undefined is not supported')_

If you could offer any sort of further advice, I'd be really appreciative.

**devices.js entry:**
``    {
        zigbeeModel: ['FB56+ZSW1HKJ2.5', 'FB56+ZSW1HKJ1.7'],
        model: 'TODO',
        vendor: 'Nue',
        description: 'Smart light switch - 2 gang',
        supports: 'TODO',
        fromZigbee: [],
        toZigbee: [],
        configure: (ieeeAddr, shepherd, coordinator, callback) => {
            const ep16 = shepherd.find(ieeeAddr, 16);
            const cfg = {direction: 0, attrId: 0, dataType: 16, minRepIntval: 0, maxRepIntval: 10, repChange: 0};

            const actions = [
                (cb) => ep16.read('genOnOff', 'onOff', cb),
                (cb) => ep16.foundation('genOnOff', 'configReport', [cfg], foundationCfg, cb),
            ];

            execute(ep16, actions, callback);
        },
    },

**Debug output:**
`  zigbee2mqtt:debug 2018-12-29 13:52:04 Received zigbee message of type 'devInterview' with data '"0x00124b0014afe1fb"'
  zigbee2mqtt:warn 2018-12-29 13:52:04 Message without device!
Sat, 29 Dec 2018 03:52:04 GMT zigbee-shepherd:request REQ --> AF:dataRequest, transId: 4
Sat, 29 Dec 2018 03:52:04 GMT zigbee-shepherd:request RSP <-- AF:dataRequest, status: 0
Sat, 29 Dec 2018 03:52:05 GMT zigbee-shepherd:af dispatchIncomingMsg(): type: dataConfirm, msg: [object Object]
Sat, 29 Dec 2018 03:52:05 GMT zigbee-shepherd:msgHdlr IND <-- AF:dataConfirm, transId: 4
  zigbee2mqtt:debug 2018-12-29 13:52:05 Received zigbee message of type 'devInterview' with data '"0x00124b0014afe1fb"'
  zigbee2mqtt:warn 2018-12-29 13:52:05 Message without device!
Sat, 29 Dec 2018 03:52:05 GMT zigbee-shepherd:request REQ --> AF:dataRequest, transId: 5
Sat, 29 Dec 2018 03:52:05 GMT zigbee-shepherd:request RSP <-- AF:dataRequest, status: 0
Sat, 29 Dec 2018 03:52:06 GMT zigbee-shepherd:af dispatchIncomingMsg(): type: dataConfirm, msg: [object Object]
Sat, 29 Dec 2018 03:52:06 GMT zigbee-shepherd:msgHdlr IND <-- AF:dataConfirm, transId: 5
  zigbee2mqtt:debug 2018-12-29 13:52:06 Received zigbee message of type 'devInterview' with data '"0x00124b0014afe1fb"'
  zigbee2mqtt:warn 2018-12-29 13:52:06 Message without device!
Sat, 29 Dec 2018 03:52:06 GMT zigbee-shepherd:request REQ --> AF:dataRequest, transId: 6
Sat, 29 Dec 2018 03:52:06 GMT zigbee-shepherd:request RSP <-- AF:dataRequest, status: 0
Sat, 29 Dec 2018 03:52:06 GMT zigbee-shepherd:af dispatchIncomingMsg(): type: dataConfirm, msg: [object Object]
Sat, 29 Dec 2018 03:52:06 GMT zigbee-shepherd:msgHdlr IND <-- AF:dataConfirm, transId: 6
  zigbee2mqtt:debug 2018-12-29 13:52:06 Received zigbee message of type 'devInterview' with data '"0x00124b0014afe1fb"'
  zigbee2mqtt:warn 2018-12-29 13:52:06 Message without device!
Sat, 29 Dec 2018 03:52:06 GMT zigbee-shepherd:request REQ --> AF:dataRequest, transId: 7
Sat, 29 Dec 2018 03:52:06 GMT zigbee-shepherd:request RSP <-- AF:dataRequest, status: 0
Sat, 29 Dec 2018 03:52:07 GMT zigbee-shepherd:af dispatchIncomingMsg(): type: dataConfirm, msg: [object Object]
Sat, 29 Dec 2018 03:52:07 GMT zigbee-shepherd:msgHdlr IND <-- AF:dataConfirm, transId: 7
  zigbee2mqtt:debug 2018-12-29 13:52:07 Received zigbee message of type 'devInterview' with data '"0x00124b0014afe1fb"'
  zigbee2mqtt:warn 2018-12-29 13:52:07 Message without device!
Sat, 29 Dec 2018 03:52:07 GMT zigbee-shepherd:request REQ --> AF:dataRequest, transId: 8
Sat, 29 Dec 2018 03:52:07 GMT zigbee-shepherd:request RSP <-- AF:dataRequest, status: 0
Sat, 29 Dec 2018 03:52:07 GMT zigbee-shepherd:af dispatchIncomingMsg(): type: dataConfirm, msg: [object Object]
Sat, 29 Dec 2018 03:52:07 GMT zigbee-shepherd:msgHdlr IND <-- AF:dataConfirm, transId: 8
  zigbee2mqtt:debug 2018-12-29 13:52:07 Received zigbee message of type 'devInterview' with data '"0x00124b0014afe1fb"'
  zigbee2mqtt:warn 2018-12-29 13:52:08 Message without device!
Sat, 29 Dec 2018 03:52:08 GMT zigbee-shepherd:request REQ --> AF:dataRequest, transId: 9
Sat, 29 Dec 2018 03:52:08 GMT zigbee-shepherd:request RSP <-- AF:dataRequest, status: 0
Sat, 29 Dec 2018 03:52:08 GMT zigbee-shepherd:af dispatchIncomingMsg(): type: dataConfirm, msg: [object Object]
Sat, 29 Dec 2018 03:52:08 GMT zigbee-shepherd:msgHdlr IND <-- AF:dataConfirm, transId: 9
  zigbee2mqtt:debug 2018-12-29 13:52:08 Received zigbee message of type 'devInterview' with data '"0x00124b0014afe1fb"'
[... 38 lines cut ...]
  zigbee2mqtt:warn 2018-12-29 13:52:13 Please see: https://koenkk.github.io/zigbee2mqtt/how_tos/how_to_support_new_devices.html
Sat, 29 Dec 2018 03:52:13 GMT zigbee-shepherd:af dispatchIncomingMsg(): type: incomingMsg, msg: [object Object]
Sat, 29 Dec 2018 03:52:13 GMT zigbee-shepherd:msgHdlr IND <-- AF:incomingMsg, transId: 0
Sat, 29 Dec 2018 03:52:13 GMT zigbee-shepherd:af dispatchIncomingMsg(): type: zclIncomingMsg, msg: [object Object]
[... cut ...]
Sat, 29 Dec 2018 03:52:24 GMT zigbee-shepherd:af dispatchIncomingMsg(): type: zclIncomingMsg, msg: [object Object]
  zigbee2mqtt:debug 2018-12-29 13:52:24 Received zigbee message of type 'attReport' with data '{"cid":"genOnOff","data":{"onOff":1}}' of device 'undefined' (0x00124b0014bdea20)
  zigbee2mqtt:warn 2018-12-29 13:52:24 Device with modelID 'undefined' is not supported.
  zigbee2mqtt:warn 2018-12-29 13:52:24 Please see: https://koenkk.github.io/zigbee2mqtt/how_tos/how_to_support_new_devices.html`

Koenkk, 2018-12-29:
That indeed looks good! Show how the modelID is not correct.

Can you:
- stop zigbee2mqtt 
- remove all entries of `0x00124b0014bdea20` from `database.db`
- start zigbee2mqtt
- repair the device, be sure to keep it close to the coordinator

highground88, 2018-12-29:
Thanks very much again @Koenkk ...

I've followed your instructions (and quite sure I've tried this before too) but seem to get the same result.
(Stopped Z2MQTT, cleared out the relevant lines containing the coordinator address, restart Z2MQTT, allow rejoin, hold pairing button on device until the pairing light appears, activate the switches a few times). I've also fully removed the database before with no change.

**Following doing above, this is the log output:**

2018-12-30 08:39:27 - info: Logging to directory: '/share/zigbee2mqtt/log/2018-12-30.08-39-26'
2018-12-30 08:39:27 - debug: Removing old log directory '/share/zigbee2mqtt/log/2018-12-29.15-12-48'
2018-12-30 08:39:49 - debug: Using zigbee-shepherd with settings: '{"net":{"panId":6754,"channelList":[11]},"dbPath":"/share/zigbee2mqtt/database.db","sp":{"baudRate":115200,"rtscts":true}}'
2018-12-30 08:39:57 - debug: Loaded state from file /share/zigbee2mqtt/state.json
2018-12-30 08:40:05 - info: Starting zigbee2mqtt version 1.0.1 (commit #unknown)
2018-12-30 08:40:05 - info: Starting zigbee-shepherd
2018-12-30 08:40:21 - info: zigbee-shepherd started
2018-12-30 08:40:21 - info: Coordinator firmware version: '20180815'
2018-12-30 08:40:21 - debug: zigbee-shepherd info: {"enabled":true,"net":{"state":"Coordinator","channel":11,"panId":"0x1a62","extPanId":"0xdddddddddddddddd","ieeeAddr":"0x00124b0014bdea20","nwkAddr":0},"firmware":{"transportrev":2,"product":0,"version":"2.6.3","revision":20180815},"startTime":1546123221,"joinTimeLeft":0}
2018-12-30 08:40:21 - info: Currently 1 devices are joined:
2018-12-30 08:40:21 - info: 0x00124b0014afe1fb (0x00124b0014afe1fb): TODO - Nue Smart light switch - 2 gang (Router)
2018-12-30 08:40:21 - info: Zigbee: disabling joining new devices.
2018-12-30 08:40:21 - info: Connecting to MQTT server at mqtt://192.168.0.101:1883
2018-12-30 08:40:21 - info: zigbee-shepherd ready
2018-12-30 08:40:21 - info: Connected to MQTT server
2018-12-30 08:40:21 - info: MQTT publish: topic 'zigbee2mqtt/bridge/state', payload 'online'
2018-12-30 08:40:21 - debug: Mounted the cieApp (epId {epId})
2018-12-30 08:41:05 - debug: Received zigbee message of type 'attReport' with data '{"cid":"genOnOff","data":{"onOff":0}}' of device 'undefined' (0x00124b0014bdea20)
2018-12-30 08:41:05 - warn: Device with modelID 'undefined' is not supported.
2018-12-30 08:41:05 - warn: Please see: https://koenkk.github.io/zigbee2mqtt/how_tos/how_to_support_new_devices.html
2018-12-30 08:41:06 - debug: Received zigbee message of type 'attReport' with data '{"cid":"genOnOff","data":{"onOff":1}}' of device 'undefined' (0x00124b0014bdea20)
2018-12-30 08:41:06 - warn: Device with modelID 'undefined' is not supported.
2018-12-30 08:41:06 - warn: Please see: https://koenkk.github.io/zigbee2mqtt/how_tos/how_to_support_new_devices.html
2018-12-30 08:41:11 - error: Failed to configure 0x00124b0014afe1fb 0x00124b0014afe1fb
2018-12-30 08:41:35 - debug: Received zigbee message of type 'devInterview' with data '"0x00124b0014afe1fb"'
2018-12-30 08:41:35 - warn: Message without device!
2018-12-30 08:41:35 - debug: Received zigbee message of type 'devInterview' with data '"0x00124b0014afe1fb"'
2018-12-30 08:41:35 - warn: Message without device!
2018-12-30 08:41:35 - debug: Received zigbee message of type 'devInterview' with data '"0x00124b0014afe1fb"'
2018-12-30 08:41:35 - warn: Message without device!
2018-12-30 08:41:35 - debug: Received zigbee message of type 'devInterview' with data '"0x00124b0014afe1fb"'
2018-12-30 08:41:35 - warn: Message without device!
2018-12-30 08:41:35 - debug: Received zigbee message of type 'devInterview' with data '"0x00124b0014afe1fb"'
2018-12-30 08:41:35 - warn: Message without device!
2018-12-30 08:41:35 - debug: Received zigbee message of type 'devInterview' with data '"0x00124b0014afe1fb"'
2018-12-30 08:41:35 - warn: Message without device!
2018-12-30 08:41:35 - debug: Received zigbee message of type 'devInterview' with data '"0x00124b0014afe1fb"'
2018-12-30 08:41:35 - warn: Message without device!
2018-12-30 08:41:35 - debug: Received zigbee message of type 'devInterview' with data '"0x00124b0014afe1fb"'
2018-12-30 08:41:35 - warn: Message without device!
2018-12-30 08:41:35 - debug: Received zigbee message of type 'devIncoming' with data '"0x00124b0014afe1fb"' of device 'FB56+ZSW1HKJ1.7' (0x00124b0014afe1fb)
2018-12-30 08:41:35 - info: Device incoming...
2018-12-30 08:41:35 - info: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"pairing","message":"device incoming"}'
2018-12-30 08:41:35 - debug: Received zigbee message of type 'devStatus' with data '"online"' of device 'FB56+ZSW1HKJ1.7' (0x00124b0014afe1fb)
2018-12-30 08:41:36 - debug: Received zigbee message of type 'attReport' with data '{"cid":"genOnOff","data":{"onOff":0}}' of device 'undefined' (0x00124b0014bdea20)
2018-12-30 08:41:36 - warn: Device with modelID 'undefined' is not supported.
2018-12-30 08:41:36 - warn: Please see: https://koenkk.github.io/zigbee2mqtt/how_tos/how_to_support_new_devices.html
2018-12-30 08:41:36 - debug: Received zigbee message of type 'attReport' with data '{"cid":"genOnOff","data":{"onOff":1}}' of device 'undefined' (0x00124b0014bdea20)
2018-12-30 08:41:36 - warn: Device with modelID 'undefined' is not supported.
2018-12-30 08:41:36 - warn: Please see: https://koenkk.github.io/zigbee2mqtt/how_tos/how_to_support_new_devices.html
[... cut ...]
            const cfg = {direction: 0, attrId: 0, dataType: 16, minRepIntval: 0, maxRepIntval: 10, repChange: 0};

            const actions = [
                (cb) => ep16.bind('genOnOff', coordinator, cb),
                (cb) => ep16.foundation('genOnOff', 'configReport', [cfg], foundationCfg, cb),
                (cb) => ep17.bind('genOnOff', coordinator, cb),
                (cb) => ep17.foundation('genOnOff', 'configReport', [cfg], foundationCfg, cb),
            ];

            execute(ep16, actions, callback);
        },
    },`

Koenkk, 2018-12-30:
`0x00124b0014bdea20` is indeed the coordinator, which is very strange.

You are sure that `Received zigbee message of type 'attReport' with data '{"cid":"genOnOff","data":{"onOff":1}}' of device 'undefined' (0x00124b0014bdea20)` this message is produced when you click the switch (I've never seen before that the message is then reported by the coordinator).

highground88, 2019-01-02:
Happy New Year @Koenkk 

I can confirm that those messages above have been produced by activating the switch, and appear to come from the configurator. In fact, I reflashed the firmware to the latest, reinstalled Z2MQTT, and re-paired the switch. Same thing is happening.  I haven't changed the device handler from that above (now sure if this is correct or perhaps part of the problem?). Very strange - @danielwelch have you perhaps seen this happening?

**Here are the new logs, switch activations occur around 13:50:30...**

2019-1-2 13:45:33 - info: Logging to directory: '/share/zigbee2mqtt/log/2019-01-02.13-45-32'
2019-1-2 13:45:33 - debug: Removing old log directory '/share/zigbee2mqtt/log/2018-12-29.15-12-55'
2019-1-2 13:45:37 - debug: Using zigbee-shepherd with settings: '{"net":{"panId":6754,"channelList":[11]},"dbPath":"/share/zigbee2mqtt/database.db","sp":{"baudRate":115200,"rtscts":true}}'
2019-1-2 13:45:43 - debug: Loaded state from file /share/zigbee2mqtt/state.json
2019-1-2 13:45:45 - info: Starting zigbee2mqtt version 1.0.1 (commit #unknown)
2019-1-2 13:45:45 - info: Starting zigbee-shepherd
2019-1-2 13:46:30 - info: zigbee-shepherd started
2019-1-2 13:46:30 - info: Coordinator firmware version: '20181024'
2019-1-2 13:46:30 - debug: zigbee-shepherd info: {"enabled":true,"net":{"state":"Coordinator","channel":11,"panId":"0x1a62","extPanId":"0xdddddddddddddddd","ieeeAddr":"0x00124b0014bdea20","nwkAddr":0},"firmware":{"transportrev":2,"product":0,"version":"2.6.3","revision":20181024},"startTime":1546400790,"joinTimeLeft":0}
2019-1-2 13:46:30 - info: Currently 0 devices are joined:
2019-1-2 13:46:30 - info: Zigbee: disabling joining new devices.
2019-1-2 13:46:30 - info: Connecting to MQTT server at mqtt://192.168.0.101:1883
2019-1-2 13:46:31 - info: zigbee-shepherd ready
2019-1-2 13:46:31 - info: Connected to MQTT server
2019-1-2 13:46:31 - info: MQTT publish: topic 'zigbee2mqtt/bridge/state', payload 'online'
2019-1-2 13:46:31 - debug: Mounted the cieApp (epId {epId})
2019-1-2 13:48:09 - debug: Received MQTT message on 'zigbee2mqtt/bridge/config/permit_join' with data 'true'
2019-1-2 13:48:09 - info: Zigbee: allowing new devices to join.
2019-1-2 13:48:09 - info: MQTT publish: topic 'zigbee2mqtt/bridge/config', payload '{"log_level":"debug","permit_join":true}'
2019-1-2 13:48:54 - debug: Received zigbee message of type 'attReport' with data '{"cid":"genOnOff","data":{"onOff":0}}' of device 'undefined' (0x00124b0014bdea20)
2019-1-2 13:48:54 - warn: Device with modelID 'undefined' is not supported.
2019-1-2 13:48:54 - warn: Please see: https://koenkk.github.io/zigbee2mqtt/how_tos/how_to_support_new_devices.html
2019-1-2 13:48:59 - debug: Received zigbee message of type 'devInterview' with data '"0x00124b0014afe1fb"'
2019-1-2 13:48:59 - warn: Message without device!
2019-1-2 13:48:59 - debug: Received zigbee message of type 'devInterview' with data '"0x00124b0014afe1fb"'
2019-1-2 13:48:59 - warn: Message without device!
2019-1-2 13:48:59 - debug: Received zigbee message of type 'devInterview' with data '"0x00124b0014afe1fb"'
2019-1-2 13:48:59 - warn: Message without device!
2019-1-2 13:48:59 - debug: Received zigbee message of type 'devInterview' with data '"0x00124b0014afe1fb"'
2019-1-2 13:48:59 - warn: Message without device!
2019-1-2 13:49:00 - debug: Received zigbee message of type 'devInterview' with data '"0x00124b0014afe1fb"'
2019-1-2 13:49:00 - warn: Message without device!
2019-1-2 13:49:00 - debug: Received zigbee message of type 'devInterview' with data '"0x00124b0014afe1fb"'
2019-1-2 13:49:00 - warn: Message without device!
2019-1-2 13:49:00 - debug: Received zigbee message of type 'devInterview' with data '"0x00124b0014afe1fb"'
2019-1-2 13:49:00 - warn: Message without device!
2019-1-2 13:49:00 - debug: Received zigbee message of type 'devInterview' with data '"0x00124b0014afe1fb"'
2019-1-2 13:49:00 - warn: Message without device!
2019-1-2 13:49:00 - debug: Received zigbee message of type 'devIncoming' with data '"0x00124b0014afe1fb"' of device 'FB56+ZSW1HKJ1.7' (0x00124b0014afe1fb)
2019-1-2 13:49:00 - info: Device incoming...
2019-1-2 13:49:00 - info: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"pairing","message":"device incoming"}'
2019-1-2 13:49:00 - debug: Received zigbee message of type 'devStatus' with data '"online"' of device 'FB56+ZSW1HKJ1.7' (0x00124b0014afe1fb)
2019-1-2 13:49:02 - info: Succesfully configured 0x00124b0014afe1fb 0x00124b0014afe1fb
2019-1-2 13:49:17 - debug: Received zigbee message of type 'devLeaving' with data '"0x00124b0014afe1fb"'
2019-1-2 13:49:17 - warn: Message without device!
2019-1-2 13:49:52 - debug: Received zigbee message of type 'attReport' with data '{"cid":"genOnOff","data":{"onOff":1}}' of device 'undefined' (0x00124b0014bdea20)
2019-1-2 13:49:52 - warn: Device with modelID 'undefined' is not supported.
2019-1-2 13:49:52 - warn: Please see: https://koenkk.github.io/zigbee2mqtt/how_tos/how_to_support_new_devices.html
2019-1-2 13:50:03 - debug: Received zigbee message of type 'devInterview' with data '"0x00124b0014afe1fb"'
2019-1-2 13:50:03 - warn: Message without device!
2019-1-2 13:50:04 - debug: Received zigbee message of type 'devInterview' with data '"0x00124b0014afe1fb"'
2019-1-2 13:50:04 - warn: Message without device!
2019-1-2 13:50:04 - debug: Received zigbee message of type 'devInterview' with data '"0x00124b0014afe1fb"'
2019-1-2 13:50:04 - warn: Message without device!
[... cut ...] 0,"epId":17,"devId":2,"inClusterList":[0,4,5,6],"outClusterList":[0],"clusters":{"genBasic":{"dir":{"value":3},"attrs":{"16393":1170014208,"hwVersion":0,"manufacturerName":"Feibit Inc co.  ","modelId":"FB56+ZSW1HKJ1.7","dateCode":"20161030        ","powerSource":1,"locationDesc":"                ","physicalEnv":0,"deviceEnabled":1}},"genGroups":{"dir":{"value":1},"attrs":{}},"genScenes":{"dir":{"value":1},"attrs":{}},"genOnOff":{"dir":{"value":1},"attrs":{"onOff":0}}}}},"_id":"PZYL316hqm259LaU"}

Koenkk, 2019-01-05:
Could you try sniffing the traffic from the device? This will give us some more insight on how it works: https://github.com/Koenkk/zigbee2mqtt/blob/dev/docs/how_tos/how_to_sniff_zigbee_traffic.md

highground88, 2019-01-05:
I can sure try, but by chance is there a Windows compatible tool to do this? Otherwise I'll try to find a spare machine to put Ubuntu on.

Do you think this would help with the coordinator issue though?  Thanks for your help as always.

Koenkk, 2019-01-06:
I would recommend using ubuntu to do this (never tried it on windows).

Probably not, but perhaps it could give us insight on why this happens.

highground88, 2019-01-09:
Sorry for the delay.... had fun getting this all to work/install/run over the last few days now......

I now have Ubuntu running on a laptop and have flashed the CC2531 with the sniffer software.  I'm doing my best to follow along, and have interpreted based on trial and error that in your sniffing guide, the command to start wireshark should be "sudo whsniff -c **11** | wireshark -k -i -". Setting this to any other channel seemingly aside from 11 results in the error "unrecognised libpcap format or not libpcap data" (same if I use the command listed). Once I have it up and sniffing, the most I seem to get is a beacon request via IEEE 802.15.4once the switch is in pairing mode (03 08 67 ff ff ff ff 07 58 b2).

EDIT: I take this back..... I've had a breakthrough - it appears that the hub and switches are operating on channel 19. I didn't try this far up the sequence. If I browse to the "FBee Cloud Gateway" page ("Nue Smart Hub device") via IP, I'm greeted with this:

> Version: |  
> -- | --
> Software Version | 2.5.5(Nov 22 2017_13:42:58)Light GW
> Hardware version | FBF109CB V1.0
> Serial No | XXXXXXXXX
> 
> IP Mode | DHCP
> IP Address | 192.168.XXX
> IP Mask | 255.255.255.0
> IP Gateway | 192.168.XXX
> MAC Address | 10:A5:D4:XXXXXXX
> 
> WAN | Connected
> FBee Cloud | Full Control
> User Name | XXXXXXXX
> 
> Current Date: | 2019-01-09
> Current Time | 18:12:23 (Time Zone:UTC +10:00)
> Upload bytes | 1783
> Download bytes | 16
> Free Mem | 14832
> Free Mini Mem | 9792
> PanID | 5443-XXXXXXXXC60D004B1200-**CH:19** Build:021203
> Scene Mode: | 1

The CH19 part at the end of PanID tipped me off on the channel number!

I have also captured the Network Encryption Key, so have access to all broadcasts in wireshark now! (Is this a common key, or best I keep private?).

Thank you for getting me this far @koenkk - how can I best help from here with this access? Shall I capture and export a .pcap file or .csv or ..?  I'm still treading water as far as understanding the Zigbee protocol, so unfortunately I'll still need a hand but I'd love to contribute as much as I can to both your project and help others with these devices.

From what I understand, the things that need finalisation with these switches for myself and a lot of others is getting them reliably onto the CC2531 platform and recognised under HA via Zigbee2MQTT - this might be device specific in some cases, due to dimming components, different numbers of gang switches, and "scene switch" devices which do not have a physical switch and still manage to retain their bind to other switches without a coordinator switched on.

I'll have a few more days I hope of time to play with this before being back at work, so please let me know what I can do to help. I'll post up some more info shortly once I work out how to get samba shares working...!

Koenkk, 2019-01-09:
What messages do you see when you press a button on the switch?

highground88, 2019-01-09:
```
No.     Time           Source                Destination           Protocol Length Info
      1 0.000000       0x806f                0x0000                ZigBee HA 52     ZCL: Report Attributes, Seq: 13

Frame 1: 52 bytes on wire (416 bits), 52 bytes captured (416 bits) on interface 0
IEEE 802.15.4 Data, Dst: 0x0000, Src: 0x806f
ZigBee Network Layer Data, Dst: 0x0000, Src: 0x806f
    Frame Control Field: 0x0248, Frame Type: Data, Discover Route: Enable, Security Data
    Destination: 0x0000
    Source: 0x806f
    Radius: 15
    Sequence Number: 141
    ZigBee Security Header
        Security Control Field: 0x28, Key Id: Network Key, Extended Nonce
        Frame Counter: 155
        Extended Source: TexasIns_00:14:af:e1:fb (00:12:4b:00:14:af:e1:fb)
        Key Sequence Number: 0
        Message Integrity Code: 3e940752
        [Key: fb030507090b0d0f00020406080a0cfb]
        [Key Label: NUE ]
ZigBee Application Support Layer Data, Dst Endpt: 11, Src Endpt: 16
    Frame Control Field: Data (0x00)
    Destination Endpoint: 11
    Cluster: On/Off (0x0006)
    Profile: Home Automation (0x0104)
    Source Endpoint: 16
    Counter: 24
ZigBee Cluster Library Frame, Command: Report Attributes, Seq: 13
    Frame Control Field: Profile-wide (0x08)
    Sequence Number: 13
    Command: Report Attributes (0x0a)
    Attribute Field

No.     Time           Source                Destination           Protocol Length Info
      2 0.002048                                                   IEEE 802.15.4 5      Enhanced Beacon[Malformed Packet]

Frame 2: 5 bytes on wire (40 bits), 5 bytes captured (40 bits) on interface 0
IEEE 802.15.4 Enhanced Beacon
[Malformed Packet: IEEE 802.15.4]
    [Expert Info (Error/Malformed): Malformed Packet (Exception occurred)]
        [Malformed Packet (Exception occurred)]
        [Severity level: Error]
        [Group: Malformed]

No.     Time           Source                Destination           Protocol Length Info
      3 0.015941       0x0000                0x806f                ZigBee HA 50     ZCL: Default Response, Seq: 13

Frame 3: 50 bytes on wire (400 bits), 50 bytes captured (400 bits) on interface 0
IEEE 802.15.4 Data, Dst: 0x806f, Src: 0x0000
ZigBee Network Layer Data, Dst: 0x806f, Src: 0x0000
    Frame Control Field: 0x0248, Frame Type: Data, Discover Route: Enable, Security Data
    Destination: 0x806f
    Source: 0x0000
    Radius: 15
    Sequence Number: 68
    ZigBee Security Header
        Security Control Field: 0x28, Key Id: Network Key, Extended Nonce
        Frame Counter: 1444585
        Extended Source: TexasIns_00:0d:c6:54:43 (00:12:4b:00:0d:c6:54:43)
        Key Sequence Number: 0
        Message Integrity Code: 9f08b79a
        [Key: fb030507090b0d0f00020406080a0cfb]
        [Key Label: NUE ]
ZigBee Application Support Layer Data, Dst Endpt: 16, Src Endpt: 11
    Frame Control Field: Data (0x00)
    Destination Endpoint: 16
    Cluster: On/Off (0x0006)
    Profile: Home Automation (0x0104)
    Source Endpoint: 11
    Counter: 160
[... 342 lines cut ...]
    Command Frame: Link Status

No.     Time           Source                Destination           Protocol Length Info
     21 13.908235      0x8f43                Broadcast             ZigBee   71     Link Status

Frame 21: 71 bytes on wire (568 bits), 71 bytes captured (568 bits) on interface 0
IEEE 802.15.4 Data, Dst: Broadcast, Src: 0x8f43
ZigBee Network Layer Command, Dst: Broadcast, Src: 0x8f43
    Frame Control Field: 0x1209, Frame Type: Command, Discover Route: Suppress, Security, Extended Source Command
    Destination: 0xfffc
    Source: 0x8f43
    Radius: 1
    Sequence Number: 119
    Extended Source: TexasIns_00:19:e1:4f:a6 (00:12:4b:00:19:e1:4f:a6)
    ZigBee Security Header
        Security Control Field: 0x28, Key Id: Network Key, Extended Nonce
        Frame Counter: 444232
        Extended Source: TexasIns_00:19:e1:4f:a6 (00:12:4b:00:19:e1:4f:a6)
        Key Sequence Number: 0
        Message Integrity Code: eb7c54ef
        [Key: fb030507090b0d0f00020406080a0cfb]
        [Key Label: NUE ]
    Command Frame: Link Status

```

highground88, 2019-01-09:
[LINK](https://www.dropbox.com/sh/vt3o8lken7vbkmx/AACekFPfg2mC7yCWGX4OcqbKa?dl=0)

Koenkk, 2019-01-09:
It seems that the messages are send to the zapp-cie, something that has been added in zigbee2mqtt 1.0.0. This also explains why the switch suddenly started to work.

Can you add the following to https://github.com/Koenkk/zigbee2mqtt/blob/master/lib/zapp/cie.js (before line 56)

```js
cieClusters.init('genOnOff', 'dir', {value: 2}); // Client Side(Output)

cieClusters.init('genOnOff', 'cmds', {
    off: (zapp, argObj, cb) => {
        console.log("Hello there! this is off");
        console.log(zapp, argObj);
    },
    on: (zapp, argObj, cb) => {
        console.log("Hello there! this is ON");
        console.log(zapp, argObj);
    },
});
```

highground88, 2019-01-10:
Thanks, looks promising. Is this zapp-cie change the likely culprit that has also caused the router device messages to appear as if they are coming from the coordinator perhaps?

I will have to look at setting up another HA device today, as I'm on Hassio and don't believe I can make those changes?

highground88, 2019-01-10:
Hi @Koenkk ,

I think I have done as you suggested via a second Hasbian instance and a fork of your dev build with the modification to the cie.js file. This is the output: [PASTEBIN LINK](https://pastebin.com/EVRnmRr3)
It appears that the coordinator is still passing on the switch activation message unfortunately.

The device.js I used is below - again I really have no idea what I shoud be putting in this still, but I have included it for your reference.

```
 {
        zigbeeModel: ['FB56+ZSW1HKJ2.5', 'FB56+ZSW1HKJ1.7'],
        model: 'TODO',
        vendor: 'Nue',
        description: 'Smart light switch - 2 gang',
        supports: 'on/off',
        fromZigbee: [fz.nue_power_state, fz.ignore_onoff_change, fz.generic_state],
        toZigbee: [tz.on_off],
        configure: (ieeeAddr, shepherd, coordinator, callback) => {
        const ep16 = shepherd.find(ieeeAddr, 16);
        const cfg = {direction: 0, attrId: 0, dataType: 16, minRepIntval: 0, maxRepIntval: 10, repChange: 0};

            const actions = [
                (cb) => ep16.read('genOnOff', 'onOff', cb),
                (cb) => ep16.foundation('genOnOff', 'configReport', [cfg], foundationCfg, cb),
        ];

        execute(ep16, actions, callback);
      },
    },

```

Koenkk, 2019-01-10:
Can you post the whole log? (I'm also interested in the startup part)

Koenkk, 2019-01-10:
No luck yet, can you try with:

```js
cieClusters.init('genOnOff', 'dir', {value: 1}); // Client Side(Output)

cieClusters.init('genOnOff', 'cmds', {
    off: (zapp, argObj, cb) => {
        console.log("Hello there! this is off");
        console.log(zapp, argObj);
    },
    on: (zapp, argObj, cb) => {
        console.log("Hello there! this is ON");
        console.log(zapp, argObj);
    },
});
```

Koenkk, 2019-01-11:
That's indeed the good procedure, I've added extra logging to your fork, could you update and provide the logging again?

Koenkk, 2019-01-13:
The current problem is that we don't know that it come from the switch. In the logging I can't find any identifiers. I'm out of ideas again..

highground88, 2019-01-14:
At least we have got some kind of reaction out of the switches I guess now. I  still stumped why this switch isn't  working but the other handlers already in shepherd for the other switches do.

Being that these seem to work in SmartThings, would this thread shed any secrets for you perhaps? The distributor '3asmarthome' is also active on that forum so I will reach out to him again.

https://community.smartthings.com/t/instruction-and-device-handler-for-smartthings-hub-controlling-nue-zigbee-in-ceiling-light-controller/127109

I will wire up some different model switches perhaps today and see if anything comes up. I have ordered a second cc2531 so that one can stay connected to Zigbee2Mqtt whilst the other is used as a sniffer. By all means if you have any other ideas or anytjkng else you want me to try please let me know as I'm not quite ready to give up yet! :-)

Thanks Koen

highground88, 2019-01-14:
I've hooked up a different model switch (1 gang + dimmer up/down) which shows as a FB56+ZSC05HG1.0 [Product Link](https://www.3asmarthome.com/zigbee-smart-dimmer-switch) and I've created a new dump (just added the model number in behind the previous 2 switches in the devices.js file).

[LOG FROM FB56+ZSC05HG1.0](https://pastebin.com/qv6hd7G4)

Not sure if it's helpful or not. The other switch was powered off at the time.  Pressing any of the buttons resulted in this output:

```
 zigbee-shepherd:msgHdlr IND <-- ZDO:srcRtgInd +31s
  zigbee-shepherd:af dispatchIncomingMsg(): type: incomingMsg, msg: [object Object] +12ms
  zigbee-shepherd:msgHdlr IND <-- AF:incomingMsg, transId: 0 +10ms
  zigbee-shepherd:af dispatchIncomingMsg(): type: zclIncomingMsg, msg: [object Object] +2ms
  zigbee-shepherd:af dispatchIncomingMsg(): type: incomingMsg, msg: [object Object] +6s
  zigbee-shepherd:msgHdlr IND <-- AF:incomingMsg, transId: 0 +10ms
  zigbee-shepherd:af dispatchIncomingMsg(): type: zclIncomingMsg, msg: [object Object] +2ms
  zigbee-shepherd:af dispatchIncomingMsg(): type: incomingMsg, msg: [object Object] +3s
  zigbee-shepherd:msgHdlr IND <-- AF:incomingMsg, transId: 0 +10ms
  zigbee-shepherd:af dispatchIncomingMsg(): type: zclIncomingMsg, msg: [object Object] +2ms
  zigbee-shepherd:af dispatchIncomingMsg(): type: incomingMsg, msg: [object Object] +978ms
  zigbee-shepherd:msgHdlr IND <-- AF:incomingMsg, transId: 0 +10ms
```
From what I understand the Smart Things guys diddn't have too much trouble with the other switches, but this dimmer seemed a little more complex.

Koenkk, 2019-01-14:
Can you add to: https://github.com/Koenkk/zigbee-shepherd/blob/master/lib/components/af.js#L689 `console.log(msg.zclMsg.cmdId)` and provide the log?

highground88, 2019-01-15:
Sure:  [LOG with console.log(msg.zclMsg.cmdId)](https://pastebin.com/mMnEWV5a)

Both model devices re-paired.

highground88, 2019-01-15:
[New 'COMPLETE' LOG](https://pastebin.com/sNqtsRkz)

Koenkk, 2019-01-15:
That doesn't contain any information unfortunately, what code are you using for this device? 

There is a donation link here: https://github.com/Koenkk/zigbee2mqtt/blob/master/README.md

Koenkk, 2019-01-16:
Can you provide the database.db of FB56+ZSC05HG1.0

highground88, 2019-01-16:
Here you go, thanks:

```
{"id":3,"type":"Router","ieeeAddr":"0x00124b001545a430","nwkAddr":8910,"manufId":0,"manufName":"Feibit Inc co.  ","powerSource":"Mains (single phase)","modelId":"FB56+ZSC05HG1.0","epList":[11,13],"status":"online","joinTime":1547518256,"endpoints":{"11":{"profId":260,"epId":11,"devId":257,"inClusterList":[0,3,4,5,6,8],"outClusterList":[],"clusters":{"genBasic":{"dir":{"value":1},"attrs":{}},"genIdentify":{"dir":{"value":1},"attrs":{}},"genGroups":{"dir":{"value":1},"attrs":{}},"genScenes":{"dir":{"value":1},"attrs":{}},"genOnOff":{"dir":{"value":1},"attrs":{}},"genLevelCtrl":{"dir":{"value":1},"attrs":{}}}},"13":{"profId":49246,"epId":13,"devId":57694,"inClusterList":[4096],"outClusterList":[4096],"clusters":{"lightLink":{"dir":{"value":3},"attrs":{}}}}},"_id":"4UtIx3zVqzgjDM6y"}
{"id":2,"type":"Router","ieeeAddr":"0x00124b0014afe1fb","nwkAddr":14173,"manufId":0,"manufName":"Feibit Inc co.  ","powerSource":"Mains (single phase)","modelId":"FB56+ZSW1HKJ1.7","epList":[16,17],"status":"online","joinTime":1547518291,"endpoints":{"16":{"profId":260,"epId":16,"devId":2,"inClusterList":[0,4,5,6],"outClusterList":[0],"clusters":{"genBasic":{"dir":{"value":3},"attrs":{}},"genGroups":{"dir":{"value":1},"attrs":{}},"genScenes":{"dir":{"value":1},"attrs":{}},"genOnOff":{"dir":{"value":1},"attrs":{"onOff":1}}}},"17":{"profId":260,"epId":17,"devId":2,"inClusterList":[0,4,5,6],"outClusterList":[0],"clusters":{"genBasic":{"dir":{"value":3},"attrs":{"16393":1287651328,"hwVersion":0,"manufacturerName":"Feibit Inc co.  ","modelId":"FB56+ZSW1HKJ1.7","dateCode":"20161030        ","powerSource":1,"locationDesc":"                ","physicalEnv":0,"deviceEnabled":1}},"genGroups":{"dir":{"value":1},"attrs":{}},"genScenes":{"dir":{"value":1},"attrs":{}},"genOnOff":{"dir":{"value":1},"attrs":{"onOff":0}}}}},"_id":"SYv4N45UlvOeo59l"}
{"id":1,"type":"Coordinator","ieeeAddr":"0x00124b0014bdea20","nwkAddr":0,"manufId":0,"epList":[1,2,3,4,5,6,11],"status":"online","joinTime":1547518207,"endpoints":{"1":{"profId":260,"epId":1,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{}},"2":{"profId":257,"epId":2,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{}},"3":{"profId":261,"epId":3,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{}},"4":{"profId":263,"epId":4,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{}},"5":{"profId":264,"epId":5,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{}},"6":{"profId":265,"epId":6,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{}},"11":{"profId":260,"epId":11,"devId":1024,"inClusterList":[0,1281,3],"outClusterList":[1280,1282,3],"clusters":{"genBasic":{"dir":{"value":1},"acls":{"zclVersion":"R","hwVersion":"R","manufacturerName":"R","modelId":"R","dateCode":"R","powerSource":"R","locationDesc":"RW","physicalEnv":"RW","deviceEnabled":"RW"},"attrs":{"zclVersion":1,"hwVersion":1,"manufacturerName":"sivann inc.","modelId":"hiver0001","dateCode":"20170407","powerSource":1,"locationDesc":"    ","physicalEnv":0,"deviceEnabled":1}},"ssIasAce":{"dir":{"value":1}},"ssIasZone":{"dir":{"value":2},"cmdRsps":{"enrollReq":{"exec":"_exec_"},"statusChangeNotification":{"exec":"_exec_"}}},"ssIasWd":{"dir":{"value":2}},"genIdentify":{"dir":{"value":3},"acls":{"identifyTime":"RW"},"attrs":{"identifyTime":0}}}}},"_id":"xyNF05O5uq7OmfpG"}
{"$$indexCreated":{"fieldName":"id","unique":true,"sparse":false}}
{"id":1,"type":"Coordinator","ieeeAddr":"0x00124b0014bdea20","nwkAddr":0,"manufId":0,"epList":[1,2,3,4,5,6],"status":"online","joinTime":1547518431,"endpoints":{"1":{"profId":260,"epId":1,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{}},"2":{"profId":257,"epId":2,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{}},"3":{"profId":261,"epId":3,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{}},"4":{"profId":263,"epId":4,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{}},"5":{"profId":264,"epId":5,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{}},"6":{"profId":265,"epId":6,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{}}},"_id":"xyNF05O5uq7OmfpG"}
{"id":3,"type":"Router","ieeeAddr":"0x00124b001545a430","nwkAddr":8910,"manufId":0,"manufName":"Feibit Inc co.  ","powerSource":"Mains (single phase)","modelId":"FB56+ZSC05HG1.0","epList":[11,13],"status":"offline","joinTime":null,"endpoints":{"11":{"profId":260,"epId":11,"devId":257,"inClusterList":[0,3,4,5,6,8],"outClusterList":[],"clusters":{"genBasic":{"dir":{"value":1},"attrs":{}},"genIdentify":{"dir":{"value":1},"attrs":{}},"genGroups":{"dir":{"value":1},"attrs":{}},"genScenes":{"dir":{"value":1},"attrs":{}},"genOnOff":{"dir":{"value":1},"attrs":{}},"genLevelCtrl":{"dir":{"value":1},"attrs":{}}}},"13":{"profId":49246,"epId":13,"devId":57694,"inClusterList":[4096],"outClusterList":[4096],"clusters":{"lightLink":{"dir":{"value":3},"attrs":{}}}}},"_id":"4UtIx3zVqzgjDM6y"}
{"id":2,"type":"Router","ieeeAddr":"0x00124b0014afe1fb","nwkAddr":14173,"manufId":0,"manufName":"Feibit Inc co.  ","powerSource":"Mains (single phase)","modelId":"FB56+ZSW1HKJ1.7","epList":[16,17],"status":"offline","joinTime":null,"endpoints":{"16":{"profId":260,"epId":16,"devId":2,"inClusterList":[0,4,5,6],"outClusterList":[0],"clusters":{"genBasic":{"dir":{"value":3},"attrs":{}},"genGroups":{"dir":{"value":1},"attrs":{}},"genScenes":{"dir":{"value":1},"attrs":{}},"genOnOff":{"dir":{"value":1},"attrs":{"onOff":1}}}},"17":{"profId [... cut ...] lusterList":[0,3,4,5,6,8],"outClusterList":[],"clusters":{"genBasic":{"dir":{"value":1},"attrs":{}},"genIdentify":{"dir":{"value":1},"attrs":{}},"genGroups":{"dir":{"value":1},"attrs":{}},"genScenes":{"dir":{"value":1},"attrs":{}},"genOnOff":{"dir":{"value":1},"attrs":{}},"genLevelCtrl":{"dir":{"value":1},"attrs":{}}}},"13":{"profId":49246,"epId":13,"devId":57694,"inClusterList":[4096],"outClusterList":[4096],"clusters":{"lightLink":{"dir":{"value":3},"attrs":{}}}}},"_id":"4UtIx3zVqzgjDM6y"}
```

Koenkk, 2019-01-16:
Can you add the following to `devices.js`:

```js
    {
        zigbeeModel: ['FB56+ZSC05HG1.0'],
        model: 'TODO',
        vendor: 'Nue',
        description: 'ZigBee smart dimmer switch',
        supports: 'on/off, brightness',
        fromZigbee: [],
        toZigbee: [],
        configure: (ieeeAddr, shepherd, coordinator, callback) => {
            const device = shepherd.find(ieeeAddr, 11);
            const actions = [
                (cb) => device.bind('genOnOff', coordinator, cb),
                (cb) => device.bind('genLevelCtrl', coordinator, cb),
            ];

            execute(device, actions, callback);
      },
    },
```

- In `configuration.yaml` set `log_level: debug` (https://koenkk.github.io/zigbee2mqtt/configuration/configuration.html) 
- Start zigbee2mqtt *without* `DEBUG=`
- After 10 seconds, press all the button on the device
- Provide the log

Koenkk, 2019-01-16:
Are you sure the readRsp comes from the button click? These can happen randomly, therefore it could look like that it comes from the button click which is not the case.

Can you try with:

```js
    {
        zigbeeModel: ['FB56+ZSC05HG1.0'],
        model: 'TODO',
        vendor: 'Nue',
        description: 'ZigBee smart dimmer switch',
        supports: 'on/off, brightness',
        fromZigbee: [],
        toZigbee: [],
        configure: (ieeeAddr, shepherd, coordinator, callback) => {
            const device = shepherd.find(ieeeAddr, 11);
            const cfg = {direction: 0, attrId: 0, dataType: 16, minRepIntval: 0, maxRepIntval: 3, repChange: 0};
            const actions = [
                (cb) => device.bind('genOnOff', coordinator, cb),
                (cb) => device.bind('genLevelCtrl', coordinator, cb),
                (cb) => device.foundation('genOnOff', 'configReport', [cfg], foundationCfg, cb),
                (cb) => device.report('genLevelCtrl', 'currentLevel', 0, 3, 0, cb),
            ];

            execute(device, actions, callback);
      },
    },
```

highground88, 2019-01-16:
Sure. For info the "readRsp" is definitely a button press - it just happens that this dimming device rapidly sends the command even when briefly held down, hence why so many,

```
> zigbee2mqtt@1.0.1 start /opt/zigbee2mqtt
> node index.js

  zigbee2mqtt:info 1/16/2019, 11:11:06 AM Logging to directory: '/opt/zigbee2mqtt/data/log/2019-01-16.11-11-06'
  zigbee2mqtt:debug 1/16/2019, 11:11:07 AM Using zigbee-shepherd with settings: '{"net":{"panId":6754,"channelList":[11],"precfgkey":[1,3,5,7,9,11,13,15,0,2,4,6,8,10,12,13]},"dbPath":"/opt/zigbee2mqtt/data/database.db","sp":{"baudRate":115200,"rtscts":true}}'
  zigbee2mqtt:debug 1/16/2019, 11:11:07 AM Loaded state from file /opt/zigbee2mqtt/data/state.json
  zigbee2mqtt:info 1/16/2019, 11:11:07 AM Starting zigbee2mqtt version 1.0.1 (commit #9605e72)
  zigbee2mqtt:info 1/16/2019, 11:11:07 AM Starting zigbee-shepherd
  zigbee2mqtt:info 1/16/2019, 11:11:08 AM zigbee-shepherd started
  zigbee2mqtt:info 1/16/2019, 11:11:08 AM Coordinator firmware version: '20181024'
  zigbee2mqtt:debug 1/16/2019, 11:11:08 AM zigbee-shepherd info: {"enabled":true,"net":{"state":"Coordinator","channel":11,"panId":"0x1a62","extPanId":"0xdddddddddddddddd","ieeeAddr":"0x00124b0014bdea20","nwkAddr":0},"firmware":{"transportrev":2,"product":0,"version":"2.6.3","revision":20181024},"startTime":1547637068,"joinTimeLeft":0}
  zigbee2mqtt:info 1/16/2019, 11:11:08 AM Currently 2 devices are joined:
  zigbee2mqtt:info 1/16/2019, 11:11:08 AM 0x00124b001545a430 (0x00124b001545a430): TODO - Nue ZigBee smart dimmer switch (Router)
  zigbee2mqtt:info 1/16/2019, 11:11:08 AM 0x00124b0014afe1fb (0x00124b0014afe1fb): TODO - Nue Smart light switch - 2 gang (Router)
  zigbee2mqtt:warn 1/16/2019, 11:11:08 AM `permit_join` set to  `true` in configuration.yaml.
  zigbee2mqtt:warn 1/16/2019, 11:11:08 AM Allowing new devices to join.
  zigbee2mqtt:warn 1/16/2019, 11:11:08 AM Set `permit_join` to `false` once you joined all devices.
  zigbee2mqtt:info 1/16/2019, 11:11:08 AM Zigbee: allowing new devices to join.
  zigbee2mqtt:info 1/16/2019, 11:11:08 AM Connecting to MQTT server at mqtt://192.168.0.101
  zigbee2mqtt:debug 1/16/2019, 11:11:09 AM Configuring 0x00124b001545a430 (0x00124b001545a430) ...
  zigbee2mqtt:debug 1/16/2019, 11:11:09 AM Configuring 0x00124b0014afe1fb (0x00124b0014afe1fb) ...
  zigbee2mqtt:info 1/16/2019, 11:11:09 AM zigbee-shepherd ready
  zigbee2mqtt:info 1/16/2019, 11:11:09 AM Connected to MQTT server
  zigbee2mqtt:info 1/16/2019, 11:11:09 AM MQTT publish: topic 'zigbee2mqtt/bridge/state', payload 'online'
  zigbee2mqtt:debug 1/16/2019, 11:11:09 AM Mounted the cieApp (epId 11)
  zigbee2mqtt:error 1/16/2019, 11:11:10 AM Failed to configure 0x00124b0014afe1fb (0x00124b0014afe1fb) ('Error: AF data request fails, status code: 233. MAC no ack.')
defaultRsp
defaultRsp
  zigbee2mqtt:error 1/16/2019, 11:11:10 AM Failed to configure 0x00124b001545a430 (0x00124b001545a430) ('TypeError: Cannot read property 'status' of undefined')
readRsp
readRsp
readRsp
readRsp
```

Is something else in my setup possibly broken at the moment - would it be worth pulling it down again, and if so, would you like me to change the cie file or anything else? Thanks

Koenkk, 2019-01-16:
If that is the case, can you add `readRsp` to this array: https://github.com/Koenkk/zigbee-shepherd/blob/master/lib/components/af.js#L687

highground88, 2019-01-16:
I've edited this file in "/opt/zigbee2mqtt/node_modules/zigbee-shepherd/lib/components" but unfortunately the output is the same.

Koenkk, 2019-01-16:
Can you try with:
```js
    {
        zigbeeModel: ['FB56+ZSC05HG1.0'],
        model: 'TODO',
        vendor: 'Nue',
        description: 'ZigBee smart dimmer switch',
        supports: 'on/off, brightness',
        fromZigbee: [],
        toZigbee: [],
        configure: (ieeeAddr, shepherd, coordinator, callback) => {
            const device = shepherd.find(ieeeAddr, 11);
            const cfg = {direction: 0, attrId: 0, dataType: 16, minRepIntval: 0, maxRepIntval: 3, repChange: 0};
            const actions = [
                (cb) => device.bind('genOnOff', coordinator, cb),
                (cb) => device.bind('genLevelCtrl', coordinator, cb),
                (cb) => device.report('genLevelCtrl', 'currentLevel', 0, 3, 0, cb),
            ];

            execute(device, actions, callback);
      },
    },
```

highground88, 2019-01-16:
No change sorry....  [LOG](https://pastebin.com/Ydqvx64g)

highground88, 2019-01-16:
This is the resulting log pf both devices powered up on boot.

[2 devices](https://pastebin.com/De7EKhBb)

Koenkk, 2019-01-16:
Could you siff the zigbee traffic of the FB56+ZSC05HG1.0 ?

highground88, 2019-01-20:
Hi @Koenkk  - my apologies for not getting back to you sooner, life and an unexpected rebuild of my LAN got in the way....

But I think I have some good news.  My latest sniffing attempt of the FB56+ZSC05HG1.0 turned up another device trust centre key.... I had assumed this key was a single key that might be manufacturer specific across all devices, but maybe not?

The latest capture shops the attribute fields that coincide with the button presses of the FB56+ZSC05HG1.0 and also the FB56+ZSW1HKJ1.7. These are located here in pcap format with the necessary codes: [LINK](https://www.dropbox.com/sh/vt3o8lken7vbkmx/AAAB2Y4tX3dcY0Hzat7-yj4Ya/20JAN19?dl=0&subfolder_nav_tracking=1)

I'm running two instances of HA to try to debug this and currently experiencing a few issues with the Hassbian flavour, but I am about to test out your latest dev build on my Hassio build, as I see you've been busy with the code and everyone elses issues....   thanks mate.

highground88, 2019-01-20:
I'm still not entirely sure how this part works, but in analysing the pcap files I think I have determined what each devices 'supports' from the attributes response. As far as determining precisely what to put in the from/to zigbee, I'm really only guessing and using previous device entries. This is the current devices.js entry:

```
{
        zigbeeModel: ['FB56+ZSW1HKJ1.7'],
        model: 'TODO',
        vendor: 'Nue',
        description: 'Smart light switch - 2 gang',
        supports: 'Basic, On/Off, Scenes, Groups',
        fromZigbee: [fz.generic_state],
        toZigbee: [tz.on_off],
        configure: (ieeeAddr, shepherd, coordinator, callback) => {
            const ep16 = shepherd.find(ieeeAddr, 16);
            execute(ep16, [(cb) => ep16.bind('genOnOff', coordinator, cb)], () => {
                const ep17 = shepherd.find(ieeeAddr, 17);
                execute(ep17, [(cb) => ep17.bind('genOnOff', coordinator, cb)], callback);
            });
        },
    },
	{
        zigbeeModel: ['FB56+ZSC05HG1.0'],
        model: 'TESTING-DIMMER SWITCH',
        vendor: 'Nue',
        description: 'ZigBee smart dimmer switch',
        supports: 'On/Off, Level Control, Scenes, Groups, ZLL Commissioning, Power Source, Identify',
        fromZigbee: [],
        toZigbee: [],
        configure: (ieeeAddr, shepherd, coordinator, callback) => {
            const device = shepherd.find(ieeeAddr, 11);
            const cfg = {direction: 0, attrId: 0, dataType: 16, minRepIntval: 0, maxRepIntval: 3, repChange: 0};
            const actions = [
                (cb) => device.bind('genOnOff', coordinator, cb),
                (cb) => device.bind('genLevelCtrl', coordinator, cb),
                (cb) => device.report('genLevelCtrl', 'currentLevel', 0, 3, 0, cb),
            ];

            execute(device, actions, callback);
        },
    },
```

I did notice  the coordinator issue which looked like the switch messages were sent by the co-ord ID has gone, a new one perhaps has arrived. I've highlighted the true device details below (..1fb), followed by some random ID (...6f2) which seems to be the same bug as before, but with a different ID? 

```
2019-01-20T08:34:02.329Z zigbee-shepherd-converters:devices Configured '(cb) => ep17.bind('genOnOff', coordinator, cb)' with result 'OK'
  zigbee2mqtt:info 2019-1-20 18:34:02 Succesfully configured **0x00124b0014afe1fb** (0x00124b0014afe1fb)
2019-01-20T08:35:02.104Z zigbee-shepherd:af dispatchIncomingMsg(): type: incomingMsg, msg: [object Object]
2019-01-20T08:35:02.121Z zigbee-shepherd:msgHdlr IND <-- AF:incomingMsg, transId: 0
2019-01-20T08:35:02.123Z zigbee-shepherd:af dispatchIncomingMsg(): type: zclIncomingMsg, msg: [object Object]
  zigbee2mqtt:debug 2019-1-20 18:35:02 Received zigbee message of type 'attReport' with data '{"cid":"genOnOff","data":{"onOff":0}}' of device 'undefined' (**0x00124b0018df46f2**)
  zigbee2mqtt:debug 2019-1-20 18:35:02 Ignoring message from coordinator
2019-01-20T08:35:43.330Z zigbee-shepherd:af dispatchIncomingMsg(): type: incomingMsg, msg: [object Object]
2019-01-20T08:35:43.337Z zigbee-shepherd:msgHdlr IND <-- AF:incomingMsg, transId: 0
2019-01-20T08:35:43.338Z zigbee-shepherd:af dispatchIncomingMsg(): type: zclIncomingMsg, msg: [object Object]
  zigbee2mqtt:debug 2019-1-20 18:35:43 Received zigbee message of type 'attReport' with data '{"cid":"genOnOff","data":{"onOff":0}}' of device 'undefined' (0x00124b0018df46f2)
  zigbee2mqtt:debug 2019-1-20 18:35:43 Ignoring message from coordinator
2019-01-20T08:36:04.641Z zigbee-shepherd:af dispatchIncomingMsg(): type: incomingMsg, msg: [object Object]
2019-01-20T08:36:04.651Z zigbee-shepherd:msgHdlr IND <-- AF:incomingMsg, transId: 0
2019-01-20T08:36:04.655Z zigbee-shepherd:af dispatchIncomingMsg(): type: zclIncomingMsg, msg: [object Object]
  zigbee2mqtt:debug 2019-1-20 18:36:04 Received zigbee message of type 'attReport' with data '{"cid":"genOnOff","data":{"onOff":1}}' of device 'undefined' (0x00124b0018df46f2)
  zigbee2mqtt:debug 2019-1-20 18:36:04 Ignoring message from coordinator
2019-01-20T08:36:08.807Z zigbee-shepherd:af dispatchIncomingMsg(): type: incomingMsg, msg: [object Object]
2019-01-20T08:36:08.844Z zigbee-shepherd:msgHdlr IND <-- AF:incomingMsg, transId: 0
2019-01-20T08:36:08.845Z zigbee-shepherd:af dispatchIncomingMsg(): type: zclIncomingMsg, msg: [object Object]
  zigbee2mqtt:debug 2019-1-20 18:36:08 Received zigbee message of type 'attReport' with data '{"cid":"genOnOff","data":{"onOff":1}}' of device 'undefined' (0x00124b0018df46f2)
  zigbee2mqtt:debug 2019-1-20 18:36:08 Ignoring message from coordinator
```

Koenkk, 2019-01-21:
It seems we have the same issue again, it looks like the messages are send by the coordinator.

I'm wondering if you are able to control the device

Can you send to `zigbee2mqtt/0x00124b0014afe1fb/set` payload `ON` or `OFF`

highground88, 2019-01-24:
Yes!!!!  That (and 'off') now control that 2 gang (FB56+ZSW1HKJ1.7) switch! Great work!

How does the handler need to be setup (code wise) to reach the various different end-points for the different gang switches? This is something I still don't follow due to my lack of programming skills... Once I work this out I will start to compile a list of the models I have and hopefully get them up and running!

What else do you need me to try to help with in regards to some of the other bugs still present?

[119 more replies]

closed 2019-08-04 by stale[bot]