Koenkk/zigbee2mqtt #177: new mqtt command to rename devices

lolorc, 2018-07-03:
Hi,

I've just started to use your project yesterday, I might be wrong but it looks like we have to restart the daemon when we rename a device.
If this statement is correct, I think it would be useful to be able to rename a device with a mqtt command.

ciotlosm, 2018-07-03:
@lolorc if you're using home assistant you could use the unique_id functionality and just rename your devices in Home Assistant directly.

tb-killa, 2018-07-03:
@Koenkk Question: Does we could trigger zigbee2mqtt internal to reread the configuration.yaml ?
So we could build mqtt topic for rename who is internal only the friendly_name entry change and trigger the reread of the configuration.yaml.

lolorc, 2018-07-04:
> @lolorc if you're using home assistant you could use the unique_id functionality and just rename your devices in Home Assistant directly.

Yes I'm using home-assistant but I'm not using the discovery thingy. I guess I should have a look at it.
However I still think it would be useful to be able rename devices and reload config thru mqtt.

ciotlosm, 2018-07-04:
@lolorc Agree that feature would be useful, but believe me that using Home Assistant with discovery and unique ID makes renaming and managing devices much easier.

tb-killa, 2018-07-09:
Implemented rudimental rename function for settings.js
```
function RenameFriendlyName(original,replace) {
	originalName = getIDByFriendlyName(original)
	if(!originalName) {
		return null;
	}
	removeDevice(originalName)
	if (!settings.devices) {
        settings.devices = {};
    }
	settings.devices[originalName] = {friendly_name: replace, retain: false};
    writeRead();
}
```
module.exports
```
RenameFriendlyName: (original,replace) => RenameFriendlyName(original,replace),
```

Koenkk, 2018-07-24:
Merged and documented: https://github.com/Koenkk/zigbee2mqtt/wiki/MQTT-topics-and-message-structure#zigbee2mqttbridgeconfigrename

Many thanks for the implementation @tb-killa !

closed 2018-07-24 by Koenkk