Koenkk/zigbee2mqtt #178: Log firmware version on startup

Koenkk, 2018-07-03:
See @tb-killa comment from #174

tb-killa, 2018-07-03:
@Koenkk Could we extend our welcome Message:
```
2018-7-3 22:44:38 [32mINFO[39m Starting zigbee2mqtt version 0.1.0 (commit #ae41b7b)
2018-7-3 22:44:38 [32mINFO[39m Starting zigbee-shepherd
2018-7-3 22:44:40 [32mINFO[39m zigbee-shepherd started
```
with 

```
zigbee-shepherd started with loaded configuration:
```
and the Informations from
```lib/shepherd.js ``` with function ```ZShepherd.prototype.info```
???

Koenkk, 2018-07-04:
@tb-killa yes!

tb-killa, 2018-07-05:
@Koenkk Nice :)
```
2018-7-5 20:19:54 INFO zigbee-shepherd started
2018-7-5 20:19:54 DEBUG zigbee-shepherd info: {"enabled":true,"net":{"state":"Coordinator","channel":11,"panId":"0x1a62","extPanId":"0xdddddddddddddddd","ieeeAddr":"0x00124b000e896656","nwkAddr":0},"firmware":{"error":"Unable to get firmware version"},"startTime":1530814794,"joinTimeLeft":0}
2018-7-5 20:19:54 INFO Currently 3 devices are joined:
```
Have to check the above mentioned Code for Firmware Version, it seems that i miss something :(

Koenkk, 2018-07-05:
@tb-killa yes, working on that now

ryanbeaton, 2018-07-08:
So only when logging level is debug then?

tb-killa, 2018-07-08:
@ryanbeaton yes because normaly you need this Informations only if you would debug something what doesnt seems to work.

But maybe we cut this message and bring Version as normal output.?
@Koenkk what do you think?

tb-killa, 2018-08-12:
@Koenkk We should cut the message output from ``ZShepherd.prototype.info``

```
zigbee-shepherd info: {"enabled":true,"net":{"state":"Coordinator","channel":11,"panId":"0x1a62","extPanId":"0xdddddddddddddddd","ieeeAddr":"0x00124b000e896656","nwkAddr":0},"firmware":{"transportrev":2,"product":0,"version":"2.6.3","revision":20180507},"startTime":1534098600,"joinTimeLeft":0}
```

I don't think the following is needed:
``"enabled":true,`` because we know if shepherd is running or not ;)
``"state":"Coordinator"`` can we be anything but Coordinator ?
``"transportrev":2,"product":0`` we doesn´t need this
``"startTime":1534098600`` we know our startTime ;)
``"joinTimeLeft":0`` we doesn´t need this except we allow User to define own joinTime (maybe we should put this in?) ;)

Koenkk, 2018-08-13:
I think we can keep it like this. Note that the firmware version is now also logged separately as `info`.

```
  zigbee2mqtt:info 2018-8-13 19:12:16 Coordinator firmware version: '20180507'
  zigbee2mqtt:debug 2018-8-13 19:12:16 zigbee-shepherd info: {"enabled":true,"net":{"state":"Coordinator","channel":11,"panId":"0x1a62","extPanId":"0xdddddddddddddddd","ieeeAddr":"0x00124b000e89686d","nwkAddr":0},"firmware":{"transportrev":2,"product":0,"version":"2.6.3","revision":20180507},"startTime":1534180336,"joinTimeLeft":0}
```

tb-killa, 2018-08-13:
great .. doesn´t realice this information xD ..

ryanbeaton, 2018-08-14:
Nice. Will definitely help support requests.

closed 2018-12-22 by stale[bot]