Koenkk/zigbee2mqtt #179: Creating topic names

Imperial-Guard, 2018-07-04:
Would be a very nice added feature to use your own naming conventions instead of the  standard name: 

zigbee2mqtt/<DEVICE_ID>

In my case I have +12 devices which are communicating over MQTT. And still adding new devices. 
When I will add for each room a temperature sensor, doorsensor etc. 
Debugging will be difficult with all those unique id's where I don't know which device what is. 

I made for each device a unique name with my naming convention format. 
See example below:
Location | Type | Number | Description | Base mqtt topic

hallway | switch | 0001 | doorbell | hallway/switch/0001/doorbell
hallway | light | 0002 | lightswitch | hallway/light/0002/lightswitch
basement | switch | 0001 | ventilation | basement/switch/0001/ventilation
garden | sensor | 0001 | weather_station | garden/sensor/0001/weather_station

Thanks

ciotlosm, 2018-07-04:
@Imperial-Guard use the friendly_name feature.See https://github.com/Koenkk/zigbee2mqtt/wiki/Device-specific-configuration

Koenkk, 2018-07-04:
@ciotlosm , @Imperial-Guard means that the format of the messages can be changed. E.g. `blaa/blaa/DEVICE_ID`. This should be possible by allowing the user to specify a message format.

Imperial-Guard, 2018-09-06:
@Koenkk any news on this? Thanks in advance.

Koenkk, 2018-09-07:
not yet, still very limited in time due to moving

peterforeman, 2018-12-18:
Still a wanted feature. Should not be closed just yet.

closed 2019-02-23 by stale[bot]

Fenix999, 2019-03-12:
This would be a nice Feature....